emsesp/EMS-ESP #176: any send causes CRC error

oli747, 2019-08-28:
I have some issues with the EMS bus.

I have the code running on a wemos D1R2 mini. I use the EMS interface board from bbqkees (I ordered a soldered and tested one, before he started to ship the complete setup with the esp8266 included). I power the wemos with a 5V power supply, and connect to the bus using EMS screw terminal to the Nefit trendline boiler / Nefit Easy thermostat. I tried the resistor jumper setting in both positions and see no difference. B.t.w. I compiled the code on Arduino (needed to change a few include headers from <> to " " but no real code changes). 

EMS-ESP system stats:
  System logging set to Verbose
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=17
  Tx: Last poll=2.500 seconds ago, # successful write requests=0

[..]

From looking at the logging (set to verbose), I think it reads messages correctly, but it fails to send messages (any send causes a CRC error). But perhaps I interpret the logs wrong. This is a typical log:

(00:00:09.900) Sending raw: 8B 88 02 00 20 00
(00:00:09.910) Corrupt telegram: 8B 4C 00 20 74
(00:00:11.368) 0x18 -> 0x08, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(00:00:11.404) 0x08 -> 0x18, type 0x18, telegram: 08 18 18 00 07 01 80 00 00 00 00 00 60 01 89 01 D9 01 6C 00 00 0E 30 48 00 CB 00 00 (CRC=94) #data=24
(00:00:11.443) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(00:00:12.775) Sending read of type 0x02 to 0x30:telegram: 0B B0 02 00 20 (CRC=65) #data=142
(00:00:12.785) Corrupt telegram: 0B 13 00 20 65
(00:00:18.130) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 80 00 00 00 00 00 60 01 89 01 D9 01 6C 00 00 0E 30 48 00 CB 00 00 00 (CRC=6D) #data=25
<--- UBAMonitorFast(0x18)
(00:00:18.349) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 D9 01 D9 21 00 00 03 00 00 DC 14 00 26 26 00 80 00 (CRC=23) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:18.575) Sending read of type 0x02 to 0x38:telegram: 0B B8 02 00 20 (CRC=25)
(00:00:18.584) Corrupt telegram: 0B 13 00 20 25
(00:00:27.843) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(00:00:28.155) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 80 00 00 00 00 00 60 01 89 01 D9 01 6D 00 00 0E 30 48 00 CB 00 00 00 (CRC=09) #data=25
<--- UBAMonitorFast(0x18)
(00:00:28.424) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 D9 01 D9 21 00 00 03 00 00 DC 14 00 26 26 00 80 00 (CRC=23) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:29.525) Sending read of type 0x02 to 0x02:telegram: 0B 82 02 00 20 (CRC=EC)
(00:00:29.535) Corrupt telegram: 0B 30 20 20 EC
(00:00:36.368) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(00:00:37.625) Sending read of type 0x02 to 0x08:telegram: 0B 88 02 00 20 (CRC=BC)
(00:00:37.635) Corrupt telegram: 0B 4C 00 20 BC

Anybody an idea  what I should try next?

proddy, 2019-08-29:
As the board came from bbqkees I can assure you its been thoroughly tested and Rx/Tx is wired correctly. I trust this is version 1.8.x so also have a look at https://github.com/proddy/EMS-ESP/wiki/Troubleshooting.

The boiler is 0x18 and Rx is working I see. Also try querying it directly with `send 0B 90 02 00 20`

proddy, 2019-08-29:
Also, bbqkees's boards come with firmware installed. Did it work before you built the firmware yourself? And to rule out any build errors since you're not using the official pio method, try flashing with one of the firmware bin's from https://github.com/proddy/EMS-ESP/releases

oli747, 2019-08-29:
> Also, bbqkees's boards come with firmware installed. 

as I tried to explain: I ordered the bbqkees board a few months ago, when he offered just the ' level shifter' board without the esp8266 and the voltage converter. 

I will try the firmware bin, and I will try to do the raw send.

thanks for your support, it is much appreciated!

oli747, 2019-08-29:
I now have your 1.8.3 build running, result is exactly the same (so it seems the build was good).

But your remark that Rx does work and Tx does not made me try to change the tx_mode setting, and setting this to 1 worked! thanks again for your support!

proddy, 2019-08-29:
Ok great. Strange that it didn't work with tx_mode 0. If you have time would you mind trying out version 1.9. Just take the bin. The whole tx_mode has gone and there is new logic to support writing (Tx). It'll be interesting to hear if it works on your EMS bus.

oli747, 2019-08-29:
No luck with 1.9.0b5, and no tx_mode setting so I don't know how to get this to work. Here is some general info, if you want me to try some things, please let me know.

system
ESP8266 System stats:

 [APP] EMS-ESP version: 1.9.0b5
 [APP] MyESP version: 1.2.0
 [APP] Build timestamp: 2019-08-26 23:16:06
 [APP] Uptime: 0 days 0 hours 7 minutes 24 seconds
 [APP] System Load: 6%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.0.39
 [WIFI] WiFi signal strength: 70%
 [WIFI] WiFi MAC: CC:50:E3:17:2E:17
 [MQTT] is disconnected
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.1(cfd48f3)
 [SYSTEM] CPU chip ID: 0x172E17
 [SYSTEM] Core version: 2_5_2
 [SYSTEM] Boot version: 6
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Restart from terminal
 [SYSTEM] Restart count: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65536
 [FLASH] Flash chip ID: 0x164020
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 674000
 [MEM] Max OTA size: 2465792
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 24400 bytes initially | 8488 bytes used (34%) | 15912 bytes free (65%)

info
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is enabled, Shower Alert is enabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=0, # CRC errors=15
  Tx: Last poll=2.450 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: <not enabled>
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 48.9 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 9768 times
  Warm Water active time: 39 days 3 hours 22 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 7 C
  Current flow temperature: 34.7 C
  Return temperature: 35.6 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.4 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: 36.6 C
  Pump modulation: 0 %
  Burner # starts: 44426 times
  Total burner operating time: 587 days 0 hours 22 minutes
  Total heat operating time: 547 days 21 hours 0 minutes

Shower stats:
  Shower is off

autodetect
Started scan on EMS bus for known devices
Requesting type Version(0x02) from dest 0x02
Requesting type Version(0x02) from dest 0x08
Requesting type Version(0x02) from dest 0x09
Requesting type Version(0x02) from dest 0x10
Requesting type Version(0x02) from dest 0x11
Requesting type Version(0x02) from dest 0x17
Requesting type Version(0x02) from dest 0x18
Requesting type Version(0x02) from dest 0x20
Requesting type Version(0x02) from dest 0x21
Requesting type Version(0x02) from dest 0x30
Requesting type Version(0x02) from dest 0x38
Requesting type Version(0x02) from dest 0x48
(00:09:50.523) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 5B 00 00 00 00 00 60 01 91 01 E9 01 64 00 00 0E 30 48 00 CB 00 00 00 (CRC=C3) #data=25
<--- UBAMonitorFast(0x18)
(00:09:50.767) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E9 01 E9 21 00 00 03 00 00 DC 2A 00 26 28 00 80 00 (CRC=C1) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:09:52.043) Sending read of type 0x02 to 0x02, telegram: 0B 82 02 00 20 (CRC=EC)
(00:09:52.311) Corrupt telegram: 0B 82 02 00 20 EC 00 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 (CRC=00) #data=19
(00:09:52.786) 0x18 -> 0x08, type 0x16, telegram: 18 88 16 01 02 (CRC=E5) #data=1
(00:09:52.799) 0x08 -> 0x18, type 0x16, telegram: 08 18 16 01 41 41 (CRC=F7) #data=2
(00:09:52.986) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
(00:09:53.204) 0x18 -> 0x08, type 0x06, telegram: 18 08 06 00 13 08 11 1D 0D 09 03 01 (CRC=6E) #data=8
(00:09:53.436) 0x18 -> 0x08, type 0x1C, telegram: 18 88 1C 00 08 (CRC=C5) #data=1
(00:09:53.455) 0x08 -> 0x18, type 0x1C, telegram: 08 18 1C 00 91 08 0E 16 30 00 00 00 (CRC=14) #data=8
(00:09:53.614) 0x18 -> 0x08, type 0x1A, telegram: 18 08 1A 00 00 00 (CRC=7B) #data=2
(00:09:53.836) 0x18 -> 0x08, type 0x35, telegram: 18 08 35 00 11 (CRC=1C) #data=1
(00:09:54.036) 0x18 -> 0x08, type 0x33, telegram: 18 88 33 02 01 (CRC=74) #data=1
(00:09:54.048) 0x08 -> 0x18, type 0x33, telegram: 08 18 33 02 37 (CRC=BF) #data=1
(00:09:55.193) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(00:09:55.233) Corrupt telegram: 0B 88 02 00 20 BC 00 08 0B 02 00 73 05 08 00 00 00 00 00 00 (CRC=00) #data=16
(00:09:55.486) 0x08 -> all, type 0x07, telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=5A) #data=13
(00:09:55.893) Sending read of type 0x02 to 0x09, telegram: 0B 89 02 00 20 (CRC=B4)
(00:09:55.927) Corrupt telegram: 0B 89 02 00 20 B4 00 09 0B 02 00 72 01 (CRC=03) #data=9

oli747, 2019-08-30:
please let me know if you want me to test anything on 1.9, otherwise I'll revert the install to 1.8 and continue to integrate the setup in my home logging.

proddy, 2019-08-30:
thanks for offering to help. I removed tx_mode from the new web interface as I was pretty confident the new tx code (mode 2) would work for all EMS devices but looks like there are a few exceptions. I'll add it back to the 1.9 and do a build tomorrow called 1.9.0b7 so you can use that. If you have any suggestions for the project let me know!

susisstrolch, 2019-09-05:
Sorry for the late reply against that already closed issue - was busy with other projects...

(00:09:52.043)` Sending read of type 0x02 to 0x02, telegram: 0B 82 02 00 20 (CRC=EC)
(00:09:52.311) Corrupt telegram: 0B 82 02 00 20 EC 00 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 (CRC=00) #data=19

That's definitely not a problem of the sender! Instead, the <BRK> from the echo wasn't handled properly - either by timing or by protocol violation (sendig w/o poll acknowledge).

As you can see, the corrupt telegram consists of two parts: the echo of the send package and a new telegram from 0x08 to all:

echo: 0B 82 02 00 20 EC 00
telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00

The 00 at the end of the echo is either a misdetection of emsuart (which I don't believe) or simply a handling issue in ems-esp functions.

In those cases it's pretty helpfull to "set log j" -  jabbermode - which shows what's going on on the EMS-bus.

oli747, 2019-09-16:
log j
Unknown command. Use ? for help.

I guess I need to reflash to the beta first?

proddy, 2019-09-16:
I'm pretty sure Jabber was in 1.9.0 but I could be mistaken. Anyway always good to use the latest version from the dev branch which is now 1.9.1b3. The web interface should detect it and upload automagically. Otherwise grab it from the Releases page

susisstrolch, 2019-09-17:
> log j
> Unknown command. Use ? for help.
> 
> I guess I need to reflash to the beta first?

No...
> In those cases it's pretty helpfull to **"set log j"** - jabbermode - which shows what's going on on the EMS-bus.

proddy, 2019-09-27:
@oli747 if you're brave would you mind checking 1.9.1 again to see if you still get those CRC errors?

proddy, 2019-10-07:
closing - if this is still a problem please re-open

oli747, 2019-10-09:
sorry it took so long (since I cannot switch between 1.8 and 1.9 without physical access it takes more time). I've tried 1.9.2b3

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=0, # CRC errors=5
  Tx: no signal

webinterface sais: EMS Bus Connected but Tx is not working

with log v I don't see corrupted messages
(00:15:54.011) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F1 00 00 0D 30 48 00 CB 00 00 00 (CRC=90) #data=25
<--- UBAMonitorFast(0x18)
(00:15:54.230) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E8 01 E8 21 00 00 03 00 00 E0 7D 00 27 63 00 80 00 (CRC=D6) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:15:58.624) 0x18 -> 0x08, type 0x16, telegram: 18 88 16 01 02 (CRC=E5) #data=1
(00:15:58.637) 0x08 -> 0x18, type 0x16, telegram: 08 18 16 01 41 41 (CRC=F7) #data=2
(00:15:58.817) 0x18 -> 0x08, type 0x06, telegram: 18 08 06 00 13 0A 12 09 03 0D 02 01 (CRC=B5) #data=8
(00:15:59.049) 0x18 -> 0x08, type 0x1C, telegram: 18 88 1C 00 08 (CRC=C5) #data=1
(00:15:59.068) 0x08 -> 0x18, type 0x1C, telegram: 08 18 1C 00 91 08 0E 16 30 00 00 00 (CRC=14) #data=8
(00:15:59.252) 0x18 -> 0x08, type 0x1A, telegram: 18 08 1A 00 00 00 (CRC=7B) #data=2
(00:15:59.374) 0x18 -> 0x08, type 0x35, telegram: 18 08 35 00 11 (CRC=1C) #data=1
(00:15:59.624) 0x18 -> 0x08, type 0x33, telegram: 18 88 33 02 01 (CRC=74) #data=1
(00:15:59.636) 0x08 -> 0x18, type 0x33, telegram: 08 18 33 02 37 (CRC=BF) #data=1
Starting scheduled query from EMS devices
(00:16:03.986) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F0 00 00 0D 30 48 00 CB 00 00 00 (CRC=F4) #data=25
<--- UBAMonitorFast(0x18)
(00:16:04.205) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E8 01 E8 21 00 00 03 00 00 E0 7D 00 27 63 00 80 00 (CRC=D6) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:16:04.599) 0x18 -> 0x08, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(00:16:04.635) 0x08 -> 0x18, type 0x18, telegram: 08 18 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F0 00 00 0D 30 48 00 CB 00 00 (CRC=54) #data=24
(00:16:14.011) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F0 00 00 0D 30 48 00 CB 00 00 00 (CRC=F4) #data=25
<--- UBAMonitorFast(0x18)
(00:16:14.230) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E8 01 E8 21 00 00 03 00 00 E0 7D 00 27 63 00 80 00 (CRC=D6) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:16:15.724) 0x18 -> 0x08, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(00:16:15.760) 0x08 -> 0x18, type 0x18, telegram: 08 18 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F0 00 00 0D 30 48 00 CB 00 00 (CRC=54) #data=24
(00:16:24.024) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
<--- UBADevices(0x07)
(00:16:24.237) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F0 00 00 0D 30 48 00 CB 00 00 00 (CRC=F4) #data=25
<--- UBAMonitorFast(0x18)
(00:16:24.539) 0x08 -> all, type 0x19, telegram: 08 00 19 00 80 00 00 F9 80 00 00 00 00 00 00 AF 42 0C F2 7F 00 00 00 0C 12 02 00 87 DF 00 00 (CRC=D7) #data=27
<--- UBAMonitorSlow(0x19)
(00:16:24.786) 0x08 -> all, type 0x1C, telegram: 08 00 1C 00 91 08 0E 16 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4D) #data=25
(00:16:25.005) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E8 01 E8 21 00 00 03 00 00 E0 7D 00 27 63 00 80 00 (CRC=D6) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:16:34.037) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F0 00 00 0D 30 48 00 CB 00 00 00 (CRC=F4) #data=25
<--- UBAMonitorFast(0x18)
(00:16:34.255) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E8 01 E8 21 00 00 03 00 00 E0 7D 00 27 63 00 80 00 (CRC=D6) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:16:44.012) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F0 00 00 0D 30 48 00 CB 00 00 00 (CRC=F4) #data=25
<--- UBAMonitorFast(0x18)
(00:16:44.230) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E8 01 E8 21 00 00 03 00 00 E0 7D 00 27 63 00 80 00 (CRC=D6) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:16:54.012) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 00 E5 00 00 00 00 00 60 01 71 01 E8 00 F1 00 00 0D 30 48 00 CB 00 00 00 (CRC=90) #data=25
<--- UBAMonitorFast(0x18)
(00:16:54.330) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 01 E8 01 E8 21 00 00 03 00 00 E0 7D 00 27 63 00 80 00 (CRC=D6) #data=19
<--- UBAMonitorWWMessage(0x34)

log j gives huge amount of output, attached a few seconds (is that good enough?):

[log_j.txt](https://github.com/proddy/EMS-ESP/files/3708352/log_j.txt)

oli747, 2019-10-09:
b.t.w. I have 1.9.2b3 still on the flash now, so I can make more logging, but I'll soon flash it back to the (working) 1.8 release.

proddy, 2019-10-09:
Thanks for checking @oli747 . I understand having a dysfunctional system is annoying so I appreciate you trying out the latest build while 1.8.3 worked fine. In all the new releases I don't want to break backward compatible but with all the changes and the difficulty to test on physical devices some things do sometimes break. So we need to get Tx fixed with your system because we know it worked before.

So can you try `tx_mode 1` and `2`. And with each do a `autodetect quick` with logging enabled?

cheers

oli747, 2019-10-09:
tx_mode is not avilable on 1.9, is that correct? so I have to test this on 1.8 ?

what level of logging do you need?

oli747, 2019-10-09:
autodetect quick on 1.9 gives me this (log level v):

autodetect quick

(04:40:58.420) 0x18 -> 0x08, type 0x16, telegram: 18 88 16 01 02 (CRC=E5) #data=1
(04:40:58.433) 0x08 -> 0x18, type 0x16, telegram: 08 18 16 01 41 41 (CRC=F7) #data=2
(04:40:58.688) 0x18 -> 0x08, type 0x06, telegram: 18 08 06 00 13 0A 16 09 1C 0D 02 01 (CRC=CD) #data=8
(04:40:58.870) 0x18 -> 0x08, type 0x1C, telegram: 18 88 1C 00 08 (CRC=C5) #data=1
(04:40:58.889) 0x08 -> 0x18, type 0x1C, telegram: 08 18 1C 00 91 08 0E 16 30 00 00 00 (CRC=14) #data=8
(04:40:59.097) 0x18 -> 0x08, type 0x1A, telegram: 18 08 1A 00 00 00 (CRC=7B) #data=2
(04:40:59.269) 0x18 -> 0x08, type 0x35, telegram: 18 08 35 00 11 (CRC=1C) #data=1
(04:40:59.495) 0x18 -> 0x08, type 0x33, telegram: 18 88 33 02 01 (CRC=74) #data=1
(04:40:59.507) 0x08 -> 0x18, type 0x33, telegram: 08 18 33 02 37 (CRC=BF) #data=1
Starting scheduled query from EMS devices
(04:41:00.757) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D9 00 00 0D 30 48 00 CB 00 00 00 (CRC=13) #data=25
<--- UBAMonitorFast(0x18)
(04:41:00.976) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 02 2D 02 2D 21 00 00 03 00 00 E0 9C 00 27 65 00 80 00 (CRC=58) #data=19
<--- UBAMonitorWWMessage(0x34)
(04:41:01.370) 0x18 -> 0x08, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(04:41:01.406) 0x08 -> 0x18, type 0x18, telegram: 08 18 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D9 00 00 0D 30 48 00 CB 00 00 (CRC=AB) #data=24
(04:41:10.782) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D9 00 00 0D 30 48 00 CB 00 00 00 (CRC=13) #data=25
<--- UBAMonitorFast(0x18)
(04:41:11.001) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 02 2D 02 2D 21 00 00 03 00 00 E0 9C 00 27 65 00 80 00 (CRC=58) #data=19
<--- UBAMonitorWWMessage(0x34)
(04:41:12.495) 0x18 -> 0x08, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(04:41:12.531) 0x08 -> 0x18, type 0x18, telegram: 08 18 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D9 00 00 0D 30 48 00 CB 00 00 (CRC=AB) #data=24
(04:41:20.807) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D9 00 00 0D 30 48 00 CB 00 00 00 (CRC=13) #data=25
<--- UBAMonitorFast(0x18)
(04:41:21.026) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 02 2D 02 2D 21 00 00 03 00 00 E0 9C 00 27 65 00 80 00 (CRC=58) #data=19
<--- UBAMonitorWWMessage(0x34)
(04:41:23.545) 0x18 -> 0x08, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(04:41:23.581) 0x08 -> 0x18, type 0x18, telegram: 08 18 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D9 00 00 0D 30 48 00 CB 00 00 (CRC=AB) #data=24
(04:41:30.819) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
<--- UBADevices(0x07)
[EMS] Detected new EMS Device with ID 0x08
Requesting type Version(0x02) from dest 0x08
[EMS] Detected new EMS Device with ID 0x09
Requesting type Version(0x02) from dest 0x09
[EMS] Detected new EMS Device with ID 0x18
Requesting type Version(0x02) from dest 0x18
(04:41:31.032) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D8 00 00 0D 30 48 00 CB 00 00 00 (CRC=77) #data=25
<--- UBAMonitorFast(0x18)
(04:41:31.334) 0x08 -> all, type 0x19, telegram: 08 00 19 00 80 00 02 01 80 00 00 00 00 00 00 AF 44 0C F2 9E 00 00 00 0C 12 02 00 87 DF 00 00 (CRC=B8) #data=27
<--- UBAMonitorSlow(0x19)
(04:41:31.545) 0x18 -> 0x08, type 0x19, telegram: 18 88 19 00 02 (CRC=DB) #data=1
(04:41:31.558) 0x08 -> 0x18, type 0x19, telegram: 08 18 19 00 80 00 (CRC=51) #data=2
(04:41:31.607) 0x08 -> all, type 0x1C, telegram: 08 00 1C 00 91 08 0E 16 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4D) #data=25
(04:41:31.826) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 02 2D 02 2D 21 00 00 03 00 00 E0 9C 00 27 65 00 80 00 (CRC=58) #data=19
<--- UBAMonitorWWMessage(0x34)
(04:41:33.827) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(04:41:33.866) Corrupt telegram: 0B 88 02 00 20 BC 00 08 0B 02 00 73 05 08 00 00 00 00 00 00 (CRC=00) #data=16
(04:41:34.119) 0x08 -> all, type 0x07, telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=5A) #data=13
<--- UBADevices(0x07)
(04:41:34.377) Sending read of type 0x02 to 0x09, telegram: 0B 89 02 00 20 (CRC=B4)
(04:41:34.411) Corrupt telegram: 0B 89 02 00 20 B4 00 09 0B 02 00 72 01 (CRC=03) #data=9
(04:41:35.394) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
<--- UBADevices(0x07)
(04:41:37.127) Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C)
(04:41:37.165) Corrupt telegram: 0B 98 02 00 20 3C 00 18 0B 02 00 CA 02 (CRC=13) #data=9
(04:41:37.419) 0x08 -> all, type 0x07, telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=5A) #data=13
<--- UBADevices(0x07)
(04:41:38.144) 0x08 -> all, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
<--- UBADevices(0x07)
(04:41:40.857) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 BC 00 00 00 00 00 60 01 AB 02 2D 01 D8 00 00 0D 30 48 00 CB 00 00 00 (CRC=77) #data=25
<--- UBAMonitorFast(0x18)
(04:41:41.076) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 02 2D 02 2D 21 00 00 03 00 00 E0 9C 00 27 65 00 80 00 (CRC=58) #data=19
<--- UBAMonitorWWMessage(0x34)
[... cut ...]
<--- UBAMonitorFast(0x18)
(04:42:11.026) 0x08 -> all, type 0x34, telegram: 08 00 34 00 38 02 2D 02 2D 21 00 00 03 00 00 E0 9C 00 27 65 00 80 00 (CRC=58) #data=19
<--- UBAMonitorWWMessage(0x34)
(04:42:12.545) 0x18 -> 0x08, type 0x18, telegram: 18 88 18 00 18 (CRC=C5) #data=1
(04:42:12.581) 0x08 -> 0x18, type 0x18, telegram: 08 18 18 00 07 01 BB 00 00 00 00 00 60 01 AB 02 2D 01 D8 00 00 0D 30 48 00 CB 00 00 (CRC=02) #data=24
(04:42:20.757) 0x08 -> all, type 0x18, telegram: 08 00 18 00 07 01 BB 00 00 00

proddy, 2019-10-09:
Did you try different tx_mode settings with 1.9.1 or 1.9.2? I couldn’t see
from the examples below.

Also you’re sure this worked with 1.8.3? If so which tx_mode was used in
that version ?

Thanks

oli747, 2019-10-09:
I thought the 1.9 branch does no longer have a tx_mode setting? (You said so somewhere above?) The log posted above was with 1.9.2b3.

tx_mode 1 on release 1.8.3 works, and has been stable for several weeks (logging every second to influxdb)

proddy, 2019-10-09:
Tx_mode was added back as there were still issues with Junkers Heatronics3.

Tx_mode 1 on 1.8.3 is the same as tx_mode 2 now on 1.9.x. I know. I confuse
myself and others.

Your influxdb setup is interesting. Are you using HA or going direct from
the mqtt broker ?

oli747, 2019-10-10:
tx_mode 2 works with 1.9.2b3 !!

do you need any additional logging?

proddy, 2019-10-10:
awesome! thanks so much for checking. It was a bug in 1.9.0 when I decided to change "tx_mode 2" to the default "tx_mode 1" and forgot to change it one place. 

One last thing, what does `devices` show you? I'm interested in the product ID and version for your Nefit Topline Aquapower CW5 so I can add it the database.

oli747, 2019-10-10:
devices

These device IDs are on the EMS Bus: 0x08 0x09 0x18
and 3 were recognized by EMS-ESP as:
 Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.08)
 BC10 Base Controller (DeviceID:0x09 ProductID:114 Version:01.03)
 Logamatic TC100/Nefit Moduline Easy (DeviceID:0x18 ProductID:202 Version:02.19)

oli747, 2019-10-10:
b.t.w. about the influxdq setup: I wrote a mqtt gateway for influxdb, that also reduces the number of entries in the database for fields that did not change. For every mqtt field it has a conversion to influxdb measurement, and a setting how much change is needed for a new value in influxdb.

proddy, 2019-10-10:
right, nice. I'm using influxdb with gafana too but letting Home Assistance populate it. Not as efficient as your setup.

BTW I'm working on changing the thermostat topics for the publishing data since with the support for multiple Heating Circuits the ESP is running out of TCP stack. Hope it doesn't break your config too much.

If you feel the issue can be closed (CRC error related) please do...

closed 2019-10-10 by oli747