emsesp/EMS-ESP #177: 1.9.x node build error

SpaceTeddy, 2019-08-30:
Dear Paul,

I'm in trouble to build the dev 1.9.x versions, because of some errors with the new webinterface or node ...
hopefully you may help me to get my tool chain running again, becasue the new web interface looks really fantastic :)
my setup, here you are:
- macOS Mojave
- VS Code with Platform IO Home 2.3.2·Core 4.0.4a1
- Node.js v12.9.1 installed via Homebrew
- followed your 
npm install --global gulp-cli
   npm install gulp
instruction
- error messages:
```
gulp
fs.js:27
const { Math, Object } = primordials;
                         ^

ReferenceError: primordials is not defined
    at fs.js:27:26
    at req_ (/Users/Christian/Documents/development/Arduino/EMS-ESP-dev-1.9.0b4/tools/webfilesbuilder/node_modules/natives/index.js:143:24)
    at Object.req [as require] (/Users/Christian/Documents/development/Arduino/EMS-ESP-dev-1.9.0b4/tools/webfilesbuilder/node_modules/natives/index.js:55:10)
    at Object.<anonymous> (/Users/Christian/Documents/development/Arduino/EMS-ESP-dev-1.9.0b4/tools/webfilesbuilder/node_modules/graceful-fs/fs.js:1:37)
    at Module._compile (internal/modules/cjs/loader.js:936:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:947:10)
    at Module.load (internal/modules/cjs/loader.js:790:32)
    at Function.Module._load (internal/modules/cjs/loader.js:703:12)
    at Module.require (internal/modules/cjs/loader.js:830:19)
    at require (internal/modules/cjs/helpers.js:68:18)
```

any idea??

many thanks

proddy, 2019-08-31:
Download node from the URL in gave in the ChangeLog which will install version 10.16 (https://nodejs.org/dist/v10.16.3/node-v10.16.3.pkg) and see if that helps.

SpaceTeddy, 2019-09-01:
hi paul,
thanks for the hint. After installing node@10 via homebrew it works!

```
brew install node@10
```

closed 2019-09-01 by SpaceTeddy