emsesp/EMS-ESP #178: Cannot find module 'strip-ansi'

Gert05, 2019-09-04:
**Question**
Trying to upgrade from 1.8.2 to 1.9.0
Stuck with a "Cannot find module 'strip-ansi'" error.

<cut from PlatformIO terminal>
Processing debug (board: d1_mini_pro; framework: arduino; platform: espressif8266)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
node ./tools/webfilesbuilder/node_modules/gulp/bin/gulp.js --cwd ./tools/webfilesbuilder
internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module 'strip-ansi'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (c:\Users\gert\Documents\PlatformIO\Projects\EMS-ESP-master-190\tools\webfilesbuilder\node_modules\chalk\index.js:4:17)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
*** Error 1

Have installed nodejs and gulp. the command "node .\node_modules\gulp\bin\gulp.js" gives a similar error...

Any suggestions?

proddy, 2019-09-04:
check which version of npm you have (that comes installed with node).
```
C:\dev\EMS-ESP> node --version
v10.16.0
PS C:\dev\EMS-ESP> npm --version
6.10.2
```

if it fails, go the webfilesbuilder directory, remove the node_modules folder and `npm install` again

Gert05, 2019-09-05:
Thanks for the quick response. I have nodejs v10.16.3 and npm 6.9.0.
Did an `npm install `from the webfilesbuilder directory and now it works... Not sure what went wrong, but not complaining. Case closed.

closed 2019-09-05 by Gert05

Gert05, 2019-09-05:
The new web interface is looking good, by the way. Keep up the good work!