emsesp/EMS-ESP #179: Update - OTA

majdzik84, 2019-09-05:
Hi. I found information about an update via OTA but I don't know where to enter these commands?
Can someone guide me on how to update to version 1.9.0 because I wouldn't want to make a mistake?
I'm using win10.
```
For upgrading via OTA (Over The Air):

wget https://raw.githubusercontent.com/esp8266/Arduino/master/tools/espota.py
wget https://github.com/proddy/EMS-ESP/releases/latest/download/firmware_*.bin
python espota.py -i ems-esp -f firmware_*.bin
```

proddy, 2019-09-05:
If you want to go OTA then you either need PlatformIO installed (at least the command line CLI version) or use the espota.py tool as you mentioned. In both cases you will need Python installed on your windows box. Its not too complex and you wont break anything, just try it out!

BTW In version 1.9 you can now upload firmware's via the new web interface.

majdzik84, 2019-09-05:
thank you very much.

closed 2019-09-07 by proddy