emsesp/EMS-ESP #180: Change from 1.8.3 to 1.9.0 EMS-BUS issue, devices not more well identified

bstones002, 2019-09-05:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [doc](https://github.com/proddy/EMS-ESP/blob/master/README.md)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session with the logging set to Verbose mode.*

**Additional context**
*Add any other context about the problem here.*

bstones002, 2019-09-05:
Hi,
i try to change from 1.8.3 to 1.9.0, using the pre compiled binary. After beginning to do the necessary settings the system get in trouble.
The EMS Bus Connection is not really working. The device scan do not werk well ... 
Use a Buderus GB162, RC30 with the BBQKees' EMS Gateway board which work well before update.
 It look like that the as deactivate set Serial mode will actuated. The TX-Mode selection setting maybe not will.

"EMS Bus is disabled when in Serial mode. Check Settings->General Settings->Serial Port"

I am not know how i can debug it.

My some one can help me how to proceed forward.

Best regards

bstone

proddy, 2019-09-05:
Hi @bstones002, some things to try:

- in the web interface make sure Serial is disabled (in Settings->General Settings)
- set the Tx mode to 1
- reboot

if it still doesn't not detected anything go to the telnet and do a `info` and post to this issue.

bstones002, 2019-09-05:
Hi proddy, 
i have done it again and it is working fine.
But  before it failed . I will monitor this and come back if necessary.

best regards and many thanks for the fast response

bstones002, 2019-09-05:
Hi proddy,

a short time fun, then the EMS-BUS ist lost. It look like the application crashed..
After a reboot it works a short time.
[image]
[image]

bstones002, 2019-09-05:
Error Web interface
[image]

proddy, 2019-09-05:
Are you using a genuine Wemos D1 or a clone by any chance? There have been
reports of crashes with esp8266s from DO-IT.

Also noticed your thermostat isn't recognized?

bstones002, 2019-09-05:
I am use an original full kit board 0.9 from Kees, actual version (a few week old).
Was send with 1.8.x and work since 2-3 weeks well.  (I did not can go back, i did have the binary)

proddy, 2019-09-05:
ah, ok. must be something else then. When you get a connection, from telnet do a `system` and see what the memory consumption is like.

bstones002, 2019-09-05:
system
[image]

proddy, 2019-09-05:
everything looks fine. And you're still getting regular restarts?

bstones002, 2019-09-05:
Not a regular restarts,  I look more that the EMS-BUS will lost or some thing is blocking... maybe the program is fixt in a error loop ...  but then all connections (Mqtt,TCP or Http)  lost .
I need a reboot.

bstones002, 2019-09-05:
I am using it with a Home Assistant Raspberry Pi. 
Actual the HA ist down and its look like the system run well.
That it can be an MQTT based issue?

proddy, 2019-09-05:
EMS-ESP is just sending out MQTTs so doesn't really matter if anything is listening on the other end. You could go to telnet (or web) and turn off both the heartbeat and mqtt and see if that helps with the stability.

```
set mqtt_heartbeat off
set mqtt_enabled off
```

bstones002, 2019-09-05:
it is running with out break since 75 min.

proddy, 2019-09-06:
ok, leave mqtt_heartbeat off and re-enable the mqtt (mqtt_enabled) and see if it still works. Just so we can try and isolate the problem.

lmdecroos, 2019-09-07:
Hi proddy,

I have also tried the 1.9.0 binary build ( upgraded from 1.8.3) and have also issues with the stability of the EMS gateway.

It seems the new version has issue communicating to the EMS bus, sometimes it find my Buderus boiler other times it doesn't.
When accessing the web or terminal interface, the gateway becomes unstable.
The EMS CPU use is 0% or 100% when running the 1.9.0 version, when using version 1.8.3 it's around 2 a 5 %

Devices as scanned by the gateway:
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
* Setting Boiler to model Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
Device found: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
* Setting Thermostat to RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
Device found: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)

System details :
[SYSTEM] System is Stable
[SYSTEM] Board: PLATFORMIO_D1_MINI
[SYSTEM] CPU frequency: 80 MHz
[SYSTEM] SDK version: 2.2.1(cfd48f3)
[SYSTEM] CPU chip ID: 0x5E969C
[SYSTEM] Core version: 2_5_2
[SYSTEM] Boot version: 6
[SYSTEM] Boot mode: 1
[SYSTEM] Last reset reason: Exception
[SYSTEM] Last reset info: Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4000bf80 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000018 depc:0x00000000
[SYSTEM] Restart count: 8
[SYSTEM] rtcmem status: blocks:2 addr:0x60001280
[SYSTEM] rtcmem 00: 1163087990
[SYSTEM] rtcmem 01: 65544
[FLASH] Flash chip ID: 0x16400E
[FLASH] Flash speed: 40000000 Hz
[FLASH] Flash mode: DIO
[FLASH] Flash size (CHIP): 4194304
[FLASH] Flash size (SDK): 4194304
[FLASH] Flash Reserved: 4096
[MEM] Firmware size: 620544
[MEM] Max OTA size: 2519040
[MEM] OTA Reserved: 16384
[MEM] Free Heap: 26328 bytes initially | 9712 bytes used (36%) | 16616 bytes free (63%)

I'm using a BBQKees gateway, who is working stable with version 1.8.3

When trying to get the logs in the web GUI, the process hangs on step 7 of 8.

I have downgraded again to version 1.8.3, but can upgrade again to version 1.9.3 to do some more testing or collect logs

Thanks in advance

bstones002, 2019-09-07:
Hi,
this look like the same i have had after the update.
Now since 34 h it looks stable, i did not change settings. But set all 
step by step and boot after each setting window (CPU load 1% now).
But i have an issue that my RC30 will not show.

Best regards

bstones002, 2019-09-07:
Hi. 
no change, but after 35h system lost EMS-BUS .
After Reboot:
- System did not found devices
next reboot:
-EMS Bus is disabled when in Serial mode. Check Settings->General Settings->Serial Port
I did not do more then actuated a reboot from the WEB-Interface

Can a  search at EMS-BUS at a running system actuated manualy ?
After set serial Port ON and switch back to OFF, after reboot scan looks better, but system drop after 2 min and Telnet , EMS lost.
All try to do a scan etc. via Telnet end after 2-5 min with a crash.
Looks for me, that the communication to bekees board will lost, and error handling is killing.

proddy, 2019-09-07:
how do you know if the EMS bus is lost? Does the LED start flashing? If the system crashes more than 10 times each within 60 seconds of booting then it will automatically go into Serial mode so the user can debug using USB. I think this is what is happening on your system. You can see the # restarts in the 'system' command via Telnet.

I don't know why your system is unstable. I'll do so more testing here but it'll be a week or two before I do any deeper testing as I'm out of the country for work

lmdecroos, 2019-09-07:
I had a similar experience when i was testing the new version yesterday and this morning.
the telnet connection dropped freqently or froze whereof i had to do hard reboot

proddy, 2019-09-07:
and you should contact bbqkees to see if he has any ideas.

bstones002, 2019-09-07:
I will contact bbqkees again.

After several reconnects it look that the EMS-BUS scan crash. I did get any stable connection since the 35h running periods.

bstones002, 2019-09-07:
If there away to deactivated serial mode
[image]

proddy, 2019-09-07:
it says at the top of the screen shot you senf 'set serial off'

for some reason your system is really unstable and the ESP keeps crashing and restarting. I would take the Wemos D1 out of the bbqkees gateway, plug it into a PC or +5 USB charger, start it up and see if it still crashes. If it does, 'set serial on' and see if then crashes. This will help pinpoint the problem.

Also look at https://github.com/proddy/EMS-ESP/wiki/Troubleshooting

bstones002, 2019-09-07:
Hi, 

i have set the system back to F_183 and will run it the next 10 days.
I also leaving for work some days.

The 1.8.3 came up  silent and looks well ...
Only my  Thermostat did not found by search.
I set it to  0x10 what a default for RC30. 
I get now values:
Setpoint room temperature: 0.0 C ?
 (Heat control by outtemp //  floor heating  with separated thermostat)                                                                                        Current room temperature: 26.3 C (RC30 in heating case mounted)                                                                                    
Thermostat time is 16:23:45 7/9/2019   

I will do further investigation if i am back.

Thanks a lot

Thanks a lot and a well time

P.S. 
If i have time at my trip, i will try to take a look  into your source.
It's a long time ago if i have coded for hardware ...
"I am interested to stepp into study's, for my privat addition heating projects"

majdzik84, 2019-09-09:
I also get RC35 setpoint temperature 0.0 C in auto mode (1.8.3)
[image]

https://github.com/proddy/EMS-ESP/issues/172#issuecomment-522350476

proddy, 2019-09-09:
the 0.0 problem I believe I fixed in 1.9.x for the RC3000. Just sucks that it doesn't work with you guys!

majdzik84, 2019-09-09:
I just updated to 1.9.0. Unfortunately, setpoint temperature in auto mode is still 0.0 C :(
[image]

proddy, 2019-09-09:
oh, then its something else. Can you capture a few telegrams of type 0x3E - I'm interesting if any of them are longer than 12 data bytes

majdzik84, 2019-09-09:
```
(02:50:39.338) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(02:52:40.086) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(02:54:40.709) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(02:55:41.208) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(02:56:41.457) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16

** [DEBUG MODE] Warning, we missed the bus - Rx non-idle!
(00:02:09.678) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 09 11 0A 26 05 01 01 (CRC=2F) #data=8
(00:02:09.948) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
```

bbqkees, 2019-09-11:
I have been running the 1.9.0 stable on a new Gateway in parasitic mode (bus powered) for 2 days now without problems.
I upload the firmware via Platform IO to the empty Wemos chips via USB before they are mounted on the Gateway board. 

Only issue I noticed last week when installing 1.9.0 and then testing them is that TX is not active until you set TX mode (in Custom settings).
When upgrading from 1.9.0b4 and b5 to 1.9.0 the TX mode selector is empty by default, but TX works anyway.

proddy, 2019-09-11:
The tx_mode issue is fixed in 1.9.0. I didn’t bump the version.

proddy, 2019-09-14:
@majdzik84 in https://github.com/proddy/EMS-ESP/releases/tag/1.9.0

majdzik84, 2019-09-14:
@proddy  thank you very much. Unfortunately, this file does not upload to me. do you know why this is happening? The process is completed correctly but I still have version 1.9.0 :(