emsesp/EMS-ESP #181: Using MQTT to set boiler flow temp

higgers, 2019-09-13:
Hi Proddy!

I had some time tonight to do some investigation into setting the heating flow temp.  I've got a Bosch Easy Control CT200 and when weather compensation is turned on it sets the heating flow temp based on the room set point.  This means that setting the room set point to 30 degrees centrigrade results in a heating flow temp of 53 degrees, a room set point of 28 degrees results in a heating flow temp of 49 degrees, etc.  I set logging to thermostat only and grabbed the following telegrams:

```
Thermostat ->   Boiler, type 0x1A, telegram: 98 08 1A 00 35 4C (CRC=D4) #data=2 | 53
-- | --
Thermostat -> Boiler, type 0x1A,   telegram: 98 08 1A 00 31 4C (CRC=DC) #data=2 | 49
Thermostat -> Boiler, type 0x1A,   telegram: 98 08 1A 00 2B 4C (CRC=E8) #data=2 | 43
Thermostat -> Boiler, type 0x1A,   telegram: 98 08 1A 00 19 4C (CRC=8C) #data=2 | 25
Thermostat -> Boiler, type 0x1A,   telegram: 98 08 1A 00 14 4C (CRC=96) #data=2 | 20
Thermostat -> Boiler, type 0x1A,   telegram: 98 08 1A 00 00 00 (CRC=F2) #data=2 | 0

```

The fifth byte is the flow temp.  Awesomeballs!

I then noticed that it's possible to set the boiler temp from a telnet session with the "boiler flowtemp" command.  This works some of the time, the log quite often shows messages that the program has attempted it 2 out of 2 times but it's giving up and removing the message from the queue.  Is there a way to increase the number of retries?

Since I use openHAB as my HA system I could always write a rule that compares the selected flow temp from the boiler with the flow temp sent from openHAB and if they're different send an MQTT message.  Effectively building the retries into openHAB, I'm totally fine with that.

However, I can't seem to get any MQTT messages to be received by the ESP software and I'm fairly sure this is a fault with the way I have openHAB set up.  What I did want to check is if I should expect to see a message in the log when the software receives an MQTT message?  I'm not seeing any and if I knew that I should see some then that would confirm the issue is with my openHAB config.

Thanks! :)

higgers, 2019-09-13:
Can I check that this is the correct command topic:

`home/ems-esp/boiler_data/boiler_cmd_flowtemp`

higgers, 2019-09-13:
Ignore me, I'm a fool!  This is the correct topic:

`home/ems-esp/boiler_cmd_flowtemp`

And it seems to work!  Proddy and co, I could kiss you!!!!!  How can I send you a token of my appreciation? :)

proddy, 2019-09-14:
@higgers that's very kind of you. Your appreciation makes it all worthwhile. Just keep on suggesting new features and if you get chance provide some feedback on the latest 1.9.x builds. BTW I've extended the command `mqttlog` to also show which MQTT topics its currently subscribed too as this was not very intuitive.

higgers, 2019-09-17:
Could you post a link to the precompiled .bin file please?  I'm having the same compile issues I had when I last tried to compile 1.9.0b3 and raised issue #173 .

closed 2019-09-22 by proddy