emsesp/EMS-ESP #182: V1.9.x craches when using EMS protocol while scanning bus at startup

lmdecroos, 2019-09-15:
With version 1.8.3 my RC35 thermostat was recognized by using the ems protocol ( tx_mode 1)
With version 1.9.0 and 1.9.1b the ems protocol craches  the gateway :

System Logging set to None
System Logging set to Verbose
(00:00:05.704) Sending read of type 0x02 to 0x38, telegram: 0B B8 02 00 20 (CRC=25)
(00:00:06.476) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 72 02 72 21 00 00 03 00 00 98 A7 00 15 3C 00 80 00 (CRC=4A) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:06.706) Boiler -> all, type 0x18, telegram: 08 00 18 00 3D 02 89 72 00 00 00 64 60 02 67 02 72 02 82 00 00 0D 30 41 01 31 00 01 00 (CRC=A4) #data=25
<--- UBAMonitorFast(0x18)
(00:00:06.949) Sending read of type 0x02 to 0x02, telegram: 0B 82 02 00 20 (CRC=EC)
(00:00:08.112) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 03 00 00 00 00 00 00 00 00 00 00 00 (CRC=53) #data=13
(00:00:08.518) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(00:00:08.558) Boiler -> me, type 0x02, telegram: 08 0B 02 00 73 05 05 00 00 00 00 00 00 00 (CRC=D7) #data=10
<--- Version(0x02)
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
(00:00:08.636) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 03 00 00 00 00 00 00 00 00 00 00 00 (CRC=7F) #data=13
(00:00:08.892) Sending read of type 0x02 to 0x09, telegram: 0B 89 02 00 20 (CRC=B4)
(00:00:08.930) 0x09 -> me, type 0x02, telegram: 09 0B 02 00 44 02 03 (CRC=25) #data=3
<--- Version(0x02)
Device found: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
(00:00:09.240) Sending read of type 0x02 to 0x10, telegram: 0B 90 02 00 20 (CRC=7C)
(00:00:09.279) 0x10 -> me, type 0x02, telegram: 10 0B 02 00 56 15 08 (CRC=5E) #data=3
<--- Version(0x02)
Thermostat found: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
* Setting Thermostat to RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
Requesting type RC35StatusMessage_HC1(0x3E) from dest 0x10
Requesting type RC35Set_HC1(0x3D) from dest 0x10
Requesting type RC35StatusMessage_HC2(0x48) from dest 0x10
Requesting type RC35Set_HC2(0x47) from dest 0x10
Requesting type RC35StatusMessage_HC3(0x52) from dest 0x10
Requesting type RC35Set_HC2(0x51) from dest 0x10
Requesting type RC35StatusMessage_HC4(0x5C) from dest 0x10
Requesting type RC35Set_HC4(0x5B) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
(00:00:09.539) Sending read of type 0x02 to 0x11, telegram: 0B 91 02 00 20 (CRC=74)
(00:00:09.584) 0x11 -> me, type 0x02, telegram: 11 0B 02 00 47 02 00 (CRC=7C) #data=3
<--- Version(0x02)
Device found: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
(00:00:09.813) Sending read of type 0x02 to 0x17, telegram: 0B 97 02 00 20 (CRC=44)
(00:00:10.417) Boiler -> all, type 0x18, telegram: 08 00 18 00 07 02 88 00 00 00 00 40 60 02 69 02 73 02 81 00 00 0A 30 48 00 CB 00 00 00 (CRC=25) #data=25
<--- UBAMonitorFast(0x18)
(00:00:10.677) 0x11 -> Boiler, type 0x1E, telegram: 11 08 1E 00 00 CD (CRC=AF) #data=2
(00:00:10.984) Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C)

The gateway hangs every time on : Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C) and reboots itself afterwards

When using the EMS+ protocol (tx_mode 2) the gateway works stable, but doesn’t recognize my RC35 thermostat 
the EMS+ protocol recognizes only my boiler ( same for version 1.8.3)

lmdecroos, 2019-09-15:
please find the jabber log until the crash :

(00:00:08.141) ems_parseTelegram: 00 01: 8B 
(00:00:08.142) emsuart_tx_buffer: 00 01: 0B 82 02 00 20 EC 
(00:00:08.158) ems_parseTelegram: 00 02: 0B 82 02 00 20 EC 
(00:00:08.158) echo:telegram: 0B 82 02 00 20 (CRC=EC) #data=1
(00:00:08.390) ems_parseTelegram: 00 02: 90 
(00:00:08.395) ems_parseTelegram: 00 02: 10 
(00:00:08.415) ems_parseTelegram: 00 02: 89 
(00:00:08.425) ems_parseTelegram: 00 02: 09 
(00:00:08.440) ems_parseTelegram: 00 02: 8D 
(00:00:08.465) ems_parseTelegram: 00 02: 89 
(00:00:08.472) ems_parseTelegram: 00 02: 09 
(00:00:08.490) ems_parseTelegram: 00 02: 8E 
(00:00:08.515) ems_parseTelegram: 00 02: 91 
(00:00:08.519) ems_parseTelegram: 00 02: 11 
(00:00:08.540) ems_parseTelegram: 00 02: 89 
(00:00:08.549) ems_parseTelegram: 00 02: 09 
(00:00:08.565) ems_parseTelegram: 00 02: 99 
(00:00:08.569) ems_parseTelegram: 00 02: 19 
(00:00:08.590) ems_parseTelegram: 00 02: 8F 
(00:00:08.615) ems_parseTelegram: 00 02: A1 
(00:00:08.619) ems_parseTelegram: 00 02: 21 
(00:00:08.639) ems_parseTelegram: 00 02: 92 
(00:00:08.664) ems_parseTelegram: 00 02: 89 
(00:00:08.672) ems_parseTelegram: 00 02: 09 
(00:00:08.689) ems_parseTelegram: 00 02: 89 
(00:00:08.698) ems_parseTelegram: 00 02: 09 
(00:00:08.714) ems_parseTelegram: 00 02: 89 
(00:00:08.724) ems_parseTelegram: 00 02: 09 
(00:00:08.739) ems_parseTelegram: 00 02: 89 
(00:00:08.749) ems_parseTelegram: 00 02: 09 
(00:00:08.764) ems_parseTelegram: 00 02: 8B 
(00:00:08.789) ems_parseTelegram: 00 02: 90 
(00:00:08.794) ems_parseTelegram: 00 02: 10 
(00:00:08.814) ems_parseTelegram: 00 02: 89 
(00:00:08.821) ems_parseTelegram: 00 02: 09 
(00:00:08.839) ems_parseTelegram: 00 02: 93 
(00:00:08.864) ems_parseTelegram: 00 02: 89 
(00:00:08.873) ems_parseTelegram: 00 02: 09 
(00:00:08.888) ems_parseTelegram: 00 02: 94 
(00:00:08.913) ems_parseTelegram: 00 02: 91 
(00:00:08.918) ems_parseTelegram: 00 02: 11 
(00:00:08.938) ems_parseTelegram: 00 02: 89 
(00:00:08.945) ems_parseTelegram: 00 02: 09 
(00:00:08.964) ems_parseTelegram: 00 02: 99 
(00:00:08.968) ems_parseTelegram: 00 02: 19 
(00:00:08.988) ems_parseTelegram: 00 02: 95 
(00:00:09.013) ems_parseTelegram: 00 02: A1 
(00:00:09.017) ems_parseTelegram: 00 02: 21 
(00:00:09.038) ems_parseTelegram: 00 02: 96 
(00:00:09.063) ems_parseTelegram: 00 02: 89 
(00:00:09.073) ems_parseTelegram: 00 02: 09 
(00:00:09.088) ems_parseTelegram: 00 02: 89 
(00:00:09.094) ems_parseTelegram: 00 02: 09 
(00:00:09.113) ems_parseTelegram: 00 02: 89 
(00:00:09.120) ems_parseTelegram: 00 02: 09 
(00:00:09.138) ems_parseTelegram: 00 02: 89 
(00:00:09.145) ems_parseTelegram: 00 02: 09 
(00:00:09.163) ems_parseTelegram: 00 02: 8B 
(00:00:09.188) ems_parseTelegram: 00 02: 90 
(00:00:09.192) ems_parseTelegram: 00 02: 10 
(00:00:09.212) ems_parseTelegram: 00 02: 89 
(00:00:09.222) ems_parseTelegram: 00 02: 09 
(00:00:09.237) ems_parseTelegram: 00 02: 97 
(00:00:09.262) ems_parseTelegram: 00 02: 89 
(00:00:09.269) ems_parseTelegram: 00 02: 09 
(00:00:09.287) ems_parseTelegram: 00 02: 9C 
(00:00:09.312) ems_parseTelegram: 00 02: 91 
(00:00:09.316) ems_parseTelegram: 00 02: 11 
[... 221 lines cut ...]
(00:00:13.402) ems_parseTelegram: 00 01: 18 
(00:00:13.447) ems_parseTelegram: 00 01: 08 00 34 00 3D 02 45 02 45 21 00 00 03 00 00 98 EC 00 15 49 00 80 00 D4 
(00:00:13.448) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 45 02 45 21 00 00 03 00 00 98 EC 00 15 49 00 80 00 (CRC=D4) #data=19
(00:00:13.646) ems_parseTelegram: 00 01: B8 
(00:00:13.671) ems_parseTelegram: 00 01: 89 
(00:00:13.677) ems_parseTelegram: 00 01: 09 
(00:00:13.696) ems_parseTelegram: 00 01: C0 
(00:00:13.721) ems_parseTelegram: 00 01: 99 
(00:00:13.725) ems_parseTelegram: 00 01: 19 
(00:00:13.745) ems_parseTelegram: 00 01: 89 
(00:00:13.754) ems_parseTelegram: 00 01: 09 
(00:00:13.771) ems_parseTelegram: 00 01: A1 
(00:00:13.775) ems_parseTelegram: 00 01: 21 
(00:00:13.795) ems_parseTelegram: 00 01: C8 
(00:00:13.820) ems_parseTelegram: 00 01: 89 
(00:00:13.831) ems_parseTelegram: 00 01: 09 
(00:00:13.845) ems_parseTelegram: 00 01: D0 
(00:00:13.870) ems_parseTelegram: 00 01: 89 
(00:00:13.877) ems_parseTelegram: 00 01: 09 
(00:00:13.895) ems_parseTelegram: 00 01: 89 
(00:00:13.903) ems_parseTelegram: 00 01: 09 
(00:00:13.920) ems_parseTelegram: 00 01: 8B 
(00:00:13.920) emsuart_tx_buffer: 00 01: 0B 98 02 00 20 3C 
(00:00:13.935) ems_parseTelegram: 00 02: 0B 98 02 00 20 3C 
(00:00:13.936) echo:telegram: 0B 98 02 00 20 (CRC=3C) #data=1

proddy, 2019-09-16:
could you try the latest dev/1.9.1b3 and per the Wiki Troubleshooting try

- turn off MQTT
- turn off Tx (listen_mode)
- run in serial mode
- don't start the web

this will help pin point what is causing it

lmdecroos, 2019-09-17:
unfortunately no progress :

* EMS-ESP version 1.9.1b3
[FS] Checking file /myesp.json, Size: 333 bytes
[FS] System config loaded
[FS] Checking file /customconfig.json, Size: 202 bytes
[FS] Custom config loaded
[FS] Checking file /eventlog.json, Size: 2321 bytes (max is 20176)
[FS] Event log is healthy
[WEB] Web server started
[UART] Opened Rx/Tx connection
Starting auto discover of EMS devices...
Started scan on EMS bus for known devices
[WIFI] Connecting to SSID: lmdnet
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
* Setting Boiler to model Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
[WIFI] SSID 
[WIFI] CH    8
[WIFI] RSSI  -61
[WIFI] IP    
[WIFI] MAC   
[WIFI] GW    
[WIFI] MASK  
[WIFI] DNS   
[WIFI] HOST  ems-esp
[OTA] listening to ems-esp.local:8266
[MQTT] is disabled
[SYSTEM] Disabling serial port communication.
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
Device found: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
[TELNET] Telnet connection established

* Connected to: EMS-ESP version 1.9.1b3
* Hostname: ems-esp (192.168.0.121)
* WiFi SSID: lmdnet (signal 82%)
* MQTT disconnected

System Logging set to Verbose
(00:00:08.895) 0x10 -> all, type 0x06, telegram: 10 00 06 00 13 09 09 11 0C 13 01 01 (CRC=BC) #data=8
<--- RCTime(0x06)
(00:00:09.466) Sending read of type 0x02 to 0x02, telegram: 0B 82 02 00 20 (CRC=EC)
(00:00:10.414) 0x10 -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 00 CB 00 00 00 00 00 00 00 00 00 05 04 (CRC=7D) #data=16
<--- RC35StatusMessage_HC1(0x3E)
(00:00:10.985) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(00:00:11.024) Boiler -> me, type 0x02, telegram: 08 0B 02 00 73 05 05 00 00 00 00 00 00 00 (CRC=D7) #data=10
<--- Version(0x02)
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
(00:00:11.458) Sending read of type 0x02 to 0x09, telegram: 0B 89 02 00 20 (CRC=B4)
(00:00:11.495) 0x09 -> me, type 0x02, telegram: 09 0B 02 00 44 02 03 (CRC=25) #data=3
<--- Version(0x02)
Device found: BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
(00:00:11.957) Sending read of type 0x02 to 0x10, telegram: 0B 90 02 00 20 (CRC=7C)
(00:00:12.008) 0x10 -> me, type 0x02, telegram: 10 0B 02 00 56 15 08 (CRC=5E) #data=3
<--- Version(0x02)
Thermostat found: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
* Setting Thermostat to RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
Requesting type RC35StatusMessage_HC1(0x3E) from dest 0x10
Requesting type RC35Set_HC1(0x3D) from dest 0x10
Requesting type RC35StatusMessage_HC2(0x48) from dest 0x10
Requesting type RC35Set_HC2(0x47) from dest 0x10
Requesting type RC35StatusMessage_HC3(0x52) from dest 0x10
Requesting type RC35Set_HC2(0x51) from dest 0x10
Requesting type RC35StatusMessage_HC4(0x5C) from dest 0x10
Requesting type RC35Set_HC4(0x5B) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
(00:00:12.430) Sending read of type 0x02 to 0x11, telegram: 0B 91 02 00 20 (CRC=74)
(00:00:12.531) 0x11 -> me, type 0x02, telegram: 11 0B 02 00 47 02 00 (CRC=7C) #data=3
<--- Version(0x02)
Device found: WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
(00:00:12.977) Sending read of type 0x02 to 0x17, telegram: 0B 97 02 00 20 (CRC=44)
(00:00:14.321) 0x11 -> Boiler, type 0x1E, telegram: 11 08 1E 00 00 CE (CRC=AC) #data=2
(00:00:14.746) Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C)

log j ( from another session)
(00:00:16.735) ems_parseTelegram: 00 01: 8B 
(00:00:16.735) emsuart_tx_buffer: 00 01: 0B 98 02 00 20 3C 
(00:00:16.751) ems_parseTelegram: 00 02: 0B 98 02 00 20 3C 
(00:00:16.751) echo:telegram: 0B 98 02 00 20 (CRC=3C) #data=1

proddy, 2019-09-17:
did you try all the recommendations in the Troubleshooting guide?
For example do this:

- make sure you are using a genuine Wemos D1 (no clone)
- go into Telnet
- set led on
- set mqtt_enabled off
- set listen_mode on
- restart
- go into Telnet (but stop turning on verbose logging)
- does it crash? You can tell by looking at the LEDs and also 'system' will show you the uptime. If so after how many seconds before each reboot?
- if it looks fine, turn on mqtt_enabled are try again
- if it looks fine, turn on listen_mode and try again

When it crashes you may want to enable serial mode and capture the stack dump. This is all described in the wiki too

lmdecroos, 2019-09-17:
I’m using a bbqkees EMS gateway with genuine Wemos D1

As long as the gateway is in listen only is, the gateway is stable.
Once I’m setting “set listen mode off” and the gateway start sending messages then it crashes after about  10 -15 seconds.
The gateway crashes always after "Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C)”
this happens around 15 seconds after a restart of the gateway.
The gateway reboots automatically and crashes again after 10 - 15 seconds
I had a look at my boiler and spotted that my RFM20 ( wireless remote module) is also crashing after a 3 reboots of the gateway.
On my boiler  I get a “A32” error message.

So the EMS gateway send an unsupported / incorrect request  to the EMS bus and this cause a crash of the EMS gateway and EMS bus

Can you explain the 0B 98 02 00 20  command ?

proddy, 2019-09-17:
`send 0B 98 02 00 20` sends a version command (0x02) request to the device with device id (0x18) which is typically an RC35 thermostat. This is part of the autodetect startup procedure to detect which devices are on the EMS bus. It goes through all the know devices as defined in `ems_devices.h`. It's all fairly harmless since its just a query and I'm surprised it causes the ESP to reset itself. 

We can try and omit these checks from the code and see if helps. I'll tell you what to change but you'll need to build the firmware yourself. Or try issuing the send command yourself quickly before it restarts after 15 seconds?

I looked up the error messages on your Buderus GB162 and couldn't find the A32?

lmdecroos, 2019-09-17:
when is do a manual send of “ 0B 98 02 00 20 “ it hangs the EMS-gateway.
it’s easy to reproduce on version 1.8.3 and on version 1.9.x when selected the ems+ protocol

Version 1.8.3 is running very stable, therefore i assume that "0B 98 02 00 20 “ is new since version 1.9.0

yes, please let me know where i have to comment stuff in the source code and I will troubleshoot

The error code is from the MM10 module http://nl.documents2.nefit.nl/download/pdf/file/63037183.pdf <http://nl.documents2.nefit.nl/download/pdf/file/63037183.pdf>  see page 6
Sorry i found only a dutch manual

proddy, 2019-09-17:
That's really strange. I can't reproduce with 1.8.3 or the latest 1.9.1b in both tx_mode 1 and 2. I'll check with others to see if they can also reproduce.

To skip checking on 0x18,  go to `ems.cpp` and in function `ems_scanDevices()` find the line `    Device_Ids.remove(EMS_MODEL_NONE);` around line 2382, and add this afterwards:

```
Device_Ids.remove(0x18);
```

proddy, 2019-09-17:
also why are you using the EMS+ protocol (tx_mode 2). Does it still crash with tx_mode 1 ?

lmdecroos, 2019-09-18:
The A32 error is an error in communication : lost communication with the MM10 or wrong address between RC30 and MM10
This is in line with my findings of a EMS bus crash due to a wrong instruction

During troubleshooting version 1.9.x I found that when using the EMS+ protocol the gateway was stable with limited functionality.
As when using tx_mode 1, the gateway restarts every 10-15 seconds, the time to test stuff is very limited.
Therefore I’m switching to tx_mode 2 after each test. This allows me to monitor the gateway and upload ( over the air) new software.
The alternative would be to go my garage and disconnect every time the gateway.

As since yesterday I know that the putting the gateway in listen only mode also works, this is also a possibility to keep the gateway on-line

proddy, 2019-09-18:
I think we're getting closer to the root cause of the problem. If you see the EMS-ESP crashes when using tx_mode 2 with **1.8.3** and tx_mode 1 with both **1.9.0** and **1.9.1** then the issue is most likely in the new Tx code we added. I'm guessing it never ever worked for you then? Sorry! 

I'll figure out a plan with @susisstrolch to see how to troubleshoot it.

susisstrolch, 2019-09-18:
Ok, let's see..
Just tried my 'txmode2' branch from this repo (which is 1.9 before the webserver got implemented). It doesn't show any issues in txmode2 (it's running fixed in txmode2).
```
[163674] ESP8266 System stats:
[163675] 
[163675]  [APP] EMS-ESP version: 1.9.0
[163675]  [APP] MyESP version: 1.1.24
[163676]  [APP] Build timestamp: 2019-07-31 14:56:52
[163676]  [APP] Uptime: 1 days 6 hours 36 minutes 3 seconds
[163676]  [APP] System Load: 2%
```
Issuing the 0B 98 02 00 20 doesn't lead to a system reboot...
```
log j
... <jabber>...
(06:31:36.358) ems_parseTelegram: 00 01: D0 
(06:31:36.389) ems_parseTelegram: 00 01: 89 
send 0B 98 02 00 20
(06:31:36.398) ems_parseTelegram: 00 01: 09 
(06:31:36.403) ems_parseTelegram: 00 01: 90 
(06:31:36.408) ems_parseTelegram: 00 01: 10 
... <jabber>...
(06:31:36.688) ems_parseTelegram: 00 01: 89 
(06:31:36.695) ems_parseTelegram: 00 01: 09 
(06:31:36.698) ems_parseTelegram: 00 01: 8B 
(06:31:36.699) Sending raw: 0B 98 02 00 20 00
(06:31:36.699) emsuart_tx_buffer: 00 01: 0B 98 02 00 20 3C 
(06:31:36.714) ems_parseTelegram: 00 01: 0B 98 02 00 20 3C 
(06:31:36.715) echo:telegram: 0B 98 02 00 20 (CRC=3C) #data=1
(06:31:36.922) ems_parseTelegram: 00 01: 89 
(06:31:36.932) ems_parseTelegram: 00 01: 09 
(06:31:36.937) ems_parseTelegram: 00 01: 8C 
```
As you can see, the telegram is simply ignored from my system, because we don't get an answer at all.
Looking at the timestamp (echo vs. next poll) we see approx. 200ms difference, that means I don't hava a device '18'.

So, to come further, you should do the following (instead of sending the 0B 98...):
```
log r
send 0B 88 02 00 20
send 0B 88 07 00 20
```
and show the results...
The first one will tell about your system (according to the table HT3-Bus_Telegramme.html), the second one will show a bit map of all devices detected by your bus master.

susisstrolch, 2019-09-18:
@proddy, it could be a single BRK send from bus master (or from 0x18) which means something like 'shut up', which isn't handled properly.

proddy, 2019-09-18:
@susisstrolch thanks for jumping in, I can't reproduce either - it seems to open happen on certain boilers (the crash). I'll look in type 0x07 - if it reports back on what is connected to the EMS master then I don't need to ping each one individiually

susisstrolch, 2019-09-18:
It does - delivers a bit mask starting with ID8...

susisstrolch, 2019-09-18:
@lmdecroos according to your log
`(00:00:08.112) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 03 00 00 00 00 00 00 00 00 00 00 00 (CRC=53) #data=13
`
you have the devices
0x08, 0x09, 0x10, 0x11
there's no 0x18...

@proddy don't you show the reboot reason anymore?

proddy, 2019-09-18:
@susisstrolch the reboot reason is commented out for now. it wasn't reliable, like always show exceptions after a human reset or OTA update. needs fixing.

Can you help me understand the 0x07 telegram. How does 
00001011 (x0B) says x08, x09, 0x0B are set right?
And 00000011 (x03) is the next set starting at 0x10 so 0x11 and 0x12 are set?

that's so cool!

susisstrolch, 2019-09-18:
@proddy, thats pretty easy...
The 0x07 payload is a large bitfield with a 1 for each detected device.
It starts with ID8 - busmaster, at LSB.
[4] - IDx08.. IDx0F
[5] - IDx10.. IDx17
[6] - IDx18.. IDx1F
...
[18] - IDx78.. IDx7F
The LSB corresponds always to the lowest ID of the particular byte.

And - if the target doesn't understand the type 0x07 you get something like
```
(11:51:34.614) Sending raw: 0B 90 07 00 20 00
(11:51:34.701) Thermostat -> me, telegram: 10 0B 07 00 (CRC=A2)
```

susisstrolch, 2019-09-18:
It also looks like the busmaster will broadcast the type 0x07 after you query any of the other devices (or if a device doesn't exist?)

proddy, 2019-09-18:
ok got it. I'll work on the implementation. The logic then is
 - on startup, do a 'send 0B 88 07 00 20', to request type 0x07 from the boiler 0x08. All boilers are 0x08 so should be ok, unless someone has just a heatpump
 - add detection for the 0x07. If its the first time this type has been received, create a list of device IDs by looking at the bitmasks on each byte in the data telegram and send out a Version request (type 0x02) to each of them to collect the product ID and version number

this should also speed up the startup considerably.

Still interesting is why the send to an unknown device ID would crash the ESP?

susisstrolch, 2019-09-18:
> Still interesting is why the send to an unknown device ID would crash the ESP?
I guess it's some kind of sideeffect, because the busmaster will always answer with a type 07 broadcast when you send to an unknown/nonexisting device.

```
send 0B C8 02 00 20
(00:17:26.724) Sending raw: 0B C8 02 00 20 0B C8 02 00 20 00
(00:17:27.067) Boiler -> all, type 0x07, telegram: 08 00 07 00 1B 01 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=8A) #data=15
(00:17:27.780) Boiler -> all, type 0x18, telegram: 08 00 18 00 0B 00 CB 64 00 00 00 00 00 83 00 7D 00 80 00 00 00 FF 30 48 00 00 FF 00 00 83 00 (CRC=D3) #data=27
```
You should reactivate the reboot reason, maybe we can catch the real reason...

lmdecroos, 2019-09-18:
I did a few test on version 1.8.3 using tx_mode 0 and this are my logs

log v
System Logging set to Verbose
(00:01:23.383) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 2A 02 2A 21 00 00 03 00 00 98 E1 00 15 50 00 80 00 (CRC=F5) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:01:24.075) 0x21 -> all, type 0xAB, telegram: 21 00 AB 00 00 83 00 00 00 01 00 (CRC=67) #data=7
send 0B 98 02 00 20
(00:01:28.925) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 00 00 00 00 C8 00 00 00 00 00 00 00 00 11 05 04 (CRC=78) #data=16
<--- RC35StatusMessage_HC1(0x3E)
(00:01:29.235) Sending raw: 0B 98 02 00 20 00

Connection closed by foreign host.

send 0B 88 02 00 20
(00:01:16.281) 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B
(00:01:16.462) Sending raw: 0B 88 02 00 20 00
(00:01:16.471) 0B 88 02 00 20 BC
(00:01:16.502) 08 0B 02 00 73 05 05 00 00 00 00 00 00 00 D7
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
(00:01:16.842) 08 00 18 00 07 01 11 00 00 00 00 00 60 01 08 02 2A 01 1B 00 00 0A 30 48 00 CB 00 00 00 91
(00:01:17.069) 08 00 19 00 80 00 01 1C 80 00 00 00 00 00 00 65 EB 06 21 E9 00 00 00 05 89 08 00 50 9B 00 C8 C7
(00:01:17.365) 08 00 1C 00 91 0C 08 08 27 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04
(00:01:17.633) 08 00 34 00 3D 02 2A 02 2A 21 00 00 03 00 00 98 E1 00 15 50 00 80 00 F5
(00:01:17.900) 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 B7
(00:01:18.872) 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B

send 0B 88 07 00 20
(00:01:04.734) Sending raw: 0B 88 07 00 20 00
(00:01:04.744) 0B 88 07 00 20 A8
(00:01:04.777) 08 0B 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 96
(00:01:05.077) 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 B7
(00:01:06.197) 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B
(00:01:06.968) 11 08 1E 00 00 C8 AA
(00:01:09.622) 08 00 18 00 07 01 11 00 00 00 00 00 60 01 09 02 29 01 1C 00 00 0A 30 48 00 CB 00 00 00 20
(00:01:09.866) 08 00 34 00 3D 02 29 02 29 21 00 00 03 00 00 98 E1 00 15 50 00 80 00 27
(00:01:10.480) 21 00 AB 00 00 83 00 00 00 01 00 67
(00:01:16.548) 11 00 9C 00 00 C8 00 01 00 C0
(00:01:16.840) 11 08 1E 00 00 C8 AA
(00:01:17.254) 10 00 3E 00 00 00 00 00 C8 00 00 00 00 00 00 00 00 11 05 04 78
(00:01:19.610) 08 00 18 00 07 01 11 00 00 00 00 00 60 01 09 02 2A 01 1C 00 00 0A 30 48 00 CB 00 00 00 A2
(00:01:19.927) 08 00 34 00 3D 02 2A 02 2A 21 00 00 03 00 00 98 E1 00 15 50 00 80 00 F5

send 0B 98 02 00 20
(00:03:25.256) Sending raw: 0B 98 02 00 20 00

The 0B 98 02 00 20 cause a crash
The 0B 88 02 00 20 provides the details of my boiler 
The 0B 88 07 00 20 doesn’t reply

I will now try the latest 1.9 beta and comment the scanning for the 0x18 device

susisstrolch, 2019-09-18:
You have a reply for your Type 7 query:

send 0B 88 07 00 20
(00:01:04.734) Sending raw: 0B 88 07 00 20 00
(00:01:04.744) 0B 88 07 00 20 A8
(00:01:04.777) 08 0B 07 00 03 03 03 02 00 00 00 00 00 00 00 0000 96
(00:01:05.077) 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 B7
(00:01:06.197) 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B

It shows that you have 
ID08, ID09, ID10,ID11,ID18,ID19,ID21

01:04:734 send type 7
01:04:744 echo
01:04:777 response to 0x0b

01:05:077  busmaster broadcast,  disturbed
01:06:197  busmaster resends because of prev error

Looks completely ok so far...

However the reboot after your query device 18 needs further investigation...

Sent by mobile device

> Am 18.09.2019 um 21:20 schrieb lmdecroos <notifications@github.com>:
> 
> send 0B 88 07 00 20
> (00:01:04.734) Sending raw: 0B 88 07 00 20 00
> (00:01:04.744) 0B 88 07 00 20 A8
> (00:01:04.777) 08 0B 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 96
> (00:01:05.077) 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 B7
> (00:01:06.197) 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B

lmdecroos, 2019-09-18:
(00:10:07.241) Sending raw: 0B 88 07 00 20 (CRC=A8)
(00:10:07.465) 0B 88 07 00 20 A8 (CRC=90) #data=2
(00:10:09.624) 11 08 1E 00 00 C8 (CRC=AA) #data=2
(00:10:11.087) 19 00 AF 00 00 B6 00 06 06 (CRC=B7) #data=5
(00:10:11.344) 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 (CRC=9B) #data=13
(00:10:11.631) 08 00 18 00 07 01 0E 00 00 00 00 00 60 01 06 02 26 01 19 00 00 0A 30 48 00 CB 00 00 00 (CRC=63) #data=25
(00:10:11.857) 08 00 19 00 80 00 01 19 80 00 00 00 00 00 00 65 EB 06 21 E9 00 00 00 05 89 08 00 50 9B 00 C8 (CRC=CE) #data=27
(00:10:12.218) 10 21 AC 00 05 00 01 (CRC=0F) #data=3
(00:10:12.403) 08 00 1C 00 91 0C 08 08 27 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=04) #data=25
(00:10:12.671) 08 00 34 00 3D 02 25 02 25 21 00 00 03 00 00 98 E1 00 15 50 00 80 00 (CRC=44) #data=19
(00:10:19.595) 11 08 1E 00 00 C8 (CRC=AA) #data=2
(00:10:21.210) 10 88 14 00 03 (CRC=6E) #data=1
(00:10:21.221) 08 10 14 00 32 C0 69 (CRC=61) #data=3
(00:10:21.394) 08 00 18 00 07 01 0E 00 00 00 00 00 60 01 08 02 25 01 19 00 00 0A 30 48 00 CB 00 00 00 (CRC=25) #data=25
(00:10:21.662) 08 00 34 00 3D 02 25 02 25 21 00 00 03 00 00 98 E1 00 15 50 00 80 00 (CRC=44) #data=19
(00:10:24.625) 10 00 3E 00 00 00 00 00 C7 00 00 00 00 00 00 00 00 11 05 04 (CRC=24) #data=16
(00:10:29.367) 11 08 1E 00 00 C8 (CRC=AA) #data=2
(00:10:31.183) 08 00 18 00 07 01 0E 00 00 00 00 00 60 01 06 02 25 01 17 00 00 0A 30 48 00 CB 00 00 00 (CRC=AB) #data=25
(00:10:31.475) 08 00 34 00 3D 02 25 02 25 21 00 00 03 00 00 98 E1 00 15 50 00 80 00 (CRC=44) #data=19

lmdecroos, 2019-09-18:
adding Device_Ids.remove(0x18); in ems.cpp fixed the crashing ! :)

info
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=34, # CRC errors=0
  Tx: Last poll=2.689 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Nefit Topline Compact/Buderus GB162 (ProductID:115 Version:05)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 58 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 54.9 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 5456 times
  Warm Water active time: 27 days 4 hours 17 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 7 C
  Current flow temperature: 26.8 C
  Return temperature: 27.8 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.0 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 30 C
  Boiler circuit pump modulation max power: 91 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 27.9 C
  Pump modulation: 0 %
  Burner # starts: 26091 times
  Total burner operating time: 279 days 2 hours 17 minutes
  Total heat operating time: 251 days 22 hours 0 minutes
  Total UBA working time: 2309 days 18 hours 24 minutes

Thermostat stats:
  Thermostat: RC35 (ProductID:86 Version:21.08)
  Thermostat time is 22:02:42 18/09/2019
  Heating Circuit 1
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to night
  Heating Circuit 2
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to night
  Heating Circuit 3
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to night
  Heating Circuit 4
   Setpoint room temperature: 14724.0 C
   Current room temperature: 1361.6 C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to night

susisstrolch, 2019-09-18:
Please set loglevel to ‚log j‘ - it shows more details ‚under the hood‘

susisstrolch, 2019-09-18:
@proddy can you store the last received Telegramm in RTC mem and Show it on reboot?

lmdecroos, 2019-09-19:
Please find my  jabber log from 

send 0B 88 02 00 20  and send 0B 88 07 00 20

Using ems protocol on version 1.9.1b4

send 0B 88 02 00 20(00:01:30.484) ems_parseTelegram: 00 01: 10 
(00:01:30.506) ems_parseTelegram: 00 01: 91 
(00:01:30.510) ems_parseTelegram: 00 01: 11 

(00:01:30.786) ems_parseTelegram: 00 01: 09 
(00:01:30.830) ems_parseTelegram: 00 01: 08 00 34 00 3D 02 57 02 57 21 00 00 03 00 00 98 F0 00 15 52 00 80 00 5B 
(00:01:30.831) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 57 02 57 21 00 00 03 00 00 98 F0 00 15 52 00 80 00 (CRC=5B) #data=19

(00:01:31.029) ems_parseTelegram: 00 01: 89 
(00:01:31.038) ems_parseTelegram: 00 01: 09 
(00:01:31.085) ems_parseTelegram: 00 01: 08 00 18 00 07 01 34 00 00 00 00 00 60 01 14 02 57 01 47 00 00 0A 30 48 00 CB 00 00 00 EC 
(00:01:31.086) Boiler -> all, type 0x18, telegram: 08 00 18 00 07 01 34 00 00 00 00 00 60 01 14 02 57 01 47 00 00 0A 30 48 00 CB 00 00 00 (CRC=EC) #data=25
(00:01:31.278) ems_parseTelegram: 00 01: 90 

(00:01:31.477) ems_parseTelegram: 00 01: 99 
(00:01:31.552) ems_parseTelegram: 00 01: 19 00 AF 00 00 B1 00 06 05 8C 
(00:01:31.552) 0x19 -> all, type 0xAF, telegram: 19 00 AF 00 00 B1 00 06 05 (CRC=8C) #data=5
(00:01:31.740) ems_parseTelegram: 00 01: 19 

(00:01:31.801) ems_parseTelegram: 00 01: ED 
(00:01:31.857) ems_parseTelegram: 00 01: 08 00 18 00 07 01 35 00 00 00 00 00 60 01 15 02 57 01 48 00 00 09 30 48 00 CB 00 00 00 92 
(00:01:31.859) Boiler -> all, type 0x18, telegram: 08 00 18 00 07 01 35 00 00 00 00 00 60 01 15 02 57 01 48 00 00 09 30 48 00 CB 00 00 00 (CRC=92) #data=25
(00:01:32.050) ems_parseTelegram: 00 01: 89 

(00:01:32.125) ems_parseTelegram: 00 01: 90 
(00:01:32.269) ems_parseTelegram: 00 01: 10 21 AC 00 05 00 01 0F 
(00:01:32.270) Thermostat -> 0x21, type 0xAC, telegram: 10 21 AC 00 05 00 01 (CRC=0F) #data=3
(00:01:32.279) ems_parseTelegram: 00 01: 01 

(00:01:32.498) ems_parseTelegram: 00 01: 8B 
(00:01:32.499) Sending raw: 0B 88 02 00 20 (CRC=BC) #data=240
(00:01:32.499) emsuart_tx_buffer: 00 01: 0B 88 02 00 20 BC 
(00:01:32.514) ems_parseTelegram: 00 01: 0B 88 02 00 20 BC 
(00:01:32.514) echo:telegram: 0B 88 02 00 20 (CRC=BC) #data=1
(00:01:32.538) ems_parseTelegram: 00 01: 08 0B 02 00 73 05 05 00 00 00 00 00 00 00 D7 
(00:01:32.539) Boiler -> me, type 0x02, telegram: 08 0B 02 00 73 05 05 00 00 00 00 00 00 00 (CRC=D7) #data=10
Boiler found: Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
(00:01:32.772) ems_parseTelegram: 00 01: 98 
(00:01:32.776) ems_parseTelegram: 00 01: 18 
(00:01:32.815) ems_parseTelegram: 00 01: 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 B7 
(00:01:32.816) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 (CRC=B7) #data=13
(00:01:32.996) ems_parseTelegram: 00 01: 99 

(00:01:34.142) ems_parseTelegram: 00 01: 8B 
g(00:01:34.185) ems_parseTelegram: 00 01: 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B 
(00:01:34.186) Boiler -> all, type 0x07, telegram: 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 (CRC=9B) #data=13
 (00:01:34.366) ems_parseTelegram: 00 01: 90 
(00:01:34.371) ems_parseTelegram: 00 01: 10 

send 0B 88 07 00 20(00:04:02.868) ems_parseTelegram: 00 01: 09 
(00:04:02.883) ems_parseTelegram: 00 01: 89 
(00:04:02.889) ems_parseTelegram: 00 01: 09 
(00:04:02.908) ems_parseTelegram: 00 01: 8B 
(00:04:02.909) emsuart_tx_buffer: 00 01: 0B 90 3D 00 20 80 
(00:04:02.924) ems_parseTelegram: 00 02: 0B 90 3D 00 20 80 
(00:04:02.924) echo:telegram: 0B 90 3D 00 20 (CRC=80) #data=1
(00:04:03.079) ems_parseTelegram: 00 02: 10 0B 3D 00 01 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 43 
(00:04:03.081) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=43) #data=27
(00:04:03.082) emsuart_tx_buffer: 00 01: 0B 
Publishing boiler data via MQTT
(00:04:03.092) ems_parseTelegram: 00 01: 0B 

(00:04:03.356) ems_parseTelegram: 00 01: 8F 
(00:04:03.413) ems_parseTelegram: 00 01: 08 00 18 00 07 01 34 00 00 00 00 00 60 01 14 02 57 01 47 00 00 0A 30 48 00 CB 00 00 00 EC 
[... 114 lines cut ...]
(00:04:10.231) ems_parseTelegram: 00 01: 8B 
(00:04:10.231) emsuart_tx_buffer: 00 01: 0B 88 16 00 20 EC 
(00:04:10.247) ems_parseTelegram: 00 02: 0B 88 16 00 20 EC 
(00:04:10.247) echo:telegram: 0B 88 16 00 20 (CRC=EC) #data=1
(00:04:10.273) ems_parseTelegram: 00 02: 08 0B 16 00 00 1E 64 00 06 FA 0A 01 05 5B 1E 00 48 
(00:04:10.274) Boiler -> me, type 0x16, telegram: 08 0B 16 00 00 1E 64 00 06 FA 0A 01 05 5B 1E 00 (CRC=48) #data=12
(00:04:10.275) emsuart_tx_buffer: 00 01: 0B 
(00:04:10.279) ems_parseTelegram: 00 01: 0B 
(00:04:10.282) ems_parseTelegram: 00 01: 96 
g(00:04:10.330) ems_parseTelegram: 00 01: 08 00 34 00 3D 02 56 02 56 21 00 00 03 00 00 98 F0 00 15 52 00 80 00 15 
(00:04:10.331) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 56 02 56 21 00 00 03 00 00 98 F0 00 15 52 00 80 00 (CRC=15) #data=19
 (00:04:10.530) ems_parseTelegram: 00 01: 90 

(00:04:10.559) ems_parseTelegram: 00 01: 11 
(00:04:10.611) ems_parseTelegram: 00 01: 08 00 18 00 07 01 34 00 00 00 00 00 60 01 14 02 56 01 47 00 00 0A 30 48 00 CB 00 00 00 65 
(00:04:10.612) Boiler -> all, type 0x18, telegram: 08 00 18 00 07 01 34 00 00 00 00 00 60 01 14 02 56 01 47 00 00 0A 30 48 00 CB 00 00 00 (CRC=65) #data=25
(00:04:10.803) ems_parseTelegram: 00 01: 98 
(00
(00:04:11.127) ems_parseTelegram: 00 01: 8B 
(00:04:11.128) emsuart_tx_buffer: 00 01: 0B 88 14 00 20 E4 
(00:04:11.143) ems_parseTelegram: 00 02: 0B 88 14 00 20 E4 
(00:04:11.143) echo:telegram: 0B 88 14 00 20 (CRC=E4) #data=1
(00:04:11.160) ems_parseTelegram: 00 02: 08 0B 14 00 32 C4 A3 E8 
(00:04:11.160) Boiler -> me, type 0x14, telegram: 08 0B 14 00 32 C4 A3 (CRC=E8) #data=3
(00:04:11.161) emsuart_tx_buffer: 00 01: 0B

proddy, 2019-09-19:
@lmdecroos  1.9.1b5 has a different method of detecting EMS devices - can you test if that solves your crash issue?

@susisstrolch can't store a whole telegram in RTC. not enough bytes

lmdecroos, 2019-09-20:
1.9.1b5 still crashes 

Requesting type Version(0x02) from dest 0x18
(00:00:21.420) Sending read of type 0x5B to 0x10, telegram: 0B 90 5B 00 20 (CRC=01)
(00:00:21.574) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 1E 26 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=F8) #data=27
<--- RC35Set_HC4(0x5B)
(00:00:21.968) Sending read of type 0x06 to 0x10, telegram: 0B 90 06 00 20 (CRC=6C)
(00:00:22.023) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 09 07 14 21 33 04 01 (CRC=69) #data=8
<--- RCTime(0x06)
(00:00:22.098) Boiler -> all, type 0x18, telegram: 08 00 18 00 07 00 ED 00 00 00 00 00 60 00 D5 02 05 00 F4 00 00 09 30 48 00 CB 00 00 00 (CRC=11) #data=25
<--- UBAMonitorFast(0x18)
(00:00:22.416) Boiler -> all, type 0x34, telegram: 08 00 34 00 3D 02 05 02 05 21 00 00 03 00 00 98 F0 00 15 52 00 80 00 (CRC=61) #data=19
<--- UBAMonitorWWMessage(0x34)
(00:00:22.814) Sending read of type 0x02 to 0x18, telegram: 0B 98 02 00 20 (CRC=3C)

(00:00:17.526) ems_parseTelegram: 00 01: 89 
(00:00:17.532) ems_parseTelegram: 00 01: 09 
(00:00:17.583) ems_parseTelegram: 00 01: 08 00 18 00 41 01 5F 72 71 0A 10 65 60 00 CB 01 F4 01 42 01 05 0D 3D 48 00 C9 00 02 00 68 
(00:00:17.584) Boiler -> all, type 0x18, telegram: 08 00 18 00 41 01 5F 72 71 0A 10 65 60 00 CB 01 F4 01 42 01 05 0D 3D 48 00 C9 00 02 00 (CRC=68) #data=25
(00:00:17.775) ems_parseTelegram: 00 01: 8B 
(00:00:17.776) emsuart_tx_buffer: 00 01: 0B 98 02 00 20 3C 
(00:00:17.791) ems_parseTelegram: 00 02: 0B 98 02 00 20 3C 
(00:00:17.791) echo:telegram: 0B 98 02 00 20 (CRC=3C) #data=1

I will do more testing later on

proddy, 2019-09-20:
after startup, when you type 'devices' what does it say under "These device IDs are on the EMS Bus:"

proddy, 2019-09-20:
actually, I remember you have an RC35 so 0x18 is a real device on your EMS bus. We still need to see why a Tx to "0B 98 02 00 20" causes the crash because it a valid command. @susisstrolch  any ideas?

susisstrolch, 2019-09-20:
No, currently not, because we don‘t know what happened between the echo Rx and the exception. 
I vaguely remember that I sometimes saw single BRK from busmaster, maybe that’s such a case which isn‘t handled properly.

proddy, 2019-09-20:
@susisstrolch I'm thinking of refactoring the Rx piece of the UART. Right now the UART interrupt fills up the Rx buffer with a telegram and puts a call to ems_parseTelegram on a timer queue. I think the best practice is to continuously fill a Rx buffer queue and make calls from ems-esp to see if there are available telegrams for parsing. This way the Rx interrupt function can go back to IRAM and we'll have more control on viewing the incoming Rx queue. What do you think?

lmdecroos, 2019-09-20:
Devices as detected by 1.9.1b4 with ems protocol 

These 5 EMS devices were detected:
 Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

proddy, 2019-09-20:
there's a new section above that. e.g. on my system:

```

These device IDs are on the EMS Bus:
 0x08
 0x09
 0x17

These known 3 EMS devices were detected:
 Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:06.01)
 BC10 Base Controller (DeviceID:0x09 ProductID:190 Version:01.03)
 RC20/Nefit Moduline 300 (DeviceID:0x17 ProductID:77 Version:03.03)
```

I'm still not sure why your system is sending a Version (0x02) request to 0x18 ??

lmdecroos, 2019-09-20:
sorry, i don’t have the section "These device IDs are on the EMS Bus:"

devices

These 9 devices are supported as boiler units:
 MC10 Module (DeviceID:0x08 ProductID:72)
 Buderus GB172/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123)
 Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115)
 Buderus Logamax U122/Junkers Cerapur (DeviceID:0x08 ProductID:203)
 Buderus Logamax plus/GB192 (DeviceID:0x08 ProductID:208)
 Sieger BK15/Nefit Smartline/Buderus GB152 (DeviceID:0x08 ProductID:64)
 Bosch Condens 2500/Junkers Heatronics3 (DeviceID:0x08 ProductID:95)
 Nefit Proline (DeviceID:0x08 ProductID:122)
 Nefit Enviline (DeviceID:0x08 ProductID:172)

These 3 devices are supported as solar module devices:
 SM10 Solar Module (DeviceID:0x30 ProductID:73)
 SM100 Solar Module (DeviceID:0x30 ProductID:163)
 Junkers ISM1 Solar Module (DeviceID:0x30 ProductID:101)

These 15 devices are supported as other known EMS devices:
 MM10 Mixer Module (DeviceID:0x21 ProductID:69)
 WM10 Switch Module (DeviceID:0x11 ProductID:71)
 MM100 Mixing Module (DeviceID:0x20 ProductID:160)
 MM100 Mixing Module (DeviceID:0x21 ProductID:160)
 MM50 Mixing Module (DeviceID:0x21 ProductID:159)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68)
 BC10 Base Controller (DeviceID:0x09 ProductID:190)
 BC10 Base Controller (DeviceID:0x09 ProductID:114)
 BC25 Base Controller (DeviceID:0x09 ProductID:125)
 Junkers Controller (DeviceID:0x09 ProductID:152)
 Nefit Moduline Easy Connect (DeviceID:0x02 ProductID:205)
 Bosch Easy Connect (DeviceID:0x02 ProductID:206)
 EMS-OT OpenTherm converter (DeviceID:0x02 ProductID:171)
 HeatPump Module (DeviceID:0x38 ProductID:252)
 Web Gateway KM200 (DeviceID:0x48 ProductID:189)

The following telegram type IDs are supported:
 type 02 (Version)
 type 18 (UBAMonitorFast)
 type 19 (UBAMonitorSlow)
 type 34 (UBAMonitorWWMessage)
 type 33 (UBAParameterWW)
 type 14 (UBATotalUptimeMessage)
 type 15 (UBAMaintenanceSettingsMessage)
 type 16 (UBAParametersMessage)
 type 1A (UBASetPoints)
 type 1A5 (RCPLUSStatusMessage)
 type 1B9 (RCPLUSSetMessage)
 type 1AF (RCPLUSStatusMode)
 type 6F (JunkersStatusMessage)

These 16 thermostat devices are supported:
 Logamatic TC100/Nefit Moduline Easy (DeviceID:0x18 ProductID:202) can write:n
 Bosch EasyControl CT200 (DeviceID:0x18 ProductID:203) can write:n
 Bosch CW100 (DeviceID:0x18 ProductID:157) can write:n
 RC10/Nefit Moduline 100 (DeviceID:0x17 ProductID:79) can write:y
 RC20/Nefit Moduline 300 (DeviceID:0x17 ProductID:77) can write:y
 RC20F (DeviceID:0x18 ProductID:93) can write:y
 RC30/Nefit Moduline 400 (DeviceID:0x10 ProductID:78) can write:y
 RC35 (DeviceID:0x10 ProductID:86) can write:y
 RC300/RC310/Nefit Moduline 3000 (DeviceID:0x10 ProductID:158) can write:y
 Nefit Moduline 1010 (DeviceID:0x18 ProductID:165) can write:n
 Sieger ES73 (DeviceID:0x10 ProductID:76) can write:y
 Junkers FW100 (DeviceID:0x10 ProductID:105) can write:n
 Junkers FR10 (DeviceID:0x18 ProductID:111) can write:n
 Junkers FR100 (DeviceID:0x18 ProductID:105) can write:n
 Junkers FR110 (DeviceID:0x18 ProductID:108) can write:n
 Junkers FW120 (DeviceID:0x10 ProductID:192) can write:n

These 5 EMS devices were detected:
 Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
 RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
 WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

Note: if any devices are marked as 'unknown?' please report this as a GitHub issue so the EMS devices list can be updated.

proddy, 2019-09-20:
oh, pretty sure it was in b4. I'll do a new build called b5 now.

lmdecroos, 2019-09-20:
during my lunchbreak i compiled the new b5

when using EMS+ protocol :

These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
No devices detected.

When using EMS protocol the gateway hangs :
System Logging set to None
[EMS] Querying EMS Device ID 0x19
[EMS] Querying EMS Device ID 0x21
[EMS] Querying EMS Device ID 0x18
Device found: MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01) 

It seems that excluding 0x18 from scanning doesn’t anymore work in the b5 build since yesterday evening

Difficult to troubleshoot as i have only 15 seconds after a reboot :(

proddy, 2019-09-20:
thanks for checking during your lunch break! It looks like there is something with device ID 0x18 on your EMS bus. I don't know what it is. And when it's queried it crashes. I'll create a new branch with some extra debug information so we can test this specific scenario. I'll let you know where it is...

proddy, 2019-09-20:
@lmdecroos I made a new branch and customized the code for your environment so we go a little deeper into debugging. Can you compile the code in https://github.com/proddy/EMS-ESP/tree/issue182

just start it up, no need to set any settings other than wifi (if haven't done already). You will see the ems telegrams come in. 
type `query 21` and wait until the telegram is sent
then type `query 18`  and see if it explodes. Try and capture what you see last on the screen.

lmdecroos, 2019-09-20:
hope this helps 
It keeps running no crash :)

System Logging set to Verbose
(00:00:05.667) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 06 00 00 09 30 48 00 CB 00 00 00 30 
(00:00:06.664) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 08 00 00 0A 30 48 00 CB 00 00 00 E3 
(00:00:07.538) ems_parseTelegram: 00 01: 10 08 1A 00 00 00 00 00 91 
(00:00:07.660) ems_parseTelegram: 00 01: 10 11 9D 00 00 D7 
(00:00:08.049) ems_parseTelegram: 00 01: 18 08 35 00 00 00 1A 
(00:00:08.157) ems_parseTelegram: 00 01: 18 00 AF 00 83 00 09 00 00 B5 
(00:00:08.369) ems_parseTelegram: 00 01: 08 00 07 00 03 03 01 02 00 00 00 00 00 00 00 00 00 53 
(00:00:08.644) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:00:11.051) ems_parseTelegram: 00 01: 19 00 AF 00 00 B6 00 06 06 B7 
(00:00:11.333) ems_parseTelegram: 00 01: 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B 
(00:00:11.669) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 08 00 00 0A 30 48 00 CB 00 00 00 E3 
(00:00:11.987) ems_parseTelegram: 00 01: 08 00 34 00 0F 02 17 02 17 20 00 00 03 00 00 98 FB 00 15 54 00 80 00 3A 
(00:00:14.708) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FC 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 BF 
(00:00:15.704) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 08 00 00 0A 30 48 00 CB 00 00 00 E3 
(00:00:18.814) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:00:20.787) ems_parseTelegram: 00 01: 21 00 AB 00 00 83 00 00 00 01 00 67 
(00:00:21.601) ems_parseTelegram: 00 01: 08 00 34 00 0F 02 17 02 17 20 00 00 03 00 00 98 FB 00 15 54 00 80 00 3A 
(00:00:21.881) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 08 00 00 0A 30 48 00 CB 00 00 00 E3 
(00:00:23.600) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FC 02 17 01 06 00 00 09 30 48 00 CB 00 00 00 26 
(00:00:24.596) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FC 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 BF 
(00:00:28.786) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:00:30.946) ems_parseTelegram: 00 01: 10 00 48 00 00 00 00 00 B6 00 00 00 00 00 00 00 00 00 05 04 7C 
(00:00:31.588) ems_parseTelegram: 00 01: 08 00 34 00 0F 02 17 02 17 20 00 00 03 00 00 98 FB 00 15 54 00 80 00 3A 
(00:00:31.868) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 A9 
(00:00:35.158) ems_parseTelegram: 00 01: 10 08 35 00 01 01 00 
(00:00:38.557) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:00:38.693) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 A9 
(00:00:39.395) ems_parseTelegram: 00 01: 18 89 29 00 01 10 
(00:00:39.415) ems_parseTelegram: 00 01: 09 18 29 00 7B 8F 
(00:00:39.565) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 08 00 00 0A 30 48 00 CB 00 00 00 E3 
(00:00:41.600) ems_parseTelegram: 00 01: 08 00 34 00 0F 02 17 02 17 20 00 00 03 00 00 98 FB 00 15 54 00 80 00 3A 
(00:00:41.881) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FC 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 BF 
(00:00:48.834) ems_parseTelegram: 00 01: 11 00 9C 00 00 BF 00 01 00 53 
(00:00:49.127) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:00:51.571) ems_parseTelegram: 00 01: 08 00 19 00 80 00 01 06 80 00 00 00 00 00 00 65 EF 06 22 03 00 00 00 05 89 08 00 50 9B 00 BF FC 
(00:00:51.842) ems_parseTelegram: 00 01: 08 00 1C 00 91 0C 08 08 27 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 
(00:00:52.110) ems_parseTelegram: 00 01: 08 00 34 00 0F 02 17 02 17 20 00 00 03 00 00 98 FB 00 15 54 00 80 00 3A 
(00:00:52.427) ems_parseTelegram: 00 01: 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B 
(00:00:52.640) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FC 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 BF 
query 21 

(00:00:56.021) Sending raw: 0B A1 02 00 20 (CRC=ED) #data=240
(00:00:56.036) ems_parseTelegram: 00 01: 0B A1 02 00 20 ED 
(00:00:56.127) ems_parseTelegram: 00 01: 21 0B 02 00 45 02 01 D9 
(00:00:56.437) ems_parseTelegram: 00 01: 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 B7 
(00:00:57.176) ems_parseTelegram: 00 01: 10 00 06 00 13 09 00 15 0A 30 05 01 71 
(00:00:57.445) ems_parseTelegram: 00 01: 10 00 3E 00 00 00 00 00 C0 00 00 00 00 00 00 00 00 00 05 04 2A 
(00:00:58.081) ems_parseTelegram: 00 01: 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 9B 
(00:00:58.799) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:01:01.507) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FC 02 17 01 06 00 00 09 30 48 00 CB 00 00 00 26 
(00:01:01.724) ems_parseTelegram: 00 01: 08 00 34 00 0F 02 17 02 17 20 00 00 03 00 00 98 FB 00 15 54 00 80 00 3A 
(00:01:02.065) ems_parseTelegram: 00 01: 10 00 A3 00 6E 00 00 00 C0 00 C6 00 C6 7D 00 7D 00 FA 
(00:01:02.288) ems_parseTelegram: 00 01: 10 21 AC 00 05 00 01 0F 
(00:01:07.661) ems_parseTelegram: 00 01: 10 08 1A 00 00 00 00 00 91 
(00:01:07.784) ems_parseTelegram: 00 01: 10 11 9D 00 00 D7 
(00:01:08.151) ems_parseTelegram: 00 01: 18 08 35 00 00 00 1A 
(00:01:08.670) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:01:08.857) ems_parseTelegram: 00 01: 18 00 AF 00 83 00 09 00 00 B5 
[... cut ...]
(00:03:03.447) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FC 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 BF 
(00:03:04.120) ems_parseTelegram: 00 01: 08 00 18 00 07 00 FF 00 00 00 00 00 60 00 FD 02 17 01 06 00 00 0A 30 48 00 CB 00 00 00 A9 
(00:03:08.284) ems_parseTelegram: 00 01: 10 08 1A 00 00 00 00 00 91 
(00:03:08.406) ems_parseTelegram: 00 01: 10 11 9D 00 00 D7 
(00:03:08.623) ems_parseTelegram: 00 01: 11 08 1E 00 00 BF DD 
(00:03:08.755) ems_parseTelegram: 00 01: 18 08 35 00 00 00 1A

proddy, 2019-09-21:
wow. This tells us a lot
1. the Tx logic is not the problem. @susisstrolch's code is rock solid.
2. You do have something on your EMS bus with ID 0x18. It has a product ID 0x5E (94) I have no idea what it is!
3. It breaks when we query the version of this 0x18 by requesting type 0x02, which we kinda knew anyway. This is because it expects a telegram response with data like <product id> <software version major> <software version minor> but here it comes back with <0> <0> <0> followed by the the version information. Looking this up in [HT3-Bus_Telegramme.html](https://raw.githubusercontent.com/proddy/EMS-ESP/dev/doc/ems%20references/HT3-Bus_Telegramme.html) says its a second detection of bus subscribers, loosely translated from german. 

I'll make a change and let you know when you can test again.

susisstrolch, 2019-09-21:
Jep - the description of query is even for native speakers not clear...

proddy, 2019-09-21:
Found the bug. Was one stupid debug statement with an addition argument that wasn't set in function _process_Version() with this line:
```
myDebug_P(PSTR("Unrecognized device found: %s (DeviceID:0x%02X ProductID:%d Version:%s)"),
 EMS_RxTelegram->src, product_id, version);
```

could have been easily detected with static analysis tools. I'll get Travis CI up and running again to do more regular code checks.

@lmdecroos can you check the dev branch again, version b6

lmdecroos, 2019-09-21:
Well done, doesn’t crash anymore !

I have still an unknown device

   Boiler stats:
  Boiler: Nefit Topline Compact/Buderus GB162 (ProductID:115 Version:05.05)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 58 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 51.3 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 5462 times
  Warm Water active time: 27 days 4 hours 54 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 7 C
  Current flow temperature: 28.7 C
  Return temperature: 29.9 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.0 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 30 C
  Boiler circuit pump modulation max power: 91 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 30.0 C
  Pump modulation: 0 %
  Burner # starts: 26097 times
  Total burner operating time: 279 days 2 hours 54 minutes
  Total heat operating time: 251 days 22 hours 0 minutes
  Total UBA working time: 2312 days 14 hours 7 minutes

Thermostat stats:
  Thermostat: RC35 (ProductID:86 Version:21.08)
  Thermostat time is 17:30:07 21/09/2019
  Heating Circuit 1
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to night
  Heating Circuit 2
   Setpoint room temperature: ? C
   Current room temperature: ? C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to night

   These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x18 0x19 0x21
   and from those these 7 were recognized as:
    Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.05)
    BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)
    RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
    WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
    unknown? (DeviceID:0x18 ProductID:94 Version:02.01)
    RC20F (DeviceID:0x18 ProductID:93 Version:02.00)
    MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

proddy, 2019-09-21:
yup, like I said before it has ID 0x18 and a product ID 94 but I have no idea what it is! Just ignore it for now.

proddy, 2019-09-22:
closing this issue. The fix at startup is crashed - the bug was when printing out an unknown device. We also discovered the 0x02 Version telegram can be split into 3 chunks of version information so the code was also adjusted to support this in 1.9.1b5

closed 2019-09-22 by proddy