emsesp/EMS-ESP #183: Able to set warm water temperature via MQTT

higgers, 2019-09-16:
Should I expect to be able to set the warm water temperature via MQTT? I have set up my openHAB system to allow me to send the boiler_cmd_wwtemp MQTT topic and I can telnet into the Wemos and see the MQTT topic set in the log. However, typing "info" shows that the "Warm water desired temperature" attribute doesn't change.

I looked at the EMS wiki and I don't think the 6 byte 0x33 UBAParameterWW is showing up in the log. Should I expect to see it? If so, would it be possible for someone to copy part of the verbose log from a telnet session so I can see a 0x33 telegram setting the warm water desired temperature so I know what to look for.

Also, what's the difference between the attributes "Warm water selected temperature" and "Warm water desired temperature" on the info section in the telnet session?

proddy, 2019-09-17:
I'll test it tonight, it's been a while since that logic was created. The desired temperature is probably the wrong word. According to the german translated [wiki page](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#wwbetriebsart) it's described as "Target temperature thermal disinfection", so just ignore it I'd say.

does `boiler wwtemp XX` work? Do you see a successful write when in verbose logging?

proddy, 2019-09-17:
Just tested and works on my Buderus:

```
info
:
Warm Water current temperature: 56.1 C
:

log v
System Logging set to Verbose
boiler wwtemp 55
Setting boiler warm water temperature to 55 C
(00:04:10.261) Sending write of type 0x33 to 0x08, telegram: 0B 08 33 02 37 (CRC=0F)
(00:04:10.312) Sending validate of type 0x33 to 0x08, telegram: 0B 88 33 02 01 (CRC=5D)
-> Validate confirmed, last Write to 0x08 was successful
Requesting type UBAParameterWW(0x33) from dest 0x08
(00:04:10.410) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 80 00 00 00 00 00 00 00 00 00 00 00 (CRC=47) #data=13
(00:04:10.682) Boiler -> all, type 0x33, telegram: 08 00 33 00 08 FF 37 FB 00 28 00 00 46 00 FF FF 00 (CRC=FE) #data=13
<--- UBAParameterWW(0x33)
Publishing boiler data via MQTT
(00:04:10.933) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(00:04:10.990) Boiler -> me, type 0x33, telegram: 08 0B 33 00 08 FF 37 FB 00 28 00 00 46 00 FF FF 00 (CRC=F3) #data=13
<--- UBAParameterWW(0x33)
```

```
info
:
Warm Water selected temperature: 55 C
:
```

higgers, 2019-09-17:
Doesn't seem to work on my Worcester-Bosch:

```
boiler wwtemp 44
Setting boiler warm water temperature to 44 C
(21:33:43.463) Boiler -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(21:33:43.779) Boiler -> all, type 0x18, telegram: 88 00 18 00 32 00 FE 20 00 00 01 00 40 00 DB 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=D7) #data=25
<--- UBAMonitorFast(0x18)
(21:33:44.019) Boiler -> all, type 0x34, telegram: 88 00 34 00 33 00 DB 80 00 80 00 00 01 00 01 C6 AE 00 04 F8 00 (CRC=0C) #data=17
<--- UBAMonitorWWMessage(0x34)
(21:33:44.833) Sending write of type 0x33 to 0x08:telegram: 8B 08 33 02 2C (CRC=DC) #data=128
(21:33:44.894) Sending validate of type 0x33 to 0x08:telegram: 8B 88 33 02 01 (CRC=95) #data=74
Last write failed. Compared set value 0x2C with received value 0x33
...Retrying write. Attempt 1/2...
(21:33:44.978) Boiler -> all, type 0x07, telegram: 88 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=56) #data=15
(21:33:45.194) Boiler -> all, type 0x33, telegram: 88 00 33 00 08 FF 33 00 00 00 00 02 46 00 FF FF (CRC=E9) #data=12
<--- UBAParameterWW(0x33)
Publishing boiler data via MQTT
(21:33:45.456) Sending write of type 0x33 to 0x08:telegram: 8B 08 33 02 2C (CRC=DC) #data=128
(21:33:45.537) Boiler -> all, type 0x33, telegram: 88 00 33 00 08 FF 33 00 00 00 00 02 46 00 FF FF (CRC=E9) #data=12
<--- UBAParameterWW(0x33)
(21:33:45.800) Sending validate of type 0x33 to 0x08:telegram: 8B 88 33 02 01 (CRC=95) #data=128
Last write failed. Compared set value 0x2C with received value 0x33
...Retrying write. Attempt 2/2...
(21:33:45.925) Sending write of type 0x33 to 0x08:telegram: 8B 08 33 02 2C (CRC=DC) #data=74
(21:33:46.011) Boiler -> all, type 0x33, telegram: 88 00 33 00 08 FF 33 00 00 00 00 02 46 00 FF FF (CRC=E9) #data=12
<--- UBAParameterWW(0x33)
(21:33:46.300) Sending validate of type 0x33 to 0x08:telegram: 8B 88 33 02 01 (CRC=95)
Last write failed. Compared set value 0x2C with received value 0x33
Write failed. Giving up, removing from queue
```

higgers, 2019-09-17:
I've disconnected the Bosch Easy Control thermostat to check if it was overriding the commands from the Wemos but it gave the same result.

proddy, 2019-09-17:
can you test with 1.9.1b? I noticed some strange things in the telegram format like #data having very high values

higgers, 2019-09-17:
Could you point me at a precompiled 1.9.1b .bin file please?  I can't seem to compile from source due to the same issues I had compiling 1.9.0b3 which I described in issue #173 .

(I thought I'd already posted this reply here but maybe I posted to a different issue by mistake?)

proddy, 2019-09-17:
I hid it here - https://github.com/proddy/EMS-ESP/releases/tag/1.9.0

higgers, 2019-09-17:
The file on that page is firmware_release_1_9_0.bin, is that the one you want me to test?  I only ask because you mentioned 1.9.1b in your earlier post above.

proddy, 2019-09-17:
there are two .bin files, take the latest beta one

higgers, 2019-09-17:
Sorry to be a pain, but can you remind me how to flash the device from the CLI?  I've tried pio, esptool.py and espota.py and can't get any to work.

proddy, 2019-09-17:
this [link](https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware#b-using-the-pre-built-firmware) didn't help?

higgers, 2019-09-17:
Sorry, I didn't see that link.

I've run "pio run -t erase" and then run "esptool.py -p COM4 write_flash 0x00000 ./firmware_debug_1_9_1b4.bin" but when I point my browser at 192.168.4.1 I get asked for a password, which I don't know.

At first I thought that maybe I'd set a password the last time I used the 1.9 branch and configured the web frontend but surely running "pio run -t erase" would have cleared that?

proddy, 2019-09-17:
default password is 'admin'

higgers, 2019-09-17:
Finally got there, thanks for your patience.

The dashboard on the web frontend has a message in red that says, "EMS Bus Connected but Tx is not working."  I can no longer see the warm water temperature on the info section via the telnet connection.

The "boiler wwtemp 42" command doesn't seem to do anything now:

```
Setting boiler warm water temperature to 42 C
(00:04:48.776) 0x08 -> all, type 0x19, telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 00 00 00 10 CE 09 3F B4 00 00 00 07 78 F9 00 0B D3 80 00 (CRC=55) #data=27
<--- UBAMonitorSlow(0x19)
(00:04:57.768) 0x08 -> all, type 0x18, telegram: 88 00 18 00 45 01 9E 20 00 00 00 40 40 01 05 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=39) #data=25
<--- UBAMonitorFast(0x18)
(00:04:57.990) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:04:58.295) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 05 80 00 80 00 00 01 00 01 C6 BA 00 04 FB 00 (CRC=DC) #data=17
<--- UBAMonitorWWMessage(0x34)
Requesting scheduled EMS device data
(00:05:07.733) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:05:08.048) 0x08 -> all, type 0x18, telegram: 88 00 18 00 45 01 9E 20 00 00 00 40 40 01 05 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=39) #data=25
<--- UBAMonitorFast(0x18)
(00:05:08.288) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 05 80 00 80 00 00 01 00 01 C6 BA 00 04 FB 00 (CRC=DC) #data=17
<--- UBAMonitorWWMessage(0x34)
(00:05:17.726) 0x08 -> 0x13, type 0x05, telegram: 88 13 05 22 00 (CRC=80) #data=1
(00:05:18.041) 0x08 -> all, type 0x18, telegram: 88 00 18 00 45 01 9D 20 00 00 00 40 40 01 05 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=41) #data=25
<--- UBAMonitorFast(0x18)
(00:05:18.281) 0x08 -> all, type 0x34, telegram: 88 00 34 00 33 01 05 80 00 80 00 00 01 00 01 C6 BA 00 04 FB 00 (CRC=DC) #data=17
<--- UBAMonitorWWMessage(0x34)
```

proddy, 2019-09-17:
Try changing the tx_mode. 1 is the same as 2 in 1.8.x, 2 is now for EMS
plus and 3 for Junkers equipment

higgers, 2019-09-17:
Somehow, I seem to have ended up with a non-working system.

It seems like by default tx_mode is set to 1 in the 1.9.1b firmware so I didn't need to change it and the boiler wasn't detected in the same way it was in 1.8.3.  In 1.8.3 the boiler is detected within a few seconds of me setting tx_mode to 2.

However... I reverted to 1.8.3 after testing 1.9.1b but now the system restarts a few seconds after it outputs the message that it detected the boiler.  The sequence of events is:

Run "pio run -t erase" (if I'm coming from a 1.9 branch)
Flash firmware
Join laptop to EMS AP
Telnet to 192.168.4.1
run "set tx_mode 2"
wait a few seconds
see the message that the boiler is detected - even before running "autodetect"
wait a few more seconds and the LED turns off as the device reboots

I really don't know what the difference is in the setup that had been running without fault since I progressed through issue #173 and now.  I did have the Bosch Easy Control connected when I last flashed 1.8.3.  Maybe that had some soft of impact?

I've run out of time tonight but will try and tackle it again tomorrow.

proddy, 2019-09-18:
This sounds like the problem at https://github.com/proddy/EMS-ESP/issues/182 

When issuing Tx commands (i.e. to detect the EMS devices) it crashes.  I have no idea what is causing it but you're the second to report it.

Is it working now with 1.8.3? If so which tx_mode are you using?

higgers, 2019-09-18:
I can't seem to get 1.8.3 to be stable any longer. That sequence of events I described in my previous post is what happens in 1.8.3.

1.9.1b doesn't seem able to detect the boiler but is stable.
1.8.3 does detect the boiler but reboots. This is very strange because I was running 1.8.3 for a few weeks prior to flashing to 1.9.1b and it had been rock solid.

higgers, 2019-09-18:
Forgot to mention: 1.8.3 with tx_mode 2 allows me to successfully set the flow temperature via MQTT.

proddy, 2019-09-18:
did you compile 1.8.3 yourself or use the firmware bin in the releases page? There have been many updates to the 3rd party libraries (onewire, arduinojson, espressif) since 1.8.3 that may be having a side effect.

higgers, 2019-09-18:
I compiled it myself. I kept an untouched copy of the contents of the zip file I downloaded. I'll check to see if the original version of the bin file is in there when I get home tonight.

susisstrolch, 2019-09-18:
@higgers so try the txmode2 branch - which is unchanged since july...

susisstrolch, 2019-09-18:
@higgers stupid question - did you roll back your platformio.ini before compiling 1.8.3?

higgers, 2019-09-18:
@susisstrolch what do you mean by roll back?  After I unzip the source to a new folder the only change I make to platformio.ini is to set the port to either a COM port or (99% of the time) the IP address of the Wemos.

proddy, 2019-09-19:
@higgers that should be fine. There are new libraries in 1.9.x's platformio but you should be ok if you going back a version.

proddy, 2019-09-27:
also try again with the latest 1.9.1 as I'm constantly making small amendments.

higgers, 2019-10-02:
I've finally got some time to look at this tonight but, as ever, I can't get my build environment in a state that will allow me to build 1.9.1b.  Is there a precompiled .bin file available that I can download?  I can only see tags up to and including 1.9.0, I can't see 1.9.1b.

Many thanks!

proddy, 2019-10-02:
I hide the 1.9.1 betas under https://github.com/proddy/EMS-ESP/releases/tag/1.9.0 for now.

I'm working on using Travis to automatically build and upload the bins to a new stash, which would help later.,

higgers, 2019-10-02:
Ah, OK.

I've uploaded the latest release version and when logging in to the web page it shows the message "EMS Bus Connected but Tx is not working."

I've tried changing the tx mode from 1 to 2 but that causes the error "Error! Unable to read the EMS bus." to be printed in the telnet session.  Setting the tx mode back to 1 doesn't resolve it until I power cycle the Wemos.

higgers, 2019-10-02:
I've reverted to version 1.8.3 (downloaded from github because the output from my build environment continuously reboots) and set the tx mode to 2 and it autodetects the boiler (without me having to run "autodetect deep") and I can set the flow temp again.  Excellent! :)

higgers, 2019-10-03:
Quick question: does v1.8.3 support multiple DS18S20 thermometers and sending the data via MQTT?

1.8.3 seems to be the latest stable version for my system and I'd like to capture the central heating return temperature (since it doesn't seem to be sent from the boiler.) I'd like to glue a thermometer to the return pipe at the point the pipe enters the boiler.

proddy, 2019-10-03:
yes, support multiple external sensor, also via MQTT. See https://github.com/proddy/EMS-ESP/wiki/adding-external-temperature-sensors

higgers, 2019-10-03:
Excellent, thanks for letting me know.  I knew the latest version supported sensors/MQTT but didn't know which version had introduced the support.

proddy, 2019-10-04:
ok, closing this issue. If 1.9.1 still fails to detect the EMS devices on boot then please create a new bug so we can focus on resolving that.

closed 2019-10-04 by proddy