emsesp/EMS-ESP #184: Cannot upgrade from 1.9.0b1

ViNYLeq, 2019-09-17:
Hello :)

I'm using 1.9.0b1 version and not able to upgrade via new interface. After 1.9.1b3 bin file uploaded, module reboots but still can see old version 1.9.0b1. I"m connected via VPN so not able to do this using wires. Please advise.

Best regards,
Pawel

proddy, 2019-09-17:
Which browser are you using?

I would take the .bin file and upload it via OTA. Instructions are in the wiki.

When you have 1.9.1 see if the web upload works. I have seen reported issues of this failing on 1.9.0 but not able to reproduce.

ViNYLeq, 2019-09-17:
Thank you. Used Chrome and Firefox, after several retries something changed because I'm not able to log-in anymore probably because settings are lost. Now router can see new "heat-controller" network but I'm not able to connect because I'm away. After 1.8 -> 1.9 I've expected full reset, 1.9.0->1.9.1 this should not happen. Am I right?

proddy, 2019-09-17:
yes 1.9.0 -> 1.9.1 it should be backwards compatible. It's always a good idea to make a backup of your settings just in case.

If the EMS-ESP crashes 10 times in a row it will go into Access Point mode, this is probably what you're seeing. I'll be interesting to see if 1.9.1 fixes any of your problems. It's not 100% stable I know, and not 100% perfect either but its just me in my free time writing code so don't expect too much!

closed 2019-09-27 by proddy