emsesp/EMS-ESP #185: no thermostat data with Sieger ES73 since 1.9.1bx

SpaceTeddy, 2019-09-18:
Dear Paul,
After an update from master 1.9.0 to 1.9.1b4 or b5, the thermostat data for my Sieger ES73 is not read correctly. Via telnet, no temperature data is shown, via Web interface the Setpoint temperature is round about 4343°C, current temperature could be correct.
I expect, there is something with your heatcircuit change, and I would like to ask you which data/loggings you need to find the root cause. Hope I can help to fix this.

thx

proddy, 2019-09-18:
oops. I'll look into it. Does the ES73 support multiple heating circuits?

SpaceTeddy, 2019-09-18:
yes, the ES73 does support up to 4 circuits, i guess .)
The ES73 is more and less a mix of Buderus RC30 and RC35.

SpaceTeddy, 2019-09-19:
with the latest 1.9.1_b4 I can see the following:
[image]

[image]

[image]

any idea?

many thanks

SpaceTeddy, 2019-09-23:
I recompiled the b4 with:

``
// check if current temp sensor is unavailable
    if (EMS_RxTelegram->data[EMS_OFFSET_RC35StatusMessage_curr] != 0x7D) {
        EMS_Thermostat.hc[hc_num].curr_roomTemp = _toShort(EMS_OFFSET_RC35StatusMessage_curr); // is * 10
    }
``

and it works for me.

proddy, 2019-09-23:
wow, you fixed a bug!

proddy, 2019-09-23:
Can we close this now?

SpaceTeddy, 2019-09-23:
well, for the HC1 current room temperature basically yes, but the ESP detects two HC's and I have only HC1 in use.

proddy, 2019-09-23:
that shouldn't happen with the latest dev build anymore?

SpaceTeddy, 2019-09-23:
alright! :)

SpaceTeddy, 2019-09-30:
Dear Proddy,

since 1.9.1b8, the setpoint temp and the current temp is not shown in telnet and mqtt anymore. I've seen that you modified a lot between b7 and b8. Anything I can test with my ES73 to keep it running again?

thx,
chris

proddy, 2019-09-30:
is Tx working?

proddy, 2019-09-30:
oops, found the error!

SpaceTeddy, 2019-09-30:
yes, fixed!

closed 2019-09-30 by SpaceTeddy