emsesp/EMS-ESP #186: Boiler Temperature and Return Temperature

andriuskr, 2019-09-18:
Hello,

I have Buderus Logomax Plus GB1721i-24 with separate water tank.

Some values are missing in EMS gateway:
Boiler Temperature
Return Temperature
System Presure

Maybe i can fine tune something for this ?

And do i understand correctly, if i have separate water tank, connected to boiler to see its temperature, i cannot see shower info of water flow  ?

info
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is disabled, Solar Module is disabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=33, # CRC errors=0
  Tx: Last poll=3.332 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus GB172/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:06.08)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 46.9 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 5635 times
  Warm Water active time: 42 days 22 hours 16 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 5 C
  Current flow temperature: 28.2 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 45 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 20608 times
  Total burner operating time: 162 days 9 hours 44 minutes
  Total heat operating time: 119 days 11 hours 28 minutes
  Total UBA working time: 518 days 8 hours 52 minutes

Shower stats:
  Shower is off

proddy, 2019-09-18:
To be honest, I'm not sure! Could you turn on logging (telnet and `log v`) and send some examples of telegram 0x19 (UBAMonitorSlow). The temperature should be in that telegram.

andriuskr, 2019-09-18:
<--- UBAMonitorSlow(0x19)
(00:13:06.003) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(00:13:06.079) Boiler -> me, type 0x33, telegram: 08 0B 33 00 08 FF 32 FB 00 28 00 02 46 00 FF FF 00 (CRC=5E) #data=13
<--- UBAMonitorSlow(0x19)
(00:15:30.768) Boiler -> all, type 0x1C, telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4B) #data=25
(00:15:39.399) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=AC) #data=21
(00:15:39.620) Boiler -> all, type 0x34, telegram: 08 00 34 00 32 01 D5 01 D6 21 00 00 03 00 00 F1 78 00 16 03 00 80 00 (CRC=44) #data=19

andriuskr, 2019-09-18:
<--- UBAMonitorSlow(0x19)
(00:25:27.608) Boiler -> all, type 0x1C, telegram: 08 00 1C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=4B) #data=25
(00:25:27.849) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=AC) #data=21
(00:25:36.388) Boiler -> all, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B4 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=AC) #data=21
(00:25:36.609) Boiler -> all, type 0x34, telegram: 08 00 34 00 32 01 D6 01 D6 21 00 00 03 00 00 F1 78 00 16 03 00 80 00 (CRC=AC) #data=19

proddy, 2019-09-18:
Got any examples of "Boiler -> all, type 0x19". I need type 0x19 examples.

andriuskr, 2019-09-18:
<--- UBAMonitorFast(0x18)
(00:42:07.216) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=3E) #data=13
(00:42:07.597) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(00:42:07.688) Boiler -> me, type 0x19, telegram: 08 0B 19 00 80 00 80 00 80 00 00 00 00 00 00 50 80 03 91 88 00 00 00 02 A0 10 00 3A 7D 80 00 (CRC=4F) #data=27
<--- UBAMonitorSlow(0x19)

<--- UBAMonitorFast(0x18)
(00:42:22.190) Boiler -> all, type 0x19, telegram: 08 00 19 00 80 00 80 00 80 00 00 00 00 00 00 50 80 03 91 88 00 00 00 02 A0 10 00 3A 7D 80 00 (CRC=B1) #data=27
<--- UBAMonitorSlow(0x19)

<--- UBAMonitorFast(0x18)
(00:43:05.797) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=3E) #data=13
(00:43:06.169) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(00:43:06.255) Boiler -> me, type 0x19, telegram: 08 0B 19 00 80 00 80 00 80 00 00 00 00 00 00 50 80 03 91 88 00 00 00 02 A0 10 00 3A 7D 80 00 (CRC=4F) #data=27
<--- UBAMonitorSlow(0x19)

<--- UBAMonitorFast(0x18)
(00:43:21.844) Boiler -> all, type 0x19, telegram: 08 00 19 00 80 00 80 00 80 00 00 00 00 00 00 50 80 03 91 88 00 00 00 02 A0 10 00 3A 7D 80 00 (CRC=B1) #data=27
<--- UBAMonitorSlow(0x19)

<--- UBAMonitorFast(0x18)
(00:44:07.676) Boiler -> all, type 0x07, telegram: 08 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=3E) #data=13
(00:44:08.074) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(00:44:08.164) Boiler -> me, type 0x19, telegram: 08 0B 19 00 80 00 80 00 80 00 00 00 00 00 00 50 80 03 91 88 00 00 00 02 A0 10 00 3A 7D 80 00 (CRC=4F) #data=27
<--- UBAMonitorSlow(0x19)

<--- UBAMonitorFast(0x18)
(00:44:20.905) Boiler -> all, type 0x19, telegram: 08 00 19 00 80 00 80 00 80 00 00 00 00 29 00 50 81 03 91 88 00 00 00 02 A0 10 00 3A 7E 80 00 (CRC=39) #data=27
<--- UBAMonitorSlow(0x19)

proddy, 2019-09-18:
right, so the boiler temp is bytes 2 and 3 (a two byte short) starting at data position 2 in the telegram right after the header. Your values shows in all cases a x8000 which means 'unset'. Which is why the '?'.  I don't know why.

higgers, 2019-09-18:
This seems to be the same on my boiler:

```
(00:01:34.331) 0x08 -> all, type 0x19, telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 00 00 00 0F 3C 09 3B 6A 00 00 00 07 76 28 00 0A BF 80 00 (CRC=9D) #data=27
<--- UBAMonitorSlow(0x19)
```

I did notice that the onewire library is included in the code?  Does this mean it's possible to attache a temperature probe such as a DS18S20 to the Wemos and glue it to the return pipe to get the temperature?  And potentially add this to the MQTT message?

proddy, 2019-09-19:
@higgers yes, I added support for external Dallas sensors back in version 1.4. You simply daisy chain the sensors to the dallas_gpio pin (D1 default but can be set in telnet or via web). Values are shown in `info` and also sent as MQTT topic "sensors/temp_<num>/value" which is configurable too. Supports DS1822, DS18S20, DS18B20, DS1825 and also the parasite varieties.

I'll add a section to the Wiki as this is missing.

bbqkees, 2019-09-19:
If you have a Gateway you can connect those DS18B20 sensors to the connector on the side via the included cable:
[image]

If you are using the I/O pins of the Wemos directly don't forget the 4k7 pullup resistor for the sensor.

closed 2019-09-22 by proddy