emsesp/EMS-ESP #187: Dual AP and wifi client mode

mrfixit1, 2019-09-20:
Regression of issue 152 in v1.9.0. After setting the wifi client parameters and restarting, the AP is still active.

Wifi mode from system command:

ESP8266 System stats:

 [APP] EMS-ESP version: 1.9.0
 [APP] MyESP version: 1.2.0
 [APP] Build timestamp: 2019-09-20 20:13:04
 [APP] Uptime: 0 days 1 hour 3 minutes 48 seconds
 [APP] System Load: 6%
 [WIFI] Device is in AP mode with SSID ems-esp

Wifi mode and parameters from set command:
 
Stored settings:

  wifi_mode=client
  wifi_ssid=xxxxxxxx
  wifi_password=************

The open ems-esp shows in the PC's list of wifi networks. Anyone know how to turn it off for better security?

proddy, 2019-09-21:
I spent some time before researching this and to be honest have no idea why it happens. The easiest way is to disable AP completely in the code, perhaps with a compile option called -DNO_AP. If you want a quick hack comment out the line:

 `jw.enableAPFallback(true); // AP mode only as fallback`

in the function MyESP.cpp._wifi_setup()

proddy, 2019-09-22:
Could you turn on serial, connect it via USB to your PC and look at the serial monitor when it boots up? I'd like to see what is happening. If the EMS-ESP is unable to join the wifi network it will go back  to AP mode. I'll also add some more debug statements in the next beta build

mrfixit1, 2019-09-22:
Here's the serial output after restarting:

restart
* Restart ESP...

Exception (9):
              epc1=0x40201bac epc2=0x00000000 epc3=0x00000000 excvaddr=0x039f037                                                 a depc=0x00000000

                 >>>stack>>>

                            ctx: cont
                                     sp: 3ffffd70 end: 3fffffc0 offset: 01a0
                                                                            3fff                                                 ff10:  3fff6f9c 00000001 3fff1894 4020276b
                                            3fffff20:  0000000d 00000001 3fff189                                                 4 40206ab1
            3fffff30:  4025d437 3fff1bb8 3fff1bb8 4021fda4
                                                            3fffff40:  402124e8                                                  3fff3d50 3fff1bb8 3fff1c3c
                            3fffff50:  0000000d 3fff1bb8 3fff1894 40208749
                                                                            3fff                                                 ff60:  0000000d 3fff2288 3fff3e48 4022a1ba
                                            3fffff70:  00000000 3fff2288 3fff189                                                 4 3fff4024
            3fffff80:  3fff1c94 3fff3ee4 3fff1894 40208857
                                                            3fffff90:  3fffdad0                                                  3fff2288 3fff2288 4020e618
                            3fffffa0:  3fffdad0 00000000 3fff3ff4 402219ac
                                                                            3fff                                                 ffb0:  feefeffe feefeffe 3ffe8594 401007d9
                                            <<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

   * EMS-ESP version 1.9.0
[FS] System settings loaded
[FS] Custom config loaded
[WEB] Web server started
Warning! EMS bus communication disabled when Serial mode enabled. Use 'set seria                                                 l off' to start communication.
[WIFI] Connecting to SSID: MySSID
[WIFI] SSID  MySSID
[WIFI] CH    11
[WIFI] RSSI  -57
[WIFI] IP    192.168.nnn.nn
[WIFI] MAC   nn:nn:nn:nn:nn:nn
[WIFI] GW    192.168.nnn.nn
[WIFI] MASK  255.255.255.0
[WIFI] DNS   nn.nnn.nn.nn
[WIFI] HOST  ems-esp
[OTA] listening to ems-esp.local:8266
[SYSTEM] Last reset info: Fatal exception:9 flag:2 (EXCEPTION) epc1:0x40201bac epc2:0x00000000 epc3:0x00000000 excvaddr:0x039f037a depc:0x00000000
[SYSTEM] Serial port communication is enabled.
[TELNET] Telnet server started
[MQTT] Connecting to MQTT using user xxxxxxxx...
[MQTT] is connected
[MQTT] Subscribing to home/ems-esp/restart (PID 2)
[MQTT] Subscribing to home/ems-esp/start (PID 3)
[MQTT] Subscribing to home/ems-esp/thermostat_cmd_temp (PID 5)
[MQTT] Subscribing to home/ems-esp/thermostat_cmd_mode (PID 6)
[MQTT] Subscribing to home/ems-esp/thermostat_cmd_hc (PID 7)
[MQTT] Subscribing to home/ems-esp/thermostat_daytemp (PID 8)
[MQTT] Subscribing to home/ems-esp/thermostat_nighttemp (PID 9)
[MQTT] Subscribing to home/ems-esp/thermostat_holidayttemp (PID 10)
[MQTT] Subscribing to home/ems-esp/boiler_cmd_wwactivated (PID 11)
[MQTT] Subscribing to home/ems-esp/boiler_cmd_wwtemp (PID 12)
[MQTT] Subscribing to home/ems-esp/boiler_cmd_comfort (PID 13)
[MQTT] Subscribing to home/ems-esp/boiler_cmd_flowtemp (PID 14)
[MQTT] Subscribing to home/ems-esp/shower_timer (PID 15)
[MQTT] Subscribing to home/ems-esp/shower_alert (PID 16)
[MQTT] Subscribing to home/ems-esp/shower_coldshot (PID 17)

It takes a few minutes before the ems-esp appears on the LAN, despite the serial output showing that it connected immediately with the correct IP address set by a reserved rule in the router.

proddy, 2019-09-27:
I really don't know what is causing the AP to be alive while its still been given a fixed IP on your LAN.

mrfixit1, 2019-09-28:
Unfortunately commenting out the jw.enableAPFallback(true) line in the MyESP.cpp._wifi_setup() function (and recompiling and reflashing of course) didn't help, the ems-esp soft AP was still active after the client reconnected.

proddy, 2019-10-01:
So you still see both ems-esp as a wifi AP in your router and at the same time ems-esp as a hostname on your network with IP 192.168.x.x which you can telnet into

Can you use a laptop/pc and connect to the ems-esp wifi Access Point as well? If so which IP does it have?

Just to rule out its not a ghost entry in your router. I still don't see how a device can be an AP with 192.168.4.1 and also on the network with another IP address. Unless you have DHCP enabled and using the range 192.168.4.x/24 which would be very odd !

mrfixit1, 2019-10-01:
Almost: I see ems-esp as a wifi AP in my PC's list (not router). If I connect to it, the ems-esp AP IP is 192.168.4.1 and the DHCP server gives my PC's network card an IP a few numbers higher, all as expected.

The ems-esp is concurrently connected to my LAN which is definitely on a different subnet, not 4.1. With the PC back on the LAN, I can connect to the ems-esp with telnet or the new web interface using the IP address that I reserved for it in my router, again all as expected.

A scanner such as Acrylic or the Android Wifi Analyzer shows that the ems-esp AP is on the same channel as the LAN AP that it's also connected to.

So it's acting like something half way between a router and mesh.

proddy, 2019-10-02:
EMS-ESP uses the [justwifi library](https://github.com/xoseperez/justwifi). I suggest spending a few minutes creating a new PlatformIO project with the example code [here](https://github.com/xoseperez/justwifi/tree/master/examples/basic) and see if you can recreate the problem. If that works then it's something buggy in my code and we can do another deep dive.

mrfixit1, 2019-10-03:
I tried a merge of the basic.ino and utils.ino in the Arduino IDE with a static IP. The results were exactly as expected: my spare, test d1 mini connected to the network and no dual open AP appeared. So it looks to me like the justwifi library is OK.

Just to be sure, I then unplugged my live ems-esp and uploaded v1.9.0 to the test board. The results were repeatable: the dual mode reappeared. So it doesn't look like a hardware issue or upload glitch.

proddy, 2019-10-03:
wow, didn't expect that. ok, then I need to compare both and see what the problem is.

proddy, 2019-10-04:
made a really small change so the initializing sequence is exactly the same as JustWifi's basic demo which worked on your environment. Can you test? Latest 1.9.1b11

mrfixit1, 2019-10-04:
After uploading, firstly it proved very difficult to change the settings. Here's one example from a telnet session:

set serial on

Type 'restart' to activate Serial mode.
serial changed.
[FS] Failed to open system config for writing

Unknown set command or wrong number of arguments.

I tried a second board but the results were the same.

When after many attempts I finally succeeded in setting the wifi mode, ssid and password, it then proved very difficult to connect to the LAN wifi. The ems-esp just kept saying unable to connect and fell back to AP mode, despite being only 2m away from a LAN AP with a signal of around -40dBm.

After many resets this also eventually succeeded. The good news is: when all was running as it should, the ems-esp AP did not reappear!

proddy, 2019-10-05:
well that's some good news. I've noticed on some Wemos boards with PlatformIO that the reset doesn't work too well and it needs a hard reset (the little button on the side).

The problem with telnet and # commands, can you reproduce this? What are you using as the Telnet client? Again there are so many variations for example putty on Windows behaves differently to the Telnet native client on Windows10 and Termius on OSX and telnet with Ubuntu etc. Or did you just use the serial monitor from pio?

mrfixit1, 2019-10-06:
The issue with changing the settings is fully reproducible. My default telnet client is PuTTY on Windows 10, but the results were the same with the web interface on Windows 10, Linux and Android.

proddy, 2019-10-06:
can you describe again the problem?

mrfixit1, 2019-10-09:
The problem is now: difficulty changing the settings in v1.9.1b11.

Example
1. Hard reset, Windows 10 PuTTY serial port output:
* EMS-ESP version 1.9.1b11
[FS] Formatting filesystem...
[FS] Failed to format file system
[FS] File /myesp.json not found
[FS] Failed to open system config
[FS] Creating a new system config
[FS] Failed to open system config for writing
[FS] File /customconfig.json not found
[FS] Failed to open custom config
[FS] Failed to open custom config for writing
[FS] File /eventlog.json not found
[FS] Resetting event log
[WIFI] Creating access point
[WIFI] MODE AP
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   aa:bb:cc:dd:ee:ff
[SYSTEM] Disabling serial port communication.
Note: Serial output will now be disabled. Please use Telnet.

2. Connected to ems-esp AP, Windows 10 PuTTY Telnet session:
[WEB] Web server started
[TELNET] Telnet connection established

* Connected to: EMS-ESP version 1.9.1b11
* Device is in AP mode with SSID ems-esp
*
system

ESP8266 System stats:

 [APP] EMS-ESP version: 1.9.1b11
 [APP] MyESP version: 1.2.6
 [APP] Build timestamp: 2019-10-04 17:51:40
 [APP] Uptime: 0 days 0 hours 3 minutes 8 seconds
 [APP] System Load: 3%
 [WIFI] Device is in AP mode with SSID ems-esp
 [WIFI] WiFi MAC: aa:bb:cc:dd:ee:ff
 [MQTT] is disconnected
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_D1_MINI_PRO
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.1(cfd48f3)
 [SYSTEM] CPU chip ID: 0x6E3CB0
 [SYSTEM] Core version: 2_5_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: External System
 [SYSTEM] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
 [SYSTEM] Restart count: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 0
 [FLASH] Flash chip ID: 0x1840EF
 [FLASH] Flash speed: 80000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 16777216
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 793008
 [MEM] Max OTA size: 1298432
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 19448 bytes initially | 5752 bytes used (29%) | 13696 bytes free (70%)

set
Current settings:
  wifi_mode=ap
  wifi_ssid=
  wifi_password=
  mqtt_enabled=off
set wifi_mode client

Note: please 'restart' to apply new WiFi settings
wifi_mode changed.
[FS] Failed to open system config for writing

Unknown set command or wrong number of arguments.
set wifi_ssid MyLANSSID

Note: please 'restart' to apply new WiFi settings
wifi_ssid changed.
[FS] Failed to open system config for writing

Unknown set command or wrong number of arguments.
set wifi_password MyLANPW

Note: please 'restart' to apply new WiFi settings
wifi_password changed.
[FS] Failed to open system config for writing

Unknown set command or wrong number of arguments.
Error! Unable to read the EMS bus.
restart

3. PuTTY serial port again after soft reset:
* EMS-ESP version 1.9.1b11
[FS] Formatting filesystem...
[FS] Failed to format file system
[FS] File /myesp.json not found
[FS] Failed to open system config
[FS] Creating a new system config
[FS] Failed to open system config for writing
[FS] File /customconfig.json not found
[FS] Failed to open custom config
[FS] Failed to open custom config for writing
[FS] File /eventlog.json not found
[FS] Resetting event log
[WIFI] Creating access point
[WIFI] MODE AP
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   aa:bb:cc:dd:ee:ff
[SYSTEM] Disabling serial port communication.
Note: Serial output will now be disabled. Please use Telnet.

4. Connected to ems-esp AP, Windows 10 PuTTY Telnet session:
[WEB] Web server started
[TELNET] Telnet connection established

* Connected to: EMS-ESP version 1.9.1b11
* Device is in AP mode with SSID ems-esp
*
set
Current settings:
  wifi_mode=ap
  wifi_ssid=
  wifi_password=
  mqtt_enabled=off

The behaviour through the web interface is similar: after saving the changes, hitting restart and logging in again, nothing has changed.

Let me know if you think this is unrelated to the fix for the original #187 and needs a new issue.

proddy, 2019-10-09:
try:

1) refresh your repo with the latest 1.9.2dev branch. It's the same as 1.9.1 but at least we're working off the same version
2) in platformio.ini add `-DFORCE_SERIAL` to `custom_flags`
3) build firmware
4) upload
5) start serial monitor via USB. You should see messages like loading config file etc
6) do the `set` command again and see if it works
7) if it fails, your board is probably corrupted. do a `pio run -t erase` and repeat from step 4
8) if it still fails then we have a problem

BTW I trust you have a genuine Wemos D1 (and not a clone with a dodgy flash chip) and that the board parameter is correctly set in the platformio.ini file.

mrfixit1, 2019-10-09:
All of my Wemos D1 minis and pros came from their official AliExpress store.

This time it took only one attempt to change the wifi settings, but getting a connection to my LAN WAP is still difficult, even after resetting the board with pio run -t erase and uploading again.

When it finally did connect, the ems-esp AP disappeared, so the original #187 is (still) fixed, no regression.

BTW, I didn't mention this earlier because I thought it was unrelated, but for completeness: to get the code to compile I need to add lib_ignore = Time to my platformio.ini.

Here are the relevant lines from the serial port.

1. After initial upload - as expected:
             * EMS-ESP version 1.9.2b3
[FS] Checking file /myesp.json (326 bytes)
[FS] Serial is forced
[FS] System config loaded
[FS] Checking file /customconfig.json (202 bytes)
[FS] Custom config loaded
[FS] Checking file /eventlog.json (264/20840 bytes)
[FS] Event log is healthy
[WIFI] Creating access point
[WIFI] MODE AP
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   aa:bb:cc:dd:ee:ff
[SYSTEM] Serial port communication is enabled.
[WEB] Web server started
Warning! EMS bus communication disabled when Serial mode enabled. Use 'set serial off' to start communication.
[TELNET] in AP mode
[TELNET] Telnet server started
set

Current settings:

  wifi_mode=ap
  wifi_ssid=
  wifi_password=
  mqtt_enabled=off
  ...
  serial=on (this is always when compiled with -DFORCE_SERIAL)
  ntp_enabled=off

set wifi_mode client
Please 'restart' to apply new settings.
wifi_mode changed.
[FS] system config saved
set wifi_ssid MyLANSSID
Please 'restart' to apply new settings.
wifi_ssid changed.
[FS] system config saved
set wifi_password MyLANPW
Please 'restart' to apply new settings.
wifi_password changed.
[FS] system config saved

2. Soft restart:
restart
* Restart ESP...

Exception (9):
              epc1=0x40201fcc epc2=0x00000000 epc3=0x00000000 excvaddr=0x040602de depc=0x00000000

                 >>>stack>>>

                            ctx: cont
                                     sp: 3ffffd70 end: 3fffffc0 offset: 01a0
                                                                            3fffff10:  3fff6bec 3fff26f8 3fff2078 40202c47
                                            3fffff20:  0000000d 3fff26f8 00000001 4020a2dc
            3fffff30:  00000001 0000000a 3fff2674 4022bbdb
                                                            3fffff40:  4021b424 3fff3eb0 3fff2674 3fff2078
                            3fffff50:  0000000d 3fff2674 0000000d 4020d798
                                                                            3fffff60:  3fff2674 00000000 3fff3eb0 40237dd6
                                            3fffff70:  3fffdad0 00000000 3fff2078 3fff408c
            3fffff80:  3fffdad0 3fff3f4c 3fff2078 4020d8aa
                                                            3fffff90:  3fffdad0 00000000 3fff2d44 40215d41
                            3fffffa0:  3fffdad0 00000000 3ffe8594 4022e246
                                                                            3fffffb0:  feefeffe feefeffe feefeffe 40100b65
                                            <<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

   * EMS-ESP version 1.9.2b3
[FS] Checking file /myesp.json (341 bytes)
[FS] Serial is forced
[FS] System config loaded
[FS] Checking file /customconfig.json (202 bytes)
[FS] Custom config loaded
[FS] Checking file /eventlog.json (622/20808 bytes)
[FS] Event log is healthy
[WEB] Web server started
Warning! EMS bus communication disabled when Serial mode enabled. Use 'set serial off' to start communication.
[WIFI] Connecting to SSID: MyLANSSID
[WIFI] Could not connect to MyLANSSID
[WIFI] Creating access point
[WIFI] MODE AP
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   aa:bb:cc:dd:ee:ff
[SYSTEM] Serial port communication is enabled.
[TELNET] in AP mode
[TELNET] Telnet server started
set

Current settings:

  wifi_mode=ap
  wifi_ssid=MyLANSSID
  wifi_password=************
  mqtt_enabled=off
...
  serial=on (this is always when compiled with -DFORCE_SERIAL)
  ntp_enabled=off
  
3. Same again after subsequent hard resets:
             * EMS-ESP version 1.9.2b3
[FS] Checking file /myesp.json (341 bytes)
[FS] Serial is forced
[FS] System config loaded
[FS] Checking file /customconfig.json (202 bytes)
[FS] Custom config loaded
[FS] Checking file /eventlog.json (696/21776 bytes)
[FS] Event log is healthy
[WEB] Web server started
Warning! EMS bus communication disabled when Serial mode enabled. Use 'set serial off' to start communication.
[WIFI] Connecting to SSID: MyLANSSID
[WIFI] Could not connect to MyLANSSID
[WIFI] Creating access point
[WIFI] MODE AP
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   aa:bb:cc:dd:ee:ff
[SYSTEM] Serial port communication is enabled.
[TELNET] in AP mode
[TELNET] Telnet server started

4. Until after +/-5 hard resets it finally connected:
* EMS-ESP version 1.9.2b3
[FS] Checking file /myesp.json (341 bytes)
[FS] Serial is forced
[FS] System config loaded
[FS] Checking file /customconfig.json (202 bytes)
[FS] Custom config loaded
[FS] Checking file /eventlog.json (1288/21776 bytes)
[FS] Event log is healthy
[WEB] Web server started
Warning! EMS bus communication disabled when Serial mode enabled. Use 'set serial off' to start communication.
[WIFI] Connecting to SSID: MyLANSSID
[WIFI] SSID  MyLANSSID
[WIFI] CH    11
[WIFI] RSSI  -46
[WIFI] IP    192.168.nnn.nn
[WIFI] MAC   aa:bb:cc:dd:ee:ff
[WIFI] GW    192.168.xxx.xx
[WIFI] MASK  255.255.255.0
[WIFI] DNS   aa.bbb.cc.dd
[WIFI] HOST  ems-esp
[OTA] listening to ems-esp.local:8266
[MQTT] is disabled
[SYSTEM] Serial port communication is enabled.
[TELNET] Telnet server started

proddy, 2019-10-09:
>    1. Soft restart:

It remains a mystery. Perhaps more time is needed for the wifi to connect.
Try increasing the  #define MYESP_WIFI_CONNECT_TIMEOUT in MyESP.h. It’s
currently 10 seconds.

The Time lib is odd. Should not be needed.

mrfixit1, 2019-10-09:
Increasing the connect timeout to 30s has solved the connection issue. The ems-esp now connects reliably first time after every reset. Thanks for your help with this, much appreciated.

FYI these are the compile errors without the lib_ignore = Time:

Compiling .pio\build\debug\lib110\ESP Async WebServer\WebRequest.cpp.o
.pio\libdeps\debug\ESP Async WebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(tm*)':
.pio\libdeps\debug\ESP Async WebServer\src\WebHandlers.cpp:67:64: error: 'strftime' was not declared in this scope
   strftime (result,30,"%a, %d %b %Y %H:%M:%S %Z", last_modified);
                                                                ^
.pio\libdeps\debug\ESP Async WebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified(time_t)':
.pio\libdeps\debug\ESP Async WebServer\src\WebHandlers.cpp:73:60: error: 'gmtime' was not declared in this scope
   return setLastModified((struct tm *)gmtime(&last_modified));
 C                                                           ^
ompiling .pio\build\debug\lib110\ESP Async WebServer\WebResponses.cpp.o
.pio\libdeps\debug\ESP Async WebServer\src\WebHandlers.cpp: In member function 'AsyncStaticWebHandler& AsyncStaticWebHandler::setLastModified()':
.pio\libdeps\debug\ESP Async WebServer\src\WebHandlers.cpp:78:25: error: 'time' was not declared in this scope
   if(time(&last_modified) == 0) //time is not yet set
                         ^
*** [.pio\build\debug\lib110\ESP Async WebServer\WebHandlers.cpp.o] Error 1
=================================================== [FAILED] Took 119.62 seconds ===================================================

proddy, 2019-10-09:
re-opening so I can change the timeout for the wifi in case others have a similar issue.

@mrfixit1 which value worked for you? 20 seconds?

mrfixit1, 2019-10-10:
I tried 30s initially, but 20s also works OK with my config.

closed 2019-10-10 by proddy