emsesp/EMS-ESP #189: Supporting multiple ESP8266 board types

blacksocket, 2019-09-22:


blacksocket, 2019-09-22:
[image]

1.RST - Reset
2.ADC - Konwerter A/C
3.EN - Chip enable. HIGH - active
4.GPIO16; WAKE - DEEP SLEEP
5.GPIO14; HSPI_CLK
6.GPIO12; HSPI_MISO
7.GPIO13; HSPI_MOSI; UART0_CTS
8.VCC - 3.3V
9.CS0 - Chip select
10.MISO
11.GPIO9
12.GPIO10
13.MOSI
14.SCLK - Clock
15.GND
16.GPIO15; MTDO; HSPICS; UART0_RTS
17.GPIO2; UART1_TXD
18.GPIO0
19.GPIO4
20.GPIO5
21.RXD; UART0_RXD; GPIO3
22.TXD; UART0_TXD; GPIO1

blacksocket, 2019-09-23:
Mapping Wemos vs ESP8266:

PIN WEMOS | FUNKCJA | PIN ESP-8266
-- | -- | --
TX | TXD | TXD
RX | RXD | RXD
A0 |   | A0
D0 | IO: | GPIO16
D1 | IO, SCL | GPIO5
D2 | IO, SDA | GPIO4
D3 | IO, 10k Pull-up | GPIO0
D4 | IO, 10k Pull-up, BUILTIN_LED | GPIO2
D5 | IO, SCK | GPIO14
D6 | IO, MISO | GPIO12
D7 | IO, MOSI | GPIO13
D8 | IO, 10k Pull-down, SS | GPIO15
G | Ground | GND
5V | 5V | –
3V3 | 3.3V | 3.3V
RST | Reset | RST

susisstrolch, 2019-09-23:
@blacksocket it looks like you're using my old EMS-ESP gateway design.
You can change the hardware w/o big effort:

1. Replace the connection between "Extension Port" Pin2 by a 4.7k resistors soldered to the backside of the PCB
2. Connect "Extension Port" Pin2 with "Mode Port" Pin5
3. Connect "Extension Port" Pin13 with "Mode Port" Pin2

susisstrolch, 2019-09-23:
As a side node - it's not a problem of the ESP8266-12F, instead it's a flakyness of the old gateway design and easily visible (and fixing) by (re)viewing the gateway schema file.
So I'm closing this issue.

closed 2019-09-23 by susisstrolch

blacksocket, 2019-09-23:
Hi @susisstrolch nice to meet you ;-) Do you mean modify the connection like this one below?

[image]

Thank you!

susisstrolch, 2019-09-23:
Autsch... very bad typo from my side...
The 4.7k and also the EMS_RX must go to Pin4 (GPIO15) - not to UBUCK.

blacksocket, 2019-09-23:
Sure np. I am going to test this modification ;-) Many thanks!

blacksocket, 2019-09-23:
@susisstrolch it works fine after modification! Do you have maybe the redesigned pcb with this modification ?

susisstrolch, 2019-09-24:
@blacksocket Sorry, no - didn‘t find the time and still have enough old PCB laying around...