emsesp/EMS-ESP #190: Home Assistant - Climate thermostat not working

majdzik84, 2019-09-24:
Version 1.9.1b7

Climate configuration is not working properly.

When I try the following configuration I get an empty thermostat card. I also tried on HC2 - the same
```
 - platform: mqtt
   name: Thermostat HC1
   modes:
     - "auto"
     - "heat"
     - "off"

   mode_state_topic: "home/ems-esp/thermostat_data1"
   current_temperature_topic: "home/ems-esp/thermostat_data1"
   temperature_state_topic: "home/ems-esp/thermostat_data1"

   temperature_command_topic: "home/ems-esp/thermostat_cmd_temp1"
   mode_command_topic: "home/ems-esp/thermostat_cmd_mode1"

   mode_state_template: "{{ value_json.thermostat_mode1 }}"
   current_temperature_template: "{{ value_json.thermostat_currtemp1 }}"
   temperature_state_template: "{{ value_json.thermostat_seltemp1 }}"
```
[image]
When I delete the numbers indicating the circuit. I am able to switch HC1. When I add the numbers 1 or 2, there is no reaction.

proddy, 2019-09-24:
looks like a typo, try removing the 1s

```
   mode_state_template: "{{ value_json.thermostat_mode }}"
   current_temperature_template: "{{ value_json.thermostat_currtemp }}"
   temperature_state_template: "{{ value_json.thermostat_seltemp }}"
```

majdzik84, 2019-09-24:
After deleting the digits, the thermostat card appeared and shows the current temperatures.
However, I cannot change the temperature or mode. nothing happens.

The putty also does not display information about the change of mode or temperature

majdzik84, 2019-09-25:
unfortunately the bug has reappeared.
This time in NIGHT mode

[image]
**
```
(00:58:10.354) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
 (00:58:11.354) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 10 00 E9 00 00 00 00 00 00 00 00 11 05 00 (CRC=24) #data=16
(00:58:33.605) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 10 00 E9 00 00 00 00 00 00 00 00 11 05 00 (CRC=4C) #data=16
(00:58:28.480) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(00:59:28.729) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(01:01:29.478) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
(01:01:34.353) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 10 00 E9 00 00 00 00 00 00 00 00 11 05 00 (CRC=4C) #data=16

```
After switching to DAY mode, setpoint temp shows temperature day.
Return to night mode does not change - the day temperature is still shown
[image]

```
(01:11:11.218) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
(01:11:11.749) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 2D 39 26 00 28 00 02 00 05 05 2D 01 01 04 3C 06 39 01 00 3C FF 0D 05 05 03 00 (CRC=40) #data=27
(01:11:12.218) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 10 00 E8 00 00 00 00 00 00 00 00 11 05 00 (CRC=EC) #data=16
(01:11:12.625) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 03 10 16 26 0A 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 0D 05 05 02 02 (CRC=12) #data=27
(01:11:33.343) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=0B) #data=16
Publishing boiler data via MQTT
(01:11:37.719) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 10 00 E8 00 00 00 00 00 00 00 00 11 05 00 (CRC=84) #data=16
(01:12:12.092) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 04 03 00 7D 00 00 00 00 00 00 00 00 00 11 05 00 (CRC=63) #data=16
(01:12:13.092) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 10 00 E8 00 00 00 00 00 00 00 00 11 05 00 (CRC=EC) #data=16

```

proddy, 2019-09-25:
probably a bug. To be honest I never tested this :) Will do this week

majdzik84, 2019-09-25:
@Proddy You do a really great job.

I tested the modes quickly.

- Auto is ok
- Off is ok
- Heat is not ok (not setting thermostat mode to 1

[image]

Could you still check the status of the current program? On the thermostat there is always automatic mode (in earlier versions it was also).
It would be great if it showed if the mode currently turned on is auto, heat or off
[image]

proddy, 2019-09-25:
try 1.9.1b8 again. I added "heat"

as for HA showing the string, I can't change that as its a default component within Home Assistant. 

You could try using another card component, like https://github.com/nervetattoo/simple-thermostat which I've tried before. Its much more configurable. For example I added this to the lovelace UI file for my old RC30

```
         - type: custom:thermostat-card
           title: Living Room
           entity: climate.thermostat
           pending: 1
           chevron_size: 50
           no_card: false
           hvac:
             states:
               low: off
               manual: manual
               auto: auto
           attribute: operation_mode
           highlight_tap: true
```

closed 2019-09-27 by proddy