emsesp/EMS-ESP #191: Support for Buderus older devices

MarianMajor76, 2019-09-25:
Hi all,

I have a heating system consisting of the following:
- Logamax plus GB162 - 15
- Logamatic 4121 + MEC2
- Solar module FM443
- water tank Logalux PL 750 /2SW

EMS-ESP is running on wemos D1 mini with version 1.9.1b7. Due to older components those are not recognized by ems-esp. Some informations are detected (UBAMonitorWWMessage, UBAMonitorFast, UBAMonitorSlow) but not all. If it is possible to include those devices to code, specially solar module from which I would like to read some information abou solar collector temperature, flow intensity, total flow time, water tank temperature. 
On the EMS there are 3 devices by ID: "These device IDs are on the EMS Bus: 0x08 0x09 0x0C", but the "info" always says "Tx: no signal".

Is it possible to implement support for those devices?

All informations captured from telnet session is attached in log2.txt file. Don't want to paste it here as plaintext.
[log2.txt](https://github.com/proddy/EMS-ESP/files/3651551/log2.txt)

I can do fw upgrades, or trying some commands, etc and post the results.
Many thanks

proddy, 2019-10-02:
hi, could you try out the latest 1.9.1 build to see if detects anything. I think you have a mix of EMS 1.0 and EMS 2.0 (EMS+) so perhaps the tx_mode 2 will work. When we have Tx working we can work on adding the other devices if not too much work.

MarianMajor76, 2019-10-03:
hi proddy, compiled latest build, uploaded to esp an set TX mode to 2. Now 3 devices was recognized on the bus:

These device IDs are on the EMS Bus: 0x08 0x09 0x0C and 2 were recognized by EMS-ESP as:                                                                                     Nefit Topline Compact/Buderus GB162 (DeviceID:0x08 ProductID:115 Version:05.08)                                         BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.03)

That's a good sign, I'm really happy. But what is the 3rd device - 0x0C? Could it be MEC2 which is on the bus too and via this device I can see informations about boiler, solar, water tank, etc. and of course I can set parameters too. There is a lot of communication between boiler and 0x0C.

One thing I had to change in code - EMS_POLL_TIMEOUT set to 15seconds instead of 5sec. The reason was the ems_getTxCapable method. When the timeout was 5sec, "Tx: no signal" was displayed in info command. But after changing it to 15sec, there is som information about Tx "Tx: Last poll=0.75 seconds ago, # successful write requests=0". The EMS_Sys_Status.emsPollFrequency in my situation is about 7 to 10sec. Never 5sec or less.

But now Boiler stats is complete - everywhere is some info. see attachment.

The question is - where are the informations about my solar system?
On MEC2 i can see for example the total boiler time. With "log v" enabled I can clearly see the packets where the time is transmitted on the bus. But the total time for solar is not on the bus - not as readable as totoal biler time. Really don't know how the MEC2 reads the infos from solar. 

[log.txt](https://github.com/proddy/EMS-ESP/files/3687576/log.txt)

thank you for your work!

proddy, 2019-10-03:
Interesting change on the EMS_POLL_TIMEOUT. There must be some lag I think. I'll make the change permanent anyway as it doesn't do too much harm.

I haven't come across the device ID 0x0C but from the logs it does indeed look like an MEC2 Control Unit.

I'll need some more time to decipher the telegrams.

proddy, 2019-10-03:
and you can see from

```
(00:23:52.367) Sending read of type 0x02 to 0x0C, telegram: 0B 8C 02 00 20 (CRC=9C)
(00:23:52.437) 0x0C -> me, telegram: 0C 0B 02 00 (CRC=48)
```

that the MEC2 doesn't respond to a Version command, which is why its not detected.

MarianMajor76, 2019-10-04:
Yes, the MEC2 don't want to send informations abou him, he is just requiring from boiler. But how does the mec2 knows about solar. Don't know what exactly is "Solar module FM443" and if it is ems device, but probably not. Or which device know something abou solar parameter - boiler, mec2? I think we miss some other device on the bus. If 0x0c is the mec2, from where are solar informations?
if you need some debuging or try something, no problem, just let me know. I'm watching to debug outputs, but don't know what to try else.

proddy, 2019-10-04:
I would expect the Buderus SolarModule FM443 to be identified as type 0x30 like the others. Not sure it's not.

Also I noticed Tx is not working either which is why your queue is filling up.

MarianMajor76, 2019-10-04:
Yes, Tx is not working...still "# successful write requests=0" from yesterday to now. But I did not send write requests. Only read requests - type 2 -> version telegram, as you noticed above. If write requests means requests where I want to set something (boiler temperature), then this is ok, did not send that kind of telegram. I only need to read, but not only from biler, but from solar too.

proddy, 2019-10-04:
ok, but we should get Tx working. Tx is for sending telegrams and used at startup to query what is on the EMS bus. This is why some of your devices are not activated. The "write request" is for when values are sent e.g. to change settings on a boiler or thermostat.

MarianMajor76, 2019-10-04:
OK, agree. But now it is a SW or HW issue? All 4 (rx, tx, gnd, 3.3v) wires from esp are connected to bbqkees board and the board is connected to the boiler jack. bbqkees board is configured as described on his pages, esp is powered via phone charger. Checked many times and it seems ok - the board is receiving data. But can;t send...

moustic999, 2019-10-04:
There is so many issue to the tx... don t panic.
The fm443 is an extension for logamatic 4000 series, which one do you have? Some are ems, not all....

-------- Message d'origine --------

proddy, 2019-10-04:
@MarianMajor76 yes start with following the troubelshooting guide in the wiki. try different tx_mode settings and see if anything is sent using the `send` command from telnet  (with verbose logging on) and go from there.

MarianMajor76, 2019-10-04:
moustic999 - the logamatic is Logamatic 4121 + MEC2. It should be ems capable - it is. If there are some fm443 version, how can I find which version is my? Slovak buderus pages where is fm443 and other modules https://www.buderus.com/sk/sk/ocs/komercne-budovy/moduly-logamatic-4000-757585-c/

@proddy - will try. Did some experiments with older firmwares, will try again with this version.

moustic999, 2019-10-04:
Yes this one is ems capable.

Only 1 version for fm443,

Mec2 is the command unit, it should have an address.... you can try to monitor the bus and change temp on mec, messages triggered will tell you how address

-------- Message d'origine --------

MarianMajor76, 2019-10-04:
OK, tried setting the "Warm Water selected temperature:" on MEC2 and definitelly the addres for mec2 is 0x0C. The temperature was set to 45C, then changed to 49C and again back to 45C. From log it is clear, that mec sends the type 0x33 to boiler. The interesting lines:

....
(22:24:21.835) 0x0C -> Boiler, type 0x33, telegram: 0C 08 33 00 08 FF 31 FB 00 19 FF 04 46 00 FF (CRC=37) #data=11
(22:24:21.917) 0x0C -> Boiler, type 0x35, telegram: 0C 08 35 00 11 11 (CRC=9B) #data=2
(22:24:22.014) 0x0C -> Boiler, type 0x1B, telegram: 0C 08 1B 00 00 (CRC=EC) #data=1
(22:24:22.121) 0x0C -> Boiler, type 0x1A, telegram: 0C 08 1A 00 1D 64 64 00 (CRC=A6) #data=4
(22:24:22.236) Boiler -> all, type 0x33, telegram: 08 00 33 00 08 FF 31 FB 00 19 FF 04 46 00 FF (CRC=37) #data=11
<--- UBAParameterWW(0x33)
....

Then the bolier send type 0x34 to all with temp set. And again - when setting back to 45C the sequence is the same with 2d=45C temperature.

see attached mec2.txt
[mec2.txt](https://github.com/proddy/EMS-ESP/files/3691524/mec2.txt)

MarianMajor76, 2019-10-04:
@proddy - tried "send" command with differen tx mode like the wiki said. See attached files. I think, the tx_mode=2 is the right one, maybe the 3, but not 1.

[tx_3.txt](https://github.com/proddy/EMS-ESP/files/3691674/tx_3.txt)
[tx_2.txt](https://github.com/proddy/EMS-ESP/files/3691675/tx_2.txt)
[tx_1.txt](https://github.com/proddy/EMS-ESP/files/3691676/tx_1.txt)

with tx_mode 2 and increades timeout to 15s I have informations on ems dashboard with "EMS Bus Connected with both Rx and Tx active."

[system.txt](https://github.com/proddy/EMS-ESP/files/3691699/system.txt)

proddy, 2019-10-05:
yup, looks like tx_mode 2 works for you. Is the goal now to read data from the Solar Module?

MarianMajor76, 2019-10-05:
yes, informations from solar - mec2 unit shows these informations about solar:
- total working time
- collector temperatues
- tank (logalux) temperature
- solar pump - off/low/high

but there infos about aoutdoor temperatue (senzor outside), indoor temperature (senzor in mec2).
Maybe the solar working time will help - now it is 9128:37 hours and did not chamged from yesterdays logs But those information must be transmitted in different way as bioler times. 

If you could decode somethimg, that would be very helpfull (at least for me) and maybe to others.

proddy, 2019-10-05:
honestly, I don't really have the time. What you could do is take the logic for the SM50 by looking at the code and the bits I added to the wiki and see if you can find any similarities with the 0x0C telegram. When you know where the data is located I can help add the coding.

MarianMajor76, 2019-10-06:
OK, a little more searching and I think, this is the end, for me :)
Logamatic 4121 comes with CM431 controller module, ZM424 central module, FM455 function module (KSE1 for controlling one EMS boiler or wall-mounted boiler with UBA1.5), MEC2 control pad.
And MEC2 is NOT connected via EMS, but via ECOCAN bus. So the 0x0C address is not MEC2, but I think, it is a FM455 (or ZM OR CM) address, which is on both busses - ECOCAN and EMS. Everything I see o MEC2 is from CAN bus, not EMS. Everything I set on MEC2 is set onCAN bus and via FM455 it is sent to EMS UBA device - boiler.
So the result is - if I want to read and log informations seen on MEC2 (primarily solar infos), I need to connect to CAN bus, not EMS bus. 
Thanks for everyone to inspired me and I will go further - searching for CAN bus and reading from that. 
Will definitelly follow this great project.

I think, this thread can be closed :)

proddy, 2019-10-06:
ok, let us know if you find anything we can add to the project.

closed 2019-10-06 by proddy

jandrassy, 2020-04-25:
Nazdar @MarianMajor76.  I have the Logamatic 4000 too. My primary goal is to monitor the boiler so ems-esp will do, but I want to ask if you could do something with MEC and CAN?

@proddy,  thank you for this project. I have the mentioned Tx problems. I got the BBQKees gateway with 1.9.4 release. Is the EMS_POLL_TIMEOUT changed as you mentioned in comments above?

MarianMajor76, 2020-04-25:
Nazdar Juraj, EMS-ESP worked for me, but my solar is not ems friendly :) But thanks to Peter Holzleitner and his project (https://holzleitner.com/el/buderus-monitor/index-en.html) you can monitor your MEC2. My goal was only read informations from the MEC2 display, not write to the bus, or setting some parameters, but I think it is possible too. Peter's project is in python, my is for arduino and you can find it on https://github.com/MarianMajor76/ArduinoMec2.