emsesp/EMS-ESP #193: What values used for hot water and central heating?

kroon040, 2019-10-04:
Hi,

With the new moduline 3000 thermostat, when using the hot tap water en central heating is not correct  I think. What values does the use?

Gas = on , and burner on, etc.

regards Richard

proddy, 2019-10-04:
yes, this is on my todo list and its very dependent on the boiler. See https://github.com/proddy/EMS-ESP/wiki/Wish-List

The code is in ems.cpp and says:

```
/**
 * Check if hot tap water or heating is active
 * using a quick hack for checking the heating. Selected Flow Temp >= 70
 */
void _checkActive() {
    // hot tap water, using flow to check instead of the burner power
    if (EMS_Boiler.wWCurFlow != EMS_VALUE_INT_NOTSET && EMS_Boiler.burnGas != EMS_VALUE_INT_NOTSET) {
        EMS_Boiler.tapwaterActive = ((EMS_Boiler.wWCurFlow != 0) && (EMS_Boiler.burnGas == EMS_VALUE_INT_ON));
    }

    // heating
    if (EMS_Boiler.selFlowTemp != EMS_VALUE_INT_NOTSET && EMS_Boiler.burnGas != EMS_VALUE_INT_NOTSET) {
        EMS_Boiler.heatingActive = ((EMS_Boiler.selFlowTemp >= EMS_BOILER_SELFLOWTEMP_HEATING) && (EMS_Boiler.burnGas == EMS_VALUE_INT_ON));
    }
}
```

basically:

* tap water is on if the warm water flow is on and the gas is burning.
* heating is active if the selected flow temperature is greater than 70 degrees and the gas is also on
 
all these params can be seen in `info`. Would be interesting to see what values you have so we can configure for heatpumps.

kroon040, 2019-10-04:
De moduline 3000 has also an icon for hot water, so I can check , if the icon appears a debug of the thermostat, and when disappear also.

I will check some , and will post it

kroon040, 2019-10-04:
first is heating on , second off. Post later hot tap water

```
 LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=14227, # CRC errors=9
  Tx: Last poll=2.589 seconds ago, # successful write requests=6

Boiler stats:
  Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:06.01)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Intelligent
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 32.2 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 36345 times
  Warm Water active time: 47 days 6 hours 44 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 34 C
  Current flow temperature: 39.6 C
  Return temperature: 33.1 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 75 %
  Burner current power: 28 %
  Flame current: 26.8 uA
  System pressure: 1.6 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 70 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 39.4 C
  Pump modulation: 30 %
  Burner # starts: 60131 times
  Total burner operating time: 207 days 15 hours 37 minutes
  Total heat operating time: 160 days 8 hours 53 minutes
  Total UBA working time: 2004 days 12 hours 25 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000 (ProductID:158 Version:11.10)
  Thermostat time is 20:12:25 04/10/2019
  Heating Circuit 1
   Current room temperature: 19.8 C
   Setpoint room temperature: 20.0 C
   Mode is set to auto

(23:43:07.702) Thermostat -> all, type 0x06, telegram: 10 00 06 00 13 0A 14 04 0D 15 04 01 10 FF 00 (CRC=FA) #data=11
Starting scheduled query from EMS devices
(23:43:09.305) Thermostat -> me, type 0x01A5, telegram: 10 0B FF 00 01 A5 00 C6 21 28 24 00 28 21 00 89 07 03 01 00 89 00 A3 00 00 11 01 03 07 BB 00 (CRC=83) #data=25
Publishing boiler data via MQTT
(23:43:09.438) Thermostat -> me, telegram: 10 0B FF 00 01 A6 (CRC=91)
(23:43:09.711) Thermostat -> all, type 0x01A5, telegram: 10 00 FF 04 01 A5 24 (CRC=40) #data=1
(23:43:10.457) Thermostat -> me, telegram: 10 0B FF 00 01 A7 (CRC=90)
(23:43:10.959) Thermostat -> me, telegram: 10 0B FF 00 01 A8 (CRC=9F)
(23:43:11.226) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 13 0A 14 04 0D 18 04 01 10 FF 00 (CRC=06) #data=11

EMS-ESP system stats:
  System logging set to Thermostat only
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected
  Rx: # successful read requests=14247, # CRC errors=9
  Tx: Last poll=2.390 seconds ago, # successful write requests=6

Boiler stats:
  Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:06.01)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Intelligent
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 32.5 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 36345 times
  Warm Water active time: 47 days 6 hours 44 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 5 C
  Current flow temperature: 34.2 C
  Return temperature: 34.3 C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.6 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 70 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 34.4 C
  Pump modulation: 30 %
  Burner # starts: 60132 times
  Total burner operating time: 207 days 15 hours 38 minutes
  Total heat operating time: 160 days 8 hours 54 minutes
  Total UBA working time: 2004 days 12 hours 27 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000 (ProductID:158 Version:11.10)
  Thermostat time is 20:14:26 04/10/2019
  Heating Circuit 1
   Current room temperature: 19.8 C
   Setpoint room temperature: 20.0 C
   Mode is set to auto

(23:44:40.973) Thermostat -> Boiler, type 0x1A, telegram: 10 08 1A 00 00 00 64 (CRC=A0) #data=3
(23:44:41.028) Thermostat -> all, type 0x0167, telegram: 10 00 FF 00 01 67 00 00 (CRC=AB) #data=2
(23:44:41.441) Thermostat -> all, type 0x01A5, telegram: 10 00 FF 00 01 A5 00 C6 21 28 00 00 28 21 00 88 07 03 01 00 88 00 A4 00 00 11 01 03 07 BB 00 (CRC=0B) #data=25
Publishing boiler data via MQTT
(23:44:41.879) Thermostat -> all, type 0x01A5, telegram: 10 00 FF 19 01 A5 08 04 00 00 00 00 FF 00 4B 00 3C 01 FF 01 03 (CRC=E6) #data=15
(23:44:42.090) Thermostat -> all, type 0x021D, telegram: 10 00 FF 00 02 1D 00 00 0A 07 (CRC=02) #data=4
(23:44:50.668) Thermostat -> all, type 0x01A5, telegram: 10 00 FF 0D 01 A5 00 87 00 A5 (CRC=F2) #data=4
Publishing boiler data via MQTT
(23:44:50.861) Thermostat -> all, type 0x01A5, telegram: 10 00 FF 08 01 A5 00 87 (CRC=8F) #data=2

```

kroon040, 2019-10-04:
Looks like hot tap water is ok, but shower, not. What does shower check?

Tap water and?
 Warm Water current temperature: 62.1 C?
 Current flow temperature: 63.3 C

proddy, 2019-10-05:
shower logic is very basic: the hot tap water must be on for minimum 2 minutes

kroon040, 2019-10-07:
Hi,

Did some checking for heating, and looks like that with the new thermostat the same boiler also changes the selected flow temperature. Mostly he starts heating with selected flow temperature 70, and mostly after 4-5 min, het falls back to selected flow around 37 degrees

I think for the central heating for ems+ , the selected flow must be >30 degrees and gas on.

Below some info when central heating is active

```
`EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=21297, # CRC errors=10
  Tx: Last poll=2.390 seconds ago, # successful write requests=11

Boiler stats:
  Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:06.01)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Intelligent
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 41.8 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 36374 times
  Warm Water active time: 47 days 8 hours 22 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 38 C
  Current flow temperature: 37.2 C
  Return temperature: 30.2 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 75 %
  Burner current power: 28 %
  Flame current: 27.0 uA
  System pressure: 1.6 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 70 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 37.6 C
  Pump modulation: 31 %
  Burner # starts: 60206 times
  Total burner operating time: 207 days 19 hours 56 minutes
  Total heat operating time: 160 days 11 hours 34 minutes
  Total UBA working time: 2007 days 1 hours 3 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000 (ProductID:158 Version:11.10)
  Thermostat time is 08:45:16 07/10/2019
  Heating Circuit 1
   Current room temperature: 19.5 C
   Setpoint room temperature: 20.0 C
   Mode is set to auto

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=21387, # CRC errors=10
  Tx: Last poll=3.939 seconds ago, # successful write requests=13

Boiler stats:
[... 245 lines cut ...]
  Flame current: 27.4 uA
  System pressure: 1.6 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 70 C
  Boiler circuit pump modulation max power: 70 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 35.7 C
  Pump modulation: 30 %
  Burner # starts: 60220 times
  Total burner operating time: 207 days 20 hours 42 minutes
  Total heat operating time: 160 days 12 hours 13 minutes
  Total UBA working time: 2007 days 6 hours 26 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000 (ProductID:158 Version:11.10)
  Thermostat time is 14:07:10 07/10/2019
  Heating Circuit 1
   Current room temperature: 19.8 C
   Setpoint room temperature: 20.0 C
   Mode is set to auto

Shower stats:
  Shower is offa
  `
```

kroon040, 2019-10-11:
is this possible to change in the build?

I think for the central heating for ems+ , the selected flow must be >30 degrees and gas on. 

regards Richard

proddy, 2019-10-12:
ok, will add and see how this effects other non EMS+ boilers

proddy, 2019-10-12:
in 1.9.2b6

closed 2019-10-12 by proddy