emsesp/EMS-ESP #194: WEB interface loading slow/fail;

majo551, 2019-10-05:
Very often the WEB based interface does not load, or loads only as simplified unusable page.  Actually I saw the full web interface only twice...

What can be the route cause? 

The ESP module used is ESP 8266 Witty Cloud (ESP12E)

Telnet works fine, the system output:

> * EMS-ESP version 1.9.0
[FS] System settings loaded
[FS] Custom config loaded
[WEB] Web server started
[UART] Opened Rx/Tx connection
Starting auto discover of EMS devices...
Started scan on EMS bus for known devices
[WIFI] Connecting to SSID: xxxx
[WIFI] SSID  xxxx
[WIFI] CH    11
[WIFI] RSSI  -73
[WIFI] IP    10.3.1.40
[WIFI] MAC   3xxxx
[WIFI] GW    1xxx
[WIFI] MASK  255.255.255.0
[WIFI] DNS   1xxx
[WIFI] HOST  ems-esp
[OTA] listening to ems-esp.local:8266
[SYSTEM] Last reset info: Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4023ae2a epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[MQTT] is disabled
[SYSTEM] Disabling serial port communication.
[NTP] NTP internet time enabled via server pool.ntp.org

>  [WIFI] WiFi signal strength: 54%
>  [WIFI] WiFi MAC: 3.....
>  [MQTT] is disconnected
>  [NTP] Time in UTC is 00:04:51
>  [SYSTEM] System is Stable
>  [SYSTEM] Board: PLATFORMIO_ESP12E
>  [SYSTEM] CPU frequency: 80 MHz
>  [SYSTEM] SDK version: 2.2.1(cfd48f3)
>  [SYSTEM] CPU chip ID: 0x03C8A5
>  [SYSTEM] Core version: 2_5_2
>  [SYSTEM] Boot version: 31
>  [SYSTEM] Boot mode: 1
>  [SYSTEM] Last reset reason: Exception
>  [SYSTEM] Last reset info: Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4023ae2a epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
>  [SYSTEM] Restart count: 0
>  [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
>  [SYSTEM] rtcmem 00: 1163087990
>  [SYSTEM] rtcmem 01: 65536
>  [FLASH] Flash chip ID: 0x164020
>  [FLASH] Flash speed: 40000000 Hz
>  [FLASH] Flash mode: DOUT
>  [FLASH] Flash size (CHIP): 4194304
>  [FLASH] Flash size (SDK): 4194304
>  [FLASH] Flash Reserved: 4096
>  [MEM] Firmware size: 729120
>  [MEM] Max OTA size: 2408448
>  [MEM] OTA Reserved: 16384
>  [MEM] Free Heap: 20064 bytes initially | 8736 bytes used (43%) | 11328 bytes free (56%)

page result attached:
[image]

proddy, 2019-10-05:
I'm moving 1.9.1 to master - please try that. It has a few improvements but nothing major in the web area. I have noticed some issues with certain ESP8266 chips (ones branded from DO-IT for example) that cause strange things to happen with the web. I can also see your wifi strength is 50% which may also cause it.

majo551, 2019-10-05:
I will try the new release, WIFI should not be an issue, I moved and tested the unit later also placing it 30 cm away from AP... The same result

proddy, 2019-10-05:
you're right - web is not working anymore. I think has to do with the update from the web libraries. I'm looking into this now

majo551, 2019-10-05:
confirmed. Now I loaded the latest release , no web page at all . I tried both ESP WITTY CLOUD and WeMos D1 R1 (mini) modules. No Web interface now.

proddy, 2019-10-05:
can you try and rebuild again?

majo551, 2019-10-05:
well, rebuilt, but no web page comes up for me with D1

> * EMS-ESP version 1.9.1
[FS] Checking file /myesp.json (size 308 bytes)
[FS] System config loaded
[FS] Checking file /customconfig.json (size 202 bytes)
[FS] Custom config loaded
[FS] Event log is healthy
[WIFI] Creating access point
[WIFI] MODE AP
[WIFI] SSID  ems-esp
[WIFI] IP    192.168.4.1
[WIFI] MAC   82xxxC5
[SYSTEM] Disabling serial port communication.
Note: Serial output will now be disabled. Please use Telnet.

[image]

proddy, 2019-10-06:
I rolled back to an earlier version of the AsyncWebServer library. The latest update 1.2.3 from a few days ago made it all go very unresponsive. It should be ok now. Note there are known issues with these external libraries regarding Chrome on Android and Safari on iOS due to way the web socket authentication is handled.

majo551, 2019-10-06:
Confirmed: Works with IE, after the  initial  setup in AP mode  **ON PC USING IE** I can see the web page also in Android Chrome when ESP set to Client mode. I was not able to access the ESP in AP mode with Android Chrome.  Tested on a WeMos D1 unit.

lmdecroos, 2019-10-06:
When using Firefox or safari I'm unable to connect to the Web Page when wifi is in client mode
When using Chromium on my Raspberry Pi then I'm able to connect and login

When using wifi in AP mode then i'm able to connect to the 192.168.4.1 webpage  using safari on my iPad

This is since the at least since the 1.9.1b6, but was working before

proddy, 2019-10-06:
Ok, I've no issues with Chrome on both iPad, OSX and Windows10 - but will do some more testing.

proddy, 2019-10-06:
tested with Chrome and Firefox from Windows10 and MacBook/OSX and both work fine. Doesn't work on the latest iOS and iPadOS in any browser (so no iphones or ipads). I don't have Android but I remember bbqkees telling me it works.

proddy, 2019-10-09:
tested on 5 boards, from Wemos D1 clones (with AI-Thinker and DO-IT chips), generic ESP8266 dev board, Nodemcuv2 and Nodemcu0.9 and cannot reproduce.

closed 2019-10-09 by proddy

mellbo, 2019-11-04:
I have this problem too. I modifyed line 1950 from "ems-esp.cpp"
`do_publishShowerData(); // publish to MQTT`
with this one
```
if ((ems_getBusConnected()) && myESP.isMQTTConnected() && EMSESP_Settings.publish_time) {
    do_publishShowerData(); // publish to MQTT
}
```
then problem has been  dissapear.

proddy, 2019-11-05:
totally not related. If your web is slow either you're using a bad ESP chip or not building the firmware correctly.

mellbo, 2019-11-05:
> totally not related. If your web is slow either you're using a bad ESP chip or not building the firmware correctly.

i'm sure you are right because you made this project. I don't know.. but for me it works...