emsesp/EMS-ESP #195: detailed information about type 0x35 (flags)?

SpaceTeddy, 2019-10-05:
dear community,

one of my wishes would be to have control the warm water one-time loading of the ww buffer. This function can be activated by the the water tap symbol of the RC20 / ES73 /RC3x ... 
my logging shows, that my RC20RF send every minute a status command to the boiler which is usually 
```
(19:37:19.169) 0x18 -> Boiler, type 0x35, telegram: 18 08 35 00 00 00 (CRC=1A) #data=2
```
if I press the water tap button the message changes to:
```
(19:12:20.932) 0x18 -> Boiler, type 0x35, telegram: 18 08 35 00 22 22 (CRC=7C) #data=2
```
if I press again to stop the ww heating it sends:
```
(19:36:18.042) 0x18 -> Boiler, type 0x35, telegram: 18 08 35 00 02 00 (CRC=1E) #data=2
```
... but sometimes ...
```
(07:25:42.224) 0x18 -> Boiler, type 0x35, telegram: 18 08 35 00 02 22 (CRC=3C) #data=2
```

it seems to me, that the first databyte enables or disables the function 0x22=enable, 0x02=disable, but I don not really understand the second databyte, which is sometimes 0x22 and sometimes 0x00? 

has someone detailed information about type 0x35?

thanks

proddy, 2019-10-06:
@SpaceTeddy this was on my feature todo list to implement next. See https://github.com/proddy/EMS-ESP/issues/94. Perhaps work together with vanbogaertetienne

SpaceTeddy, 2019-10-07:
hi proddy,
I try to create a demo version with the OneTime water heating command, but I fail to add an extra mqtt boiler_cmd Topic. tried to use the same like the wwactivation, but in telnet mode, I never receive a mqtt command.
With mqttlog telnet command I see the new ``Topic:home/ems-esp/boiler_cmd_wwonetime`` MQTT subscription, but seems that it is not received.
I added also 
```
        // wwOneTime
        if (strcmp(topic, TOPIC_BOILER_CMD_WWONETIME) == 0) {
            if (message[0] == '1' || strcmp(message, "on") == 0) {
                ems_setWarmWaterOnetime(true);
            } else if (message[0] == '0' || strcmp(message, "off") == 0) {
                ems_setWarmWaterOnetime(false);
            }
            return;
        }
```
in the ems-esp.cpp, but no success.

ems.cpp
```
/**
 * Activate / De-activate One Time warm water 0x35
 * true = on, false = off
 */
void ems_setWarmWaterOnetime(bool activated) {
    myDebug_P(PSTR("Setting boiler warm water OneTime loading %s"), activated ? "on" : "off");

    _EMS_TxTelegram EMS_TxTelegram = EMS_TX_TELEGRAM_NEW; // create new Tx
    EMS_TxTelegram.timestamp       = millis();            // set timestamp
    EMS_Sys_Status.txRetryCount    = 0;                   // reset retry counter

    EMS_TxTelegram.action        = EMS_TX_TELEGRAM_WRITE;
    EMS_TxTelegram.dest          = EMS_Boiler.device_id;
    EMS_TxTelegram.type          = EMS_TYPE_UBAFlags;
    EMS_TxTelegram.offset        = EMS_OFFSET_UBAParameterWW_wwOneTime;
    EMS_TxTelegram.length        = EMS_MIN_TELEGRAM_LENGTH;
    EMS_TxTelegram.type_validate = EMS_ID_NONE;               // don't validate
    EMS_TxTelegram.dataValue     = (activated ? 0x22 : 0x02); // 0x22 is on, 0x02 is off for RC20RF

    EMS_TxQueue.push(EMS_TxTelegram);
}
```

Any idea what else to add to create an additional boiler_cmd?
thx

proddy, 2019-10-08:
did you subscribe to it? Look at function MQTTCallback() in `ems-esp.cpp`

SpaceTeddy, 2019-10-08:
yes, sure. You can chack it in my fork.
https://github.com/SpaceTeddy/EMS-ESP/blob/master/src/ems-esp.cpp

proddy, 2019-10-08:
looked briefly at your code additions and it should work. Add some more debug statements, like commend out line 347 in MyESP.cpp

SpaceTeddy, 2019-10-08:
already done this, but also here no incomming message. It is strange to me, because the other commands like wwactivate work well with raspi mosquitto and MQTT.fx.
Anyhow, I will start a pull request in the next days and hopefully you can xcheck it with your hardware/software setup.

thx

proddy, 2019-10-08:
ok. You could also test with MQTT Explorer (which is what I do) and force a Publish

SpaceTeddy, 2019-10-08:
checked, but still no success. :(

proddy, 2019-10-09:
I'm testing with your branch and found the problem - not in your code - the number of subscriptions is limited. I'll looking into a fix

proddy, 2019-10-09:
the problem is that there are too many Subscribes filling up the TCP stack on the ESP8266. A workaround for now is to comment on the loop in MQTTCallback() which subscribes to all the Heating Circuits. Until I've fixed https://github.com/proddy/EMS-ESP/issues/204

SpaceTeddy, 2019-10-09:
after comment on the mqtt HC loop, the OneTime water loading mqtt commands works fine.
many thanks for your quick check.  👍

proddy, 2019-10-13:
beware @SpaceTeddy I made some changes to the MQTT in dev that will break your code. It should be relatively easy to figure out. Let me know if you need help porting it over.

SpaceTeddy, 2019-10-14:
noticed. with the next master release, I'm will update my fork and my HomeAutomation (fhem). no worries.
thanks again for your strong effort help us all to control our heating systems. Hope thats eats not all your time :) 👍

proddy, 2019-10-14:
np. Please use the dev branch and post your PRs on that instead of master. I looked through your code its looks good. It'll be good if you can push the PR soon so I can finalize the 1.9.2 or even call it 1.10.0 because of your new feature addition.

proddy, 2019-10-28:
@SpaceTeddy how's this feature coming along? It'll be good to add it to the next release as more people are asking.

SpaceTeddy, 2019-10-28:
well, for my system it is working fine since weeks, but I have really some trouble with Git to stay updated  with your master or dev branch. If you could tell me a version, I would like to delete my actual fork, will create a new fork and start a pull request. This is based for the RC20RF controller and I would like to double check, if there is a different behavior, if I press the tap water button at my es73 / RC30. 
Hope to init a pull request in the next 2 days...

proddy, 2019-10-28:
you should always fork the dev branch (not a specific version or tag) and keep it in sync with the updates as frequently as you can.

proddy, 2019-11-13:
this was implemented in 1.9.3. If it needs more work please re-open

closed 2019-11-13 by proddy