emsesp/EMS-ESP #203: EMS Bus Connected but Tx is not working

PhillyGilly, 2019-10-07:
I have just installed Kees device at version 1.9.0 and connected it to the EMS bus in my Worcester Bosch Greenstar 24i boiler. I seem to have the same issue that Higgy identified in issue #60 . When I use Telnet, autodect finds nothing. Have the changes which he made to detect his Greenstar boiler been incorporated into the latest code?

proddy, 2019-10-07:
Hi, first thing is to get you on the latest 1.9.2 which has some more improvements. You can compile the firmware yourself or use the web front-end to upload. Then I think you'll need to change tx_mode to 2.

proddy, 2019-10-07:
https://github.com/proddy/EMS-ESP/releases/download/1.9.1/EMS-ESP-1_9_2b3-d1_mini-release.bin

PhillyGilly, 2019-10-07:
Thanks. I downloaded 1.9.2 and used OTA to update the device. However I am getting:
ESP8266 System stats:                                                                                                                                                        [APP] EMS-ESP version: 1.9.0
Something must have changed as Event Logging is now working, but same error message on the Dashboard.

proddy, 2019-10-08:
Something went wrong will the upload I'm guessing. Have you looked at the wiki and troubleshooting guide?

PhillyGilly, 2019-10-08:
Thanks. Yes I think I worked through those thoroughly before creating a new Issue. As I haven't got any good alternative to OTA, is there any way of ensuring that the OTA update works?

proddy, 2019-10-08:
I've had no issues using OTA. Are you doing this from the command line using py tools or via PlatformIO? Why not use the web interface?

PhillyGilly, 2019-10-08:
I am using OTA. I'm not getting any error messages. I've tried updating 1.9.0 to both 1.9.1 and 1.9.2 both are unsuccessful.

proddy, 2019-10-08:
what do you mean "using OTA"?

PhillyGilly, 2019-10-08:
I am using the device web interface 
[image]

proddy, 2019-10-08:
after you click Update wait about 2-3 mins. The progress bar is misleading and I think when it hits 100% you're clicking on the URL that resets it. That has been fixed in 1.9.1

PhillyGilly, 2019-10-08:
Thanks. I gave it a good ten minutes but it still won't move on from 1.9.0. Is there any alternative  method that is documented for updating?

proddy, 2019-10-08:
look in the wiki

PhillyGilly, 2019-10-08:
Thanks (again) I had read the section on Building and Uploading the Firmware but that wasn't what I had in mind. If I can't move forward using the Web interface, is there anything I can send you to help determine if this is a bug?

proddy, 2019-10-08:
Well, strange it doesn't work. Is anything shown in the Event Log? It should say 'Firmware Upload Started' and '...Finished'.

you could also just follow these 3 easy steps from your PC and see if there are errors?

- install Python 3.7 (https://www.python.org/downloads)
- download espota (https://raw.githubusercontent.com/esp8266/Arduino/master/tools/espota.py)
- python espota.py -i ems-esp -f EMS-ESP-1_9_2b3-d1_mini-release.bin

PhillyGilly, 2019-10-08:
[image]
As you can see it starts but doesn't finish.
Thanks for the Python primer, I'll try that later.

proddy, 2019-10-08:
it's also in the wiki under https://github.com/proddy/EMS-ESP/wiki/Building-and-Uploading-the-Firmware#b-using-the-pre-built-firmware

:-)

higgers, 2019-10-09:
Hi Proddy, I just saw your post on issue #60 so I thought I'd post here to save you a job.

I have a Worcester-Bosch 550CDi that's about 10 years old.  I've found version 1.8.3 to work best for my boiler.  Within seconds of changing the tx mode to 2 in a telnet session I see a message appear that the boiler has been detected.  The boiler info from the info command is:

```
Boiler stats:
  Boiler: Bosch Condens 2500/Junkers Heatronics3 (ProductID:95 Version:25.02)
```

Versions later than 1.8.3 don't seem to be able to send telegrams successfully so I'm unable to change boiler parameters.  With v1.8.3 I've successfully got a weather compensation algorithm working for the central heating water temperature.  I've not tried 1.9.2 yet.

proddy, 2019-10-09:
Hi @higgers , could you try 1.9.2 when you get a chance? It should work just like 1.8.3 and if it doesn't I'll make the damn thing work!

higgers, 2019-10-09:
Hahaha! I like your style!

Is tx mode 1 in v1.9.2 like tx mode 2 in v1.8.3? Meaning, when I flash 1.9.2 it should just work because tx mode 1 is the default?

proddy, 2019-10-09:
yes, in theory!

PhillyGilly, 2019-10-09:
OK guys you're running and I'm still crawling. My Python is rusty and I've
never used the command line before. When I've worked out how to tell it
where the files are I'll report back.

PhillyGilly, 2019-10-09:
I resolved my path issue by starting Python from the directory where i stored the script and bin file.
This is what happened

C:\..........\EMS-ESP>python espota.py -i ems-esp -f EMS-ESP-1_9_2b3-d1_mini-release.bin
18:47:54 [ERROR]: No response from device

The device was there as I could ping it.
I opened up 192.168.2.101 and there was a "blank" i.e. no info or version number opening screen.
I then used the web interface to do an update to 1.9.2b3 and it succeeded!

[image]

proddy, 2019-10-09:
ok great! a few questions
- your unknown device with product ID 209 and device ID 0x09 - do you know what this is?
- when installing Python with windows I think you need to set the path and also restart windows. This would have solved the path problem.
- The boiler and return temperatures are empty. To look at this further you'll need to go to Telnet and look at the traffic coming in. Let me know when you want to go deeper!
- It would be good to see if Tx works, i.e. are you able to send messages. We can also test this via the Telnet.
- Do you have a thermostat on your EMS network? I don't see it.

PhillyGilly, 2019-10-10:
1. I have connected the EMS-ESP directly on to the circuit board in the Bosch Boiler and there are no other external devices on the bus. Could it be the optional internal diverter valve kit which is a little stepper motor device? 
[image]
FWIW this means there are two flow return temperature settings, one for CH which is set at 40C and one for HW which is set at 60C
2. Thanks. Once I'd realised that I needed to import os before I could use os.getcwd() to find that my path wasn't working, I simply opened a command window, change the active directory to the one where my files were and then started Python.
3&4. Let's do it!
 5. No EMS thermostat. My system is as follows: **HW** by solar thermal Bosch MS200 (Junkers?) with gas boiler back up.  There are two 150L cylinders and the MS200 unit manages the solar inputs. The boiler runs at night if there isn't enough hot water generated in the day.  The solar thermal is running fine on an EMS bus which is not (yet) connected to the EMS-ESP. **CH** by the same gas boiler is all underfloor and managed on a zone per room basis by a Heatmiser system. The reason that I haven't joined up the two EMS systems is that Bosch don't allow for gas back up to two cylinders, so I wouldn't have enough hot water in the winter. My ambition is to use the two cylinder temperatures measured by the MS200 as inputs to the boiler control.
[image]

proddy, 2019-10-14:
great info! I'll prepare a few tests for you.

zxdavb, 2019-10-15:
I had the same error message, but it is working after installing the latest firmware. :)

I also had no success updating the firmware via the web UI (despite being very patient) - updating via  **esptool.py** was OK.

I have a shiny new EMS Bus Wi-Fi Gateway, which had arrived with a sticker saying v1.9 firmware installed. I also have a Worcester Greenstar Compact Si (exact model to be confirmed).

I did change the Tx mode, but only when I updated to **EMS-ESP-1_9_2b9-d1_mini-debug.bin** did the error message disappear. Now, on the dashboard, I have:
```
Discovered Devices:
Buderus GBx72/Nefit Trendline/Junkers Cerapur (Version:06.00 ProductID:123 DeviceID:0x08)
Junkers Controller (Version:02.06 ProductID:152 DeviceID:0x09)
```

I am pretty good with Python/Git/Linus/etc., if I can help at all, just ping me!

proddy, 2019-10-15:
@zxdavb thanks for reporting that. Which error did you have previously with 1.9.0? If more people are having similar issues with Bosch I may need to push 1.9.2 to the latest stable master.

Also is everything working for you now?

PhillyGilly, 2019-10-15:
Hi @zxdavb when you check it out can you see if you have the internal diverter kit fitted too? Basically it means you can set a higher return temperature for HW than for CH. If nothing else is obvious you will have a hot water storage cylinder and two separate return pipes.

proddy, 2019-10-15:
@PhillyGilly could you `log v` in a telnet session leave it running for a 1-2 minutes and capture all the output. Then we can try and see where the boiler temps are.

zxdavb, 2019-10-15:
> Which error did you have previously with 1.9.0? 

I can't recall _exactly_, but I think it was:
```
EMS Bus Connected but Tx is not working.
```

> when you check it out can you see if you have the internal diverter kit fitted too?

Here, I have a combi boiler (i.e. no DHW cylinder) - the HW leaves the boiler and ends up down the shower drain.

However, when I get it working here, I will be installing a second gateway in another house that does have a system boiler - the CV leaves the boiler, heats up a storage cylinder, and returns to the boiler.  It is a Worcester Greenstar i 27kw.

This will not be in the next two weeks, however.

PhillyGilly, 2019-10-15:
Hello Proddy. Yes but I will do the Telnet in about two weeks and post back here. Thanks.

proddy, 2019-10-17:
BTW in https://github.com/proddy/EMS-ESP/issues/206 @suysh has seen similar issues where certain values from the Boiler are not sent in the expected telegrams (like boiler temperature)

PhillyGilly, 2019-10-29:
System times look funny, but here two minutes telnet output.

System Logging set to Verbose
(07:57:05.185) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 01 FF 64 1E 09 01 25 D0 80 00 01 7F 80 00 00 B8 FF 00 00 00 C8 00 00 00 00 00 (CRC=F3) #data=27
<--- UBAMonitorFast(0x18)
(07:57:05.393) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:05.678) Boiler -> all, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 13 00 00 00 D8 00 D7 80 00 00 80 00 80 00 80 00 00 (CRC=F9) #data=21
(07:57:05.901) Boiler -> all, type 0x34, telegram: 88 00 34 00 0A 01 7F 01 7F 00 00 00 03 00 00 1C C4 00 03 43 00 80 00 (CRC=B3) #data=19
<--- UBAMonitorWWMessage(0x34)
(07:57:15.101) Boiler -> all, type 0x34, telegram: 88 00 34 00 0A 01 80 01 80 00 00 00 03 00 00 1C C4 00 03 43 00 80 00 (CRC=F1) #data=19
<--- UBAMonitorWWMessage(0x34)
(07:57:15.310) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 04 64 1E 09 01 25 D0 80 00 01 80 80 00 00 B8 FF 00 00 00 C8 00 00 00 00 00 (CRC=67) #data=27
<--- UBAMonitorFast(0x18)
(07:57:15.568) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:15.804) Boiler -> all, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 13 00 00 00 D8 00 D7 80 00 00 80 00 80 00 80 00 00 (CRC=F9) #data=21
(07:57:24.085) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 08 64 1E 09 01 25 D0 80 00 01 7F 80 00 00 B8 FF 00 00 00 C8 00 00 00 00 00 (CRC=EF) #data=27
<--- UBAMonitorFast(0x18)
(07:57:24.318) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:25.135) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 08 64 1E 09 01 25 D0 80 00 01 7F 80 00 00 B8 FF 00 00 00 C8 00 00 00 00 00 (CRC=EF) #data=27
<--- UBAMonitorFast(0x18)
(07:57:25.368) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:25.604) Boiler -> all, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 13 00 00 00 D8 00 D8 80 00 00 80 00 80 00 80 00 00 (CRC=D7) #data=21
(07:57:25.827) Boiler -> all, type 0x34, telegram: 88 00 34 00 0A 01 7F 01 7F 00 00 00 03 00 00 1C C4 00 03 43 00 80 00 (CRC=B3) #data=19
<--- UBAMonitorWWMessage(0x34)
(07:57:33.085) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 0E 64 1E 09 01 24 D0 80 00 01 80 80 00 00 A5 FF 00 00 00 CC 00 00 00 00 00 (CRC=C1) #data=27
<--- UBAMonitorFast(0x18)
(07:57:33.318) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:34.160) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 0F 64 00 01 01 24 D0 80 00 01 80 80 00 00 00 FF 00 00 00 CC 00 00 00 00 00 (CRC=AB) #data=27
<--- UBAMonitorFast(0x18)
(07:57:34.368) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:35.085) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 10 64 00 01 01 24 D0 80 00 01 80 80 00 00 00 FF 00 00 00 CC 00 00 00 00 00 (CRC=90) #data=27
<--- UBAMonitorFast(0x18)
(07:57:35.293) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:35.529) Boiler -> all, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 13 00 00 00 D8 00 D8 80 00 00 80 00 80 00 80 00 00 (CRC=D7) #data=21
(07:57:35.752) Boiler -> all, type 0x34, telegram: 88 00 34 00 0A 01 7F 01 80 00 00 00 03 00 00 1C C4 00 03 43 00 80 00 (CRC=F4) #data=19
<--- UBAMonitorWWMessage(0x34)
(07:57:43.110) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 06 64 00 01 01 24 D0 80 00 01 80 80 00 00 00 FF 00 00 00 CC 00 00 00 00 00 (CRC=76) #data=27
<--- UBAMonitorFast(0x18)
(07:57:43.318) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:45.085) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 02 03 64 00 01 01 24 D0 80 00 01 80 80 00 00 00 FF 00 00 00 CC 00 00 00 00 00 (CRC=64) #data=27
<--- UBAMonitorFast(0x18)
(07:57:45.293) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
(07:57:45.535) Boiler -> all, type 0x19, telegram: 88 00 19 00 00 4D 80 00 80 00 00 00 02 5C 00 47 DB 01 C3 16 00 00 00 01 A6 52 00 44 98 80 00 (CRC=AA) #data=27
<--- UBAMonitorSlow(0x19)
(07:57:45.833) Boiler -> all, type 0x1C, telegram: 88 00 1C 00 93 02 0D 13 27 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=8D) #data=25
(07:57:46.054) Boiler -> all, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 13 00 00 00 D8 00 D8 80 00 00 80 00 80 00 80 00 00 (CRC=D7) #data=21
(07:57:46.327) Boiler -> all, type 0x34, telegram: 88 00 34 00 0A 01 80 01 80 00 00 00 03 00 00 1C C4 00 03 43 00 80 00 (CRC=F1) #data=19
<--- UBAMonitorWWMessage(0x34)
(07:57:46.545) Boiler -> all, type 0x07, telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=B5) #data=13
<--- UBADevices(0x07)
(07:57:48.185) Boiler -> all, type 0x18, telegram: 88 00 18 00 30 01 FD 64 00 01 01 24 D0 80 00 01 80 80 00 00 00 FF 00 00 00 CC 00 00 00 00 00 (CRC=A5) #data=27
<--- UBAMonitorFast(0x18)
(07:57:48.393) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11
<--- UBAMonitorFast(0x18)
[... cut ...] nitorFast(0x18)                                                                                               (07:59:25.570) Boiler -> all, type 0x18, telegram: 88 00 18 1B 00 00 00 00 00 00 00 00 00 00 00 (CRC=5C) #data=11       <--- UBAMonitorFast(0x18)                                                                                               (07:59:25.780) Boiler -> all, type 0x2A, telegram: 88 00 2A 00 00 00 00 00 13 00 00 00 D8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=22) #data=21 
log n

proddy, 2019-10-30:
Why did you say the system times look funny? It’s just elapsed time since the device was booted. Unless it’s wrong!?

PhillyGilly, 2019-10-30:
I just checked the log and there was a system restart that I wasn't aware of. So no big problem. Should have checked the log before commenting :-)