emsesp/EMS-ESP #205: EMS Bus Connected but Tx is not working (Buderus Logamax GB062)

peclik, 2019-10-09:
My boiler is Buderus Logamax GB062.
EMS-ESP won't send any telegram after boiler detection.

It seems the problem is this piece of code:
```
           // check to see if its a Junkers Heatronic 3, which has a different poll'ing logic
            if (EMS_Boiler.product_id == EMS_PRODUCTID_HEATRONIC) {
                EMS_Sys_Status.emsIDMask     = 0x80;
                EMS_Sys_Status.emsPollAck[0] = EMS_ID_ME ^ EMS_Sys_Status.emsIDMask;
            }
```
At start, emsIDMask is 0, so EMS-ESP is able correctly detect data poll for it (0x8B). After "Bosch/Heatronic" is detected, emsIDMask is set to 0x08 which ends in no more telegram transmission due to undetected poll with this code:
```
           if ((value ^ 0x80 ^ EMS_Sys_Status.emsIDMask) == EMS_ID_ME) {
```

After commenting-out the HEATRONIC test block, EMS-ESP 1.9.2 works like a charm (tx_mode 2), correctly detecting even thermostat and mixing device (which where not in 1.8.1).

Sample handshake:

```
(00:00:09.128) 08 00 34 00 34 01 F9 01 F9 81 00 00 03 00 00 2B 7C 00 05 9C 00 (CRC=AA) #data=17
(00:00:09.390) 08 00 19 00 00 6B 80 00 80 00 FF FF 00 00 00 1F C7 01 34 18 00 0000 01 08 9C 00 1A 2B 00 D5 (CRC=79) #data=27
poll 8B
(00:00:09.916) Sending read of type 0x07 to 0x08, telegram: 0B 88 07 00 20 (CRC=A8)
(00:00:09.931) 0B 88 07 00 20 (CRC=A8) #data=1
(00:00:09.968) 08 0B 07 00 03 01 00 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=BF)#data=15
[EMS] Detected new EMS Device with ID 0x08
[EMS] Detected new EMS Device with ID 0x09
[EMS] Detected new EMS Device with ID 0x10
[EMS] Detected new EMS Device with ID 0x20
(00:00:10.031) 08 00 07 00 0B 01 00 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=3B)#data=15
poll 8B
(00:00:10.291) Sending read of type 0x02 to 0x08, telegram: 0B 88 02 00 20 (CRC=BC)
(00:00:10.306) 0B 88 02 00 20 (CRC=BC) #data=1
(00:00:10.338) 08 0B 02 00 5F 2D 0A 00 00 00 00 00 00 00 (CRC=8C) #data=10
EMS Device recognized as Boiler: Bosch Condens 2500/Junkers Heatronic 3 (DeviceID:0x08 ProductID:95 Version:45.10)
(00:00:11.032) 08 00 07 00 03 01 00 01 00 00 00 00 00 00 00 00 00 00 00 (CRC=8B)#data=15
poll 8B
... no more telegrams from 0B
```

peclik, 2019-10-09:
Is special handling of product id 95 really necessary when there is the following auto-detection code?
```
    if (EMS_Sys_Status.emsTxStatus == EMS_TX_REV_DETECT) {
        if ((length >= 5) && (telegram[length - 1] == _crcCalculator(telegram, length))) {
            EMS_Sys_Status.emsTxStatus   = EMS_TX_STATUS_IDLE;
            EMS_Sys_Status.emsIDMask     = telegram[0] & 0x80;
```

proddy, 2019-10-10:
hi @peclik. Firstly thanks for reporting this and looking into the code to find the actual root cause. That's really helpful.

I always thought anything with a product ID of 95 (0x5F) would be a Heatronics device but now you've just disproved that with your Buderus Logamax GB062 ! 

And you're absolutely correct in stating the piece of code "`if (EMS_Boiler.product_id == EMS_PRODUCTID_HEATRONIC)`"  is now obsolete as we test the headers on the first incoming telegram. I'll remove it.