emsesp/EMS-ESP #210: Buderus Logamax plus GB112-11/19 with Buderus Logamatic ERC

Gisbert1, 2019-10-12:
Dear proddy,

I have the above boiler and control unit, both are from the year 2000.
As I do not find both devices on the list of supported devices, I like to ask if there is a chance that your EMS-ESP software together with the EMS gateway can work.

Is there a way to find out whether my devices are compatible to the EMS gateway and your EMS-ESP? I have some basic experience in microelectronic but no professional equipment, only basic equipment like multimeter, I have access to a local microcontroller group with equipment. Can you guide me to find out?

I'm searching so long to make my boiler smart but was not succesful so far. In the meantime I added thermocouples on the outside of the lines and more of these kind of things to get some information from the boiler. Idealwise I would like to change settings and this will be the case for a new boiler once the existing will break (now 19 years old, but still in good shape and working without troubles).

Best Regards
Gisbert

proddy, 2019-10-12:
Hi @Gisbert1 - did a quick search on the web and found [this](https://www.mikrocontroller.net/topic/205249). Looks like your boiler is UBA1 compliant and EMS is only supported with UBA3 and higher. Worth reading the whole thread though for possible workarounds as I only briefly glanced at it (in google translate!)

Gisbert1, 2019-10-12:
Hi proddy,

with this information it becomes clear that my devices are designed for EMS.
But you helped me a lot with the link which included another link: [http://www.picbasic.nl/frameload.htm?http://www.picbasic.nl/thermostaat_nefit.htm](url).
They broad up the idea to tune the desired temperature in an similar thermostat like the ERC. I will use this idea but I will do it slightly different:
I'm going to use the same digital potentiometer X9C103 (0.1-10 kOhm) which I control with an ESP8266. There is already in Arduino a library for the potentiometer available which make things easier. I have already such a circuit for the outside temperature in place where I use a X9C104 (1-100 kOhm).

Best Regards
Gisbert

proddy, 2019-10-13:
ok, good luck! Let us know how you get on.

closed 2019-10-13 by proddy