emsesp/EMS-ESP #211: Get data via USB or serial with wifi switched off?

timoon, 2019-10-14:
I have the Wemos D1 mini connected to bqkees' board and it is plugged in USB of a raspberry 1. Over Wifi I can see that the Buderus EMS telegrams are recognized and everything seems to work. I would like to know if there are any ways to directly communicate between wemos and the raspberry without Wifi.

proddy, 2019-10-14:
Unfortunately no. The standard way to talk between a rpi and ESP is using the serial/TTL port but since there is only one on the ESP8266 and used to read the EMS bus it'll won't work.  I've tried to find a way around this in the past but I couldn't get it to work reliably. There are a few people who have successfully used the board with a rpi (without my code). I can't recall who but maybe bbqkees can point you in the right direction.

bbqkees, 2019-10-15:
I have listed some Raspberry Pi examples here: https://github.com/bbqkees/EMS-Wiki
However, none of those examples are as feature rich as EMS-ESP.

closed 2019-10-17 by proddy