emsesp/EMS-ESP #212: WIP: Some boilers do not report water pressure

zxdavb, 2019-10-18:
**Bug description**
My combi boiler (Worcester Greenstar 30 Si NG) does not report water pressure.

```
info

EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is disabled, Solar Module is disabled, Mixing Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected, protocol: Junkers HT3
  Rx: # successful read requests=3373, # CRC errors=0
  Tx: Last poll=2.599 seconds ago, # successful write requests=0

Boiler stats:
    ...
  System pressure: ? bar
    ...
 ```
 
**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by from the telnet session using the `system` command and `info`.*

**Additional context**
*Add any other context about the problem here.*

*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/proddy/EMS-ESP/issues) (both open and closed)*
* *searched the [wiki help pages](https://github.com/proddy/EMS-ESP/wiki/Troubleshooting)*

*Completing this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

zxdavb, 2019-10-18:
If pressure is still available to the bus, then why has it been removed from telegram 0x18?

Assuming that pressure is available to a precision of 0.01 (as it is on another boiler I own, an Intergas Eco RF 36), then that data would no longer fit into a single byte - so it would have to have been move to another (new?) telegram.

Supposing the new telegram is only available polled, rather than periodically broadcast to the bus...

... then the system would be behaving as seen.

** Next steps

The step step would be to identify this new telegram, and poll it.?

proddy, 2019-10-18:
The system pressure in the UBA Monitor Fast (0x18) telegram shows a value 0xFF meaning "not available" on some Worcester model boilers.

It is available on a Bosch Condens GC9000iWM for example but not all the Worcester Greenstars.

Likewise Boiler Temperature and Boiler Return Temperature are not always available on both these models. I expect the data is somewhere in a separate telegram but without details it will be hard to find as none of this EMS protocol is publically available.

What you could try is capture all the logs from the Boiler with one Pressure setting, then up the Pressure setting and capture a new set. Then painstakingly going through all the telegrams looking for possible changes.

zxdavb, 2019-10-18:
> What you could try is capture all the logs from the Boiler with one Pressure setting, then up the Pressure setting and capture a new set. Then painstakingly going through all the telegrams looking for possible changes.

As a trial, I have done this via a `log v` for 1 hour.  I found telegrams 07, 14, 16, 18, 19, 1C, 2A, 33, 34.

I did not find anything, but will do 24 hours & have another go.

moustic999, 2019-11-06:
One question, Can you see the pressure value in a menu on the thermostat ? 
if yes, Can you tell me in which menu / sub menu... that can help to identify the right telegram.
if no, I fear that this value is not reported to the bus !

zxdavb, 2019-11-08:
I have a **Worcester Bosch Greenstart 27i System Compact**.

From the `device` command:
```
These device IDs are on the EMS Bus: 0x08 0x09
and 2 were recognized by EMS-ESP as:
 Buderus GBx72/Nefit Trendline/Junkers Cerapur (DeviceID:0x08 ProductID:123 Version:06.02)
 Junkers Controller (DeviceID:0x09 ProductID:152 Version:03.01)
```
... and from the `info` command:
```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is disabled, Solar Module is disabled, Mixing Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected, protocol: Junkers HT3
  Rx: # successful read requests=28, # CRC errors=0
  Tx: Last poll=2.699 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur (ProductID:123 Version:06.02)
  Central heating: active
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Eco
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: off
  Selected flow temperature: 60 C
  Current flow temperature: 52.1 C
  Return temperature: 46.1 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 31 %
  Flame current: 27.7 uA
  System pressure: ? bar
  System service code:  (200)
  Heating temperature setting on the boiler: 60 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 60 %
  Boiler temperature: 57.0 C
  Pump modulation: 72 %
  Burner # starts: 9696 times
  Total burner operating time: 62 days 22 hours 51 minutes
  Total heat operating time: 62 days 22 hours 51 minutes
  Total UBA working time: 321 days 13 hours 37 minutes
```

zxdavb, 2019-11-08:
> One question, Can you see the pressure value in a menu on the thermostat ?
> if yes, Can you tell me in which menu / sub menu... that can help to identify the right telegram.
> if no, I fear that this value is not reported to the bus !

My two WB boiler appear/behave identically.  They have an analog pressure dial, and no pressure is available via the digital display.  However, from their manuals, there is a clear indication of water-pressure error codes, so I still suspect it's a matter of working out what telegram to request.

glitter-ball, 2019-11-18:
I think a lot of WB boilers have no water pressure sensor at all. I have a Greenstar 24i System ErP and added a Sense II to it. The Sense II constantly complained that the water pressure sensor was faulty. Bosch came and changed the HCM (heat control module) in the boiler for a later software version and the problem went away. Looking inside, there's no sensor fitted, so the boiler relies on an analogue pressure gauge and that's it.

PhillyGilly, 2020-01-28:
Hello @proddy . As you know I have exactly the same Boiler/Sense II combination as @glitter-ball. When I saw the above I also checked and the only pressure sensor is connected to the front panel gauge by a length of black capillary tube. I would suggest that this issue can be closed next time you are doing house keeping.

proddy, 2020-01-28:
great! thanks, forgot about this one.

closed 2020-01-28 by proddy