emsesp/EMS-ESP #213: Supported ESP Hardware:ESP-12E

mellbo, 2019-10-19:
https://components101.com/sites/default/files/component_pin/ESP12E-Pinout.png
I have one bare ESP-12E module, and I want to know where can I connect TX/RX with this module.

bbqkees, 2019-10-20:
TX is on GPIO15 and RX is on GPIO13.

proddy, 2019-10-20:
https://github.com/proddy/EMS-ESP/wiki/Assembly-and-Configuration

mellbo, 2019-10-21:
[image]

Please look again at this picture. Seems TX on GPIO02 and RX on GPIO08.
I have any signal on GPIO15/GPIO13. I scaned output with one logic analyzer.
But I have signal to GPIO08...

Please answer me...I need information before I made my PCB.

proddy, 2019-10-21:
The code switches the UART0 to use 13 and 15 as not to interfere with any USB devices connected. I'll add a note to the Wiki to explain this.

https://arduino-esp8266.readthedocs.io/en/latest/reference.html#serial

proddy, 2019-10-21:
and a few others use this on a custom board with a bare esp12e so it should work. You can try it yourself using a breadboard and a few components. This also is explained in detail on my Wiki.

mellbo, 2019-10-21:
I make my PCB with your instruction.
But. Look my friend. Serial.swap() switch UART0 to UART1. At my ESP-12E UART0 is on GPIO01/GPIO03, and UART1 with GPIO02/GPIO08. This pin is hardware UART0 and UART1. I don`t know...but I burn firmware in ESP-12E and I check with logic analyzer, or usbtoUART convertor. In any case, no result. When I read GPIO08 I read some data... 

I hope you are right.

proddy, 2019-10-21:
I would just forget the swap and comment out line 147 from `emsuart.cpp` and use the default pins

mellbo, 2019-10-22:
You have right about my question. With Serial.swap() TX/RX move on GIPO13/GPIO15.
Seems my photo with ESP-12E pinout is wrong.

mellbo, 2019-10-25:
i have other problems now. for Bosch Condens 2500W with any tx_mode I have just received data. EMS port connected but TX not. Are you sure about R11, R10 and orientation capacitor C8 from schematic Rev 1.1 with super capacitor?

bbqkees, 2019-10-25:
Ditch the supercap, reduce R4 and R5 to 360 or 330 and rotate C8 if you used an elco.

mellbo, 2019-10-25:
i will be try. but I want to know about R10, R11. Is ok to be set 1K both of them?...

bbqkees, 2019-10-25:
Ok now I see they are 1k. ->R10 10k, R11 100k.

mellbo, 2019-10-25:
whell..when I change R4=R5=330R or 360R TX appear OK. but boiler (BOSCH CONDENS 2500W) do never stop the pump. seems like pulse is to loong and boiler try to starting heating, like on/off thermostat. meibe I need try other values. more that 180R and less 235R ?

mellbo, 2019-10-25:
if I change EMSUART_TX_BRK_WAIT parameter, not help for this situation ?
please, someone tell me if I can or not to starting heating with "send xx"

bbqkees, 2019-10-25:
So it appears you want to use EMS-ESP as a stand alone thermostat?
This is not possible.

mellbo, 2019-10-25:
i need to change by EMS just temperature of warm water, heating retur temperature, start and stop heating, and some data like warm water in use or not..etc.
 
with [8B 08 33 02 {40..60}] - i can change warm water temperature.
**why I can`t start/stop heating from EMS ?** 

in other side, i say on top:
"_whell..when I change R4=R5=330R or 360R TX appear OK. but boiler (BOSCH CONDENS 2500W) do never stop the pump. seems like pulse is to loong and boiler try to starting heating, like on/off thermostat. meibe I need try other values. more that 180R and less 235R ?_"

bbqkees, 2019-10-25:
The thermostat controls the heating. EMS-ESP only sets the temperature on a (supported) thermostat.

mellbo, 2019-10-25:
very bad newz. but ...why ? just because we don`t know the code or...another reason ?
I just need to find the code or why can`t start/stop heating with SEND command ...?

proddy, 2019-10-28:
yes, because we don't know the code. The project is to interact with known EMS devices. If you're planning to simulate the complex logic of a thermostat that periodically sends commands to the Boiler that is going to be very difficult and dangerous if you don't fully understand exactly what you're doing. This is also no the intent of this GitHub project.

I suggest borrow a cheap Bosch compatible Thermostat and see what commands its sending to your UBA Master (Boiler) and start there.

closed 2019-10-28 by proddy