emsesp/EMS-ESP #214: mqtt "thermostat_cmd" does not work

SpaceTeddy, 2019-10-19:
dear proddy, 
I'm close to switch to the 1.9.2 Version with different thermostat command structure as 1.9.1 . For my Home Automation Tool, I try to use the *thermostat_cmd* command, which is described here: [MQTT commands](https://github.com/proddy/EMS-ESP/wiki/MQTT#receiving-mqtt-commands), but if I send ***...thermostat_cmd daytemp 20.0*** the telnet console presents an error: [MQTT] Invalid command from topic thermostat_cmd, payload daytemp 20.0 error InvalidInput
can <ou confirm this behaviour??

thx.

proddy, 2019-10-20:
checking....

proddy, 2019-10-20:
did a quick test with MQTT Explorer and it worked

topic: `home/ems-esp/thermostat_cmd`
payload: `{ "cmd":"daytemp", "data": 20 }`

I noticed I didn't update the Wiki on the new format. Sorry!

closed 2019-10-20 by proddy

SpaceTeddy, 2019-10-20:
great, works fine! 
thanks for updating the Wiki