emsesp/EMS-ESP #218: Releases BIN filename

stbuerger, 2019-10-22:
**Question**
In 1.9.2 (actually starting with 1.9.1) the name of the bin file changed to "EMS-ESP-1_9_2-d1_mini-release.bin" whereas in earlier releases up to 1.9.0 it used to be "firmware_release_1_9_0.bin".
Is this relevant to HW compatibility of the FW ?

I'm experiencing stability & connectivity issues since updated from 1.9.0 to 1.9.2 yesterday

HW is the D1 minion/with bbqkees board

proddy, 2019-10-22:
I just renamed the builds and will start doing nightly builds for each board type (d1 mini, nodemcuv2 etc).

What kind of issues are you experiencing with 1.9.2?

There have been other reports too with the telnet and web becoming non-reponsive after 24hrs. I'm running some stability tests now.

proddy, 2019-10-22:
https://github.com/proddy/EMS-ESP/issues/217

stbuerger, 2019-10-22:
> There have been other reports too with the telnet and web becoming non-responsive after 24hrs. I'm running some stability tests now.

In fact I updated (1.9.0 => 1.9.2) yesterday (espota.py) because Web-OTA didn't work.
After update & (multiple) reset EMS-ESP worked from ~~ 18:00 until 02:00 next morning.
At this time MQstrictlyTT stopped working.
Today around 10:00 I was still able to connect telnet - mqtt and http no longer connected.
Nothing changed after a telnet restart. After a full powercycle even telnet & ping no longer work evenso the blue led is on.

I'm not  strictly depending on the device other than for monitoring/statistics.
So I can wait for your recommendations or can re-flash 1.9.1 for the time beeing.

thx
-sb

proddy, 2019-10-22:
It would be useful if you could see if 1.9.1 exhibits the same behavior. It will help pinpoint the root cause of the problem. thanks

stbuerger, 2019-10-22:
I've flashed 1.9.1 and started device at 20:00 this evening.
Will close this issue #218 as it was initially a question about bin file naming.
Will check device tomorrow and report results in issue #217.

closed 2019-10-22 by stbuerger