emsesp/EMS-ESP #219: Unknow DeviceID:0x10

pcfrickel, 2019-10-23:
### Unknow DeviceID:0x10
As prompted in the telnet session I would like to report a new device.
Output from command "devices":
` These device IDs are on the EMS Bus: 0x08 0x09 0x10`
`and 3 were recognized by EMS-ESP as:`
`Sieger BK15/Nefit Smartline/Buderus GB152 (DeviceID:0x08 ProductID:64 Version:02.07)`
`BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.00)`
`unknown? (DeviceID:0x10 ProductID:67 Version:02.08)`
`You have a device is that is not known by EMS-ESP. Please report this as a GitHub issue so we can  expand the EMS device library.`

EMS-ESP Version 1.9.2
The Boiler is a Buderus GB132 (ID 0x08)
I think ID 0x10 is the thermostat RC30

King regards

proddy, 2019-10-23:
that's awesome, thanks for reporting!

closed 2019-10-23 by proddy

proddy, 2019-12-03:
@pcfrickel I think you're using an RC30 right? In relation to issue #248 can you check whether the current room temperature is reported correctly?