emsesp/EMS-ESP #220: Support Buderus GB125 and external temperature sensor

bobzomer, 2019-10-25:
My EMS-ESP setup detects correctly my RC300 thermostat, but neither my GB125 boiler nor my external temperature sensor.

With telnet:
```
These device IDs are on the EMS Bus: 0x08 0x09 0x10
and 3 were recognized by EMS-ESP as:
 unknown? (DeviceID:0x08 ProductID:133 Version:01.44)
 unknown? (DeviceID:0x09 ProductID:230 Version:37.01)
 RC300/RC310/Moduline 3000/Bosch CW400 (DeviceID:0x10 ProductID:158 Version:18.06)

You have a device is that is not known by EMS-ESP. Please report this as a GitHub issue so we can expand the EMS device library.
```

How can I help you add the support for those devices?

proddy, 2019-10-25:
Thanks for reporting. It has been added to the latest dev build 1.9.3b5. Can you build the firmware youself?

proddy, 2019-10-25:
yes, typo. fixed now. So for the wrong boiler temps could you go into the telnet, turn on verbose logging `log v` and capture a few 0x18 (UBAMonitorFast) and 0x19 (UBAMonitorSlow) telegrams so I can look at the data format.

When you mentioned the external temp sensor I thought it was a home-grown Dallas sensor, not a separate EMS device. Do you have details on the device? If you type `info` in the telnet does it show the outside temperature?

bobzomer, 2019-10-25:
A few minutes log...
[log.txt](https://github.com/proddy/EMS-ESP/files/3772193/log.txt)

The outdoor temperature is not shown in `info`:

```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Mixing Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=2254, # CRC errors=217
  Tx: Last poll=3.226 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus GB133 (ProductID:133 Version:01.44)
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: ? C
  Warm Water current tap water flow: ? l/min
  Warm Water # starts: ? times
  Warm Water 3-way valve: ?
  Selected flow temperature: ? C
  Current flow temperature: ? C
  Return temperature: ? C
  Gas: ?
  Boiler pump: ?
  Fan: ?
  Ignition: ?
  Circulation pump: ?
  Burner selected max power: ? %
  Burner current power: ? %
  Flame current: ? uA
  System pressure: ? bar
  System service code: ??
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Boiler temperature: ? C
  Pump modulation: ? %
  Burner # starts: ? times
  Total UBA working time: 150 days 16 hours 23 minutes

Thermostat stats:
  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400 (ProductID:158 Version:18.06)
  Thermostat time is 15:05:10 25/10/2019
  Heating Circuit 1
   Current room temperature: 20.3 C
   Setpoint room temperature: 19.5 C
   Mode is set to auto
```

proddy, 2019-10-25:
I see lots of unrecognized telegram types like 0xE4, 0xE9 and 0xD1 from the boiler which explains why they cannot be interpreted. They also don't look like EMS+ messages. Do you know what is special about this GB125? Is it new? Never seen this before...

bobzomer, 2019-10-25:
This has been installed in my house earlier this year, but I don't know about the product itself.

moustic999, 2019-10-25:
It is a brand new buderus heater. Messaging is done via ems+, no ems anymore.
You can control it via command of rc300, but getting values from boiler will requires work to decode new messages....
Take a log of 20 30 minutes, to see how messages are repeated.... that is a first step. Before logging check amount of starts, and check after logging... finding this value in message will help identifying which packet is used to

-------- Message d'origine --------

bobzomer, 2019-10-25:
Here is a longer log.
[long_log.txt](https://github.com/proddy/EMS-ESP/files/3772958/long_log.txt)
I don't know where to look for the number of starts...

moustic999, 2019-10-30:
thermostat is known, so you can retrieve HC values via 01A5 messages. almost sure that write to it will work too : 01B9 ;-) 

now to retrieve the boilers value
0xE4  is sent cyclicly every 60 seconds. seems some value are decreasing in it, not every minutes, but regularly, I would be tempted to say it is a temperature decreasing, but which one ? that need to be checked real time with value on the heater

bobzomer, 2019-10-30:
To be checked during a longer period, but the first two bytes of an E4 message seems to be the heater temperature. 
```
(00:12:04.865) Boiler -> all, type 0xE4, telegram: 08 00 E4 00 10 20 30 59 00 CC 29 02 3F 64 00 02 00 00 00 00 00 80 00 00 00 FF 00 02 3F 80 00 (CRC=EE) #data=27
(00:12:05.016) Boiler -> all, type 0xE4, telegram: 08 00 E4 1B 02 44 80 00 01 E9 80 00 (CRC=48) #data=8
```
0x244 = 580, reported as 58°C on heater HMI. I checked in front of the heater during one or two minutes (while the heater was active, so the temperature was moving fast), and it seems correct.

bobzomer, 2019-10-30:
To checked also, but the E9 message could contain the hot water temperature. Displayed as 55.9°C on my thermostat, so 0x022F (= 559).
```
(00:20:12.532) Boiler -> all, type 0xE9, telegram: 08 00 E9 00 37 02 2F 02 2F 00 00 00 00 46 3C 00 00 01 00 0A 8E 00 01 5C 00 00 00 37 00 37 (CRC=FF) #data=26
```

proddy, 2019-10-30:
nice work!

bobzomer, 2019-11-01:
I think I get the outdoor temperature, in D1 messages. The outdoor temperature displayed by the thermostat has just increased from 8.5°C to 9°C at the same time as we got a switch from 0x0057 (= 87) to 0x0058 (= 88).
```
12:42:20.562) Boiler -> all, type 0xD1, telegram: 08 00 D1 00 00 57 (CRC=90) #data=2
(12:43:20.543) Boiler -> all, type 0xD1, telegram: 08 00 D1 00 00 57 (CRC=90) #data=2
(12:44:20.573) Boiler -> all, type 0xD1, telegram: 08 00 D1 00 00 58 (CRC=9F) #data=2
(12:45:20.529) Boiler -> all, type 0xD1, telegram: 08 00 D1 00 00 58 (CRC=9F) #data=2
(12:46:20.635) Boiler -> all, type 0xD1, telegram: 08 00 D1 00 00 58 (CRC=9F) #data=2
(12:47:20.541) Boiler -> all, type 0xD1, telegram: 08 00 D1 00 00 58 (CRC=9F) #data=2
(12:48:20.622) Boiler -> all, type 0xD1, telegram: 08 00 D1 00 00 58 (CRC=9F) #data=2
```

bobzomer, 2019-11-01:
Another match in E4 message.
```
(13:16:01.463) Boiler -> all, type 0xE4, telegram: 08 00 E4 1B 01 53 80 00 01 60 80 00 (CRC=7A) #data=8
```
The second part of this message contains the smoke temperature (0x160 = 352, displayed as 35.2°C on HMI).

bobzomer, 2019-11-02:
Here is a summary of all my findings:

**E4 message seems to be heater-related:**
When offset (byte 4) is 0 (ex. `08 00 E4 00 10 20 30 59 00 CC 1E 01 02 64 00 02 00 00 00 00 00 80 00 00 00 FF 00 01 02 80 00`)
* Status code in byte 9 (CC = 204)
* Target heater temperature in byte 10 (1E = 30°C)
* Real heater temperature in 0.1°C in bytes 11-12 and 27-28 (0102 = 25.8°C)
* Minimum burner power in byte 13 (64 = 100%)
* Real burner power in byte 14 (here 0 = 0%, when burner is active 64 = 100%)
* Flame current in 0.1 µA in bytes 23-24 (here 0000 = 0µA, when burner is active 020F = 52.7µA)

When offset (byte 4) is 1B (ex. `08 00 E4 1B 01 00 80 00 00 F8 80 00`)
* Smoke temperature in 0.1°C in bytes 8-9 (00F8 = 24.8°C)

**E9 message seems to be water-related** 
(ex. `08 00 E9 00 37 02 03 02 03 00 00 00 00 46 3C 00 00 01 00 0A C5 00 01 62 00 00 00 37 00 37`):
* Real water temperature in 0.1°C in bytes 5-6 and bytes 7-8 (0203 = 51.5°C)

**D1 message seems to be related to outdoor temperature sensor** (ex. `08 00 D1 00 00 80`)
* Outdoor temperature in 0.1°C in bytes 4-5 (0080 = 12.8°C)

proddy, 2019-11-03:
nice work @bobzomer . We should add this to the Wiki, or you can I have give you edit privileges? 

Then how do you suggest we implement this? We could detect the boiler type and correctly set the external/outdoor temp and the status code since these values are already captured and sent to MQTT. What about the other values? Are they important to monitor and send?

Also watch out for values like 0x8000 (`80 00`) as EMS uses this a lot to indicate an unknown value. Like the D1 message.

bobzomer, 2019-11-04:
For the wiki, as you want. If you want, I can copy-paste my message in your wiki... :-)

For the implementation, you are the master. The minimum burner power does not seem interesting. The real burner power seems interesting only to detected the burner is active. I don't know if the flame current is relevant to anything, as I'm not a heater-expert.

Regarding the different temperatures, as you send as a JSON array with MQTT, maybe we can just send them if you have them?

Regarding the outdoor temperature, do you plan to send the value as any other Dallas-type temperature sensor?

proddy, 2019-11-04:
@bobzomer check out this [link](https://github.com/Th3M3/buderus_ems-wiki/blob/master/Quelle_08.md) too which I had forgotten about. @Th3M3 worked a lot of these messages out before, like the E3, E4, E9, D1 and others. Worth comparing notes.

theloukou, 2019-11-05:
I can provide logs for double checking telegram types and data if needed.
Regarding implementation, would it be possible to detect boiler type, and switch which telegrams it decodes/forwards? I suppose since @Th3m3 seems to have done a lot of the guesswork (i don't speak german to be sure), the real trouble would be to code the proper decoding.

proddy, 2019-11-05:
Could you verify https://github.com/proddy/EMS-ESP/wiki/MC110-controller and provide some example telegrams. Then I'll start implementing.

theloukou, 2019-11-07:
**0xD1**: reports correctly, bad value for me. I suppose no outdoor sensor is present.
**0x14**: correct
**0xBF**: 2=0xEA, 235, for my BG122
         4=0x00 for me, instad of 0x11 on your wiki
**0xE3/0xE4**: seems different for me. different lengths, not sure i decode temps correctly. only thing i can definitelyconfirm is inE4, bytes 8-9/24-25 are actually right temps.
**0xE9**: byte 1 is correct, not sure about the rest
Also, my data frequencies are a bit different to those on the wiki, but i dont think that this is a huge factor

Data i compared is as follows: Heating temp setpoint to 31C, hot water to 55C.
At timestamp _08:10:45_ actual water temp was 46.5 on screen
At timestamp _08:15:49_ actual water temp was 58.8 on screen
At timestamp _08:17:19_ actual water temp was 59.2 on screen
[ems log.txt](https://github.com/proddy/EMS-ESP/files/3818428/ems.log.txt)

proddy, 2019-11-07:
thanks @theloukou . From all this data what is most important to send via MQTT do you think?

theloukou, 2019-11-07:
I would personally like to have water/heating temps, temp setpoints, status for heating and water, and maybe system pressures? Also modification of setpoints and heating on/off would be a welcome addition.
Obviously, the more the merrier, but let's not be needy :D

Also, if you could maybe point me out to where you do the decoding, and the mqtt stuff, i would like to try and mess around with the code myself, for my own needs, and if it's good enough, you could use some/all of it. if anything comes out useable of course.

proddy, 2019-11-07:
https://github.com/proddy/EMS-ESP/wiki/how-the-code-works

proddy, 2019-11-07:
it's all in ems-esp.cpp

closed 2019-11-14 by proddy