emsesp/EMS-ESP #222: Unknown Junkers devices

DaGolle, 2019-10-27:
Hi Proddy,

First of all, love your work, I was able to get EMS-ESP working using @bbqkees' EMS gateway.

Telnet _autodetect deep_ revealed the following:

```
These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x20 0x21 0x38
and 6 were recognized by EMS-ESP as:
 Bosch Condens 2500/Junkers Heatronic 3 (DeviceID:0x08 ProductID:95 Version:18.13)
 unknown? (DeviceID:0x09 ProductID:95 Version:18.13)
 RC300/RC310/Moduline 3000 (DeviceID:0x10 ProductID:158 Version:33.03)
 unknown? (DeviceID:0x20 ProductID:161 Version:29.03)
 unknown? (DeviceID:0x21 ProductID:161 Version:29.03)
 Bosch CW100 (DeviceID:0x38 ProductID:157 Version:41.05)
```

I have the following devices installed:
1. Junkers Cerapur Top 42-3 ZWB (DeviceID:0x08 I assume)
2. Junkers CW400 Thermostat (DeviceID:0x10?)
3. Junkers CW100 Thermostat (DeviceID:0x38?)
4. Junkers MM200 (http://documents.junkers.com/download/pdf/file/6720818577.pdf)

As far as I can tell, the MM200 is not discovered and probably corresponds to one (or more?) of the unknown devices.

Would be nice if I could read the information from the MM200 and include it in Home Assistant.
Let me know if there is anything that I can do/provide to help.

Best regards

proddy, 2019-10-27:
Thanks for reporting this @DaGolle. I've added the missing devices to the latest dev build.

Hopefully the MM logic for the MM100 should work with the MM200. Give it a try and you hopefully should see some results in the `info` screen and also web and MQTT. 

BTW Next time you can just do a `autodetect quick` from a Telnet session.

proddy, 2019-11-03:
CW400 is also added

DaGolle, 2019-11-06:
After installing firmware 1.9.3, this is the output of the _autodetect quick_ command:

`[EMS] Detected new EMS Device with ID 0x08
[EMS] Detected new EMS Device with ID 0x09
[EMS] Detected new EMS Device with ID 0x10
[EMS] Detected new EMS Device with ID 0x20
[EMS] Detected new EMS Device with ID 0x21
[EMS] Detected new EMS Device with ID 0x38
EMS Device recognized as Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers  (DeviceID:0x08 ProductID:95 Version:18.13)
EMS Device recognized as ?: unknown? (DeviceID:0x09 ProductID:95 Version:18.13)
EMS Device recognized as Thermostat: RC300/RC310/Moduline 3000/Bosch CW400 (DeviceID:0x10 ProductID:158 Version:33.03)
EMS Device recognized as ?: unknown? (DeviceID:0x20 ProductID:161 Version:29.03)
EMS Device recognized as ?: unknown? (DeviceID:0x21 ProductID:161 Version:29.03)
EMS Device recognized as Thermostat: Bosch CW100 (DeviceID:0x38 ProductID:157 Version:41.05)`

proddy, 2019-11-06:
the changes are in the dev build, 1.9.4

DaGolle, 2020-01-31:
After installing firmware 1.9.4, this is the output of the _autodetect_ command:
`New EMS device recognized as a UBAMaster: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x08 ProductID:95 Version:18.13)
New EMS device recognized as a Controller: Bosch Condens 2500/Buderus Logamax GB062/Junkers Cerapur Top/Worcester Greenstar i/Generic HT3 (DeviceID:0x09 ProductID:95 Version:18.13)
New EMS device recognized as a Thermostat: RC300/RC310/Moduline 3000/Bosch CW400/W-B Sense II (DeviceID:0x10 ProductID:158 Version:33.03)
New EMS device recognized as a Mixing Module: MM200 Mixing Module (DeviceID:0x20 ProductID:161 Version:29.03)
New EMS device recognized as a Mixing Module: MM200 Mixing Module (DeviceID:0x21 ProductID:161 Version:29.03)
New EMS device recognized as a Heat Pump: Buderus RC200/Bosch CW100/Junkers CW100 (DeviceID:0x38 ProductID:157 Version:41.05)`

proddy, 2020-01-31:
This has also been improved in 1.9.5 with extra support for the mixer modules do worth trying that release too.

closed 2020-01-31 by proddy