emsesp/EMS-ESP #224: unknown devices

oorotor, 2019-10-27:
Hi,

first of all: great project: I'm using it with bbqkees EMS-ESP Gateway. But I have a few questions:

1. Due to a message in a telnet output, i would like to post following unknown devices, maybe it is possible to support them in the future?
unknown? (DeviceID:0x10 ProductID:147 Version:18.02) --> This should be my Junkers FR50 
unknown? (DeviceID:0x09 ProductID:95 Version:35.00) --> ??? this Device has the same version and ProductID as my boiler Bosch Condens 2500/Buderus Logamax GB062/Junkers  (DeviceID:0x08 ProductID:95 Version:35.00), but i don't know the which device this is.

2. fhem is running on a raspberry pi with an MQTT broker, but as soon as i activate the TX in the EMS-ESP webinterface, the boiler data seems to be send not peridically (missing mqtt messages in fhem for more than a few hours (blue region in screenshot), as soon as TX is disabled the mqtt messages are send periodically (orange region in screenshot). I guess this is a misbehaviour, isn't it? 
[image]

Here are some more data:
Used EMS-ESP version: 1.9.3
Used EMS-ESP Gateway: bbqkees hardware version V1.3 with a generic Wemos D1 Mini V2 type ESP8266

Thanks for the support and the development of this tool!

proddy, 2019-10-27:
Thanks @oorotor for reporting these devices. I've add the FR50 and made it use the same logic behavior as the FR10 although I can't guarantee it will work. It's in the latest dev build so have a try and if you're not getting correct values back from the thermostat we'll need to deep dive into the telegrams. The 0x09 device is the Junkers controller which was added earlier too the dev builds.

I'm now doing nightly builds of the development/beta builds if you're not comfortable compiling and building the firmware yourself.

As for the MQTT, this is normal behaviour. As explained in the [wiki](https://github.com/proddy/EMS-ESP/wiki/MQTT#which-data-is-getting-published) MQTT topics are only published if the data has changed. I can see in your case (and also some others which use time-based data stores like innoDB and grafana) it would be more appropriate to always send the data.

I could make this an additional flag? Something like `publish_always` ?

proddy, 2019-10-27:
I added it anyway. Was a small change. 1.9.4b1

oorotor, 2019-10-30:
I tested the 1.9.4b2 version, the publish_always option is working fine for me, but after a few hours I'm faced with a reported issue of disconnected mqtt #204 (my findigs are documented in issue #204).

Regarding the FR50 device: 
Setpoint Temperature and Current Temperature are working fine, there is only a minor issue with the displayed mode names but for me this is totally acceptable:
"night/non heating" mode: auto is shown in the webinterface
"day/heating" mode: undefined is shown in the webinterface
in auto mode (switching automatically between day/night mode): auto (night) or undefined (day) is shown in the webinterface.

proddy, 2019-10-30:
for the web interface modes, does it show correct when you type `info` from the Telnet interface? What should the correct names be?

oorotor, 2019-11-01:
On telnet it is shown as "Mode is unknown". I analysed the 6F-type message on the EMS-bus, it looks like there are two modes transmitted:

example message:
90 | 00 | FF | 00 | 00 | 6F | 03 | 02 | 00 | DC | 00 | D4

the 7th byte (here: 03) stands for the day/night mode:
03 = day
02 = night

the 8th byte (here: 02) stands for the auto/manual mode:
01 = manual
02 = auto

I saw in the code, that for other thermostats (e.g. RC35) there is only one mode (auto, night, day), so i guess it's a little bit more effort to add two modes instead.

proddy, 2019-11-05:
ok, I'll see if I can add this to the code

proddy, 2019-11-05:
can you check if the correct modes (manual/auto) and day/night are shown?

moustic999, 2019-11-06:
@proddy For these model as well as the RC35, modes are working like that :
In auto, there is 2 temp level that are switched via the calendar : day and night temperature levels.
Then you can switch to day mode or switch to night mode, these modes keep the temperature at the selected level.
There is no real "manual" mode.
it is more 3 mode : auto, night, day.

On some others thermostats, like RC300, there are only 2 modes :  auto and manual.

is that help ?

proddy, 2019-11-06:
@moustic999 could you see if its correct for RC35? I can only check with my RC20 that has off, manual and auto.

moustic999, 2019-11-06:
Please see the attached picture of the manual with explanation of modes...

-------- Message d'origine --------

oorotor, 2019-11-06:
@proddy:
Via telnet the correct day mode is displayed, but the mode (auto/manual) is not displayed:

Thermostat stats:
  Thermostat: Junkers FR50 (ProductID:147 Version:18.02)
  Thermostat time is 20:21:03 06/11/2019
  Heating Circuit 1
   Current room temperature: 21.1 C
   Setpoint room temperature: 22.5 C
   Day Mode is set to day

On the webpage, "mode" with "undefined" is displayed and no "day mode".

proddy, 2019-11-06:
oops, forgot some code. try the new build if you can

oorotor, 2019-11-09:
Now the modes (daymode and mode) are displayed correctly via telnet (daymode is not transmitted via mqtt or shown on webpage) - tested with 1.9.4b7.

I just recognized that there is another "day mode" (I was wondering why the day mode byte has no byte=01), so mode "byte=01" is the "AntiFreeze-Mode" (maybe similar to Holiday-Mode?), in this mode the selTemp is set to 5°C and cannot be changed.

the 7th bytestands for the day/night mode:
01 = holiday
02 = night
03 = day

the 8th byte stands for the auto/manual mode:
01 = manual
02 = auto

proddy, 2019-11-09:
@oorotor I'm not sending the daymode via MQTT. I didn't think it would be useful. But can easily add. Also the modes on the Buderus RCs use bits to determine the modes. Are you sure the values 1,2 and 3 are not bit masks for something else?

@moustic999 With the RC35 I thought there was also a summer mode which is a different byte.

oorotor, 2019-11-17:
@proddy just wanted to inform, that i have updated to 1.9.4b13 today, and afterwards the mode and day mode is not listed correctly anymore with the FR50, also the setpoint room temperature is not correct anymore. downgraded to 1.9.4.b9 again and it seems to work fine with this version again.

1.9.4b9
Thermostat stats:
Thermostat: Junkers FR50 (ProductID:147 Version:18.02)
Thermostat time is 12:21:06 17/11/2019
Heating Circuit 1
Current room temperature: 22.3 C
Setpoint room temperature: 22.5 C
Mode is set to manual
Day Mode is set to day

1.9.4.b13
Thermostat stats:
Thermostat: Junkers FR50 (DeviceID:0x10 ProductID:147 Version:18.02)
Thermostat time is 12:24:06 17/11/2019
Heating Circuit 1
Current room temperature: 22.3 C
Setpoint room temperature: 112.5 C
Mode is set to day

proddy, 2019-11-17:
ok, thanks for reporting. I'll take a look

proddy, 2019-11-17:
made a change, not sure if it'll fix it. Can you test?

oorotor, 2019-11-17:
Yes, with 1.9.4b14 it is working again:
Thermostat stats:
Thermostat: Junkers FR50 (DeviceID:0x10 ProductID:147 Version:18.02)
Thermostat time is 19:54:00 17/11/2019
Heating Circuit 1
Current room temperature: 24.2 C
Setpoint room temperature: 22.5 C
Mode is set to auto
Day Mode is set to day

But in this version it seems that i have some issues with the webinterface, e.g. there was no password popup and therefore i was not able to login to the webinterface for several minutes after restart (also after poweroff), also the webinterface seems to stop working after some time. With telnet all seems to work fine ans also the mqtt messages are send.

proddy, 2019-11-17:
yes, known issue with the web interface. I'm trying to find what caused it but its difficult. I suspect memory problems somewhere. 

https://github.com/proddy/EMS-ESP/issues/239#issuecomment-554620207

Can I close this issue?

oorotor, 2019-11-17:
Yes you can close this issue!
Thanks!

closed 2019-11-17 by proddy