emsesp/EMS-ESP #226: Unidentified device: Worcester Bosch Greenstar i System (18i)

gcareid, 2019-10-29:
Devices:
    Bosch Condens 2500/Buderus Logamax GB062/Junkers (Version:40.10 ProductID:95 DeviceID:0x08)
    unknown? (Version:40.10 ProductID:95 DeviceID:0x09)

I assume the other device is a Drayton DT10RF which is plugged into the boiler, but I don't actually use - I have a hardwired 240V control which my software system actuates manually.  Return temperatures and boiler pressure are missing and the dump below is taken with the central heating system running.

[TELNET] Connected to EMS-ESP version 1.9.3. Type ? for commands.
log v

System Logging set to Verbose
info

_[1mEMS-ESP system stats:_[22m
  System logging set to Verbose
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is disabled, Solar Module is disabled, Mixing Module is disabled, Shower Timer is disabled, Shower Alert is disabled

_[1mEMS Bus stats:_[22m
  Bus is connected, protocol: Junkers HT3
  Rx: # successful read requests=276, # CRC errors=0
  Tx: Last poll=0.906 seconds ago, # successful write requests=0

_[1mBoiler stats:_[22m
  Boiler: Bosch Condens 2500/Buderus Logamax GB062/Junkers Heatronic 3 (ProductID:95 Version:40.10)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 40 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 41.0 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 1585 times
  Warm Water active time: 19 days 5 hours 0 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 75 C
  Current flow temperature: 68.2 C
  Return temperature: ? C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 100 %
  Burner current power: 0 %
  Flame current: 0.1 uA
  System pressure: ? bar
  System service code:  (0)
  Heating temperature setting on the boiler: 75 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 10 %
  Boiler temperature: ? C
  Pump modulation: 100 %
  Burner # starts: 35722 times
  Total burner operating time: 361 days 8 hours 24 minutes
  Total heat operating time: 342 days 3 hours 24 minutes
  Total UBA working time: 1814 days 10 hours 58 minutes

(_[0;36m00:52:19.178_[0m) _[0;32mBoiler -> all, type 0x18, telegram: 88 00 18 00 4B 02 A9 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=27) #data=25_[0m
<--- UBAMonitorFast(0x18)
(_[0;36m00:52:19.387_[0m) _[0;32mBoiler -> all, type 0x34, telegram: 88 00 34 00 28 01 98 01 98 A0 00 00 03 00 00 6C 0C 00 06 31 00 (CRC=B7) #data=17_[0m
<--- UBAMonitorWWMessage(0x34)
[MQTT] Connecting to MQTT...
[MQTT] TCP Disconnected
(_[0;36m00:52:29.176_[0m) _[0;32mBoiler -> all, type 0x18, telegram: 88 00 18 00 4B 02 A7 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=79) #data=25_[0m
<--- UBAMonitorFast(0x18)
(_[0;36m00:52:29.385_[0m) _[0;32mBoiler -> all, type 0x34, telegram: 88 00 34 00 28 01 99 01 99 A0 00 00 03 00 00 6C 0C 00 06 31 00 (CRC=28) #data=17_[0m
<--- UBAMonitorWWMessage(0x34)
(_[0;36m00:52:39.174_[0m) _[0;32mBoiler -> all, type 0x18, telegram: 88 00 18 00 4B 02 A6 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=A6) #data=25_[0m
<--- UBAMonitorFast(0x18)
(_[0;36m00:52:39.383_[0m) _[0;32mBoiler -> all, type 0x34, telegram: 88 00 34 00 28 01 9B 01 9B A0 00 00 03 00 00 6C 0C 00 06 31 00 (CRC=0F) #data=17_[0m
<--- UBAMonitorWWMessage(0x34)
(_[0;36m00:52:49.171_[0m) _[0;32mBoiler -> all, type 0x18, telegram: 88 00 18 00 4B 02 A5 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=DE) #data=25_[0m
<--- UBAMonitorFast(0x18)
(_[0;36m00:52:49.381_[0m) _[0;32mBoiler -> all, type 0x34, telegram: 88 00 34 00 28 01 9B 01 9B A0 00 00 03 00 00 6C 0C 00 06 31 00 (CRC=0F) #data=17_[0m
<--- UBAMonitorWWMessage(0x34)
(_[0;36m00:52:59.157_[0m) _[0;32mBoiler -> all, type 0x07, telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=E6) #data=15_[0m
<--- UBADevices(0x07)
(_[0;36m00:52:59.388_[0m) _[0;32mBoiler -> all, type 0x18, telegram: 88 00 18 00 4B 02 A4 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=01) #data=25_[0m
<--- UBAMonitorFast(0x18)
(_[0;36m00:52:59.597_[0m) _[0;32mBoiler -> all, type 0x34, telegram: 88 00 34 00 28 01 99 01 99 A0 00 00 03 00 00 6C 0C 00 06 31 00 (CRC=28) #data=17_[0m
<--- UBAMonitorWWMessage(0x34)
(_[0;36m00:52:59.827_[0m) _[0;32mBoiler -> all, type 0x19, telegram: 88 00 19 00 80 00 80 00 80 00 FF FF 00 64 00 8B 8A 07 F0 98 00 00 00 07 84 8C 00 85 59 80 00 (CRC=7B) #data=27_[0m
<--- UBAMonitorSlow(0x19)
Starting scheduled query from EMS devices
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(_[0;36m00:53:00.422_[0m) _[0;36mSending read of type 0x18 to 0x08, telegram: 8B 88 18 00 20 (CRC=1C)_[0m
(_[0;36m00:53:00.481_[0m) _[0;33mBoiler -> me, type 0x18, telegram: 88 0B 18 00 4B 02 A4 64 00 01 02 20 C0 80 00 80 00 80 00 FF FF FF 00 00 00 00 00 00 00 (CRC=B2) #data=25_[0m
<--- UBAMonitorFast(0x18)
(_[0;36m00:53:00.532_[0m) _[0;32mBoiler -> all, type 0x07, telegram: 88 00 07 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=56) #data=15_[0m
<--- UBADevices(0x07)
(_[0;36m00:53:00.760_[0m) _[0;36mSending read of type 0x19 to 0x08, telegram: 8B 88 19 00 20 (CRC=18)_[0m
(_[0;36m00:53:00.827_[0m) _[0;33mBoiler -> me, type 0x19, telegram: 88 0B 19 00 80 00 80 00 80 00 FF FF 00 64 00 8B 8A 07 F0 98 00 00 00 07 84 8C 00 85 59 80 00 (CRC=85) #data=27_[0m
<--- UBAMonitorSlow(0x19)
(_[0;36m00:53:00.886_[0m) _[0;36mSending read of type 0x33 to 0x08, telegram: 8B 88 33 00 20 (CRC=B0)_[0m
(_[0;36m00:53:00.940_[0m) _[0;33mBoiler -> me, type 0x33, telegram: 88 0B 33 00 08 FF 28 00 00 23 00 02 46 00 FF FF (CRC=BF) #data=12_[0m
<--- UBAParameterWW(0x33)
(_[0;36m00:53:01.010_[0m) _[0;36mSending read of type 0x16 to 0x08, telegram: 8B 88 16 00 20 (CRC=24)_[0m
(_[0;36m00:53:01.060_[0m) _[0;33mBoiler -> me, type 0x16, telegram: 88 0B 16 00 FF 4B 64 00 00 F6 03 01 03 64 0A 04 (CRC=0C) #data=12_[0m
<--- UBAParametersMessage(0x16)
(_[0;36m00:53:01.135_[0m) _[0;36mSending read of type 0x14 to 0x08, telegram: 8B 88 14 00 20 (CRC=2C)_[0m
(_[0;36m00:53:01.180_[0m) _[0;33mBoiler -> me, type 0x14, telegram: 88 0B 14 00 27 DE 53 (CRC=73) #data=3_[0m
<--- UBATotalUptimeMessage(0x14)
(_[0;36m00:53:01.501_[0m) _[0;32mBoiler -> all, type 0x07, telegram: 88 00 07 00 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=E6) #data=15_[0m
<--- UBADevices(0x07)
^]
telnet> quit
Connection closed.

proddy, 2019-10-29:
I think your request is to add the Greenstar to the text? Apart from that work everything is working as normal?

gcareid, 2019-10-29:
Not quite - return temperature, system pressure, boiler temperature etc are all missing.  I'm not sure what the logic for "central heating: off" but the pump and burner are definitely running during this dump.

proddy, 2019-10-29:
Ah I see. There have been other issues related to the Worcester Bosch using different EMS telegrams from the standards. We haven't figured out which ones yet so any help would be welcome.

gcareid, 2019-10-29:
More than happy to help in any way I can.  I am still getting to grips with the firmware code.  What would be helpful to you in debugging?

proddy, 2019-10-29:
great. It's a notoriously long process of trial and error I'm afraid. For example if you know what the system pressure or boiler temperature values are (as displayed in the Boiler's menu) take a look at the broadcasted telegrams in the EMS-ESP log (typically 0x18 and 0x19) and see if you can spot the value. Make a physical change on the boiler if you can and compare again. Then rinse and repeat. What is tricky is that some values may be stored as single bytes multiplied by either 10 or 2 and some as 2-byte shorts multiplied by 10 or 100.

This is what I've been doing for the last 12 months for Buderus, with a lot of help from other projects and people. These new Worcester range boilers are naughty.

Also look at https://github.com/proddy/EMS-ESP/issues/212.

proddy, 2019-10-29:
I don't know if its in a separate telegram or not, it could well be hidden in an existing one and just not yet mapped. So far I've only seen reports from people with Greenstar's. Perhaps join forces with @zxdavb and compare notes?

zxdavb, 2019-10-30:
(sorry, I have been busy on other repos!)  It may make sense to merge these two issues?

My premise is in https://github.com/proddy/EMS-ESP/issues/212#issuecomment-543597578 - WB changed some vales to two bytes, so they no longer fit in the original telegrams (can are not periodically broadcast, although there is some evidence they can be polled)...

If this is the case, then it's a matter of finding the telegram type, and polling it! 

Thus, the next step is guesswork, and I am hoping is not likely to cause any damage to the boiler.

gcareid, 2019-10-30:
I had a brief check last night, and my boiler responds to polls of the following telegram types (hex):

    01 02 03 04 07 10 11 14 15 16 18 19 22 29 33 34

but I couldn't see any values that looked like they were changing as I would expect the return temperature to.

I also don't seem to be able to change the flow temperature on the boiler either - the write fails regardless of which tx_mode I use.  Have you had any success here?

zxdavb, 2019-10-30:
> Have you had any success here?

No, I haven't had time to try...

zxdavb, 2019-10-30:
FYI: https://www.britishgas.co.uk/smart-home/boiler-iq.html, and
https://www.ebay.co.uk/itm/British-Gas-IQ-CL01-Boiler-Diagnostics-Module-Home-Care-Smart-Boiler-7738110150-/202314664639

gcareid, 2019-10-30:
Definitely worth thinking about.  Do you have any other control devices plugged into the boiler?  Mine has the WB/Drayton DT10RF wireless receiver plugged in to the front panel, which I haven't got rid of yet as it still controls the hot water timing.  I think (not 100% sure) that uses the FX Bus, not EMS.  I'm just wondering whether the two might be interfering somehow.

zxdavb, 2019-10-30:
No other device. I use a Honeywell 2-wire control system.

This is a photo of teh back of teh unit - I don't know what the FX bus is...

[image]

PhillyGilly, 2019-11-05:
[image]
I have similar situation which Proddy moved forward under Issue #203 (thanks).

proddy, 2019-11-10:
> (sorry, I have been busy on other repos!) It may make sense to merge these two issues?
> 
> My premise is in [#212 (comment)](https://github.com/proddy/EMS-ESP/issues/212#issuecomment-543597578) - WB changed some values to two bytes, so they no longer fit in the original telegrams (which are not periodically broadcast, although there is some evidence they can be polled)...
> 
> If this is the case, then it's a matter of finding the telegram type, and polling it!
> 
> Thus, the next step is guesswork, and I am hoping is not likely to cause any damage to the boiler.

@zxdavb I was thinking, would it help if the telegram lengths were longer? I cap them at around 27 bytes.

gcareid, 2019-11-10:
It would certainly be worth considering.  There doesn't seem to be anything in the telegrams the boiler sends periodically, and I've tried polling all possible telegrams without success.

zxdavb, 2019-11-12:
I am not sure I can help at the moment - I put my EMS gateway on a boiler 90 mins drive away, and forgot to open port 22 on the router (although I can connect it it via HTTP).

I am happy to upgrade the firmware & see what happens.

glitter-ball, 2019-11-18:
My Greenstar 24i System ErP has no water pressure sensor, no boiler temperature sensor and no return sensor. In the boiler itself, there's a flow temperature sensor and that's it. I have weather compensation so I can get the outside temperature. I also have hot water control so have water temperature from a cylinder sensor. But that's it! The W-B boilers are clearly designed to need the minimum of sensor inputs to work sensibly, so if the control algorithm doesn't need a particular sensor, it's not fitted. Annoying, but there we are. I might think about using the DS temperature sensor input to pickup return temperature with a 1-wire sensor - that would be useful.

PhillyGilly, 2019-11-18:
Thanks glitter-ball. Sounds like we have identical boilers. There is an unused NTC input on the circuit board next to the DHW input. I haven't experiment yet, but it might be worth seeing if it is live (has voltage across it) and addressable?

glitter-ball, 2019-11-19:
Returning to the original post...

Devices:
Bosch Condens 2500/Buderus Logamax GB062/Junkers (Version:40.10 ProductID:95 DeviceID:0x08)
unknown? (Version:40.10 ProductID:95 DeviceID:0x09)

I assume the other device is a Drayton DT10RF which is plugged into the boiler, but I don't actually use - I have a hardwired 240V control which my software system actuates manually

I don't think the Drayton DT10RF has an EMS bus interface. Looking at the installation instructions, there's a 7-way ribbon cable between the DT10RF and the boiler PCB. If it was an EMS device, it could work with only 2 wires. I suspect the DT10RF has relay outputs which drive inputs on the boiler PCB to fire up the boiler in central heating or hot water mode.

This then leaves the question as to what the mystery bus item is and I have something similar on my system. It's something in the boiler, either on the main PCB or elsewhere. Both pump and fan in mine modulate with boiler output, so maybe one of them is on the EMS bus?

proddy, 2019-11-19:
This 0x09 device is a controller and already added to the 1.9.4 dev build. See `ems_devices.h`. Most boilers have one embedded. Use `log` and see what it's transmitting. I don't think its useful.

glitter-ball, 2019-11-29:
> Not sure what you mean by DS but there is an unused NTC input on the circuit board next to the DHW input which is intended for a low loss header temperature measurement.

I’m using bbqkees’ gateway which has a separate input for a Dallas Semiconductor temperature sensor. So I might fit this to the return pipe to provide that to Home Assistant.

PhillyGilly, 2019-11-29:
Thanks for that clarification. 
Does anybody know what is coming back in the 0x18 message on bytes 15-16? I'm guessing that Proddy is using the Bruderus message as described by Domotic (see [https://domoticproject.com/ems-bus-bruderus-nefit-boiler/] ) who says they're not used.
But in the EMS Wiki [https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme] says they are "DL-Erhitzer?" (and my German isn't up to translating that).
I've noted different values in those bytes 00 ED and 01 80 and 02 D0 but I can't relate them to anything.

proddy, 2019-11-29:
@PhillyGilly I actually used that german wiki when I started. The rest is trial and error. The bytes 15-16 is the storage-cell temperature. I think its a sort of mixer.  The value is a short so 0x00ED is 23.7 degrees.

glitter-ball, 2019-11-29:
I've just realised what the unknown device 0x09 on my system may be. It reports as:
` unknown? (DeviceID:0x09 ProductID:209 Version:01.03)
`
In the info menu of my boiler it reports the HMI Version as 1.03. So... I'm wondering whether the control panel on the front of the boiler is actually a separate EMS device on the bus communicating with the boiler PCB. Since I don't really change anything on the boiler panel, there's nothing for host 0x09 to transmit. 

I've just tried changing the HW setpoint from 60ºC to 59ºC and these packets appeared:
`(02:26:56.236) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 01 FF (CRC=29) #data=1
(02:26:56.262) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 02 3B (CRC=EB) #data=1
(02:26:56.287) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 09 00 (CRC=C6) #data=1
`
Reversing the change generates:
`(02:29:14.833) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 01 FF (CRC=29) #data=1
(02:29:14.859) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 02 3C (CRC=EC) #data=1
(02:29:14.884) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 09 00 (CRC=C6) #data=1
`
So the last byte of the second telegram (3B / 3C) is carrying the HW setpoint temperature from the boiler HMI to the boiler controller. Goodness knows what all the other bytes do...

PhillyGilly, 2019-11-29:
> @PhillyGilly I actually used that german wiki when I started. The rest is trial and error. The bytes 15-16 is the storage-cell temperature. I think its a sort of mixer. The value is a short so 0x00ED is 23.7 degrees.

Proddy. Apologies. That'll teach me not to guess! 
Glitter-ball. Have you compared the message with the UBAParameterWW  in the EMS-Wiki? WW-Solltemperatur is HW set point.

proddy, 2019-11-29:
> Proddy. Apologies. That'll teach me not to guess!
> Glitter-ball. Have you compared the message with the UBAParameterWW in the EMS-Wiki? WW-Solltemperatur is HW set point.

I recently discovered [this Junkers HT3 implementation](https://github.com/norberts1/hometop_HT3/blob/master/HT3/sw/lib/ht_discode.py) and use it as a cheat sheet to compare findings.

PhillyGilly, 2019-12-02:
> I've just realised what the unknown device 0x09 on my system may be. It reports as:
> `unknown? (DeviceID:0x09 ProductID:209 Version:01.03)`
> In the info menu of my boiler it reports the HMI Version as 1.03. So... I'm wondering whether the control panel on the front of the boiler is actually a separate EMS device on the bus communicating with the boiler PCB. Since I don't really change anything on the boiler panel, there's nothing for host 0x09 to transmit usually.
> 
> I've just tried changing the HW setpoint from 60ºC to 59ºC and these packets appeared:
> `(02:26:56.236) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 01 FF (CRC=29) #data=1 (02:26:56.262) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 02 3B (CRC=EB) #data=1 (02:26:56.287) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 09 00 (CRC=C6) #data=1 `
> Reversing the change generates:
> `(02:29:14.833) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 01 FF (CRC=29) #data=1 (02:29:14.859) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 02 3C (CRC=EC) #data=1 (02:29:14.884) 0x09 -> Boiler, type 0x33, telegram: 89 08 33 09 00 (CRC=C6) #data=1 `
> So the last byte of the second telegram (3B / 3C) is carrying the HW setpoint temperature from the boiler HMI to the boiler controller. Goodness knows what all the other bytes do...

Glitter-ball. Have you compared the message with the UBAParameterWW  in the EMS-Wiki? WW-Solltemperatur is HW set point (literally Warm Water Should be Temperature).

PhillyGilly, 2019-12-03:
> > @PhillyGilly I actually used that german wiki when I started. The rest is trial and error. The bytes 15-16 is the storage-cell temperature. I think its a sort of mixer. The value is a short so 0x00ED is 23.7 degrees.
> 
> Proddy. Apologies. That'll teach me not to guess!

Sill searching for my second temperature measurement.
Am I right in reading 0x18 bytes 14-15 as being the same as 0x34 bytes 5-6? and or 7-8?

(00:12:34.739) Boiler -> all, type 0x18, telegram: 88 00 18 00 34 02 07 64 2E 09 03 25 D0 80 00 **02 71** 80 00 00 E4 FF 00 00 00 C8 00 00 00 00 00 (CRC=84) #data=27
(00:12:35.980) Boiler -> all, type 0x34, telegram: 88 00 34 00 3C **02 71** **02 71** 21 00 00 03 00 00 20 7F 00 03 84 00 80 00 (CRC=D8) #data=19

According to the EMS-wiki they are Temperatur (DL-Erhitzer?) Warmwasser Temperatur Ist °C Warmwasser Temperatur Ist 2. Fühler

I suspect that the DL-Erhitzer?  storage-cell temperature is the hot water storage cylinder which is heated by a heat exchanger coil. Does that make sense.

proddy, 2020-02-02:
@PhillyGilly we may be luck. Norbert has extended his library to include all the Solar Module telegrams. See https://github.com/norberts1/hometop_HT3/blob/master/HT3/docu/HT3-Bus_Telegramme.pdf

closed 2020-07-06 by proddy