emsesp/EMS-ESP #227: Provide config example for FHEM

moustic999, 2019-10-31:
as we have some config for Home Assistant and some others,
It should be fine to have also some configuration example for FHEM
(www.fhem.de)

oorotor, 2019-11-01:
I'm using the MQTT2_SERVER in fhem, so if you are using this also and activate the autocreate attribute, then the mqtt-devices should be setup automatically.
My implementation in fhem for testing is very simple (caution, this example is without any encryption/user/password):

define m2s MQTT2_SERVER 1883 global
attr m2s room MQTT

define MQTT2_ems_esp MQTT2_DEVICE ems_esp
attr MQTT2_ems_esp IODev m2s
attr MQTT2_ems_esp readingList ems_esp:EMS/ems-esp/status:.* status\
ems_esp:EMS/ems-esp/start:.* start\
ems_esp:EMS/ems-esp/heartbeat:.* { json2nameValue($EVENT) }\
ems_esp:EMS/ems-esp/shower_data:.* { json2nameValue($EVENT) }\
ems_esp:EMS/ems-esp/boiler_data:.* { json2nameValue($EVENT) }\
ems_esp:EMS/ems-esp/tapwater_active:.* tapwater_active\
ems_esp:EMS/ems-esp/heating_active:.* heating_active\
ems_esp:EMS/ems-esp/thermostat_data:.* { json2nameValue($EVENT) }
attr MQTT2_ems_esp room MQTT

moustic999, 2019-11-01:
I have mqtt2_client, but works the same. I have the readings created under a mqtt2_device. That s the easy part. Now I want to have publish enabled as well as a convenient way of displaying data ....
A template for mqtt2_device would be great....

-------- Message d'origine --------

moustic999, 2019-11-01:
beginning of a config : 

define buderus.thermostat MQTT2_DEVICE
setuuid buderus.thermostat 5dbc5776-f33f-c58d-6657-0d77e45cd1e175d6
attr buderus.thermostat IODev mqttBroker
attr buderus.thermostat devStateIcon auto:time_automatic:heat manual:time_manual_mode:auto
attr buderus.thermostat eventMap /auto:auto/heat:manual
attr buderus.thermostat readingList mqttBroker:home/ems-esp/thermostat_data:.* { json2nameValue($EVENT) }
attr buderus.thermostat room Chauffage
attr buderus.thermostat setList auto home/ems-esp/thermostat_cmd_mode1 auto\
heat home/ems-esp/thermostat_cmd_mode1 manual\
temp home/ems-esp/thermostat_cmd_temp1 $EVTPART1
attr buderus.thermostat stateFormat hc1_mode

moustic999, 2019-11-30:
I made some progress 

`defmod buderus_thermostat MQTT2_DEVICE
attr buderus_thermostat IODev mqttBroker
attr buderus_thermostat devStateIcon set.auto:edit_settings:mode+manual set.manual:edit_settings:mode+auto auto:time_automatic:mode+manual manual:time_manual_mode:mode+auto  .*set.*°C:edit_settings
attr buderus_thermostat icon temp_control
attr buderus_thermostat jsonMap hc1_mode:mode hc1_seltemp:desired-temp
attr buderus_thermostat readingList mqttBroker:home/ems-esp/thermostat_data:.* { json2nameValue($EVENT, '',$JSONMAP) }
attr buderus_thermostat room Chauffage
attr buderus_thermostat setList mode:uzsuSelectRadio,auto,manual home/ems-esp/thermostat_cmd_mode1 $EVTPART1\
desired-temp:slider,15.0,0.5,30.0,1 home/ems-esp/thermostat_cmd_temp1 $EVTPART1
attr buderus_thermostat setStateList on off
attr buderus_thermostat stateFormat mode\
desired-temp °C
attr buderus_thermostat webCmd mode:desired-temp`

This allow to manage the thermostat, changing temp and mode. It works for a RC300, maybe it will require some adaptation for RC35, such as the modes

closed 2020-09-12 by proddy