emsesp/EMS-ESP #228: Change Thermostat temp for hc=2

suysh, 2019-11-01:
I have 2 heating circuits hc=1 and hc=2.
To change the temp for circuit hc=2, In a telnet session I do:
`Thermostat temp 2 20`
Then I receive
`Setting new thermostat temperature to 20.0 for heating circuit 2 type 0 (0=auto,1=night,2=day,3=holiday)`
But always circuit 1 is changed instead of 2. Is this a copy/paste typo in the code ?

proddy, 2019-11-01:
what thermostat are you using?

the logic is quite complex and it has to take into account all the 20+ variations of thermostat brands and types and also figure out whether its auto, night, day, manual modes

suysh, 2019-11-02:
I'm using a Bosch CW400.
See #206 for my full system.

proddy, 2019-11-03:
CW400 not implemented yet. Its part of #206 and we'll use that for tracking. See comment when I said "Let's fix that first and then move on to the CW400 Thermostat which has different device telegrams. "

closed 2019-11-03 by proddy