emsesp/EMS-ESP #229: Starting heating

mellbo, 2019-11-03:
I want to starting heating with this project without one thermostat. My boiler is **Bosch Condens 2500W.**

I would like to find this function in this project or I get some helps for this idea.

proddy, 2019-11-03:
questions
- when you say heating, do you mean hot tap water or central heating? These combi boilers have different functions for both
- how do turn the heating/hot water on now? What ever manually process you do now, do it and see if there is traffic on the EMS bus you can mimic
- when you said you looked through the wiki's I guess you missed reading about the 1A telegram [here](https://emswiki.thefischer.net/doku.php?id=wiki:ems:telegramme#wwbetriebsart)?

The 1A telegram takes two values, one for the power for hot water and one for heating. For example sending `0B 08 1A 00 5A 64 00 00` will turn the power to 90% and 100% which effectively "turns on" the boiler. You need to look into the manual of your boiler to find the max values. Like wise `send 0B 08 1A 00 00 00 00 00` turns the boiler "off". This is EMS+ and works for my boiler, I don't know whether it will work on the Bosch. 
 
Please like/star my GitHub project for some appreciation of the support you're getting. thank you.

mellbo, 2019-11-03:
- when I say start heating I say start central heating to radiators. 
- in present I made one DIY smart home system, calculate temperature from all room, then command central just with simple on/off relay. but I want to controled more. like power and set temperature  of water in central heating...
- when I say about wiki, I not missing about 0x1A. but I tried with no result with 0x1A.

I have a misunderstanding. in document says at offset 00 we have "boiler setpoint temperature" and not "power request HK" and the suggested code is 0B 08 1A 00 **5A** 64 00 00 where 5A would not be "boiler setpoint temperature"?

proddy, 2019-11-03:
5A is 90% power to the boiler heating circuit (not hot water tap). See the link I sent

mellbo, 2019-11-03:
```
try 01 :: Corrupt telegram
(00:42:38.834) Sending raw: 0B 08 1A 00 5A 64 00 00 (CRC=CF)
(00:42:38.856) Corrupt telegram: 0B 84 1A 00 5A 64 00 00 (CRC=CF) #data=4

try 02 :: Corrupt telegram
(00:45:22.928) Sending raw: 0B 08 1A 00 43 50 00 00 (CRC=D7)
(00:45:22.951) Corrupt telegram: 0B 84 1A 00 43 50 00 00 (CRC=D7) #data=4

try 03
(00:47:00.126) Sending raw: 8B 88 1A 00 43 50 00 00 (CRC=CA)
(00:47:07.407) Boiler -> all, type 0x18, telegram: 88 00 18 00 35 01 14 43 00 00 01 00 40 01 22 80 00 80 00 FF FF FF 00 00 00 00 00 00 20 (CRC=B4) #data=25
<--- UBAMonitorFast(0x18)
```
anyway. i`m still confused with `0x1A`
I know that... : 
```
 [source][destination][offset][begin 5][begin 6][begin7][begin8]
      0B            08            00         5A         64          00          00
```

and in link `0x1A` description say:
  [begin 5] = Boilert setpoint temperature
  [begin 6] = 0..100 power request HK
  [begin 7] = 0..100 power requirement WW
  [begin 8] = always 0

so. what I have at [begin 5] ...

mellbo, 2019-11-03:
this log raw appear when I put central in setup mode and select maximum power for heating central to 67. (max set to 82)[menu setup at: L1->1.A1]

```
(00:32:05.622) 88 00 2A 00 00 00 00 00 03 02 00 01 98 00 00 80 00 00 80 00 80 00 80 00 (CRC=F6) #data=20
(00:32:06.401) 88 00 2A 00 00 00 00 00 03 02 00 01 98 00 48 80 00 00 80 00 80 00 80 00 (CRC=53) #data=20
(00:32:06.651) 88 00 2A 00 00 00 00 00 03 02 00 01 98 00 A8 80 00 00 80 00 80 00 80 00 (CRC=31) #data=20
(00:32:06.901) 88 00 2A 00 00 00 00 00 03 02 00 01 98 00 FC 80 00 00 80 00 80 00 80 00 (CRC=DE) #data=20
(00:32:07.151) 88 00 2A 00 00 00 00 00 03 02 00 01 98 01 2C 80 00 00 80 00 80 00 80 00 (CRC=C5) #data=20
(00:32:07.401) 88 00 2A 00 00 00 00 00 03 02 00 01 98 01 50 80 00 00 80 00 80 00 80 00 (CRC=B5) #data=20
```

proddy, 2019-11-03:
i was mistaken then, begin 5 is the temp (90%) and begin 6 is 100% power.

mellbo, 2019-11-03:
anyway...with `0B 08 1A 00 5A 64 00 00` system return 'corrupt telegram' , and with `BB 08 1A 00 5A 64 00 00` system does not return any information, but boiler does nothing...

I dont know how to put boiler in TEST mode.. 

You made this project very well, I like your ideea. Meibe in the future you find my answer....

proddy, 2019-11-03:
if you get corrupt telegram try different tx_mode settings

moustic999, 2019-11-04:
in your case the tx_mode should be set to 3, because this heater is a heatronic-like.
few questions for you in order to provide some help : 

You don't have any thermostat, but today how does it work ? how do you make it to heat ? 

What do you expect by "start heating" ?

mellbo, 2019-11-04:
> in your case the tx_mode should be set to 3, because this heater is a heatronic-like.
> few questions for you in order to provide some help :
> 
> You don't have any thermostat, but today how does it work ? how do you make it to heat ?
> 
> What do you expect by "start heating" ?

my friend, i don't have thermostat with EMS bus. in present central heating is started with ON/OFF by one personal diy project controlled from http web.

At this boiler (Bosch Condens 2500W) EMS line is same line for ON/OFF heating central.

Now, can you help me..?

bbqkees, 2019-11-05:
It could very well be that the boiler does not accept heating messages from source 0x0B, only from 0x10/0x17 which are real thermostat ID's. 
Just buy a cheap second hand EMS thermostat and do some logging to see how this works.

higgers, 2019-11-06:
Are you able to look inside your boiler at the wiring connections? Are there wiring terminals labelled L, LR and LS? These can be used with a switch to turn the heating circuit on and off.

A simple room thermostat uses these connections to determine whether to turn the heating on and off. If the thermometer in the room thermostat measures the temperature as less than the temperature selected by the user then the thermostat turns on the heating.

It may be possible to determine which telegram to use to turn the heating circuit on and off, this is what a smart thermostat like a Bosch Easy Control does, but using a physical switch might be a quicker solution.

mellbo, 2019-11-07:
Bosch Condens 2500W, use same line (2 wire) for BUS or Manual start heating. So, if the BUS is strapped, then central start heating. But if I shorted port, then I have no communication on the bus..

mellbo, 2019-11-10:
please those who have a **Bosch Condens 2500W** boiler to send me a raw log in which the communication with the thermostat will be surprised when the boiler goes into heating mode. I do not have a thermostat to make these logs myself

mellbo, 2019-12-13:
How can I send to TX this format of message:  **0x18 [BRK]** Just that.
I check with logic analyzor and CR10 send every 100ms this code: **0x18[BRK]** .. 7ms Pause then **0x98[BRK]**

proddy, 2019-12-14:
What you’re seeing is the poll. I think I explain it in the wiki.

mellbo, 2019-12-14:
It seems **EMS_Sys_Status.emsPollEnabled** remaining false.

I should set the value :    EMS_Sys_Status.emsPollEnabled = true; ?

mellbo, 2019-12-15:
anyway. ems-esp dies not reply to central when she send the poll.

closed 2019-12-15 by mellbo