emsesp/EMS-ESP #230: MQTT base prefix ignored for commands

davidw3591, 2019-11-03:
Hi, I changed the MQTT base prefix to "homeassistant".

Commands send to for example **homeassitant/ems-esp/thermostat_cmd_mode** are ignored.
Commands send to **home/ems-esp/thermostat_cmd_mode** arrive in EMS-ESP.

All status data does arrive in **homeassistant/ems-esp/#**

mqttlog command in telnet outputs this information:
[image]

proddy, 2019-11-03:
I see the subscriptions are incorrect. Can you upgrade to the latest version?

I just tested with setting the MQTT base and it works here. The correct topic to use would be `homeassistant/ems-esp/thermostat_cmd_mode1`. The heat controller is appended to the end to support the newer thermostats.

If your broker still listens to home/ems-esp its probably because of an old retain message. Mosquitto does and you'll need to delete them. I will change the code anyway to detect this and ignore.

thanks for reporting

davidw3591, 2019-11-03:
Thanks, will try to upgrade next week.

> I just tested with setting the MQTT base and it works here. The correct topic to use would be `homeassistant/ems-esp/thermostat_cmd_mode1`. The heat controller is appended to the end to support the newer thermostats.
thermostat_cmd_mode1 doesn't seem to work?
I can only set the mode with thermostat_cmd_mode

moustic999, 2019-11-03:
I don't agree for this, why everything should be related to home-assistant ? 
I think ems-esp should remains as generic as possible. 
everbody does not use home-assistant.

moustic999, 2019-11-03:
there is already a change made from "manual" to "heat" ! Stupid stuff, my thermostat has auto and manual, I don't care HA decide to constraint to use what they decided. please keep ems-esp, automation software agnostic and generic so everybody can use it.

Or better implement something as esphome did, a generic mqtt mode or a dedicated HA mode, that you can choose at setup.

moustic999, 2019-11-03:
that could be also possible to set the prefix in the web gui

proddy, 2019-11-03:
@moustic999 is right. The mapping should be done by the automation software and not EMS-ESP.

davidw3591, 2019-11-03:
Guys, this bugreport has nothing to do with homeassistant. I just used the GUI option to change the MQTT prefix and it didn't work as intended. Indeed I worked around this issue by changing the mapping in the automation software.

davidw3591, 2019-11-03:
> Thanks, will try to upgrade next week.
>
>
> thermostat_cmd_mode1 doesn't seem to work?
> I can only set the mode with thermostat_cmd_mode

@proddy Any thoughts on this?

proddy, 2019-11-03:
Which version are you on?

davidw3591, 2019-11-03:
1.9.3

proddy, 2019-11-04:
should work. try again with the dev version if you can

proddy, 2019-11-13:
not reproducible and been open for 2 weeks, closing.

closed 2019-11-13 by proddy