emsesp/EMS-ESP #231: fixed IP ?

bennybubble, 2019-11-03:
Hi,

Is it possible to define a fixed IP address for the EMS-ESP firmware? 

-ben

proddy, 2019-11-03:
no, but I can add that feature.

bennybubble, 2019-11-03:
That would be great since I like to have all my devices at fixed IP rather then DHCP.

proddy, 2019-11-04:
its common to use DHCP to assign a fixed IP. Is there a reason why you don't use this?

bbqkees, 2019-11-04:
You could just make a DHCP reservation. Standard stuff of every modern router.
If you don't use DCHP reservations and set a static IP on the ESP, the router might hand out that same IP address to another device if its inside its DHCP pool and f.i. the ESP was inactive for a while.

stbuerger, 2019-11-05:
I like to vote for fixed IP too.
I'm thinking of putting all IoT devices into separate (isolated) network.
@bbqkees - you would wonder how "flexible" current ("modern") consumer routers are ;-) ;-) ;-)

proddy, 2019-11-05:
This feature is not top of my list at the moment. I would mean adding 4 more parameters (ip address, gateway address, netmask and DNS) to the code as both telnet `set` parameters and also the Web UI with all the additional conditional logic around it. It's quite a bit of extra code which takes time to write and test. And the firmware sizes increases too.

Any more votes?

higgers, 2019-11-06:
I use DHCP to assign a fixed IP address outside of the DHCP range used for dynamically assigned addresses.

lmdecroos, 2019-11-07:
I use fixed IP addresses and don't assign a gateway for devices that doesn't need internet access

proddy, 2019-11-09:
@bennybubble could you try it out and let me know if it works?

lmdecroos, 2019-11-09:
It doesn’t work for me :(

I have set the IP adres in the web GUI, when checking the back-up config file the ip adress is there. But it seems that DHCP is taken over.
My gateway is still taken the IP from the DHCP server

  "wmode": 0,
    "staticip": "192.168.0.21",
    "gatewayip": "",
    "nmask": "255.255.255.0",
    "dnsip": “"

I would assume that you need to add a selection between DHCP and static IP
on which base have you disabled DHCP ?

proddy, 2019-11-09:
I had the same thing. I could be that some wifi routers don't allow a fixed IP. Does it work with other devices on your network that use a fixed IP address?

lmdecroos, 2019-11-10:
tested on my laptop  running debian and i had no issue to configure a static ip address while using wifi
are you sure you are also changing the mode from DHCP to static ?

bennybubble, 2019-11-10:
Just installed the beta version and fixed ip works perfect!

Many thanks for this!

proddy, 2019-11-10:
closing this feature

closed 2019-11-10 by proddy

lmdecroos, 2019-11-10:
strange behaviour

works now also for me when i’m entering a gateway IP address
when only entering the static ip address and net mask, the interface is not going in static mode but is still using DHCP mode

proddy, 2019-11-11:
this is normal behavior in the WiFi library I'm using. For non-DHCP an IP, gateway and dns are needed. https://github.com/xoseperez/justwifi/blob/8bb2c16f5b91f56de191cb8ff96a816fb672cfd2/src/JustWifi.cpp#L663

lmdecroos, 2019-11-11:
I saw in the scrip that the gateway is mandatory. Some other implementation are also requiring these 3 parameters, for others it’s working with only the IP adres and netmask.
As in the Web GUI no fields are shown as mandatory I hoped and assumed the  gateway was not required.

Anyway, when providing an ip address of an non existing device for the GW, the EMS—ESP will never be able accessible from other another subnet or the internet

proddy, 2019-11-11:
I checked espurna and esphome (https://esphome.io/components/wifi.html) and they all require a mandatory gateway