emsesp/EMS-ESP #234: Unkown Device - Buderus Logamax Plus GB122-24 KD H

theloukou, 2019-11-04:
The boiler cannot be recognized from the ESP.
Log from telnet is below:

These device IDs are on the EMS Bus: 0x08
and 1 were recognized by EMS-ESP as:
 unknown? (DeviceID:0x08 ProductID:234 Version:05.03)

proddy, 2019-11-04:
why did you close this? Shall I add the boiler to the list?

theloukou, 2019-11-04:
I closed it because or wrong tag, i reopened as issue #235.
Add the boiler if possible though :)
Also, the telegram tags my boiler sends are mostly 0xE4 and 0xE9, which dont seem to be of a know type. Should i open a knew topic for help with this? Am i just doing something wrong? :P

proddy, 2019-11-04:
keep using #235 . I'll add it to the device list and this should fix the boiler connection

closed 2019-11-04 by proddy