emsesp/EMS-ESP #236: Regular EMS message types are different?

theloukou, 2019-11-04:
My boiler seems to publish different type of messages for _UBAMonitorFast_ and _UBAMonitorSlow_, and maybe _UBAMonitorWWMessage_. The only type that i can actually get a proper response is **0x14**, the _UBATotalUptimeMessage_.
I get from telnet the types **0xE3**, **0xE4**, **0xE9** every 10 seconds. From googling around i found this [https://github.com/Th3M3/buderus_ems-wiki/blob/master/Quelle_08.md](url)
The aformentioned types are in there, and are related to the messages stated at the start.
Is there any way to add messages that the ESP undestands? Or replace the ones it already understands with mine? 

Boiler is Buderus Logamax Plus GB122 (ProductID:234 Version:05.03)

proddy, 2019-11-04:
same as https://github.com/proddy/EMS-ESP/issues/235 and https://github.com/proddy/EMS-ESP/issues/220

we're working on adding support these new boilers (GB122 and GB125)

proddy, 2019-11-05:
Using @220 to track progress of implementing this.

closed 2019-11-05 by proddy