emsesp/EMS-ESP #237: MQTT publish doesn't update all topics

lmdecroos, 2019-11-09:
I'm testing version 1.9.4b9 and spotted that some of the MQTT topics as thermostat_data are only published once, shortly after startup. They are not anymore published afterwards

I tested also the new option "publish_always=on" and this didn't made any difference.
Even after rebooting with publish_always=on, some of the topics are only published once and never updated afterwards

MQTT publish log:
  Timestamp:07:07:39 Topic:status Payload:online
  Timestamp:07:07:39 Topic:start Payload:start
  Timestamp:07:07:39 Topic:shower_data Payload:{"timer":"0","alert":"0"}
  Timestamp:07:23:45 Topic:boiler_data Payload:{"wWComfort":"Hot","wWSelTemp":56,"wWDesiredTemp":70,"selFlowTemp":53,"selBurnPow":100,"curBurnPow":25,"pumpMod":36,"wWCircPump":0,"wWCurTmp":49.1,"wWCurFlow":0,"curFlowTemp":54.3,"retTemp":49.9,"switchTemp":51.6,"sysPress":1.6,"boilTemp":53,"wWActivated":"on","wWOnetime":"off","burnGas":"on","flameCurr":8.6,"heatPmp":"on","fanWork":"on","ignWork":"off","wWCirc":"off","heating_temp":76,"pump_mod_max":91,"pump_mod_min":30,"wWHeat":"off","wWStarts":5568,"wWWorkM":39905,"UBAuptime":3400321,"burnStarts":26535,"burnWorkMin":407695,"heatWorkMin":367790,"ServiceCode":"-H","ServiceCodeNumber":200}
  Timestamp:07:23:45 Topic:tapwater_active Payload:0
  Timestamp:07:23:45 Topic:heating_active Payload:1
  Timestamp:07:07:48 Topic:thermostat_data Payload:{"hc1":{"daytemp":19,"nighttemp":15,"holidayttemp":17,"heatingtype":1,"mode":"auto"},"hc2":{"seltemp":13.5,"currtemp":14.3,"daytemp":18,"nighttemp":13.5,"holidayttemp":17,"heatingtype":1,"circuitcalctemp":5}}

publish_time=120
publish_always=on

proddy, 2019-11-09:
I’ll check in a minute

lmdecroos, 2019-11-09:
It took 20 minutes before the first update

now it’s publishing every 15  a 30 seconds even there is no change

I’m not sure about the expected behaviour of the new functionlity

proddy, 2019-11-09:
can't reproduce, but I'm looking into the code again. The recent change doesn't touch how MQTT publishes. It could be that the CRC library I used for checking payload changes has been modified.

lmdecroos, 2019-11-09:
It could be a coincidence but I believe the publish  in beta 1.9.4b9 started to work for all topics after i logged in to web GUI

I upgraded from the 1.9.1 release to latest beta.

I have now downgraded back to 1.9.1 ( as I still need the old MQTT topics to my 2 HC’s) and MQTT is now publishing every 15  a 30 seconds instead of every 120 seconds
Even when changing the publish_time from 120 to 240, MQTT publishes every 20 seconds

It seems that the web GUI settings are not fully aligned with the console settings.
As i’m in favour of console, I will always console instead of  a web ui

proddy, 2019-11-09:
I prefer Telnet too. I built the webUI for easy of use.

Anyway how's it always worked is that MQTT messages will be sent out when the payload has changed regardless of the `publish_time` setting.  So this is typically between 10-20 seconds. The `publish_time` is used to force a regular update of everything (boiler, shower, thermo, heatbeat etc) even if nothing has changed. It does sound strange I agree so I may need to re-think if its useful.

I tested with the latest b9 and use MQTT explorer which shows the incoming MQTT messages, also the values that have changed since the last publish and the timestamp history. I can see changes coming in.

moustic999, 2019-11-09:
At start-up it is important to send everything. Afterwards sending only when bus emits should be enough, because bus already emits cyclicly....

-------- Message d'origine --------

lmdecroos, 2019-11-09:
ok, then maybe I’m confused regarding the publishing update frequency

proddy, 2019-11-09:
@moustic999 this is how it works now. MQTT data is always sent when there are values, which is typically 1-2 seconds after boot up when the EMS bus is connected. I made a change now for 1.9.4b10 which
- removes publish_always
- makes publish_time <seconds> manually publish at the given frequency, or with a value of `0` go to automatic mode and send as soon as there is new data to send.

@lmdecroos in my testing I can't reproduce what you found in the latest dev release. Or find anything strange in the code. It could have been issues upgrading from a very old 1.9.1 release. If you need help getting to 1.9.4 let me know.

lmdecroos, 2019-11-09:
> makes publish_time manually publish at the given frequency, or with a value of 0 go to automatic mode and send as soon as there is new data to send.

@proddy thanks for testing, I will maybe try to simulate the strange behaviour from this morning later on
I'm stuck on version 1.9.1 since the MQTT topics has been changed from 1.9.2 onwards and the parsing for Domoticz has not yet updated. I'm waiting for bbqkees or someone else to amend the parsing plug-in

proddy, 2019-11-09:
I'll give bbqkees a nudge. 1.9.5 is coming out soon!

lmdecroos, 2019-11-09:
I tested b10 and when setting publish_time to 0 then after a reboot the gateway is sometimes taken the “wrong” thermostat.
but the updates are ok :)

My main thermostat is an RC35, the RC20 is only to control HC2

I have never seen this behaviour before

the RC20 has less details than the RC35
What is circuitcalctemp ?

 Thermostat: RC20RF (ProductID:93 Version:02.00)
Thermostat time is 23:16:51 09/11/2019
Heating Circuit 1
 Current room temperature: 18.9 C
 Setpoint room temperature: 15.0 C
Heating Circuit 2
 Current room temperature: 13.9 C
 Setpoint room temperature: 13.5 C
 Timestamp:23:17:43 Topic:thermostat_data Payload:{"hc1":{"seltemp":15,"currtemp":18.9,"circuitcalctemp":5},"hc2":{"seltemp":13.5,"currtemp":13.9,"circuitcalctemp":5}}

Thermostat stats:
Thermostat: RC35 (ProductID:86 Version:21.08)
Thermostat time is 23:18:01 09/11/2019

Timestamp:23:38:37 Topic:thermostat_data Payload:{"hc1":{"daytemp":19,"nighttemp":15,"holidayttemp":17,"heatingtype":1,"mode":"auto"},"hc2":{"daytemp":18,"nighttemp":13.5,"holidayttemp":17,"heatingtype":1,"mode":"night”}}

proddy, 2019-11-09:
oh. The code doesn't handle more than 1 thermostat. I'll need to think how best to handle this.

lmdecroos, 2019-11-11:
@proddy my RC35 is the main thermostat, this one use use to program my 2 HC’s and boiler and control the temperature of my HC1
the RC20 is the thermostat that is use to measure and control the temperature of the rooms heated by HC2

It’s only strange that from the first use of the EMS gateway  ( version 1.8.3) until 1.9.4b9 the EMS  gateway used always the information from RC35.
only in 1.9.5b10 and when  publish_time is set to 0 after some reboots ( for testing) te EMS gateway takes my RC20 as main thermostat and ignore the values of the RC35

i spotted that you have created a new feature to support more thermostats per system, i’m not sure if this related to this.
But the EMS software  should always use the main thermostat of the system, as all information should be available in the main thermostat

hope this makes sense and i don’t create unnecessary confusion

proddy, 2019-11-11:
@lmdecroos how can EMS-ESP detect what the main thermostat is?

proddy, 2019-11-13:
@lmdecroos in the EMS bus I'm guessing you'll have two EMS devices (RC20 and RC35) but sending messages. RC20 will pick up the HC1 controls and RC35 all 4 HCs. I therefore would expect a conflict. Can you confirm this?

proddy, 2019-11-13:
This bug was related to MQTT which is fixed, so closing. Please use issue #238 to answer my last question.

lmdecroos, 2019-11-16:
@proddy yes this is correct with the small correction that the RC35 is controlling HC1 and is the main device for programming the boiler. The RC20 is controlling the temperature of HC2

In case you have multiple HC's you will always having a dedicated thermostat per HC

proddy, 2019-11-16:
@lmdecroos so in your situation is there still a problem with 1.9.4b that I need to fix? Are you seeing the  correct HC1 and HC2 values? If not then I expect you're getting two sets of HC2 data, one from the RC20 and one from the RC35.

proddy, 2019-11-20:
Please use #238 for this

closed 2019-11-20 by proddy