emsesp/EMS-ESP #238: Support for multiple Thermostats (with multiple heating circuits)

proddy, 2019-11-10:
Some people have more than one Thermostat to control different heating circuits.

https://github.com/proddy/EMS-ESP/issues/237#issuecomment-552144215

martinSezman, 2019-11-23:
I have two thermostats on EMS bus - RC310 and RC200.
RC310 can control (calculate parameters) up to four HCs.
RC200 can control only one HC (if works alone) or works as slave with RC300/RC310.
Regarding Buders documentation for multiple HC these is a need to have one master (and only one) RC300/RC310 and additional termostat for every additional HC. In this case thermostat (RC200) works as thermometer and additional control panel for some zone. But every calculations are made by RC300/RC310.
So I think esp-ems should give higher priority for some detected thermostat devices like RC300, RC310,RC35,RC30 prior to RC200, RC100, RC25 and RC10.

All command to HC control should be send to master thermostat.

lmdecroos, 2019-11-24:
I have 2 thermostats a RC35 and a RCF20
The RC35 is the master thermostat and should be used to read out all settings.
Until version version 1.9.4b9, the gateway was always reading out the values of the RC35.
Since version 1.9.4b10, the gateway started to use my RC20 as main thermostat to readout the values. As the RC20 doesn't have all details, the information became less detailed.
With version 1.9.4b9 the gateway is choosing  "randomly" the RC35 or RC20 as master, after a reboot the master could been changed.
In version: 1.9.4b17, the gateway is always choosing the wrong one :(
In case there is no logic to determine who is the master thermostat, then this should become a setting

proddy, 2019-11-30:
So how do we solve this?
- does it make sense to support and show all the thermostats and their values in the `info` command as separate devices?
- should we create an option to choose the master device id, like `set thermostat_master <product id>`

martinSezman, 2019-11-30:
- all thermostat in info - it will be very use full
- set thermostat_master  - great idea. In my opinion this parameter could have also default option "auto". "auto" could automatically use choose master - thermostat with the lowest DeviceID on EMS bus

lmdecroos, 2019-12-04:
I have no idea what has been changed since 1.9.4 as all previous versions took the correct (RC35) thermostat as main device.
Is there a “master thermostat” parameter in the EMS protocol or will you need to create a setting in the ems gateway?
I would assume that the master thermostat is always controlling HC1 and has the details from the other HC's

proddy, 2019-12-04:
@lmdecroos that's an interesting approach. There must be something to tell which thermostat is the master. I'll have a look around and see where this may be held.

tbr, 2019-12-15:
Hi, my installation has the same issue. Two thermostats on one bus, the ems-esp takes the wrong thermostat (the slave) and as a consequence, setting the temperature does not work. My particular system has an RC200 and and RC300. The RC300 is the master that manages two circuits (floor and radiators).

Now, I made a small patch in the code (where the thermostat is detected), to check for the 'writable' flag of each detected thermostat. By selecting the writable one, the esp is able to correctly select the master thermostat on the bus... the patch (in src/ems.cpp) is minimal, and I will do a PR after some more testing.

proddy, 2019-12-15:
@tbr contributions always welcome, thanks! Although could you do this on the dev branch?

tbr, 2019-12-17:
@proddy I agree with the 'better' solution to be able to select the master as configuration. In any case, here is a PR to get things working in the meanwhile. Thank you all for this great project...

PS: I messed up some commit/branch/PR stuff first. I hope this PR is OK and without mistakes.

martinSezman, 2019-12-20:
two thermostats RC310 and RC200, Please take a look how it looks in 1.9.5b3 WebUI:  [image]

tbr, 2019-12-20:
On my mini-d1, 1.9.4 and later always crash somehow (after a while). It seems like a heap problem somewhere :(

lmdecroos, 2019-12-21:
Build and OTA uploaded  EMS-ESP-1_9_5b3-d1_mini.bin  and my ESP is not anymore booting, a power cycle doesn’t helps
The led blinks every 5 seconds …
Anyway to recover ?

lmdecroos, 2019-12-21:
Downgraded to 1.9.3 via USB and fully recovered :)

proddy, 2019-12-21:
Did you ever try 1.9.4 when it was in dev/beta?

lmdecroos, 2019-12-21:
from 1.9.4b10 onwards the software takes for my installation the wrong thermostat, therefore I tested the following versions only shortly

proddy, 2019-12-21:
ah right. I really need to fix this (or apply the patch PR)

proddy, 2019-12-30:
Implemented a new set option called `set master_thermostat <product id>`. Typing `set master_thermostat` will list available ids to use. Can someone test?

tbr, 2019-12-30:
I flashed the new build for testing. Will let you know if it works (or not). Seems to work...

Two things:
- The help text misses the set master_thermostat command. Maybe you can add it?
- It would be nice to add the option to the web UI?

Also, I need to restart the system after setting a new master thermostat before it gets used. I expect this is normal (and it is not a problem...).

Thanks for the work :)

proddy, 2019-12-30:
* typing `set` shows you the commands. It's added.
* In the Web UI do you mean like putting a "*" in the devices list to show which one is master?
* I'll make a change so that it does a device rescan after the 'set' command, so no need to reboot

martinSezman, 2019-12-30:
many thanks @proddy !
it works! 

only in WebUI Setpoint Temperature, Current Temperature and Mode is not available as screen shot below
[image]

when info in console shows
[image]

lmdecroos, 2019-12-30:
My compiled binary (EMS-ESP-1_9_5b7-d1_mini.bin )  is again not booting, similar as i have with a previous 1_9_5 beta build
I was previous able to compile dev branched up to 1.9.4bx

Has there anything changed in the way the dev branch needs to be compiled ?
I have to make the following change in the platformio.ini config file:

[platformio]
;default_envs = release
default_envs = debug

i’m compiling on Raspberry 

thanks in advance

proddy, 2019-12-30:
ah ok - I think I know the problem

proddy, 2019-12-30:
@lmdecroos nothing has changed. Best to grab the image from the github releases page, or just use the web interface to automatically upgrade to the latest dev build

lmdecroos, 2019-12-30:
I’m trying the latest dev to test the master_thermostat command.

my successful compiled binary is 725952 bytes EMS-ESP-1_9_5b7-d1_mini.bin  but the precompiled version on github is only 616768 bytes for the EMS-ESP-dev-d1_mini.bin

The good news is that the set master_thermostat in the b7 works :)
many thanks for this fix
 

Should i see 2 mixing circuits as i have 2 full independent HC’s

Mixing module stats:
  Switch temperature: 26.3 C
  Mixing Circuit 1
   Current flow temperature: 3353.6 C
   Setpoint flow temperature: 0 C
   Current pump modulation: 0 %

looks very hot :(

proddy, 2019-12-31:
you should be seeing the two HCs under Thermostat when typing `info`. I'm not sure why it shows up as a Mixer?

lmdecroos, 2020-01-02:
I’m now testing 1.9.5b8 and after a while ( several hours) not only my two HC’s  are listed but als the non existing HC 3 and HC4

Thermostat stats:
  Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
  Thermostat time is 06:17:27 02/01/2020
  Heating Circuit 1
   Current room temperature: 16.4 C
   Setpoint room temperature: 19.0 C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to auto
   Day Mode is set to day
  Heating Circuit 2
   Current room temperature: 13.7 C
   Setpoint room temperature: 13.5 C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to night
   Day Mode is set to night
  Heating Circuit 3
   Current room temperature: ? C
   Setpoint room temperature: ? C
   Program is set to Summer mode
   Day temperature: ? C
   Night temperature: ? C
   Vacation temperature: ? C
   Mode is set to auto
  Heating Circuit 4
   Current room temperature: ? C
   Setpoint room temperature: ? C
   Program is set to Summer mode
   Day temperature: ? C
   Night temperature: ? C
   Vacation temperature: ? C
   Mode is set to auto 

I spotted this yesterday after running for about 1 day and again this morning with an uptime of 20 hours

When rebooting the gateway, only 2 HC’s are listed as expected

proddy, 2020-01-02:
I expect you are getting messages from HC3 and HC4 which are empty and EMS-ESP is thinking you have an active HC there. These will be from telegrams 0x51 and 0x52 (HC2) and 0x5B and 0x5C (HC2).  See if you see these in the logs. Or just ping them with 'thermostat read 51' for all 4. Then I can put the check into the code.

lmdecroos, 2020-01-02:
@Proddy : the 2 non-used  HC’s are added after a change in thermostat mode eg from auto to day   or from night to day

thermostat read 51

(12:35:31) 0B 90 51 00 20 
(12:35:32) 10 0B 51 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 
Fetching data from EMS devices
Starting scheduled MQTT publish...
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
Publishing mixing device data via MQTT
[MQTT] Error publishing to home/ems-esp/mixing_data with payload {"hc1":{"flowTemp":3353.6,"setflowTemp":0,"pumpMod":0}} [error 0]
(12:35:35) 0B 90 51 00 20 
(12:35:35) 10 0B 51 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 

Manual change fro, Auto to day mode on RC35 for HC1
(12:36:20) 10 00 51 07 01 
Fetching data from EMS devices
(12:36:35) 0B 90 51 00 20 
(12:36:35) 10 0B 51 00 00 1E 26 22 00 28 00 01 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02

proddy, 2020-01-02:
51 is HC3 so why is it coming back with information? I thought you said you had only 1 and 2 enabled.

lmdecroos, 2020-01-02:
Yes i have only HC1 and HC2
I’m able to simulate the issue with version 1.9.5b  by changing between day / night / auto on my MC35 master thermostat or as part of the active schedule.
when using Domoticz and changing  between day / night / auto by MQTT commands  I’m NOT able to simulate the issue

1.9.5.b8

(16:02:29) Thermostat -> All, type 0x3E, telegram: 10 00 3E 00 00 00 1E 00 C2 00 00 00 00 00 00 00 00 11 05 00 (CRC=31) #data=16
(16:02:29) Thermostat -> All, type 0x3D, telegram: 10 00 3D 07 02 (CRC=E1) #data=1
(16:02:30) Thermostat -> All, type 0x51, telegram: 10 00 51 07 02 (CRC=48) #data=1
(16:02:30) Thermostat -> All, type 0x5B, telegram: 10 00 5B 07 02 (CRC=60) #data=1
(16:02:33) Thermostat -> All, type 0x48, telegram: 10 00 48 00 00 00 1B 00 8B 00 00 00 00 00 00 00 00 11 05 00 (CRC=68) #data=16
(16:02:34) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 6E 11 00 00 C2 00 C8 00 C8 7D 00 7D 00 (CRC=67) #data=13
(16:02:35) Thermostat -> Switching Module, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
(16:02:37) Thermostat -> All, type 0x06, telegram: 10 00 06 00 14 01 10 02 01 1E 03 00 (CRC=5E) #data=8
(16:02:38) Thermostat -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 00 00 00 00 (CRC=91) #data=4
(16:02:46) Thermostat -> Mixing Module, type 0xAC, telegram: 10 21 AC 00 05 00 01 (CRC=0F) #data=3
Fetching data from EMS devices
Starting scheduled MQTT publish...
Publishing boiler data via MQTT
Publishing thermostat data via MQTT
Publishing mixing device data via MQTT
(16:03:04) Thermostat -> Me, type 0x3E, telegram: 10 0B 3E 00 04 02 26 00 C2 00 00 00 00 00 00 00 00 11 05 00 (CRC=54) #data=16
(16:03:04) Thermostat -> Me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=BE) #data=27
(16:03:06) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 00 00 1B 00 8B 00 00 00 00 00 00 00 00 11 05 00 (CRC=00) #data=16
(16:03:06) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 01 (CRC=20) #data=2
(16:03:07) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27
(16:03:08) Thermostat -> Me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
(16:03:09) Thermostat -> Me, type 0x51, telegram: 10 0B 51 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=77) #data=27
i(16:03:09) Thermostat -> Me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
nfo(16:03:10) Thermostat -> Me, type 0x5B, telegram: 10 0B 5B 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=05) #data=27

1.9.3

(00:04:10.096) Thermostat -> me, type 0x3E, telegram: 10 0B 3E 00 00 00 1E 00 C6 00 00 00 00 00 00 04 00 11 05 00 (CRC=12) #data=16
(00:04:10.634) Thermostat -> me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=BE) #data=27
Publishing thermostat data via MQTT
(00:04:10.824) Thermostat -> all, type 0x3D, telegram: 10 00 3D 07 02 (CRC=E1) #data=1
(00:04:11.076) Thermostat -> all, type 0x51, telegram: 10 00 51 07 02 (CRC=48) #data=1
(00:04:11.730) Thermostat -> me, type 0x48, telegram: 10 0B 48 00 00 00 1B 00 8A 00 00 00 00 00 00 00 00 11 05 00 (CRC=C8) #data=16
(00:04:11.826) Thermostat -> all, type 0x5B, telegram: 10 00 5B 07 02 (CRC=60) #data=1
(00:04:13.011) Thermostat -> me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27
(00:04:13.605) Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
(00:04:14.262) Thermostat -> me, type 0x51, telegram: 10 0B 51 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=77) #data=27
(00:04:14.855) Thermostat -> me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
(00:04:15.512) Thermostat -> me, type 0x5B, telegram: 10 0B 5B 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=05) #data=27
(00:04:16.078) Thermostat -> me, type 0x06, telegram: 10 0B 06 00 14 01 10 02 0B 14 03 00 (CRC=81) #data=8
** Warning, we missed the bus - Rx non-idle!
(00:04:22.097) Thermostat -> all, type 0x3E, telegram: 10 00 3E 00 04 02 26 00 C6 00 00 00 00 00 00 04 00 11 05 00 (CRC=77) #data=16
(00:04:25.223) Thermostat -> all, type 0x48, telegram: 10 00 48 00 00 00 1B 00 8A 00 00 00 00 00 00 00 00 11 05 00 (CRC=A0) #data=16
(00:04:26.342) Thermostat -> all, type 0xA3, telegram: 10 00 A3 00 6E 11 00 00 C8 00 CE 00 CC 7D 00 7D 00 (CRC=44) #data=13

I don’t have the issue with version 1.9.3 or older.
I would assume it’s introduced with the “master_thermostat” setting 

I have seen a few crashes with 1.9.5b8 when entering logging mode or asking for info

[SYSTEM] Last reset info: Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x40003b4d epc2:0x00000000 epc3:0x00000000 excvaddr:0x02ce0182 depc:0x00000000

proddy, 2020-01-02:
I see the problem now. It's this following telegram from when HC3 is queried:

```
Thermostat -> me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
```

because when a 0x52 comes in I just assume there is a HC3 active. Which now is not always the case. I've changed the code to look at the first data byte and if it's zero I will ignore. I'm basing this off your 0x3E telegram (HC1) who's first byte is 0x04 meaning in automatic mode. Hope that helps, thanks for working with me on fixing this.

lmdecroos, 2020-01-02:
you are welcome

can you guide my why my compiled dev branches are not working? They are not booting up…  
Compiling ends successful but the size of the program is more than 100kbyte larger than a release version

Linking .pio/build/release/EMS-ESP-1_9_4-d1_mini.elf
Building .pio/build/release/EMS-ESP-1_9_4-d1_mini.bin
Retrieving maximum program size .pio/build/release/EMS-ESP-1_9_4-d1_mini.elf
Checking size .pio/build/release/EMS-ESP-1_9_4-d1_mini.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=======   ]  66.8% (used 54740 bytes from 81920 bytes)
PROGRAM: [======    ]  58.6% (used 611612 bytes from 1044464 bytes)
Creating BIN file ".pio/build/release/EMS-ESP-1_9_4-d1_mini.bin" using ".pio/build/release/EMS-ESP-1_9_4-d1_mini.elf"

Linking .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf
Retrieving maximum program size .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf
Checking size .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [=======   ]  68.5% (used 56120 bytes from 81920 bytes)
PROGRAM: [=======   ]  69.1% (used 721780 bytes from 1044464 bytes)
Building .pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.bin
Creating BIN file ".pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.bin" using ".pio/build/debug/EMS-ESP-1_9_5b9-d1_mini.elf" 

The DATA size looks ok, but the PROGRAM size is huge

I tested it on two different Debian PC’s and both are ending with the same result. A compiled release version works ok, a compiled dev version doesn’t work
I’ m aways starting with a clean cloned git  folder

proddy, 2020-01-02:
sure. I can build on my linux distro and compare sizes. But not to pollute
this issue can you create a new GitHub issue?

lmdecroos, 2020-01-02:
version 1.9.5b9 is only listing my HC2 but with the values of HC1

Thermostat stats:
  Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
  Thermostat time is 22:14:24 02/01/2020
  Heating Circuit 2
   Current room temperature: 20.7 C
   Setpoint room temperature: 15.0 C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to night
   Day Mode is set to night

Mixing module stats:
  Switch temperature: 19.1 C
  Mixing Circuit 1
   Current flow temperature: ? C
   Setpoint flow temperature: 0 C
   Current pump modulation: 0 %

Timestamp:22:18:07 Topic:thermostat_data Payload:{"hc2":{"seltemp":19,"currtemp":20.7,"daytemp":18,"nighttemp":13.5,"holidayttemp":17,"heatingtype":1,"circuitcalctemp":5,"mode":"night"}}
Timestamp:22:18:07 Topic:mixing_data Payload:{"hc1":{"flowTemp":3276.8,"setflowTemp":0,"pumpMod":0}}

proddy, 2020-01-02:
oh not good.

lmdecroos, 2020-01-03:
any progress / roll-back on this ?
with 1.9.5b11 i’m still missing my HC2

proddy, 2020-01-03:
I can't see what is causing this. I'll need a dump of the data. Or you could look at the code and try and figure it out.

lmdecroos, 2020-01-03:
i would need the diff or changeling between  1.9.5b8 and  1.9.5b9

lmdecroos, 2020-01-05:
with b13 and b14 I have back my 2 HC’s but the current room temperature and set point room temperature are not shown for HC2
the data is available in telegram 0x48

Thermostat stats:
  Thermostat: RC35 (DeviceID:0x10 ProductID:86 Version:21.08)
  Thermostat time is 12:07:13 05/01/2020
  Heating Circuit 1
   Current room temperature: 19.3 C
   Setpoint room temperature: 19.0 C
   Day temperature: 19.0 C
   Night temperature: 15.0 C
   Vacation temperature: 17.0 C
   Mode is set to auto
   Day Mode is set to day
  Heating Circuit 2
   Current room temperature: ? C
   Setpoint room temperature: ? C
   Program is set to Summer mode
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to night

Mixing module stats:
  Switch temperature: 28.6 C
  Mixing Circuit 1
   Current flow temperature: ? C
   Setpoint flow temperature: 0 C
   Current pump modulation: 0 %

   (12:08:04) Thermostat -> Me, type 0x3E, telegram: 10 0B 3E 00 04 02 26 00 C1 00 00 00 00 00 00 00 64 11 1F 00 (CRC=2A) #data=16
   (12:08:05) Thermostat -> Me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=BE) #data=27
   (12:08:05) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 00 00 1B 00 89 00 00 00 00 00 00 00 00 11 05 00 (CRC=89) #data=16
   (12:08:06) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27
   (12:08:07) Thermostat -> Me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=B6) #data=16
   (12:08:07) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 01 (CRC=20) #data=2
   (12:08:07) Thermostat -> Me, type 0x51, telegram: 10 0B 51 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=77) #data=27
   (12:08:08) Thermostat -> Me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 11 00 00 (CRC=C0) #data=16
   (12:08:09) Thermostat -> Me, type 0x5B, telegram: 10 0B 5B 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=05) #data=27
   (12:08:10) Thermostat -> Me, type 0x06, telegram: 10 0B 06 00 14 01 0C 05 08 07 06 00 (CRC=04) #data=8
   (12:08:14) Thermostat -> Mixing Module, type 0xAC, telegram: 10 21 AC 00 05 00 01 (CRC=0F) #data=3
   (12:08:16) Thermostat -> All, type 0x06, telegram: 10 00 06 00 14 01 0C 05 08 0D 06 00 (CRC=8B) #data=8
   (12:08:24) Thermostat -> UBAMaster, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
   (12:08:24) UBAMaster -> Thermostat, type 0x14, telegram: 08 10 14 00 35 24 81 (CRC=44) #data=3
   (12:08:24) Thermostat -> All, type 0xA2, telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=99) #data=14
   (12:08:28) Thermostat -> All, type 0x3E, telegram: 10 00 3E 00 04 02 26 00 C1 00 00 00 00 00 00 00 64 11 1F 00 (CRC=42) #data=16
   (12:08:30) Thermostat -> All, type 0x48, telegram: 10 00 48 00 00 00 1B 00 89 00 00 00 00 00 00 00 00 11 05 00 (CRC=E1)

lmdecroos, 2020-01-05:
update

the current room temperature and setpoint for my HC 2 becomes available after switching from mode day / night / auto 

  Heating Circuit 2
   Current room temperature: 17.6 C
   Setpoint room temperature: 18.0 C
   Day temperature: 18.0 C
   Night temperature: 13.5 C
   Vacation temperature: 17.0 C
   Mode is set to night
   Day Mode is set to day

so it seems that it’s need a trigger to start collecting these values

until now I have only seen the 2 active HC’s with b14,  the bug that it's showing 4 HC seems to be fixed

proddy, 2020-01-05:
Looks like the logic needs to change. See line 1165 in ems.cpp (https://github.com/proddy/EMS-ESP/blob/daf24af323b3fec595614d43ad29ebe9f176d048/src/ems.cpp#L1195) . Perhaps this is wrong. What you could do is add some myDebug(...) statements in to trace what is happening?

lmdecroos, 2020-01-05:
Please find some logging details

HC1 -> Auto / Day
Mode is set to auto
Day Mode is set to day
0B 90 3E 00 20 (CRC=8C)
10 0B 3E 00 04 02 26 00 BD 00 00 00 00 00 00 00 64 11 1C 00 (CRC=F5) #data=16

HC1 -> auto / night
Mode is set to auto
Day Mode is set to night
0B 90 3E 00 20 
10 0B 3E 00 04 00 1E 00 BE 00 00 00 00 00 00 00 00 00 05 00 

HC1  -> Manual set to Day
Mode is  set to night
Day Mode is set to day
0B 90 3E 00 20 (CRC=8C)
10 0B 3E 00 00 00 1E 00 BE 00 00 00 00 00 00 00 00 11 1C 00 (CRC=B2) #data=16

HC1 - Manual set to night
Mode is set to night
Day Mode is set to night
0B 90 3E 00 20 (CRC=8C)
10 0B 3E 00 00 00 1E 00 BF 00 00 00 00 00 00 00 00 11 05 00 (CRC=48) #data=16

HC2   -> Auto Mode / night
Mode is set to auto
Day Mode is set to night
0B 90 48 00 20 (CRC=4D)
10 0B 48 00 04 00 1B 00 B0 00 00 00 00 00 00 0C 00 00 05 00 (CRC=17) #data=16

HC2 -> Manual Mode set to Night
Mode is set to night
Day Mode is set to night
0B 90 48 00 20 (CRC=4D)
10 0B 48 00 00 00 1B 00 B9 00 00 00 00 00 00 0C 00 00 05 00 (CRC=79) #data=16

HC2 -> Manual Mode set to Day
Mode is set to night
Day Mode is set to day
0B 90 48 00 20 (CRC=4D)
10 0B 48 00 80 02 24 00 BC 00 00 00 00 00 00 0A 00 00 05 00 (CRC=89) #data=16

The below triggers a change in listing from 2 to 4 HC's in the telnet console

Manual action on RC35 push on Day button ( from auto)
(22:18:44) Thermostat -> All, type 0x3D, telegram: 10 00 3D 07 01 (CRC=E2) #data=1
(22:18:44) Thermostat -> All, type 0x51, telegram: 10 00 51 07 01 (CRC=4B) #data=1
(22:18:45) Thermostat -> All, type 0x5B, telegram: 10 00 5B 07 01 (CRC=63) #data=1

After a reboot of the gateway Manual action on RC35 push on Auto button ( from Day)
(22:21:24) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 03 03 02 00 00 00 00 00 00 00 00 00 (CRC=9B) #data=13
(22:21:29) Thermostat -> All, type 0x3D, telegram: 10 00 3D 07 02 (CRC=E1) #data=1
(22:21:29) Thermostat -> All, type 0x51, telegram: 10 00 51 07 02 (CRC=48) #data=1
(22:21:30) Switching Module -> UBAMaster, type 0x1E, telegram: 11 08 1E 00 00 F1 (CRC=93) #data=2
(22:21:30) Thermostat -> All, type 0x5B, telegram: 10 00 5B 07 02 (CRC=60) #data=1
(22:21:31) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 02 23 00 00 00 00 00 60 01 72 02 62 02 32 00 00 0C 30 48 00 CB 00 00 00 (CRC=9F) #data=25
(22:21:31) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0F 02 62 02 62 20 00 00 03 00 00 9F 28 00 16 20 00 80 00 (CRC=D4) #data=19

changing from mode on my RC20RF has no effect on the amount of listed HC’s , it stays always on 2

proddy, 2020-01-06:
@lmdecroos made some changes based on your tests - in 1.9.5b15. Hopefully now the jump from having 2 to 4 HCs is fixed. Also data should be coming in quicker as I removed the check for only broadcasts on RC35 status messages. Let me know...

lmdecroos, 2020-01-09:
just build  1.9.5b16 and will test it during the next days

lmdecroos, 2020-01-10:
The issue that 4 HC’s are discovered ( after changing the mode for HC1) is fixed 

But the behaviour regarding HC2 is still not changed

I have to change the mode (Auto / day/ night) for my HC2 before the “current room temperature” and “ Setpoint room temperature” are recognised for the first time, thereafter they are  updating correctly

Heating Circuit 2
 Current room temperature: ? C
 Setpoint room temperature: ? C
 Program is set to Summer mode
 Day temperature: 18.0 C
 Night temperature: 13.5 C
 Vacation temperature: 17.0 C
 Mode is set to night

 Heating Circuit 2
 Current room temperature: 14.7 C
 Setpoint room temperature: 13.5 C
 Day temperature: 18.0 C
 Night temperature: 13.5 C
 Vacation temperature: 17.0 C
 Mode is set to auto
 Day Mode is set to night

proddy, 2020-01-10:
EMS-ESP should regularly fetch data from the thermostats, also at startup. For an RC35 it will send requests to telegrams 0x3D and 0x3E for HC1 and 0x47 and 0x48 for HC2. So not sure why you're not seeing any data. Can you manually test this with 
```
thermostat read 47
thermostat read 48
```
and see what comes back and whether the data values are refreshed.

lmdecroos, 2020-01-10:
Hi,

The issue has been introduced with the “set master_thermostat” command.

I assume that there is a logical issue introduced in te code : the “set master_thermostat” forces the software to read-out the values of my master thermostat RC35 ( device 86). 
As this is the master thermostat, the software knows that their are 2 HC’s and is able to extract all the programmed values from my RC35 for both HC's

The software seems not taken in account that he has to interpret the values of HC2 from mu RC20RF.

The software is  only  triggered when i’m changing the mode on my RC20RF between auto/ day/ night. this is the trigger to interpret the values from my RC20RF.
once this trigger has been executed the values are updates every time there is an update from the boiler or fetch from the software

This evening I had an uptime of more than 12 hours and the realtime values of my HC2 where still not captured :

[APP] Uptime: 0 days 12 hours 25 minutes 33 seconds

 Heating Circuit 2
 Current room temperature: ? C
 Setpoint room temperature: ? C
 Program is set to Summer mode
 Day temperature: 18.0 C
 Night temperature: 13.5 C
 Vacation temperature: 17.0 C
 Mode is set to night

 thermostat read 47
 (23:36:25) Thermostat -> All, type 0x3E, telegram: 10 00 3E 00 04 00 1E 00 BB 00 00 00 00 00 00 00 00 00 05 00 (CRC=DF) #data=16
 (23:36:26) Thermostat -> All, type 0x48, telegram: 10 00 48 00 00 00 1B 00 91 00 00 00 00 00 00 00 00 00 05 00 (CRC=9F) #data=16
 (23:36:27) Thermostat -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 00 00 00 00 (CRC=91) #data=4

 Requesting type RC35Set_HC2(0x47) from dest 0x10
 (23:36:28) Thermostat -> Switching Module, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
 (23:36:28) Sending read of type 0x47 to 0x10, telegram: 0B 90 47 00 20 (CRC=71)
 (23:36:28) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27
 (23:36:28) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 0B 03 03 02 00 00 00 00 00 00 00 00 00 (CRC=B7) #data=13
 Fetching data from EMS devices
 Requesting type RC35StatusMessage_HC1(0x3E) from dest 0x10
 Requesting type RC35Set_HC1(0x3D) from dest 0x10
 Requesting type RC35StatusMessage_HC2(0x48) from dest 0x10
 Requesting type RC35Set_HC2(0x47) from dest 0x10
 Requesting type RC35StatusMessage_HC3(0x52) from dest 0x10
 Requesting type RC35Set_HC2(0x51) from dest 0x10
 Requesting type RC35StatusMessage_HC4(0x5C) from dest 0x10
 Requesting type RC35Set_HC4(0x5B) from dest 0x10
 Requesting type RCTime(0x06) from dest 0x10
 Requesting type UBAMonitorFast(0x18) from dest 0x08
 Requesting type UBAMonitorSlow(0x19) from dest 0x08
 Requesting type UBAParameterWW(0x33) from dest 0x08
 Requesting type UBAParametersMessage(0x16) from dest 0x08
 Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
 (23:36:29) Sending read of type 0x3E to 0x10, telegram: 0B 90 3E 00 20 (CRC=8C)
 (23:36:29) Thermostat -> Me, type 0x3E, telegram: 10 0B 3E 00 04 00 1E 00 BB 00 00 00 00 00 00 00 00 00 05 00 (CRC=B7) #data=16
 (23:36:30) Sending read of type 0x3D to 0x10, telegram: 0B 90 3D 00 20 (CRC=80)
 (23:36:30) Thermostat -> Me, type 0x3D, telegram: 10 0B 3D 00 01 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 02 (CRC=BE) #data=27
 (23:36:30) Sending read of type 0x48 to 0x10, telegram: 0B 90 48 00 20 (CRC=4D)
 (23:36:30) Thermostat -> Me, type 0x48, telegram: 10 0B 48 00 00 00 1B 00 91 00 00 00 00 00 00 00 00 00 05 00 (CRC=F7) #data=16
 (23:36:31) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 3F 00 00 00 00 00 60 00 E1 02 07 01 4C 00 00 0C 30 48 00 CB 00 00 00 (CRC=60) #data=25
 (23:36:31) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0F 02 07 02 07 20 00 00 03 00 00 9F 60 00 16 29 00 80 00 (CRC=7B) #data=19
 (23:36:31) Sending read of type 0x47 to 0x10, telegram: 0B 90 47 00 20 (CRC=71)
 (23:36:31) Thermostat -> Me, type 0x47, telegram: 10 0B 47 00 01 1B 24 22 00 28 00 00 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 02 01 (CRC=BA) #data=27
 (23:36:32) Sending read of type 0x52 to 0x10, telegram: 0B 90 52 00 20 (CRC=25)
 (23:36:32) Thermostat -> Me, type 0x52, telegram: 10 0B 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=F2) #data=16
 (23:36:32) Sending read of type 0x51 to 0x10, telegram: 0B 90 51 00 20 (CRC=29)
 (23:36:33) Thermostat -> Me, type 0x51, telegram: 10 0B 51 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=77) #data=27
 (23:36:33) Sending read of type 0x5C to 0x10, telegram: 0B 90 5C 00 20 (CRC=1D)
 (23:36:33) Thermostat -> Me, type 0x5C, telegram: 10 0B 5C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=84) #data=16
 (23:36:33) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 3F 00 00 00 00 00 60 00 E1 02 07 01 4D 00 00 0C 30 48 00 CB 00 00 00 (CRC=04) #data=25
 (23:36:34) Sending read of type 0x5B to 0x10, telegram: 0B 90 5B 00 20 (CRC=01)
 (23:36:34) Thermostat -> Me, type 0x5B, telegram: 10 0B 5B 00 00 1E 26 22 00 28 00 02 00 05 05 2D 01 01 04 4B 05 4B 01 00 3C FF 11 05 05 03 02 (CRC=05) #data=27
 (23:36:34) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 3F 00 00 00 00 00 60 00 E1 02 07 01 4C 00 00 0C 30 48 00 CB 00 00 00 (CRC=60) #data=25
[... 15 lines cut ...]
 (23:36:40) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 3F 00 00 00 00 00 60 00 E1 02 06 01 4C 00 00 0C 30 48 00 CB 00 00 00 (CRC=E9) #data=25
[... cut ...]
 (23:37:03) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 6E 00 00 00 BB 00 C1 00 C1 7D 00 7D 00 (CRC=24) #data=13
 (23:37:04) Switching Module -> UBAMaster, type 0x1E, telegram: 11 08 1E 00 01 13 (CRC=73) #data=2
 (23:37:10) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 07 01 3F 00 00 00 00 00 60 00 E3 02 07 01 4C 00 00 0C 30 48 00 CB 00 00 00 (CRC=4C) #data=25
 (23:37:11) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 0F 02 07 02 07 20 00 00 03 00 00 9F 60 00 16 29 00 80 00 (CRC=7B) #data=19

proddy, 2020-01-31:
@lmdecroos I'm looking into this issue and still don't understand the problem. As I mentioned earlier can you try
```
thermostat read 47
thermostat read 48
```

and tell me if the HC1 and HC2 values are refreshed correctly?

proddy, 2020-02-02:
HC1 and HC2 should work now. I'm refactoring all the code for v2 into C++ classes and have all ems devices inherit a base class which will enable EMS-ESP to support multiple devices, including thermostats and their heating circuits.

proddy, 2020-03-08:
multiple HCs support in 1.9.5

closed 2020-03-08 by proddy