emsesp/EMS-ESP #239: Controlling Bosch EasyControl CT200

DasBootU96, 2019-11-13:
**Question**
I recently got a new Nefit TrendLine II HRC30 which came with a Bosch EasyControl CT200. Curious about its operation, bought and connected a bbqkees MS Bus Wi-Fi Gateway.

Everything mostly works as expected. However, the heating circuit shows 0.0 C

Although I am aware that others have reported the same issue and I know you are quite busy, can you say anything about whether or not this will be addressed? 

Also, I am aware that CT200 is currently 'read-only', but how realistic is it to ever have write, e.g. setting a setpoint, access?

Lets assume it will be a while before CT200 gets write support, in the interim, would it be possible to add a second cheap thermostat that does support writing? 

For example, could I add a [cheap second hand Nefit Moduline 100](https://www.marktplaats.nl/q/nefit+moduline+100/), connect it to the same bus as the CT200 (like somewhere out of sight near the boiler) which should allow me to use home automation (Home Assistant) to set temperature using the Moduline 100 device while at the same time also allow control using CT200 in the livingroom and optionally also via the EasyControl app. Is something like that possible?

thanks for all your efforts!

**Screenshots**
n/a

**Device information**
```
info

EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Mixing Module is disabled, Shower Timer is disable
d, Shower Alert is disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=1488, # CRC errors=3
  Tx: Last poll=3.665 seconds ago, # successful write requests=2

Boiler stats:
  Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Worcester Greenstar Si (ProductID:123 Version:07.00)
  Hot tap water: off
  Central heating: active
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 38.8 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 157 times
  Warm Water active time: 0 days 1 hours 1 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 46 C
  Current flow temperature: 45.8 C
  Return temperature: 38.4 C
  Gas: on
  Boiler pump: on
  Fan: on
  Ignition: off
  Circulation pump: off
  Burner selected max power: 75 %
  Burner current power: 30 %
  Flame current: 24.0 uA
  System pressure: 2.0 bar
  System service code: -H (200)
  Heating temperature setting on the boiler: 75 C
  Boiler circuit pump modulation max power: 70 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 47.1 C
  Pump modulation: 36 %
  Burner # starts: 723 times
  Total burner operating time: 1 days 8 hours 54 minutes
  Total heat operating time: 1 days 7 hours 53 minutes
  Total UBA working time: 5 days 5 hours 51 minutes

Thermostat stats:
  Thermostat: Bosch EasyControl CT200 (ProductID:203 Version:02.08)
  Heating Circuit 1
   Current room temperature: 0.0 C
   Setpoint room temperature: 0.0 C
   Mode is unknown 

autodetect deep

System Logging set to None
Starting a deep EMS device scan. This can take up to 2 minutes. Please wait...
[EMS] Detected new EMS Device with ID 0x08
[EMS] Detected new EMS Device with ID 0x09
[EMS] Detected new EMS Device with ID 0x18
EMS Device recognized as Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Wor (DeviceID:0x08 ProductID:123 Version:
07.00)
EMS Device recognized as Other: BC10 Base Controller (DeviceID:0x09 ProductID:190 Version:01.04)
EMS Device recognized as Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Wor (DeviceID:0x08 ProductID:123 Version:
07.00) **already active**
EMS Device recognized as Other: BC10 Base Controller (DeviceID:0x09 ProductID:190 Version:01.04) **already active**
EMS Device recognized as Thermostat: Bosch EasyControl CT200 (DeviceID:0x18 ProductID:203 Version:02.08)
EMS Device recognized as Other: BC10 Base Controller (DeviceID:0x09 ProductID:190 Version:01.04) **already active**
Finished the deep EMS device scan.

These device IDs are on the EMS Bus: 0x08 0x09 0x18
and 3 were recognized by EMS-ESP as:
 Buderus GBx72/Nefit Trendline/Junkers Cerapur/Wor (DeviceID:0x08 ProductID:123 Version:07.00)
 BC10 Base Controller (DeviceID:0x09 ProductID:190 Version:01.04)
 Bosch EasyControl CT200 (DeviceID:0x18 ProductID:203 Version:02.08)
```

**Additional context**
n/a

proddy, 2019-11-13:
Hi @DasBootU96, great that you're using the software.

The 0.0 degrees problem looks like a bug. These temperature values should be sent on a type 0x0A telegram. Could you do a `log t` and capture a few? if you don't see any data coming in try requesting it with 'thermostat read 0A'. `log v` will give you more info. It's been a while since the Easy code was touched so there may still be a few glitches around.

I'd love to support Easy but haven't figured out how to communicate to it. We can only listen to broadcasts but there must be way.

Also see if you can upgrade to 1.9.4 and test that.

DasBootU96, 2019-11-13:
Here are the logs you asked for: [0x0A_log_captures.zip](https://github.com/proddy/EMS-ESP/files/3843056/0x0A_log_captures.zip), both with a 'log t' and a 'log v' option set. I let it run for a few minutes and changed the temperature of the thermostat a degree up or down every few minutes while it was dumping the log.

I very much hope you find something useful and please let me know if I can be of more assistance.

I'm using 1.9.3 at the moment and I can only find a[ beta  1.9.4b12.](https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build). Do you mean that development build, not fully tested. To be used  at your own risk.? :)

proddy, 2019-11-13:
I see the 0x0A telegrams are empty (all values 0x00) so Bosch is hiding these details somewhere else. What you could try is set the temp on the thermostat (make a note of what it is) then change it by 0.5 degrees, wait 10 seconds and set it back while logging is enabled. 

the dev versions are the latest greatest and with all the fixes. I try to release a new version every month. The 1.9.4 is close and I need beta testers

DasBootU96, 2019-11-13:
Attached are the logs: [0x0A_log_tempchange_23_to_23.5_and_back_capture.zip](https://github.com/proddy/EMS-ESP/files/3843471/0x0A_log_tempchange_23_to_23.5_and_back_capture.zip)

procedure used:
* Start 'log v', initial thermostat temp was 21.0C
* Ran 'thermostat read 0A'
* Set manually thermostat temp to 23.0C
* After a bit more than 10 seconds ran 'thermostat read 0A'
* Set manually thermostat temp to 23.5C
* After a bit more than 10 seconds ran 'thermostat read 0A'
* Set manually thermostat temp back to 23.0C
* After a bit more than 10 seconds ran 'thermostat read 0A'
* stop log

After creating that log, updated to 1.9.4b12 and the webpage opened once and I could see the new version and all config settings were retained. Then rebooted via the webpage and after that the webfrontend kind of loads, but no password prompt.

Telnet works and I can see MQTT being updated, but heating circuit bit is no longer present. When running 'info', output is different and 'autodetect' returns nothing.

Unless you have ideas, I'll be rolling back tomorrow to 1.9.3.

```
info

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: disabled, Solar Module: disabled, Mixing Module: disabled
  Shower Timer: disabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=19, # CRC errors=0
  Tx: Last poll=3.266 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Buderus Logamax U122/Junkers Cerapur (DeviceID:0x18 ProductID:203 Version:02.08)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: off
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 60 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 46.1 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 162 times
  Warm Water active time: 0 days 1 hours 4 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 5 C
  Current flow temperature: 50.1 C
  Return temperature: 45.4 C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 2.0 bar
  System service code: 0H (203)
  Heating temperature setting on the boiler: 75 C
  Boiler circuit pump modulation max power: 70 %
  Boiler circuit pump modulation min power: 30 %
  Boiler temperature: 47.8 C
  Pump modulation: 0 %
  Burner # starts: 743 times
  Total burner operating time: 1 days 10 hours 5 minutes
  Total heat operating time: 1 days 9 hours 1 minutes
  Total UBA working time: 5 days 9 hours 40 minutes

autodetect deep
               Unknown command or wrong number of arguments. Use ? for help.
autodetect

Started scanning the EMS bus for known devices

proddy, 2019-11-13:
look at the telegrams and see if you can spot a pattern.

telegram type 0x0A will always be zero.

 `autodetect deep`  has been removed.

 I see your thermostat wasn't detected (see line Thermostat: disabled at the top of info) Best to power on/off the ESP and try again. 

web behaviour is normal. trying to run a web server on a 4MB microcontroller has its sacrifices.

DasBootU96, 2019-11-13:
If you mean a pattern in EasyStatusMessage(0x0A), it has indeed many 00s
```
Requesting type EasyStatusMessage(0x0A) from dest 0x18
(01:35:58.492) Thermostat -> Boiler, type 0x23, telegram: 18 08 23 00 2A (CRC=7F) #data=1
(01:35:59.404) Sending read of type 0x0A to 0x18, telegram: 0B 98 0A 00 20 (CRC=1C)
(01:35:59.534) Thermostat -> me, type 0x0A, telegram: 18 0B 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 (CRC=2F) #data=26
<--- EasyStatusMessage(0x0A)
```

I've powered off/on the ESP without success; the webpage does not fully load.

So, in regard to controlling Bosch EasyControl CT200, does this mean there is nothing to be done to allow for EMS-ESP to set a setpoint temperature?

proddy, 2019-11-14:
Did you use the 1.9.4 pre-built firmware or did you build it yourself?

For reading thermostat temperature values from the CT200 someone (you?) needs to figure out where the values are stored and we can implement it. Same for setting values, someone needs to figure out how to commuicate EMS to these newer Easy web thermostats. Keep in mind there is not a community of developers behind this. It's just me in my spare time and I only own an old RC20 thermostat. This is how GitHub and open-source work - everyone collaborates and it only costs you time.

DasBootU96, 2019-11-14:
Used the [prebuild 1.9.4](https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-dev-d1_mini.bin) version.  

Today I wiped the Wemos D1 and flashed again. Same behavior: It comes up in AP mode and I can access the web page and change the WIFI settings to client mode and reboot. 

Next I can access the web page and configure MQTT + NTP server nl.pool.ntp.org, then a reboot and I can no longer access the webpage; a password prompt never pops up.

I did notice different communication protocols were available, I left it at default: EMS I think it was.

I very much appreciate your work, I was just wondering how realistic my expectations should be :)

mellbo, 2019-11-14:
but do you know what IP after put in mode Client?

proddy, 2019-11-14:
@DasBootU96 it could be a bug. I'll try and re-create it tonight using a fresh Wemos D1. It may be to do with the upgrade from 1.9.3 to 1.9.4.

DasBootU96, 2019-11-15:
Just now tried the [latest development build](https://github.com/proddy/EMS-ESP/releases/download/travis-dev-build/EMS-ESP-dev-d1_mini.bin): as soon as I enable MQTT and reboot, the webpage is no longer accessible.

Using these telnet commands:
```
set mqtt_enabled off
restart
```
and the webpage is accessible again.

proddy, 2019-11-15:
Thanks for trying @DasBootU96 but I can't reproduce. I use the webUI to download the latest dev build, upload the firmware, power on/off the ESP and it works, with MQTT enabled. Which browser are you using and also when you say 'no longer accessible' just it load at all do you only see the menu?

proddy, 2019-11-16:
@DasBootU96 I was able to reproduce the error. Something's broken in the websocket code again. I'm looking into it.

proddy, 2019-12-30:
WebUI is being re-written using ReactJS and rest endpoints instead of web sockets, It'll be in the next major version of EMS-ESP

closed 2019-12-30 by proddy

cristianrinaldini, 2020-11-27:
Hi DasBootU96,

did you find any way to control Bosch EasyControl CT200?

Thanks

DasBootU96, 2020-11-27:
Hi, 

Nope, I've bought a second hand Nefit moduline 3000 which can be controlled with ems-esp using home assistant.

There seems to be no development on the HA forums for CT200, API stuff from Bosch seem to have stopped as well.

cristianrinaldini, 2020-11-27:
Hi,

thanks for the feedback.

But can you use this Nefit moduline 3000 in parallel with CT200?
I'm going to buy CT200+10 valves....

My target would be just to be able to switch th boiler off from Home
Assistant when the Alrm system is on (means for sure nobody at home)

Thanks
Ciao
Cristian

proddy, 2020-11-27:
@cristianrinaldini take a look at https://emsesp.github.io/docs/#/Supported-EMS-Devices?id=thermostats

if it helps

DasBootU96, 2020-11-27:
I've never tried to connect both thermostats, I assume that would introduce multiple heating circuits. Proddy knows more about this, but using a supported thermostat from the list together with a ct200 on the same bus and expect it to work as a single heating circuit is not possible.

cristianrinaldini, 2020-11-29:
Hi,

thanks for info.

My target is to be able to switch the boiler ( Junkers Cerapur Acu Smart )
on and off from Home Assistant.
I was thinking to drive one of the following contact with a relay:
1) Connection of an on / off for room chronothermostat (without voltage)
2) 1 wire of EMS bus
3) drain pump connector (should switch off the burner)
Which one do you think better fit and is fully safe for boilers and
thermostat?

Is it possible to drive a digital out on ESP with EMS-ESP (via MQTT)?
Would be very useful and avoid the need to use a sencons ESP just for that.

Thanks a lot for your help

proddy, 2020-11-30:
@cristianrinaldini yes you can send the cmd 'gpio' via MQTT to EMS-ESP to switch a gpio pin low/high. But be careful of power and use an isolated relay. This has been discussed before in https://github.com/proddy/EMS-ESP/issues/375