emsesp/EMS-ESP #240: Unidentified device :  unknown? (DeviceID:0x09 ProductID:209 Version:01.03)

glitter-ball, 2019-11-18:
I have a W-B Greenstar 24i System ErP and W-B Sense II control. With EMS-ESP connected, these are identified as:

Buderus Logamax plus/GB192/Bosch Condens GC9000 (DeviceID:0x08 ProductID:208 Version:01.01) RC300/RC310/Moduline 3000/Bosch CW400 (DeviceID:0x10 ProductID:158 Version:33.03)

There is nothing else on the EMS Bus. But, EMS-ESP finds this item too:

 unknown? (DeviceID:0x09 ProductID:209 Version:01.03)

I can only assume this is some other device within the boiler control PCB. How can I help track down what it is?

proddy, 2019-11-19:
It's most likely a controller unit. Most boilers have one embedded.I'll add it to the database along with the other correct device names. If you want to see what it's sending out trace the logs with telnet (`log v`)

glitter-ball, 2019-11-29:
Starting to get to grips with all this now, thanks to help here *8)

I think I’m getting all the usable data from my boiler now, so need to start looking at the thermostat - a Bosch Sense II. Although ‘log v’ is showing EMS+ data from the unit on ID 0x10, I don’t think any of it’s appearing on MQTT.

Am I missing something or is there a need for more work on this thermostat? I’m very happy to capture some packets and attempt to decode them to add to the knowledge here.

proddy, 2019-11-29:
If the thermostat is not recognized it won't send any data to MQTT. Type `info` and see if at the top it says Thermostat Enabled. Also do you see any thermostat values in the `info` at the bottom. Typing `mqttlog` will show you what data is being sent via MQTT. Start there if you can. I see more people using the same boiler/thermostat configuration so would like to get this working.

glitter-ball, 2019-11-29:
Thanks for the reply. Here's the info dump...
info

EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is disabled, Thermostat is disabled, Solar Module is disabled, Mixing Module is disabled, Shower Timer is disabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected, protocol: Junkers HT3
  Rx: # successful read requests=0, # CRC errors=0
  Tx: no signal

Boiler stats:
  Boiler: <not enabled>
  Hot tap water: off
  Central heating: off
  Warm Water activated: ?
  Warm Water circulation pump available: ?
  Warm Water selected temperature: ? C
  Warm Water desired temperature: ? C
  Warm Water current temperature: 40.1 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 2397 times
  Warm Water active time: 16 days 16 hours 40 minutes
  Warm Water 3-way valve: on
  Selected flow temperature: 5 C
  Current flow temperature: 35.2 C
  Return temperature: ? C
  Gas: off
  Boiler pump: off
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 0 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: ? bar
  System service code:  (203)
  Heating temperature setting on the boiler: ? C
  Boiler circuit pump modulation max power: ? %
  Boiler circuit pump modulation min power: ? %
  Outside temperature: 7.5 C
  Boiler temperature: ? C
  Pump modulation: 0 %
  Burner # starts: 8272 times
  Total burner operating time: 64 days 20 hours 54 minutes
  Total heat operating time: 48 days 4 hours 14 minutes

mqttlog

MQTT publish log:
  Topic:status Payload:online
  Topic:start Payload:start
  Topic:heartbeat Payload:{"version":"1.9.3","IP":"192.168.1.203","rssid":84,"load":1,"uptime":4275,"freemem":62,"MQTTdisconnects":41}
  Topic:shower_data Payload:{"timer":"0","alert":"0"}
  Topic:boiler_data Payload:{"selFlowTemp":5,"selBurnPow":0,"curBurnPow":0,"pumpMod":0,"outdoorTemp":7.7,"wWCurTmp":39.5,"wWCurFlow":0,"curFlowTemp":34,"burnGas":"off","flameCurr":0,"heatPmp":"off","fanWork":"off","ignWork":"off","wWCirc":"off","wWHeat":"on","wWStarts":2397,"wWWorkM":24040,"burnStarts":8272,"burnWorkMin":93414,"heatWorkMin":69374,"ServiceCode":"","ServiceCodeNumber":203}
  Topic:tapwater_active Payload:0
  Topic:heating_active Payload:0

MQTT subscriptions:
  Topic:home/ems-esp/restart
  Topic:home/ems-esp/start
  Topic:home/ems-esp/temp1
  Topic:home/ems-esp/mode1
  Topic:home/ems-esp/temp2
  Topic:home/ems-esp/mode2
  Topic:home/ems-esp/temp3
  Topic:home/ems-esp/mode3
  Topic:home/ems-esp/temp4
  Topic:home/ems-esp/mode4
  Topic:home/ems-esp/thermostat_cmd
  Topic:home/ems-esp/boiler_cmd
  Topic:home/ems-esp/boiler_cmd_wwactivated
  Topic:home/ems-esp/boiler_cmd_wwtemp
  Topic:home/ems-esp/generic_cmd
  Topic:home/ems-esp/shower_data

I've started looking at the telegrams from 0x10.  The Sense II seems to be sending three different types of 0x01A5 telegram. The first (length 25) ties up well with the published format on the wiki. It's then immediately followed by another 0x01A5 of length 15, which is currently a mystery. It then also sends short 0x01A5 messages (length 2 or 4). I'll have a go at trying to work out what these latter types have in them.

It's also sending 0x06 which are labelled as RCTime. I can find the date and time in this OK. I'll attach a few more Sense II packets to this post in case they're useful for now.
[Sense II packets.txt](https://github.com/proddy/EMS-ESP/files/3905420/Sense.II.packets.txt)

proddy, 2019-11-29:
it says "Thermostat is disabled" - so there is your answer why nothing is being sent. Is the thermostat  detected? What does `autodetect` do?

glitter-ball, 2019-11-29:
OK - developments...

My new RPi3 arrived today so I've now built a new HA instance on that. I screwed up my networking so ended up factory resetting the EMS-ESP gateway too.

At some point in all this, the thermostat stuff sprang into life...

```
EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Mixing Module is disabled, Shower Timer is disabled, Shower Alert is disabled

Thermostat stats:

  Thermostat: RC300/RC310/Moduline 3000/Bosch CW400 (ProductID:158 Version:33.03)
  Thermostat time is 19:43:23 29/11/2019
  Heating Circuit 1
   Current room temperature: 18.7 C
   Setpoint room temperature: 18.5 C
   Mode is set to auto
```
Which is all a bit weird but I'm not complaining *8)

It seems, then, that ProductID:158 also covers the W-B 'Sense II' stat, if you want the model string to get even longer! I am running version NF33.03 - I've seen newer ones with NF33.04.

I have weather comp and the stat calculates and displays target flow temp. So... it would be interesting to see if this is in a telegram somewhere - always assuming that the stat passes this to the boiler. Or... does it pass burner output power instead? More digging required.

proddy, 2019-11-29:
cool, I'll add the WB Sense II.

I guess when you factory reset the EMS is removed some old settings from the SPIFFS filesystem storage left over from previous 1.9.x builds. Maybe, dunno. but it works now so that's progress.

Those other values must be in the EMS telegrams somewhere. This [python script](https://github.com/norberts1/hometop_HT3/blob/master/HT3/sw/lib/ht_discode.py) is a good reference but hard to read.

logiclabs, 2019-12-02:
I have same unknown device "unknown? (Version:01.04 ProductID:209 DeviceID:0x09)" being discovered on a **Worcester Bosch Greenstar 30i ErP boiler** (GC: 47-406-62).

Also have undefined Boiler and Return temps

glitter-ball, 2019-12-02:
The unknown device is almost certainly the boiler’s control panel. If you change a setting on it, you’ll see telegrams from device 0x09. You can prove this by going into the boiler’s info menu and checking the HMI Version is 1.04, matching device 0x09.
If your boiler is like mine inside, you won’t have boiler or return temps. There’s only a flow sensor. Not sure what ‘boiler temp’ would measure anyway...
GB

proddy, 2019-12-30:
@glitter-ball is this still an issue?

glitter-ball, 2019-12-30:
Thanks - not now that I've worked out that device 0x09 is actually the boiler's HMI panel. Just wondering how the info dump can be made more informative than 'unknown'?

Is 0x09 always going to be the control panel? Or is the ProductID more useful - 209 appears to be the control panel in W-B Greenstar ErP boilers (and possibly others).

Are the IDs allocated in a defined way? 0x08 boiler controller, 0x09 boiler HMI panel, 0x10 thermostat etc?

proddy, 2019-12-30:
It's a bit of a mystery how the device_d (e.g. 0x09, 0x010) works. I've been slowly building up a library in [ems_devices.h](https://github.com/proddy/EMS-ESP/blob/dev/src/ems_devices.h) and used 0x09 as a 'Controller' type. I think they also call them Modems in the Bosch/Junkers world.

I'll add product ID 209 to the list so at least it won't show up as 'Unknown' anymore.

glitter-ball, 2019-12-31:
Thanks. I Wonder if it’s better described as “W-B ErP boiler control panel”, to distinguish it from the boiler control PCB?

Maybe there’s other boilers treating the controls in the boiler as a separate EMS device, albeit in the same assembly as the control PCB. 

I may take a closer look next time I have mine open to see how it’s arranged physically.

glitter-ball, 2019-12-31:
Took a few minutes today to look at the boiler control panel in a W-B Greenstar 24i ErP. It's connected to the boiler PCB by a ribbon cable with, perhaps, eight lines on it (should have counted...)

Slightly surprised by this. I'd expected the HMI panel to be treated as an EMS bus device, as it appears to behave as one (device ID 0x09), distinct from the boiler control PCB (device ID 0x08).
So, the extra connections on the ribbon cable are a mystery.

The control panel PCB also holds the blue front panel LED, which might account for at least one extra line (although it could be bus driven via the control panel). I might do some more digging to see, for example, what telegrams the control panel sends in 'chimney sweep' mode.

proddy, 2020-01-07:
think we can close this one now right?

glitter-ball, 2020-01-08:
Yep

closed 2020-01-08 by glitter-ball