emsesp/EMS-ESP #241: Web Interface fails to load with 1.9.4 beta

proddy, 2019-11-20:
The web interface is in the latest 1.9.4 dev branch is problematic. It fails to load sometimes. I'm looking into what has caused this.

proddy, 2019-11-22:
this I think is fixed in 1.9.4b16 but there is still a problem when hitting the browser reload (F5) because web sockets are not closed. I'll fix this in the web refactor.

FrankRenp, 2019-11-26:
At least with 1.9.4b10 reload into a new window (CTRL Reload) worked with Firefox. I had
an uptime of 12 days 23 hours 3 minutes 13 seconds (so the MQTT topic seems to be solved, thanks).

With Shift-Reload (just tested) I caused a restart some minutes ago.

proddy, 2019-12-01:
fixed in b20

proddy, 2019-12-08:
it's crawled back, darn it

glitter-ball, 2019-12-31:
I may be behind the times on this, but...

Have just upgraded from 1.9.3 to the production 1.9.4 release. When I attempted to log on to the web interface with Chrome, the Dashboard page was blank after password entry. All the settings were back at defaults, despite them all being correct when I checked them in a telnet session.

I've done a restart via telnet and, on the first retry with the web interface, it was still broken. When I tried again a couple of minutes later, all was OK and has remained so.

The only lasting odd thing I've noticed is that the info icons on the Time Settings web page don't bring up tip bubbles. Tip bubbles on other pages are OK.

proddy, 2020-01-26:
It's a known issue and will be addressed in the web re-write in v2.0

closed 2020-01-26 by proddy