emsesp/EMS-ESP #244: Blank webinterface

balk77, 2019-11-22:
**Question**
When I connect to the webinterface it remains blank, except for the menu on the left. I am unable to enter WiFi credentials etc. I had to enter those via telnet.
I am running 1.9.3 but had the same issue on 1.9.0.
I am connecting to the Wemos D1 Mini through an ip address because the hostname does not work. 
I am on OS X 10.13.6, Chrome and Firefox gave the same results

**Screenshots**
[image]

**Additional**
I have erased the esp `esptool.py -p /dev/ttyUSB1 erase_flash` but that did not help.

bbqkees, 2019-11-22:
Try accessing the web interface on a Windows or Android device.

proddy, 2019-11-22:
This is known issue. See https://github.com/proddy/EMS-ESP/wiki/Troubleshooting

I'm working on re-writing the web code..still a few weeks before its done.

balk77, 2019-11-22:
I had found that known issue indeed so I intentionally did not try it in Safari. 

I am getting the same blank page on Chrome/Edge @ Win10. Also blank on Chrome/Android. Firefox.Android gives another error.

[image]
I am very happy with all the good work you guys are doing for this project! I am happy to contribute by testing for instance.

proddy, 2019-11-22:
Which version of EMS-ESP are you using? I'll run some tests tonight and try and see what is causing this.

balk77, 2019-11-22:
I am running 1.9.3 but I had the same issue with 1.9.0. could it be a faulty Wemos? I will try later today

proddy, 2019-11-22:
Wait until I've published a new build of 1.9.4 today/tomorrow and try that.

balk77, 2019-11-22:
tried the 1.9.0 and 1.9.3 versions on a different Wemos D1 Mini. Under 1.9.0 I was able to change the wifi settings in the web interface. But, after reconnecting it was blank again. Let me know if I need to test 1.9.4.

proddy, 2019-11-22:
could you try the latest dev build? Take it from https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build (make sure its the build from today though)

balk77, 2019-11-23:
I have installed Version 1.9.4b17 on a spare Wemos. Makes no difference. Page is blank (tested on Android only, FF and chrome)

proddy, 2019-11-23:
I don't have any android devices. @bbqkees can you replicate this behaviour?

balk77, 2019-11-23:
Now also tested on Win10 (Chrome) and OS X (Chrome/FF). Same blank page.

bbqkees, 2019-11-23:
If all versions don't work for you on any device including release versions which work fine for almost everyone, its not the firmware itself that is the problem but likely wrong build settings or bad Wemos chips.
Which Wemos devices are you using exactly?

balk77, 2019-11-23:
I also tend to think in that direction :(
I have a Wemos D1 Mini 2.3.0 connected to the EMS board and tried with same result on another 2.3.0 board. I have just tried another board version 3.1.0, fresh from its package. No luck.

The interesting thing is that when I connect for the first time directly in AP mode, the interface works fine. I could select my network with the _scan_ function. After a restart, it does not work anymore. Next, if I set the device to AP mode via Telnet, the interface is blank and I cannot use it to connect to my network.

I have tried Chrome and FF on my Mac, on an incognito tab, in safe mode. I have forced refreshed the web page. All without success. 1.9.3 and 1.9.4b17

In all cases I have used the precompiled firmware. I will now try to compile 1.9.3 and load it on the 3.1.0 board.

bbqkees, 2019-11-26:
I have someone else with the same problem now. At first boot everything is fine but after setting Wi-Fi nothing works anymore.

jefft4, 2019-11-26:
That's me (thanks, Kees, I was just trying the dev version before posting!)
Similar issue here, with 1.9.3 and 1.9.4b17 :-(  Browsers are firefox and chrome on Windows, same problem with both.  I've cleared browser cache, re-flashed the D1, tried another D1, etc, no joy.

One interesting observation: the UI is ok while the ESP is in AP mode; problems start when I put it into CLIENT mode... 

Nothing showing on firewall or PC, so looks like the problems are at the ESP end somehow.  No clues on the serial output (I disabled EMS to get full serial reporting), says that it's connected to wifi and started the web server and telnet, that's all.

Jeff

proddy, 2019-11-26:
> I have someone else with the same problem now. At first boot everything is fine but after setting Wi-Fi nothing works anymore.

do you mean the WebUI doesn't work, but the other services like wifi, mqtt, telnet and of course the EMS reading/writing is still all operational?

@jefft4 if the WebUI fails in Wifi client Mode but seems to work in AP then we're getting somewhere and I'll try and reproduce that scenario. I'm in the middle of slowly re-writing all those core services anyway so not putting too much effort in making the web work for everyone. I'll fix what I can.

jefft4, 2019-11-26:
Kees is referring to me, so details are as per my post.  
There's nothing coming from MQTT as far as I've seen - I was monitoring the broker for a good ten minutes when I had configured it, but I can't be sure that it was configured at all as I set up wifi at the same time.  
When it was connected to the boiler, the LED went steady after 2-3 seconds, which I guess is a positive sign.  I've not put the interface back onto the boiler since removing it to diagnose the web UI problems, so I'm not sure whether it's talking properly to EMS.  I can re-connect it and look at the EMS details (with it in AP mode) probably tomorrow evening, if you think that might be relevant.

Let me know if I can get you any more info to help, or try out new code for you... happy to load debug versions and post the logs etc.

proddy, 2019-11-27:
@jefft4 ok clear. there are known issues with the WebUI so I would just not use that for now and stick to telnet for setting up the configuration.

jefft4, 2019-11-27:
Ok, will give that a try this evening, see if I can get it online and sending MQTT.

Out of curiosity... If I open up the source code, could I easily disable the web authentication and would that avoid the known issue with websockets auth?  I don't desperately need security on something that's inside the building, connected to private wi-fi.  Security on ESP8266 is weak anyway as it can't handle encryption.

J.

https://github.com/proddy/EMS-ESP/issues/244#issuecomment-559101717

proddy, 2019-11-27:
good point. I did try that and also replacing the web sockets with REST end points but it didn't help. I think perhaps the web content is just too large (it's 500kb stored in flash memory) and the sdk can't read it fast enough. Still investigating though. My new version is in reactjs and around 200kb and if I switch to preactjs it'll be 100kb.

jefft4, 2019-11-27:
Mmm... 500k probably is quite a lot to read from the flash memory. The flash access is serial, and possibly a blocking I/O operation too, could simply be locking up the processor for too long.

It's definitely one of the prettier UIs I've seen run on an ESP!  Reducing the size sounds like a good move.  If that doesn't solve it, maybe need to look at loading the UI in several, smaller pieces?

https://github.com/proddy/EMS-ESP/issues/244#issuecomment-559116020

bbqkees, 2019-11-27:
The last few weeks I have been sending out all Gateways with 1.9.3. I only got one issue like this.
Perhaps the clone Wemos needs more specific build settings.

proddy, 2019-11-28:
The web is actually served in chunks which are in the src/webh directory. The sizes are not to big after checking:
 glyphicons is 23k 
 index.html is 2.4k
 myesp.html is 4.3k
 myesp.js 5.6k
 required.css is 20k
 required.js is 45k

I suspect the problem is somewhere else. I'm still looking

bbqkees, 2019-11-28:
I checked the web interface on my current testbed with a Lolin V3.1.1 and firmware 1.9.3 a few times yesterday and it crashed once, showing only the menu. After it rebooted itself it was fine again.

proddy, 2019-11-28:
1.9.4b17 uses older udp/tcp and arduino core libraries which seem to have fixed the web reload (F5) blank screen issue on my test environment.

closed 2019-12-01 by proddy

balk77, 2020-01-01:
Hi, I have just installed 1.9.4 but the issue is still there. Still with FF and Chrome on Mac OSX and Android. Do you have any clue?

bbqkees, 2020-01-01:
Have you tried it with a genuine Lolin Wemos V3?
Some clones have the same issue with particular firmware versions.

proddy, 2020-01-01:
refresh bug is still there, on all browsers and I'm not going to fix it. I'm going to throw away the web interface in the next major version as its too problematic.