emsesp/EMS-ESP #245: Cannot log in using Firefox 70.0.1 (64-bit)

glitter-ball, 2019-11-24:
I set up my EMS-ESP module using a laptop with Chrome, with no issues. I've now tried to log into it from another machine running Firefox 70.0.1 and have run into problems.

The password prompt appears and dots appear as I type the password. When I hover over the login button it highlights and when I click it the password text box goes from having a blue border to a grey one. Then, absolutely nothing happens.

No issues telnetting in with Putty. I'm running version 1.9.3

balk77, 2019-11-25:
[Duplicate of 244](244)
Are you using Windows or something else?

proddy, 2019-11-25:
Tried on Win10 64bit with Firefox 70.0.1 and it works, but with version 1.9.4

closed 2019-11-25 by proddy