emsesp/EMS-ESP #246: EMS-ESP / Home Assistant setup

glitter-ball, 2019-11-24:
Is there a more detailed guide anywhere to getting EMS-ESP up and running with Home Assistant? I've installed HA and realised that I need to use MQTT as the 'glue' between EMS-ESP and HA and that both need a broker.  Thereafter things get a bit sketchy! Thanks.

balk77, 2019-11-24:
Hier staan de voorbeelden hoe je het aan HA kan toevoegen. Voor zover ik weet kan je het **niet** als integration toevoegen. Handmatig dus.

https://github.com/proddy/EMS-ESP/tree/master/doc/home%20assistant

glitter-ball, 2019-11-24:
OK - I've made some progress. I've cut and pasted all the .yaml files across to HA and the config validates OK.

I've used the internal MQTT broker in HA and in developer tools can see the MQTT listener receiving stuff:

Message 7 received on ems-esp/boiler_data at 8:15 PM:

{
    "selFlowTemp": 29,
    "selBurnPow": 100,
    "curBurnPow": 0,
    "pumpMod": 96,
    "outdoorTemp": 10,
    "wWCurTmp": 60.7,
    "wWCurFlow": 0,
    "curFlowTemp": 27.6,
    "burnGas": "off",
    "flameCurr": 0,
    "heatPmp": "on",
    "fanWork": "off",
    "ignWork": "off",
    "wWCirc": "off",
    "wWHeat": "off",
    "wWStarts": 2378,
    "wWWorkM": 23837,
    "burnStarts": 8145,
    "burnWorkMin": 92370,
    "heatWorkMin": 68533,
    "ServiceCode": "",
    "ServiceCodeNumber": 204
}

But all the values in HA are still not available...
[image]

The entires in sensor.yaml seem to match those in the MQTT listen dump above, so I'm not sure why it's not rendering them?

proddy, 2019-11-25:
- Use the [1.9.4 beta](https://github.com/proddy/EMS-ESP/releases/tag/travis-dev-build) which has updated HA configs files in the folder @balk77 pointed out (dank je)
- Look at [the wiki page](https://github.com/proddy/EMS-ESP/wiki/home-assistant). If it needs improvement let me know.

glitter-ball, 2019-11-25:
With a fresh mind, found the issue quickly today. mqtt_base on the ems-esp was set to "". Once I set it to "home" some of the MQTT points lined up and data started appearing. Just need to tidy it up now.

As regards the wiki, it would be good for newbies (like me) if it explained a bit more along the lines of...

1) Get EMS-ESP working on the bus.
2) Set MQTT base="home" ready for HA.
3) Install HA and an MQTT broker (if not using the HA internal one).
4) Configure EMS-ESP's MQTT attributes to reach the HA broker (address, password etc.)
5) Update the HA .yaml files with the entries from doc/home assistant
6) In HA, use Developer Tools, MQTT, Listen To a Topic to check MQTT data.  Set the Topic to Subscribe to as "#" and then Start Listening. Check MQTT is received and that attribues match those in .yaml files.

balk77, 2019-11-25:
@glitter-ball do I see the old frontend of Home Assistant? Does it still work?

Happy to hear that this issue is now resolved 👍  Can you please close the issue?

proddy, 2019-11-25:
it's strange that the mqtt_base wasn't set to "home" as default. I'll look into that.

glitter-ball, 2019-11-27:
I installed HA on a PC to thrash the issues out. I've got another RPi3 on order and will migrate to that. I don't know if this explains the older version of HA? It's whatever version is available for  Python under Windows.

closed 2019-11-28 by proddy