emsesp/EMS-ESP #248: Thermostat RC30 / WebUI showing "undefined"

og9gjz567, 2019-11-27:
**Bug description**
Im using a RC30 (DeviceID:0x10 ProductID:67 Version:02.08) and only the 2nd Heating Circuit, the 1st one isn't in use (I don't know why ;) ). On the webUI Dashboard the Thermostat area is blue and the Table shows the following contents:

[image]

The info command shows this:

_Thermostat stats:
  Thermostat: RC30 (ProductID:67 Version:02.08)
  Thermostat time is 12:08:25 27/11/2019
  Heating Circuit 2
   Current room temperature: ? C
   Setpoint room temperature: 24.0 C
   Mode is unknown_

And I'm not able to set the tempeature via MQTT. I've used HA -> Developertools -> MQTT:
Topic: ESP-Heizung/temp2
Payload: 23.5
If I do Topic: ESP-Heizung/restart and Payload:1 my ESP restarts.

_mqttlog:
MQTT subscriptions:
  Topic:ESP-Heizung/restart
  Topic:ESP-Heizung/start
  [...]
  Topic:ESP-Heizung/temp2
  Topic:ESP-Heizung/mode2_

**Device information**
system:
_ESP8266 System stats:

 [APP] EMS-ESP version: 1.9.3
 [APP] MyESP version: 1.2.12
 [APP] Build timestamp: 2019-11-17 18:32:24
 [APP] Uptime: 0 days 0 hours 48 minutes 44 seconds
 [APP] System Load: 4%
 [WIFI] WiFi Hostname: ESP-Heizung
 [WIFI] WiFi IP: 192.168.188.42
 [WIFI] WiFi signal strength: 66%
 [WIFI] WiFi MAC: 2C:F4:32:4A:05:43
 [MQTT] is connected (heartbeat enabled)
 [NTP] Time in UTC is 11:18:57
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_D1_MINI
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.1(cfd48f3)
 [SYSTEM] CPU chip ID: 0x4A0543
 [SYSTEM] Core version: 2_5_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Restart from terminal
 [SYSTEM] Restart count: 0
 [SYSTEM] # TCP disconnects: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65536
 [FLASH] Flash chip ID: 0x1640EF
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 629168
 [MEM] Max OTA size: 2510848
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 26696 bytes initially | 12416 bytes used (46%) | 14280 bytes free (53%)_

info:
_EMS-ESP system stats:
  System logging set to None
  LED is on, Listen mode is off
  Boiler is enabled, Thermostat is enabled, Solar Module is disabled, Mixing Module is enabled, Shower Timer is enabled, Shower Alert is disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=399, # CRC errors=1
  Tx: Last poll=2.243 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (ProductID:64 Version:02.07)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 53 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 53.3 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 23507 times
  Warm Water active time: 252 days 5 hours 25 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 45 C
  Current flow temperature: 42.1 C
  Return temperature: 41.8 C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 70 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.5 bar
  System service code: 0Y (204)
  Heating temperature setting on the boiler: 90 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 50 %
  Outside temperature: 9.4 C
  Boiler temperature: 42.3 C
  Pump modulation: 98 %
  Burner # starts: 248214 times
  Total burner operating time: 2038 days 17 hours 50 minutes
  Total heat operating time: 1786 days 12 hours 25 minutes
  Total UBA working time: 5273 days 21 hours 20 minutes

Thermostat stats:
  Thermostat: RC30 (ProductID:67 Version:02.08)
  Thermostat time is 12:14:27 27/11/2019
  Heating Circuit 2
   Current room temperature: ? C
   Setpoint room temperature: 24.0 C
   Mode is unknown

Mixing module stats:
  Switch temperature: 5094.4 C

Shower stats:
  Shower is off_

**Additional context**
I've ordered the complete gateway last week. If necessary i can temporarily make the gateway public accessable

proddy, 2019-11-27:
Can you try again with 1.9.4

og9gjz567, 2019-11-27:
Yes I'll give it a try. Where can I download the "dev"-FW?

proddy, 2019-11-27:
https://github.com/proddy/EMS-ESP/releases

og9gjz567, 2019-11-27:
Thanks a lot! I've updated to version 1.9.4b17
The webUI still shows undefined, the telnet output shows:
**Thermostat stats:**
  _Thermostat: RC30 (DeviceID:0x10 ProductID:67 Version:02.08)
  Thermostat time is 20:58:05 27/11/2019
  Heating Circuit 2
   Current room temperature: ? C_

og9gjz567, 2019-11-27:
And devices shows this:
These device IDs are on the EMS Bus: 0x08 0x09 0x10 0x11 0x13 0x21
and 6 were recognized by EMS-ESP as:
 Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductID:64 Version:02.07)
 BC10/RFM20 Receiver (DeviceID:0x09 ProductID:68 Version:02.00)
 RC30 (DeviceID:0x10 ProductID:67 Version:02.08)
 WM10 Switch Module (DeviceID:0x11 ProductID:71 Version:02.00)
 unknown? (DeviceID:0x13 ProductID:82 Version:01.01)
 MM10 Mixer Module (DeviceID:0x21 ProductID:69 Version:02.01)

**You have a device is that is not yet known by EMS-ESP. Please report this as a GitHub issue so we can expand the EMS device library.**

proddy, 2019-11-27:
that's interesting. I've never tested reading data from heating circuit 2 on a RC30. Could you fire up telnet do a `log w 41` and capture a few telegrams with the type ID 0x41. This is where the data is contained.

og9gjz567, 2019-11-28:
**log w 41**

System Logging set to Watch mode
Fetching data from EMS devices
(08:09:09) 0B 90 41 00 20
(08:09:09) 10 0B 41 00
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(08:10:08) 0B 90 41 00 20
(08:10:08) 10 0B 41 00
Fetching data from EMS devices
(08:11:09) 0B 90 41 00 20
(08:11:09) 10 0B 41 00
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(08:12:08) 0B 90 41 00 20
(08:12:08) 10 0B 41 00

proddy, 2019-11-28:
ok thats what I thought. you're not getting any thermostat data in. can you do the same with `log t` which captures all the thermostat messages

og9gjz567, 2019-11-28:
**Here's my "log t":** 

```
System Logging set to Thermostat only
(12:29:56) Thermostat -> Switching Module, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
(12:30:00) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 07 11 01 (CRC=91) #data=3
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(12:30:10) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
(12:30:11) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)
(12:30:12) Thermostat -> Me, type 0x06, telegram: 10 0B 06 00 13 0B 0C 1C 19 33 03 00 (CRC=C6) #data=8
(12:30:15) Thermostat -> All, type 0x48, telegram: 10 00 48 00 04 02 30 7D 00 00 00 27 2F 37 00 00 64 11 29 (CRC=70) #data=15
(12:30:16) Thermostat -> UBAMaster, type 0x14, telegram: 10 88 14 00 03 (CRC=6E) #data=1
(12:30:17) UBAMaster -> Thermostat, type 0x14, telegram: 08 10 14 00 73 E7 56 (CRC=0D) #data=3
(12:30:21) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 11 (CRC=30) #data=2
(12:30:34) Thermostat -> Mixing Module, type 0xAC, telegram: 10 21 AC 00 29 64 01 (CRC=77) #data=3
(12:30:34) Thermostat -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 2E 64 00 00 (CRC=71) #data=4
(12:30:46) Thermostat -> All, type 0x06, telegram: 10 00 06 00 13 0B 0C 1C 1A 1A 03 00 (CRC=DD) #data=8
(12:30:56) Thermostat -> Switching Module, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
(12:31:00) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 07 11 01 (CRC=91) #data=3
Fetching data from EMS devices
(12:31:10) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
(12:31:10) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)
(12:31:11) Thermostat -> Me, type 0x06, telegram: 10 0B 06 00 13 0B 0C 1C 1A 32 03 00 (CRC=DA) #data=8
(12:31:15) Thermostat -> All, type 0x48, telegram: 10 00 48 00 04 02 30 7D 00 00 00 27 2F 37 00 00 64 11 29 (CRC=70) #data=15
(12:31:21) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 11 (CRC=30) #data=2
[NTP] Internet time: 11:31:29 UTC on 28/11. Local time: 12:31:29 CET
(12:31:34) Thermostat -> Mixing Module, type 0xAC, telegram: 10 21 AC 00 29 64 01 (CRC=77) #data=3
(12:31:34) Thermostat -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 2E 64 00 00 (CRC=71) #data=4
(12:31:47) Thermostat -> All, type 0x06, telegram: 10 00 06 00 13 0B 0C 1C 1B 1B 03 00 (CRC=D1) #data=8
(12:31:48) Thermostat -> All, type 0xA2, telegram: 10 00 A2 00 00 00 00 00 00 00 00 00 00 00 (CRC=18) #data=10
(12:31:57) Thermostat -> Switching Module, type 0x9D, telegram: 10 11 9D 00 00 (CRC=D7) #data=1
(12:32:01) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 07 11 01 (CRC=91) #data=3
Fetching data from EMS devices
Publishing boiler data via MQTT
Publishing hot water and heating states via MQTT
Publishing thermostat data via MQTT
(12:32:10) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
(12:32:10) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)
(12:32:11) Thermostat -> Me, type 0x06, telegram: 10 0B 06 00 13 0B 0C 1C 1B 32 03 00 (CRC=D2) #data=8
(12:32:14) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 00 1B (CRC=66) #data=1
(12:32:14) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 00 43 30 01 20 00 71 3B BC 00 00 00 00 (CRC=2B) #data=12
(12:32:14) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 0C 1B (CRC=7E) #data=1
(12:32:14) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 0C 43 30 01 20 00 71 3B BC 00 00 00 00 (CRC=11) #data=12
(12:32:14) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 18 1B (CRC=56) #data=1
(12:32:14) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 18 43 30 01 20 93 07 09 1D 09 00 00 00 (CRC=10) #data=12
(12:32:14) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 24 1B (CRC=2E) #data=1
(12:32:14) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 24 43 30 01 20 93 07 09 1D 07 00 00 00 (CRC=B2) #data=12
(12:32:15) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 30 1B (CRC=06) #data=1
(12:32:15) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 30 43 30 01 21 93 07 15 1A 2C 00 00 00 (CRC=7F) #data=12
(12:32:15) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 3C 1B (CRC=1E) #data=1
(12:32:15) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 3C 36 41 00 E3 93 07 0D 09 2D 00 00 00 (CRC=74) #data=12
(12:32:15) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 48 18 (CRC=F5) #data=1
(12:32:15) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 48 39 48 01 0B 91 07 06 06 03 00 00 00 (CRC=F2) #data=12
(12:32:15) Thermostat -> UBAMaster, type 0x10, telegram: 10 88 10 54 0C (CRC=D9) #data=1
(12:32:15) UBAMaster -> Thermostat, type 0x10, telegram: 08 10 10 54 36 41 00 E3 90 05 08 03 09 00 00 00 (CRC=B1) #data=12
(12:32:16) Thermostat -> All, type 0x48, telegram: 10 00 48 00 04 02 30 7D 00 00 00 27 2F 37 00 00 64 11 29 (CRC=70) #data=15
(12:32:16) Thermostat -> UBAMaster, type 0x11, telegram: 10 88 11 00 1B (CRC=62) #data=1
(12:32:16) UBAMaster -> Thermostat, type 0x11, telegram: 08 10 11 00 32 45 00 CF 00 71 3B BC 00 00 00 00 (CRC=27) #data=12
(12:32:16) Thermostat -> UBAMaster, type 0x11, telegram: 10 88 11 0C 1B (CRC=7A) #data=1
(12:32:16) UBAMaster -> Thermostat, type 0x11, telegram: 08 10 11 0C 32 45 00 CF 00 71 3B BC 00 00 00 00 (CRC=1D) #data=12
(12:32:17) Thermostat -> UBAMaster, type 0x11, telegram: 10 88 11 18 1B (CRC=52) #data=1
(12:32:17) UBAMaster -> Thermostat, type 0x11, telegram: 08 10 11 18 32 45 00 CF 00 71 3B BC 00 00 00 00 (CRC=53) #data=12
[... cut ...]
(12:55:10) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
(12:55:11) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)
(12:55:11) Thermostat -> All, type 0xA3, telegram: 10 00 A3 00 08 11 01 (CRC=AD) #data=3
(12:55:11) Thermostat -> Me, type 0x06, telegram: 10 0B 06 00 13 0B 0C 1C 32 33 03 00 (CRC=87) #data=8
(12:55:26) Thermostat -> All, type 0x48, telegram: 10 00 48 00 04 02 30 7D 00 00 00 27 2F 37 00 00 64 11 29 (CRC=70) #data=15
(12:55:27) Thermostat -> UBAMaster, type 0x35, telegram: 10 08 35 00 11 11 (CRC=30) #data=2
```

proddy, 2019-11-29:
I see data coming in on telegram type 0x48, which is Heat Controller 2 on a RC35. Are you sure you have an Buderus RC30?

og9gjz, 2019-11-29:
Hi,
i've uploaded two pictures, I think it's a RC30 (the RC35 has an matrix Display and another Menu|OK Button:
https://ibb.co/4sh7jvN
https://ibb.co/SxFK01m

proddy, 2019-11-29:
ok, doesn't really matter. I'll take your sample data and see about making it work. The values seem to be in the 0x48 telegram. I see the set point temperature is 24 degrees and the current temperature 12.5 degrees. Pretty cold in your house!

og9gjz, 2019-11-29:
Thank you! Yap, setpoint is 24 degrees. I don't know where the 12.5 degrees came from, but this isn't the right temp :D:D

proddy, 2019-11-29:
oh that's right. The current temp is the 4th byte of the data which has the value 0x7D meaning "not set". So not sure how to read the current temp. But you should be seeing the setpoint in mqtt and also the `info` command?

og9gjz, 2019-11-29:
The actual room temperature ist 19.7. I've added the output of "log v" as file, so we can analyze where the temperature is hidden ;) Or can you explain how i'll find it in the telegram raw data?
Is it this one: (15:31:41) 11 08 1E 00 01 B3 (because 179 = 0xB3)?

**info**

EMS-ESP system stats:
  System logging set to None
  LED: on, Listen mode: off
  Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: e                                                                nabled
  Shower Timer: enabled, Shower Alert: disabled

EMS Bus stats:
  Bus is connected, protocol: Buderus
  Rx: # successful read requests=3798, # CRC errors=12
  Tx: Last poll=2.143 seconds ago, # successful write requests=0

Boiler stats:
  Boiler: Sieger BK13,BK15/Nefit Smartline/Buderus GB1x2 (DeviceID:0x08 ProductI                                                                D:64 Version:02.07)
  Hot tap water: off
  Central heating: off
  Warm Water activated: on
  Warm Water circulation pump available: on
  Warm Water comfort setting: Hot
  Warm Water selected temperature: 50 C
  Warm Water desired temperature: 70 C
  Warm Water current temperature: 47.3 C
  Warm Water current tap water flow: 0.0 l/min
  Warm Water # starts: 23517 times
  Warm Water active time: 252 days 7 hours 9 minutes
  Warm Water 3-way valve: off
  Selected flow temperature: 48 C
  Current flow temperature: 52.9 C
  Return temperature: 52.4 C
  Gas: off
  Boiler pump: on
  Fan: off
  Ignition: off
  Circulation pump: off
  Burner selected max power: 70 %
  Burner current power: 0 %
  Flame current: 0.0 uA
  System pressure: 1.4 bar
  System service code: 0Y (204)
  Heating temperature setting on the boiler: 90 C
  Boiler circuit pump modulation max power: 100 %
  Boiler circuit pump modulation min power: 50 %
  Outside temperature: 4.2 C
  Boiler temperature: 47.3 C
  Pump modulation: 100 %
  Burner # starts: 248310 times
  Total burner operating time: 2039 days 22 hours 55 minutes
  Total heat operating time: 1787 days 15 hours 46 minutes
  Total UBA working time: 5276 days 0 hours 51 minutes

Thermostat stats:
  Thermostat: RC30 (DeviceID:0x10 ProductID:67 Version:02.08)
  Thermostat time is 15:23:37 29/11/2019
  Heating Circuit 2
   Current room temperature: ? C
   Setpoint room temperature: **24.0 C**

Mixing module stats:
  Switch temperature: **4428.8 C**

Shower stats:
  Shower is off

proddy, 2019-11-29:
19.7 would typically be stored as a short (2 bytes) of value 197 (its multiplied by 10), so look for a value 0xC5 in the telegram.

proddy, 2019-11-29:
@og9gjz that switch temp is a bug. can you do a 'boiler read 19' with logging verbose so I can see if there is a value

og9gjz, 2019-11-30:
@proddy Ok, I've found:

(15:30:41) 11 08 1E 00 01 C5
(15:30:59) 08 00 18 00 30 01 CA 46 00 01 01 20 60 80 00 01 D8 01 C5 00 00 0E 30 59 00 CC 00 00 00
Is this the room temperature? :)
Here's the output of boiler read:

**boiler read 19
log v**

`

System Logging set to Verbose
(09:20:28) Switching Module -> UBAMaster, type 0x1E, telegram: 11 08 1E 00 02 22 (CRC=44) #data=2
(09:20:29) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4E 46 14 09 01 25 60 80 00 01 E7 02 39 00 17 0F 2D 48 00 C8 00 02 00 (CRC=FF) #data=25
<--- UBAMonitorFast(0x18)
(09:20:29) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 32 01 E6 01 E7 21 00 01 03 00 05 8B 6D 00 5B E1 (CRC=4A) #data=16
<--- UBAMonitorWWMessage(0x34)
(09:20:30) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4E 46 14 09 01 25 60 80 00 01 E7 02 39 00 17 0E 2D 48 00 C8 00 02 00 (CRC=7F) #data=25
<--- UBAMonitorFast(0x18)
(09:20:31) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4E 46 14 09 01 25 60 80 00 01 E7 02 37 00 17 0E 2D 48 00 C8 00 02 00 (CRC=35) #data=25
<--- UBAMonitorFast(0x18)
(09:20:32) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4F 46 14 09 01 25 60 80 00 01 E7 02 37 00 17 0F 2D 48 00 C8 00 02 00 (CRC=6A) #data=25
<--- UBAMonitorFast(0x18)
(09:20:33) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4E 46 14 09 01 25 60 80 00 01 E7 02 37 00 17 0F 2D 48 00 C8 00 02 00 (CRC=B5) #data=25
<--- UBAMonitorFast(0x18)
(09:20:34) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4F 46 14 09 01 25 60 80 00 01 E7 02 37 00 17 0E 2D 48 00 C8 00 02 00 (CRC=EA) #data=25
<--- UBAMonitorFast(0x18)
(09:20:38) Switching Module -> All, type 0x9C, telegram: 11 00 9C 00 02 22 00 01 00 (CRC=FF) #data=5
(09:20:38) Switching Module -> UBAMaster, type 0x1E, telegram: 11 08 1E 00 02 22 (CRC=44) #data=2
(09:20:38) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4E 46 14 09 01 25 60 80 00 01 E7 02 36 00 17 0E 2D 48 00 C8 00 02 00 (CRC=51) #data=25
<--- UBAMonitorFast(0x18)
Fetching data from EMS devices
Requesting type RC30StatusMessage(0x41) from dest 0x10
Requesting type RC30Set(0xA7) from dest 0x10
Requesting type RCTime(0x06) from dest 0x10
Requesting type UBAMonitorFast(0x18) from dest 0x08
Requesting type UBAMonitorSlow(0x19) from dest 0x08
Requesting type UBAParameterWW(0x33) from dest 0x08
Requesting type UBAParametersMessage(0x16) from dest 0x08
Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
(09:20:39) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 32 01 E7 01 E6 21 00 01 03 00 05 8B 6D 00 5B E1 (CRC=89) #data=16
<--- UBAMonitorWWMessage(0x34)
(09:20:39) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4E 46 14 09 01 25 60 80 00 01 E6 02 37 00 17 0F 2D 48 00 C8 00 02 00 (CRC=3C) #data=25
<--- UBAMonitorFast(0x18)
(09:20:39) Sending read of type 0x41 to 0x10, telegram: 0B 90 41 00 20 (CRC=69)
(09:20:39) Thermostat -> Me, telegram: 10 0B 41 00 (CRC=2E)
(09:20:40) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 0B 0B 00 02 00 00 00 00 00 00 00 00 00 (CRC=0D) #data=13
<--- UBADevices(0x07)
(09:20:40) Sending read of type 0xA7 to 0x10, telegram: 0B 90 A7 00 20 (CRC=DA)
(09:20:40) Thermostat -> Me, telegram: 10 0B A7 00 (CRC=FB)
(09:20:40) Sending read of type 0x06 to 0x10, telegram: 0B 90 06 00 20 (CRC=6C)
(09:20:40) Thermostat -> Me, type 0x06, telegram: 10 0B 06 00 13 0B 09 1E 10 10 05 00 (CRC=8E) #data=8
<--- RCTime(0x06)
(09:20:41) Thermostat -> All, type 0x48, telegram: 10 00 48 00 04 02 30 7D 00 00 00 27 2F 37 00 00 64 11 31 (CRC=68) #data=15
<--- RC35StatusMessage_HC2(0x48)
(09:20:41) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
(09:20:41) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 36 02 4E 46 14 09 01 25 60 80 00 01 E7 02 39 00 17 0F 2D 48 00 C8 00 02 00 (CRC=4C) #data=25
<--- UBAMonitorFast(0x18)
(09:20:42) Sending read of type 0x19 to 0x08, telegram: 0B 88 19 00 20 (CRC=D0)
(09:20:42) UBAMaster -> Me, type 0x19, telegram: 08 0B 19 00 FF EA 02 47 80 00 00 00 00 62 03 CA 24 2C D6 23 00 00 00 27 4A B6 03 6E 43 (CRC=FF) #data=25
<--- UBAMonitorSlow(0x19)
(09:20:42) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4D 46 14 09 01 25 60 80 00 01 E7 02 39 00 17 0F 2D 48 00 C8 00 02 00 (CRC=87) #data=25
<--- UBAMonitorFast(0x18)
(09:20:42) Sending read of type 0x33 to 0x08, telegram: 0B 88 33 00 20 (CRC=78)
(09:20:42) UBAMaster -> Me, type 0x33, telegram: 08 0B 33 00 08 FF 32 FB 00 28 FF 03 46 00 FF (CRC=17) #data=11
<--- UBAParameterWW(0x33)
(09:20:43) Sending read of type 0x16 to 0x08, telegram: 0B 88 16 00 20 (CRC=EC)
(09:20:43) UBAMaster -> Me, type 0x16, telegram: 08 0B 16 00 FF 5A 46 00 06 FA 0A 01 05 64 32 02 (CRC=AC) #data=12
<--- UBAParametersMessage(0x16)
[... 74 lines cut ...]
<--- UBAMonitorWWMessage(0x34)
(09:21:19) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4D 46 14 09 01 25 60 80 00 01 E7 02 36 00 17 0E 2D 48 00 C8 00 02 00 (CRC=29) #data=25
<--- UBAMonitorFast(0x18)
(09:21:21) Thermostat -> UBAMaster, type 0x1A, telegram: 10 08 1A 00 36 64 00 00 (CRC=B1) #data=4
(09:21:23) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4D 46 14 09 01 25 60 80 00 01 E6 02 35 00 17 0F 2D 48 00 C8 00 02 00 (CRC=8C) #data=25
<--- UBAMonitorFast(0x18)
(09:21:23) Thermostat -> Mixing Module, type 0xAC, telegram: 10 21 AC 00 31 64 01 (CRC=17) #data=3
(09:21:24) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4D 46 14 09 01 25 60 80 00 01 E6 02 34 00 17 0E 2D 48 00 C8 00 02 00 (CRC=68) #data=25
[... cut ...]
(09:21:29) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 32 01 E6 01 E7 21 00 01 03 00 05 8B 6D 00 5B E1 (CRC=4A) #data=16
<--- UBAMonitorWWMessage(0x34)
(09:21:29) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4B 46 14 09 01 25 60 80 00 01 E7 02 2D 00 17 0F 2D 48 00 C8 00 02 00 (CRC=E8) #data=25
<--- UBAMonitorFast(0x18)
(09:21:31) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 36 02 4B 46 14 09 01 25 60 80 00 01 E7 02 2C 00 17 0E 2D 48 00 C8 00 02 00 (CRC=0C) #data=25
<--- UBAMonitorFast(0x18)`

proddy, 2019-12-01:
switch temp should be fixed. Still looking for the missing current temp on HC2

og9gjz567, 2019-12-01:
Is there something i can do?

proddy, 2019-12-01:
yes, I can't do it from here. You'll need to note down the current room temperature shown in the thermostat, then telnet into EMS-ESP, `log t` and grab say 30-60 seconds of logs. Look for telegrams of type 0x4? and in those a hex value of *10 of what the temp is. E.g. if its 20.5 degrees look for 0x00 and 0xCD (2 bytes). Please don't send logs to this github issue - I don't have the time to parse it!

og9gjz, 2019-12-01:
Hi,
the current room temperature is 17,7°, but I'm not able to find it i this string:
(17:47:52) Thermostat -> All, type 0x48, telegram: 10 00 48 00 4C 02 30 7D 00 00 00 27 2F 37 00 00 00 11 05 (CRC=98) #data=15
This is the only line with 0x4? and repeated three times (17:45:51 / 17:46:52 and 17:47:52)

You said 205 is 0xCD in Hex, so 177 is 0xB1. But I didnt't find it, or 0xB0 or 0xB2 in the log... :(

proddy, 2019-12-01:
very strange. You see in the 0x48 which is the monitor for HC2 where normally the room temperature is, yours shows "7D 00" which means its coming from somewhere else. We just don't know where! I would have expected to be on type ID 0x47 for HC1.

og9gjz, 2019-12-01:
I've found these three lines (Is it the one @18:17:10?:

```
(18:16:30) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 73 F9 **B1** (CRC=9D) #data=3
(18:17:10) Switching Module -> All, type 0x9C, telegram: 11 00 9C 00 01 **B1** 00 01 00 (CRC=33) #data=5
(18:17:10) Switching Module -> UBAMaster, type 0x1E, telegram: 11 08 1E 00 01 **B1** (CRC=D1) #data=2
```

martinSezman, 2019-12-03:
hi.
@og9gjz567 the RC30 thermostat is placed on  boiler (suspect not in the room). How do you want to read room temperature? Which device give you this information?

og9gjz567, 2019-12-03:
Yap. @martinSezman  At the moment the RC30 is placed in my Boiler, you can see it here: https://ibb.co/4sh7jvN
If I go to the menu of the RC30 I can see the actual room temperature. In spring i'll place the RC30 in my living room. Now I'm not in the mood to do some risky tasks.. ;)
I think my boiler is in "Außentemperaturgeführte Heizungssteuerung" mode. I don't know why there is only HC2 and the 1st one isn't in use...

proddy, 2019-12-03:
> I've found these three lines (Is it the one at 18:17:10?:
> 
> ```
> (18:16:30) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 73 F9 B1 (CRC=9D) #data=3
> (18:17:10) Switching Module -> All, type 0x9C, telegram: 11 00 9C 00 01 B1 00 01 00 (CRC=33) #data=5
> (18:17:10) Switching Module -> UBAMaster, type 0x1E, telegram: 11 08 1E 00 01 B1 (CRC=D1) #data=2
> ```

Yes, it could well be type 0x9C. That value B1 is 17.7 degrees?

og9gjz, 2019-12-06:
Hi, i've checked the log today. At the moment the display says 17,8 degrees. But I didn't find 0xB2 in the logs... Is it possible that the RC30 doesn't transmit the temperature to the uba/boiler because the actual mode is ""Außentemperaturgeführte Heizungssteuerung". I guess that the uba calculates the boiler temperature depending on the current outside temp?

proddy, 2019-12-06:
there is a temperature sensor in the RC30 which is used to fetch the current room temperature. I don't think it has anything to do with the boiler.

Since your RC30 is mounted in the boiler itself I'm guessing even if you knew the room temperature it would be pretty pointless as its only showing the temperature of where the boiler is located right?

glitter-ball, 2019-12-15:
It sounds as though HC2 is running with outdoor-influenced weather compensation. Unless there is a thermostat or sensor in the area heated by HC2, you have no way of measuring the temperature in that room! Instead, the system is relying on a heating curve and the outside temperature to calculate a suitable value for the flow temperature in HC2.

In some systems, only HC1 has a room temperature sensor with the thermostat and the system has to control HC2, HC3 and HC4 with basic weather compensation. Maybe that's what is happening here.

closed 2020-03-08 by proddy