emsesp/EMS-ESP #249: RC100/Moduline 1010 not writeable

Craftyalchemist, 2019-11-29:
Hi,

Thanks for your amazing software.

I hope you can clarify something for me. 
I have a RC100/Moduline 1010 (DeviceID:0x18 ProductID:165 Version:40.04)  
When I send the command via Telnet:
thermostat temp hc1 21   
it returns                                                                                                                                                                                                                       Write not supported yet for this Thermostat model.
However, from my understanding of the wiki https://github.com/proddy/EMS-ESP/wiki/supported-ems-devices, it should be supported for both reading and writing.  
Additonally on the instruction manual for the thermostat it says EMS 2.0, however if I set the mode to EMS+ (which I thought is also EMS 2.0) in the EMS-ESP webclient, then it says EMS Bus Connected but Tx is not working.
Additonally, and what I assume is related, I can't send an MQTT command to the thermostat via Home Assistant.
Any help would be appreciated, thanks.

I

proddy, 2019-11-29:
Hi @Craftyalchemist . Thanks for the kudos. A few things to try:
- which version of EMS-ESP are you using? Try the latest 1.9.4b which has improved support for EMS+
- I haven't looked at the RC100 for a while. It may just work. Take a look at https://github.com/proddy/EMS-ESP/wiki/RC3xx-Thermostats and see if you're receiving the same types like 0x1A5 and 0x1B9. If so try `send 0B 10 FF 08 01 B9 32` to set the temp to 25 degrees.
- Yes, EMS+ is the same as EMS 2.0.
- Use `mqttlog` to see if the thermostat packages are being sent. If not do a `info` and make sure it says "Thermostat Enabled" at the top.

Craftyalchemist, 2019-11-29:
Thanks for the quick reply.

-So if I'm in EMS+ mode and I get the following from info

`EMS-ESP system stats:                                                                                                                                                         System logging set to None                                                                                                                                                  LED: on, Listen mode: off                                                                                                                                                   1 external temperature sensor found                                                                                                                                         Boiler: disabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled                                                                                      Shower Timer: disabled, Shower Alert: disabled                                                                                                                                                                                                                                                                                                        EMS Bus stats:                                                                                                                                                                Bus is connected, protocol: Buderus                                                                                                                                         Rx: # successful read requests=126, # CRC errors=53                                                                                                                         Tx: Last poll=4.599 seconds ago, # successful write requests=0                                                                                                                                                                                                                                                                                        Boiler stats:                                                                                                                                                                 Boiler: <not enabled>                                                                                                                                                       Hot tap water: off                                                                                                                                                          Central heating: off                                                                                                                                                        Warm Water activated: ?                                                                                                                                                     Warm Water circulation pump available: ?                                                                                                                                    Warm Water selected temperature: ? C                                                                                                                                        Warm Water desired temperature: ? C                                                                                                                                         Warm Water current temperature: 35.4 C                                                                                                                                      Warm Water current tap water flow: 0.0 l/min                                                                                                                                Warm Water # starts: 45610 times                                                                                                                                            Warm Water active time: 26 days 21 hours 1 minutes                                                                                                                          Warm Water 3-way valve: off                                                                                                                                                 Selected flow temperature: 5 C                                                                                                                                              Current flow temperature: 31.5 C                                                                                                                                            Return temperature: 31.8 C                                                                                                                                                  Gas: off                                                                                                                                                                    Boiler pump: on                                                                                                                                                             Fan: off                                                                                                                                                                    Ignition: off                                                                 [... cut ...] AMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21                              (14:52:22) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 48 80 00 21 00 00 01 00 00 97 2D 00 B2 2A 00 80 00 (CRC=0D) #data=19                                    <--- UBAMonitorWWMessage(0x34)`

- I can see a 01xA5 telegram but If I try `send 0B 10 FF 08 01 B9 32` nothing seems to happen, no errors and no change on the thermostat

proddy, 2019-11-29:
just stick to telnet for now. Your thermostat is device ID 0x18 I think so the command should be
`send 0B 18 FF 08 01 B9 32`

Craftyalchemist, 2019-11-29:
But then if I'm in EML mode

From the info command I get
`EMS-ESP system stats:                                                                                                                                                         System logging set to None                                                                                                                                                  LED: on, Listen mode: off                                                                                                                                                   1 external temperature sensor found                                                                                                                                         Boiler: enabled, Thermostat: enabled, Solar Module: disabled, Mixing Module: disabled                                                                                       Shower Timer: disabled, Shower Alert: disabled                                                                                                                                                                                                                                                                                                        EMS Bus stats:                                                                                                                                                                Bus is connected, protocol: Buderus                                                                                                                                         Rx: # successful read requests=24, # CRC errors=0                                                                                                                           Tx: Last poll=3.775 seconds ago, # successful write requests=0                                                                                                                                                                                                                                                                                        Boiler stats:                                                                                                                                                                 Boiler: Buderus GBx72/Nefit Trendline/Junkers Cerapur/Worcester Greenstar Si/27i (DeviceID:0x08 ProductID:123 Version:06.08)                                                Hot tap water: off                                                                                                                                                          Central heating: active                                                                                                                                                     Warm Water activated: on                                                                                                                                                    Warm Water circulation pump available: off                                                                                                                                  Warm Water comfort setting: Hot                                                                                                                                             Warm Water selected temperature: 60 C                                                                                                                                       Warm Water desired temperature: 70 C                                                                                                                                        Warm Water current temperature: 37.0 C                                                                                                                                      Warm Water current tap water flow: 0.0 l/min                                                                                                                                Warm Water # starts: 45610 times                                                                                                                                            Warm Water active time: 26 days 21 hours 1 minutes                                                                                                                          Warm Water 3-way valve: off                                                                                                                                                 Selected flow temperature: 44 C                                                                                                                                             Current flow temperature: 43.4 C                                                                                                                                            Return temperature: 31.8 C                                                                                                                                                  Gas: on                                                                                                                                                                     Boiler pump: on                                                                                                                                                             Fan: on                                                                                                      [... cut ...]                            External temperature sensors:                                                                                                                                                 Sensor #1 DS18B20: 18.50 C `   
  
I can also see 01xA5 telegrams in this mode in the verbose log.

I've sent the command you suggested in EMS mode but didn't change anything.

I'm trying to change back to EML+ to test the command but it's a bit unstable and doesn't always regognise the thermostat.

proddy, 2019-11-29:
show me the `log v` when you issue the send command so I can see what is happening, I don't know what EML mode is. Also try with different tx_mode settings.

Craftyalchemist, 2019-11-29:
Hi,

Thanks again, your help is much appreciated.

So I mean the TX mode when I say EML or EML+.

So here is the `log v` when the TX mode is 1.

send 0B 18 FF 08 01 B9 32

> (00:02:19.610) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 7C 80 00 21 00 00 01 00 00 97 33 00 B2 2E 00 80 00 (CRC=85) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (00:02:19.843) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 79 00 00 00 00 40 40 01 7C 80 00 01 53 00 00 0F 30 48 00 CB 00 00 00 (CRC=E1) #data=25
> <--- UBAMonitorFast(0x18)
> (00:02:20.176) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (00:02:20.784) Sending raw: 0B 18 FF 08 01 B9 32
> (00:02:29.611) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (00:02:29.834) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 7B 80 00 21 00 00 01 00 00 97 33 00 B2 2E 00 80 00 (CRC=14) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (00:02:30.144) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 77 00 00 00 00 40 40 01 7B 80 00 01 53 00 00 0F 30 48 00 CB 00 00 00 (CRC=DD) #data=25
> <--- UBAMonitorFast(0x18)
> (00:02:39.622) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 76 00 00 00 00 40 40 01 7B 80 00 01 54 00 00 0F 30 48 00 CB 00 00 00 (CRC=27) #data=25
> <--- UBAMonitorFast(0x18)
> (00:02:39.847) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 76 00 00 00 00 40 40 01 7B 80 00 01 54 00 00 0F 30 48 00 CB 00 00 00 (CRC=27) #data=25
> <--- UBAMonitorFast(0x18)
> (00:02:40.140) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 80 00 01 73 80 00 00 00 00 00 01 0F 76 04 8F 15 00 00 00 03 F7 E2 00 5D 48 00 00 (CRC=13) #data=27
> <--- UBAMonitorSlow(0x19)
> (00:02:40.417) UBAMaster -> All, type 0x1C, telegram: 08 00 1C 00 93 01 14 0E 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=92) #data=25
> (00:02:40.635) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (00:02:40.999) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 7B 80 00 21 00 00 01 00 00 97 33 00 B2 2E 00 80 00 (CRC=14) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (00:02:41.311) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
> <--- UBADevices(0x07)
> (00:02:49.751) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 75 00 00 00 00 40 40 01 7A 80 00 01 54 00 00 0F 30 48 00 CB 00 00 00 (CRC=49) #data=25
> <--- UBAMonitorFast(0x18)
> (00:02:49.972) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (00:02:50.239) Thermostat -> UBAMaster, type 0x1A, telegram: 18 08 1A 00 00 00 00 (CRC=F6) #data=3
> (00:02:50.345) Thermostat -> All, type 0x01A5, telegram: 18 00 FF 00 01 A5 00 B7 21 24 00 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 (CRC=1B) #data=22
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (00:02:50.593) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 7A 80 00 21 00 00 01 00 00 97 33 00 B2 2E 00 80 00 (CRC=4C) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (00:02:50.965) Thermostat -> All, type 0x01A5, telegram: 18 00 FF 16 01 A5 07 27 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 03 (CRC=1D) #data=18
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (00:02:59.717) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 79 80 00 21 00 00 01 00 00 97 33 00 B2 2E 00 80 00 (CRC=A4) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (00:02:59.937) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 74 00 00 00 00 40 40 01 79 80 00 01 54 00 00 0F 30 48 00 CB 00 00 00 (CRC=AC) #data=25
> <--- UBAMonitorFast(0x18)
> Fetching data from EMS devices
> Requesting type RCPLUSStatusMessage_HC1(0x1A5) from dest 0x18
> Requesting type RCPLUSStatusMessage_HC2(0x1A6) from dest 0x18
> Requesting type RCPLUSStatusMessage_HC3(0x1A7) from dest 0x18
> Requesting type RCPLUSStatusMessage_HC4(0x1A8) from dest 0x18
> Requesting type RCTime(0x06) from dest 0x18
> Requesting type UBAMonitorFast(0x18) from dest 0x08
> Requesting type UBAMonitorSlow(0x19) from dest 0x08
> Requesting type UBAParameterWW(0x33) from dest 0x08
> Requesting type UBAParametersMessage(0x16) from dest 0x08
> Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
> (00:03:00.195) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (00:03:01.255) Sending read of type 0x1A5 to 0x18, telegram: 0B 98 FF 00 20 01 A5 (CRC=00)
> (00:03:01.397) Thermostat -> Me, type 0x01A5, telegram: 18 0B FF 00 01 A5 00 B7 21 24 00 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 07 27 00 (CRC=74) #data=25
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (00:03:01.538) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=5A) #data=13
> <--- UBADevices(0x07)
> (00:03:01.730) Sending read of type 0x1A6 to 0x18, telegram: 0B 98 FF 00 20 01 A6 (CRC=03)
> (00:03:01.782) Thermostat -> Me, telegram: 18 0B FF 00 01 A6 (CRC=88)
> (00:03:02.104) Sending read of type 0x1A7 to 0x18, telegram: 0B 98 FF 00 20 01 A7 (CRC=02)
> (00:03:02.156) Thermostat -> Me, telegram: 18 0B FF 00 01 A7 (CRC=89)
> (00:03:02.430) Sending read of type 0x1A8 to 0x18, telegram: 0B 98 FF 00 20 01 A8 (CRC=0D)
[... cut ...]
> (00:03:04.590) UBAMaster -> Me, type 0x16, telegram: 08 0B 16 00 FF 50 64 00 06 FA 0A 01 05 46 1E 00 00 00 00 00 00 00 00 00 00 00 00 00 28 00 3C (CRC=1D) #data=27
> <--- UBAParametersMessage(0x16)
> (00:03:04.904) Sending read of type 0x14 to 0x08, telegram: 0B 88 14 00 20 (CRC=E4)
> (00:03:04.970) UBAMaster -> Me, type 0x14, telegram: 08 0B 14 00 17 02 C2 (CRC=88) #data=3
> <--- UBATotalUptimeMessage(0x14)
> (00:03:06.029) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
> <--- UBADevices(0x07)

Craftyalchemist, 2019-11-29:
And then if I'm in TX mode 2

>send 08 18 FF 08 01 b9 32
> (20:20:40) Sending raw: 08 18 FF 08 01 B9 32
> (20:20:40) UBAMaster -> Thermostat, type 0x01B9, telegram: 08 18 FF 08 01 B9 32 (CRC=73) #data=1
> (20:20:47) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 B8 80 00 21 00 00 01 00 00 97 47 00 B2 35 00 80 00 (CRC=60) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (20:20:47) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 CA 64 22 09 01 25 40 01 B8 80 00 01 6A 00 EC 11 2D 48 00 C8 00 02 00 (CRC=23) #data=25
> <--- UBAMonitorFast(0x18)
> (20:20:48) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C0 00 C0 80 00 00 80 00 80 00 80 00 00 (CRC=C3) #data=21
> (20:20:57) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C0 00 C0 80 00 00 80 00 80 00 80 00 00 (CRC=C3) #data=21
> (20:20:58) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 B8 80 00 21 00 00 01 00 00 97 47 00 B2 35 00 80 00 (CRC=60) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (20:20:58) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 CA 64 22 09 01 25 40 01 B8 80 00 01 6B 00 EE 11 2D 48 00 C8 00 02 00 (CRC=75) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:05) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 CC 64 22 09 01 25 40 01 B8 80 00 01 6B 01 17 11 2D 48 00 C8 00 02 00 (CRC=DE) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:07) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 CE 64 22 09 01 25 40 01 B7 80 00 01 6B 01 16 11 2D 48 00 C8 00 02 00 (CRC=B2) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:08) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 CE 64 22 09 01 25 40 01 B7 80 00 01 6B 01 15 11 2D 48 00 C8 00 02 00 (CRC=99) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:08) UBAMaster -> All, type 0x19, telegram: 08 00 19 00 80 00 01 D2 80 00 00 00 00 23 01 0F 82 04 8F 6E 00 00 00 03 F8 27 00 5D 4D 00 00 (CRC=DB) #data=27
> <--- UBAMonitorSlow(0x19)
> (20:21:08) UBAMaster -> All, type 0x1C, telegram: 08 00 1C 00 93 01 14 0E 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=92) #data=25
> (20:21:08) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 C0 00 C0 80 00 00 80 00 80 00 80 00 00 (CRC=C3) #data=21
> (20:21:09) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 B7 80 00 21 00 00 01 00 00 97 47 00 B2 35 00 80 00 (CRC=03) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (20:21:09) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 03 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=76) #data=13
> <--- UBADevices(0x07)
> (20:21:13) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 D6 64 22 09 01 25 40 01 B7 80 00 01 6B 01 11 11 2D 48 00 C8 00 02 00 (CRC=16) #data=25
> <--- UBAMonitorFast(0x18)
> Fetching data from EMS devices
> Publishing boiler data via MQTT
> Publishing hot water and heating states via MQTT
> Publishing external sensor data via MQTT
> (20:21:16) Thermostat -> UBAMaster, type 0x1A, telegram: 18 08 1A 00 2F 64 64 (CRC=E6) #data=3
> (20:21:16) Thermostat -> All, type 0x01A5, telegram: 18 00 FF 00 01 A5 00 B3 21 24 2F 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 (CRC=DD) #data=22
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (20:21:16) Thermostat -> All, type 0x01A5, telegram: 18 00 FF 16 01 A5 06 FE 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 02 (CRC=83) #data=18
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (20:21:17) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 DA 64 22 09 01 25 40 01 B7 80 00 01 6B 00 FC 11 2D 48 00 C8 00 02 00 (CRC=59) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:18) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 BD 00 BE 80 00 00 80 00 80 00 80 00 00 (CRC=C9) #data=21
> (20:21:18) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 B7 80 00 21 00 00 01 00 00 97 47 00 B2 35 00 80 00 (CRC=03) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (20:21:27) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 B7 80 00 21 00 00 01 00 00 97 47 00 B2 35 00 80 00 (CRC=03) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (20:21:28) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 D6 64 1E 09 01 25 40 01 B7 80 00 01 6B 00 B5 11 2D 48 00 C8 00 02 00 (CRC=E6) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:28) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B1 00 B1 80 00 00 80 00 80 00 80 00 00 (CRC=DD) #data=21
> (20:21:28) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 D5 64 1E 09 01 25 40 01 B7 80 00 01 6C 00 B3 11 2D 48 00 C8 00 02 00 (CRC=ED) #data=25
> <--- UBAMonitorFast(0x18)
> ** Warning, we missed the bus - Rx non-idle!
> (20:21:32) Thermostat -> All, type 0x01B9, telegram: 18 00 FF 0A 01 B9 24 (CRC=3A) #data=1
> <--- RCPLUSSetMessage(0x1B9)
> (20:21:35) Thermostat -> All, type 0x01B9, telegram: 18 00 FF 0A 01 B9 24 (CRC=3A) #data=1
> <--- RCPLUSSetMessage(0x1B9)
> (20:21:36) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 CB 64 1E 09 01 25 40 01 B7 80 00 01 6C 00 A4 11 2D 48 00 C8 00 02 00 (CRC=30) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:37) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 2F 01 CA 64 1E 09 01 25 40 01 B7 80 00 01 6C 00 A5 11 2D 48 00 C8 00 02 00 (CRC=F6) #data=25
> <--- UBAMonitorFast(0x18)
> (20:21:38) Corrupt telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 B1 00 B1 80 00 00
> (20:21:47) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 B5 80 00 21 00 00 01 00 00 97 47 00 B2 35 00 80 00 (CRC=B3) #data=19
[... cut ...]
> (20:23:08) UBAMaster -> All, type 0x1C, telegram: 08 00 1C 00 93 01 14 0E 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 (CRC=92) #data=25
> (20:23:08) Thermostat -> All, type 0xA2, telegram: 18 00 A2 00 00 00 00 00 00 (CRC=02) #data=5
> (20:23:08) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 BB 00 BB 80 00 00 80 00 80 00 80 00 00 (CRC=39) #data=21
> (20:23:09) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 B2 80 00 21 00 00 01 00 00 97 47 00 B2 35 00 80 00 (CRC=22) #data=19
> <--- UBAMonitorWWMessage(0x34)

proddy, 2019-11-29:
One last shot, try `send 0B 18 FF 08 01 B9 32`

what we're doing is changing the origin from 08 (the boiler) to our device (0B)

Craftyalchemist, 2019-12-02:
So I gave it a go but no to avail....
Here's the log just in case

> send 0B 18 FF 08 01 B9 32
> 
> (21:11:14) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 5B 80 00 21 00 00 01 00 00 97 A4 00 B2 A9 00 80 00 (CRC=1D) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (21:11:14) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 0D 00 00 00 00 20 40 01 5B 80 00 01 0F 00 00 10 30 48 00 CB 00 00 00 (CRC=57) #data=25
> <--- UBAMonitorFast(0x18)
> (21:11:14) Sending raw: 0B 18 FF 08 01 B9 32
> (21:11:15) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (21:11:24) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (21:11:24) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 58 80 00 21 00 00 01 00 00 97 A4 00 B2 A9 00 80 00 (CRC=F5) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (21:11:24) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 0E 00 00 00 00 20 40 01 58 80 00 01 10 00 00 10 30 48 00 CB 00 00 00 (CRC=2D) #data=25
> <--- UBAMonitorFast(0x18)
> (21:11:25) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 0E 00 00 00 00 20 40 01 58 80 00 01 10 00 00 10 30 48 00 CB 00 00 00 (CRC=2D) #data=25
> <--- UBAMonitorFast(0x18)
> (21:11:34) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 0F 00 00 00 00 20 40 01 56 80 00 01 10 00 00 10 30 48 00 CB 00 00 00 (CRC=36) #data=25
> <--- UBAMonitorFast(0x18)
> (21:11:34) Thermostat -> UBAMaster, type 0x1A, telegram: 18 08 1A 00 00 00 64 (CRC=92) #data=3
> (21:11:34) Thermostat -> All, type 0x01A5, telegram: 18 00 FF 00 01 A5 00 BC 21 26 00 00 26 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 (CRC=08) #data=22
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (21:11:35) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> (21:11:35) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 56 80 00 21 00 00 01 00 00 97 A4 00 B2 A9 00 80 00 (CRC=CE) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (21:11:35) Thermostat -> All, type 0x01A5, telegram: 18 00 FF 16 01 A5 07 5B 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 03 (CRC=57) #data=18
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (21:11:44) UBAMaster -> All, type 0x34, telegram: 08 00 34 00 3E 01 54 80 00 21 00 00 01 00 00 97 A4 00 B2 A9 00 80 00 (CRC=7E) #data=19
> <--- UBAMonitorWWMessage(0x34)
> (21:11:44) UBAMaster -> All, type 0x18, telegram: 08 00 18 00 05 01 0F 00 00 00 00 20 40 01 54 80 00 01 12 00 00 10 30 48 00 CB 00 00 00 (CRC=D2) #data=25
> <--- UBAMonitorFast(0x18)
> (21:11:44) UBAMaster -> All, type 0x2A, telegram: 08 00 2A 00 00 00 00 00 00 00 00 00 F8 00 00 80 00 00 80 00 80 00 80 00 00 (CRC=EF) #data=21
> Fetching data from EMS devices
> Requesting type RCPLUSStatusMessage_HC1(0x1A5) from dest 0x18
> Requesting type RCPLUSStatusMessage_HC2(0x1A6) from dest 0x18
> Requesting type RCPLUSStatusMessage_HC3(0x1A7) from dest 0x18
> Requesting type RCPLUSStatusMessage_HC4(0x1A8) from dest 0x18
> Requesting type RCTime(0x06) from dest 0x18
> Requesting type UBAMonitorFast(0x18) from dest 0x08
> Requesting type UBAMonitorSlow(0x19) from dest 0x08
> Requesting type UBAParameterWW(0x33) from dest 0x08
> Requesting type UBAParametersMessage(0x16) from dest 0x08
> Requesting type UBATotalUptimeMessage(0x14) from dest 0x08
> Publishing boiler data via MQTT
> Publishing hot water and heating states via MQTT
> Publishing thermostat data via MQTT
> Publishing external sensor data via MQTT
> (21:11:50) Sending read of type 0x1A5 to 0x18, telegram: 0B 98 FF 00 20 01 A5 (CRC=00)
> (21:11:50) Thermostat -> Me, type 0x01A5, telegram: 18 0B FF 00 01 A5 00 BC 21 26 00 00 26 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 07 5B 00 (CRC=14) #data=25
> <--- RCPLUSStatusMessage_HC1(0x1A5)
> (21:11:50) Sending read of type 0x1A6 to 0x18, telegram: 0B 98 FF 00 20 01 A6 (CRC=03)
> (21:11:50) Thermostat -> Me, telegram: 18 0B FF 00 01 A6 (CRC=88)
> (21:11:50) UBAMaster -> All, type 0x07, telegram: 08 00 07 00 0B 00 01 00 00 00 00 00 00 00 00 00 00 (CRC=5A) #data=13
> <--- UBADevices(0x07)
> (21:11:51) Sending read of type 0x1A7 to 0x18, telegram: 0B 98 FF 00 20 01 A7 (CRC=02)
> (21:11:51) Thermostat -> Me, telegram: 18 0B FF 00 01 A7 (CRC=89)
> (21:11:51) Sending read of type 0x1A8 to 0x18, telegram: 0B 98 FF 00 20 01 A8 (CRC=0D)
> (21:11:51) Thermostat -> Me, telegram: 18 0B FF 00 01 A8 (CRC=86)
> (21:11:52) Sending read of type 0x06 to 0x18, telegram: 0B 98 06 00 20 (CRC=2C)
> (21:11:52) Thermostat -> Me, telegram: 18 0B 06 00 (CRC=E0)
> (21:11:52) Sending read of type 0x18 to 0x08, telegram: 0B 88 18 00 20 (CRC=D4)
> (21:11:52) UBAMaster -> Me, type 0x18, telegram: 08 0B 18 00 05 01 10 00 00 00 00 20 40 01 52 80 00 01 12 00 00 10 30 48 00 CB 00 00 00 (CRC=D1) #data=25

What thermostat do you recommend?

Craftyalchemist, 2019-12-11:
Hey,
Thanks for your help. I got a Moduline 300 and it works perfectly.
I don't know if it was the particular firmware on my RC100/Moduline 1010 but maybe the Wiki should be updated.

proddy, 2019-12-12:
I'll re-open this as I still want to support writing on the 1010. I think we were quite close.

Craftyalchemist, 2019-12-12:
OK, if you need any assistance checking against the thermostat then let me know.

proddy, 2020-03-08:
RC100 and the moduline 1000 and 1010 are fully supported now in 1.9.5b52

closed 2020-03-08 by proddy