emsesp/EMS-ESP #250: Websocket can't be parsed - no memory

LineF, 2019-12-02:
Hi,

I've compiled ems-esp but not yet connected to ems hardware.
Being on the Web-UI I can use everything, but when I want to save my settings they aren't saved.
Having a look on the serial interface shows me the following error message each time pressing a "save" button:

[WEB] Couldn't parse WebSocket message, error NoMemory

This is the complete boot message:
* EMS-ESP version 1.9.4b21
[FS] System config loaded (479 bytes)
[FS] Custom config loaded (201 bytes)
[WEB] Web server started
[SYSTEM] Free Heap: 23928 bytes initially | 4080 bytes used (17%) | 19848 bytes free (82%)
Warning! EMS bus communication disabled when Serial mode enabled. Use 'set serial off' to start communication.
[WIFI] Connecting to SSID: xxx...
[WIFI] Connected to SSID xxx (hostname: ems-esp, IP: 192.168.1.30)
[OTA] Listening to firmware updates on ems-esp.local:8266
[SYSTEM] Serial port communication is enabled
[NTP] NTP internet time enabled via server pool.ntp.org with timezone 2
[TELNET] Telnet server started
[NTP] Internet time: 20:33:28 UTC on 2/12. Local time: 21:33:28 CET
[TELNET] Connected to EMS-ESP version 1.9.4b21. Type ? for commands.
System Logging set to None
[WEB] Couldn't parse WebSocket message, error NoMemory

Build environment:
Windows 10, PlatformIO on Visual Studio Code, PIO Home version 3.0.1 core 4.1.0, Node.JS version v12.13.1

Hardware:
NodeMCU v2

proddy, 2019-12-02:
can you try the precompiled firmware binary? this will rule out any strange config you may have in your build environment. instructions are in the wiki

LineF, 2019-12-02:
yes, same problem with precompiled firmware binary (EMS-ESP version 1.9.4b21)

LineF, 2019-12-02:
even made a factory reset and reconfigured from scratch (client mode, ssid/pwd set, serial enable) but still the same problem.

proddy, 2019-12-02:
thats strange. when running, how much free mem do you have after a `system` command?

LineF, 2019-12-02:
system

ESP8266 System stats:

 [APP] EMS-ESP version: 1.9.4b21
 [APP] MyESP version: 1.2.22
 [APP] Build timestamp: 2019-11-30 12:51:11
 [APP] Uptime: 0 days 0 hours 8 minutes 46 seconds
 [APP] System Load: 1%
 [WIFI] WiFi Hostname: ems-esp
 [WIFI] WiFi IP: 192.168.1.30
 [WIFI] WiFi signal strength: 100%
 [WIFI] WiFi MAC: 5C:CF:7F:8B:EF:69
 [MQTT] is disconnected
 [SYSTEM] System is Stable
 [SYSTEM] Board: PLATFORMIO_NODEMCUV2
 [SYSTEM] CPU frequency: 80 MHz
 [SYSTEM] SDK version: 2.2.1(cfd48f3)
 [SYSTEM] CPU chip ID: 0x8BEF69
 [SYSTEM] Core version: 2_5_2
 [SYSTEM] Boot version: 31
 [SYSTEM] Boot mode: 1
 [SYSTEM] Last reset reason: Factory reset
 [SYSTEM] Restart count: 0
 [SYSTEM] # TCP disconnects: 0
 [SYSTEM] rtcmem status: blocks:2 addr:0x60001280
 [SYSTEM] rtcmem 00: 1163087990
 [SYSTEM] rtcmem 01: 65536
 [FLASH] Flash chip ID: 0x1640E0
 [FLASH] Flash speed: 40000000 Hz
 [FLASH] Flash mode: DIO
 [FLASH] Flash size (CHIP): 4194304
 [FLASH] Flash size (SDK): 4194304
 [FLASH] Flash Reserved: 4096
 [MEM] Firmware size: 611504
 [MEM] Max OTA size: 2527232
 [MEM] OTA Reserved: 16384
 [MEM] Free Heap: 22960 bytes initially | 8960 bytes used (39%) | 14000 bytes free (60%)

proddy, 2019-12-02:
actually, forget that. I'm able to reproduce the bug. Looking for a fix....

LineF, 2019-12-02:
ok, thanks!

proddy, 2019-12-02:
made a fix in 1.9.4b22. Please try it out. Thanks for reporting the error. The problem was the JSON file is getting larger and I hadn't intialized enough static memory to hold the contents.

LineF, 2019-12-02:
ok, problem solved!
No more "no memory" errors and saving configuration on the Web-UI is working now.
Thanks very much! Great work your project!
BTW: I'm running an EMS-GW based on AVR/Ethersex and nearly same connecting hardware since three years now, but decoding EMS+ messages is not supported, neither sending messages to the boiler. So I will give your solution a try.
Martin

proddy, 2019-12-03:
cool, let me know how you get on

closed 2019-12-03 by proddy