emsesp/EMS-ESP #251: Can't build from dev branch

theloukou, 2019-12-03:
I can't seem to build the dev branch from PlatformIO on Win10.
Using the guide on the wiki to build the release branch works just fine. The dev branch gives out this error:

`Cloning into 'd:\Lucifer\Arduino\EMS-ESP\.pio\libdeps\debug\_tmp_installing-hj04l7yx-package'...
PermissionError: [WinError 5] Access is denied: 'd:\\Lucifer\\Arduino\\EMS-ESP\\.pio\\libdeps\\debug\\_tmp_installing-hj04l7yx-package\\.git\\objects\\0a\\30a4a9f4aeadf6c13959155e99f843239b4b2c':
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\BrokenBeat\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\BrokenBeat\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\BrokenBeat\.platformio\platforms\espressif8266@2.2.2\builder\main.py", line 212:
    target_elf = env.BuildProgram()
  File "C:\Users\BrokenBeat\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\BrokenBeat\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 497:
    return self._install_from_tmp_dir(_tmp_dir, requirements)
  File "C:\Users\BrokenBeat\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 583:
    shutil.move(tmp_dir, pkg_dir)
  File "C:\Users\BrokenBeat\.platformio\python37\lib\shutil.py", line 578:
    rmtree(src)
  File "C:\Users\BrokenBeat\.platformio\python37\lib\shutil.py", line 516:
    return _rmtree_unsafe(path, onerror)
  File "C:\Users\BrokenBeat\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\BrokenBeat\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\BrokenBeat\.platformio\python37\lib\shutil.py", line 395:
    _rmtree_unsafe(fullname, onerror)
  File "C:\Users\BrokenBeat\.platformio\python37\lib\shutil.py", line 400:
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\Users\BrokenBeat\.platformio\python37\lib\shutil.py", line 398:
    os.unlink(fullname)`

Tried giving full access to the project folder to all system users, but nothing changes.

proddy, 2019-12-03:
hard to tell what the problem is. It's definitely in your environment. Try cleaning use the 'Clean' option in PlatformIO. The dev build is built nightly from scratch via Travis-CI so the code and scripts are fine. I just tested here on Win10 and it works fine too, even after deleting the whole .pio folder. I noticed you're using 'debug' as a target - perhaps your platformio.ini is different?

theloukou, 2019-12-04:
After clicking the build button time after time, the point at which the "access denied" error kept changing, everytime appearing after a different library.
Now the log stops after these messages:

`Compiling .pio\build\release\libec2\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
In file included from src\MyESP.cpp:10:0:
src\MyESP.h:22:25: fatal error: uuid/common.h: No such file or directory
 #include <uuid/common.h>
                         ^
compilation terminated.
*** [.pio\build\release\src\MyESP.cpp.o] Error 1
In file included from src\Ntp.cpp:6:0:
src\MyESP.h:22:25: fatal error: uuid/common.h: No such file or directory
 #include <uuid/common.h>
                         ^
compilation terminated.
*** [.pio\build\release\src\Ntp.cpp.o] Error 1
In file included from src\ems-esp.cpp:11:0:
src\MyESP.h:22:25: fatal error: uuid/common.h: No such file or directory
 #include <uuid/common.h>
                         ^
compilation terminated.
*** [.pio\build\release\src\ems-esp.cpp.o] Error 1
In file included from src\ems.cpp:10:0:
src\MyESP.h:22:25: fatal error: uuid/common.h: No such file or directory
 #include <uuid/common.h>
                         ^
compilation terminated.
*** [.pio\build\release\src\ems.cpp.o] Error 1
In file included from src\ems_utils.h:11:0,
                 from src\ems_utils.cpp:8:
src\MyESP.h:22:25: fatal error: uuid/common.h: No such file or directory
 #include <uuid/common.h>
                         ^
compilation terminated.
*** [.pio\build\release\src\ems_utils.cpp.o] Error 1`

It seems like a library is missing? All these will the stable branch still compiles just fine...

theloukou, 2019-12-04:
Sidenote, my Arduino IDE ESP8266 platform definition has version 2.6.2 installed, while in PlatformIO version 2.5.2 is mentioned. Could this be a culprit?

proddy, 2019-12-04:
that is strange. I'll do some more tests later on Win10. Are you sure you're using the same platformio.ini file?

theloukou, 2019-12-04:
Everything i'm using though is the cloned zip from your branch, haven't changed a thing yet. Your platformio.ini doesn't include those libs though. Maybe you have them locally so everything compiles there?

Adding the dependencies did the trick though on my machine. Maybe just add them in general? If someone could try compiling the dev branch on their machines we could have more "test data".

proddy, 2019-12-04:
I'll add them to platformio. The latest pio should work out the lib chain, and it works on Travis CI which uses docker to build from scratch on Ubuntu. Thanks for reporting.

closed 2019-12-04 by theloukou